openwrt/openwrt #13704: uhttpd, garbled non-english text

fyvfy, 2023-10-15:
### Describe the bug

wrong encoding / garbled text when trying to open in browser utf-8 text served by openwrt uhttpd

### OpenWrt version

r19803-9a599fee93

### OpenWrt target/subtarget

blah

### Device

blah

### Image kind

Official downloaded image

### Steps to reproduce

1. in openwrt terminal create text file
\# cat /www/utf8.txt 
```
ÆØẞ
```
(characters ae, slashed o, sharp s)
$ base64 /www/utf8.txt 
```
w4bDmOG6ng==
```
2. open that file in browser
```
https://192.168.1.1/utf8.txt
```

### Actual behaviour

browser shows
```
Ã†Ã˜áºž
```

### Expected behaviour

browser should show
```
ÆØẞ
```

### Additional info

need something similar to #3554

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-10-16:
uhttpd conforms respective standards (or busybox httpd) Programming questions better go to forums.
Some other web servers avail explicitly set standard encoding when not needed.

https://www.rfc-editor.org/rfc/rfc2616#section-3.4.1
https://www.w3.org/International/articles/http-charset/index

jow-, 2023-10-19:
uhttpd does not send a character encoding header for static files, as it has no means to determine the contained charset.

You can either serve an HTML file and declare the charset using a ` <meta charset="UTF-8">` header or include a BOM in your UTF-8 plain text file to aid browser charset detection heuristics:

```print '\xEF\xBB\xBFÆØẞ' > /www/utf8.txt```

Adding a BOM causes Firefox and Chrome to display `ÆØẞ` instead of `Ã†Ã˜áºž`.

closed 2023-10-19 by jow-