openwrt/openwrt #13705: DNS Hijacking Regression from 22.03 to 22.05

Spamfast, 2023-10-16:
### Describe the bug

I have two Netgear WAC124's as border gateways on different sites but both connected to Zen's Internet static-IPv4/IPv6/PD48 service via a Zen FRITZ!Box 5370 VDSL routers.

Currently one is still running 22.03.5 but I've just upgraded the other to 22.05.0 which is mostly running fine.

However I've spotted that the rules I am using for DNS hijacking on my DMZ subnet (where things like Chromecasts, FireTV sticks, cellphones etc. live) are no longer effective on 22.05.0.

The IPv4 rules in /etc/config/firewall are thus, with the first one providing the hikack and the second rewriting the DNS server IP address the DNS clients see in the replies. `192.168.26.1` is the IPv4 address of the 22.03.5 router on its DMZ network. On the 22.05.0 router the rules are identical except that the router's DMZ address is `192.167.27.1`.

```
config redirect
        option dest 'dmz'
        option target 'DNAT'
        option src 'dmz'
        option src_ip '!192.168.26.1'
        option src_dport '53'
        option dest_ip '192.168.26.1'
        option dest_port '53'
        option src_dip '!192.168.26.1'
        option name 'Intercept-DNS-IPv4-Dmz'

config nat
        list proto 'tcp'
        list proto 'udp'
        option src 'dmz'
        option dest_ip '192.168.26.1'
        option dest_port '53'
        option target 'MASQUERADE'
        option name 'Intercept-DNS-IPv4-Dmz-Masq'
``` 
Here is a dig from a host on the 22.03.5 DMZ LAN. As can be seen, even though it's attempting to use Cloudflare's global server `1.1.1.1` it is able to resolve local names.
```
samedge@hephaestus ~
$ ssh nano.ns.lan dig -p 53 -b 192.168.26.183 ceres.f2.lan @1.1.1.1

; <<>> DiG 9.18.12-0ubuntu0.22.04.3-Ubuntu <<>> -p 53 -b 192.168.26.183 ceres.f2.lan @1.1.1.1
;; global options: +cmd
;; Got answer:
;; ->>HEADER<<- opcode: QUERY, status: NOERROR, id: 10024
;; flags: qr aa rd ra; QUERY: 1, ANSWER: 2, AUTHORITY: 0, ADDITIONAL: 1

;; OPT PSEUDOSECTION:
; EDNS: version: 0, flags:; udp: 1232
;; QUESTION SECTION:
;ceres.f2.lan.                  IN      A

;; ANSWER SECTION:
ceres.f2.lan.           300     IN      A       192.168.15.114
ceres.f2.lan.           300     IN      A       192.168.15.115

;; Query time: 28 msec
;; SERVER: 1.1.1.1#53(1.1.1.1) (UDP)
;; WHEN: Mon Oct 16 11:01:53 UTC 2023
;; MSG SIZE  rcvd: 73
```
Here is the same dig from a host on the 22.05.0 DMZ LAN. In this case, it appears that the DNS server address is not being hijacked as can be seen by the timezone offsets etc. in the reply comments.
```
$ ssh bapple.f2.lan dig -p 53 -b 192.168.27.183 ceres.f2.lan @1.1.1.1

; <<>> DiG 9.11.5-P4-5.1+deb10u9-Raspbian <<>> -p 53 -b 192.168.27.183 ceres.f2.lan @1.1.1.1
;; global options: +cmd
;; Got answer:
;; ->>HEADER<<- opcode: QUERY, status: NXDOMAIN, id: 54563
;; flags: qr rd ra ad; QUERY: 1, ANSWER: 0, AUTHORITY: 1, ADDITIONAL: 1

;; OPT PSEUDOSECTION:
; EDNS: version: 0, flags:; udp: 1232
;; QUESTION SECTION:
;ceres.f2.lan.                  IN      A

;; AUTHORITY SECTION:
.                       86400   IN      SOA     a.root-servers.net. nstld.verisign-grs.com. 2023101600 1800 900 604800 86400

;; Query time: 22 msec
;; SERVER: 1.1.1.1#53(1.1.1.1)
;; WHEN: Mon Oct 16 12:04:36 BST 2023
;; MSG SIZE  rcvd: 116
```
Here are the versions from the 22.03.5 and 22.05.0 installations.
```
$ ssh hippolyta nft -V # 22.03.5
nftables v1.0.2 (Lester Gooch)
  cli:          no
  json:         yes
  minigmp:      yes
  libxtables:   no
$ ssh hippolyta opkg list-installed | grep firewall
firewall4 - 2022-10-18-7ae5e14b-1
luci-app-firewall - git-23.093.42704-1c01c64
$ ssh hippolyta opkg list-installed | grep nft
kmod-nft-core - 5.10.176-1
kmod-nft-fib - 5.10.176-1
kmod-nft-nat - 5.10.176-1
kmod-nft-offload - 5.10.176-1
libnftnl11 - 1.2.1-2
nftables-json - 1.0.2-2.1
```
```
$ ssh charybdis nft -V # 22.05.0
nftables v1.0.8 (Old Doc Yak #2)
  cli:          no
  json:         yes
  minigmp:      yes
  libxtables:   no
$ ssh charybdis opkg list-installed | grep firewall
firewall4 - 2023-03-23-04a06bd7-1
luci-app-firewall - git-23.208.40260-9504081
$ ssh charybdis opkg list-installed | grep nft
kmod-nft-core - 5.15.134-1
kmod-nft-fib - 5.15.134-1
kmod-nft-nat - 5.15.134-1
kmod-nft-offload - 5.15.134-1
libnftnl11 - 1.2.6-1
nftables-json - 1.0.8-1
```

### OpenWrt version

22.05.0

### OpenWrt target/subtarget

ramips/mt7621

### Device

Netgear WAC124

### Image kind

Official downloaded image

### Steps to reproduce

See description.

### Actual behaviour

No hijack.

### Expected behaviour

Hijack.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

obsy, 2023-10-16:
22.05.0? Should be 23.05.0

Spamfast, 2023-10-16:
I forgot to say - the resultant rules in the nftables ruleset are identical as well, apart from the router IP addresses of course.

> 22.05.0? Should be 23.05.0

Oh poop. Yes, that's what I mean! Thanks. :-)

Spamfast, 2023-10-16:
```
        chain dstnat_dmz {
                ip saddr != 192.168.26.1 ip daddr != 192.168.26.1 tcp dport 53 counter packets 0 bytes 0 dnat ip to 192.168.26.1:53 comment "!fw4: Intercept-DNS-IPv4-Dmz"
                ip saddr != 192.168.26.1 ip daddr != 192.168.26.1 udp dport 53 counter packets 153 bytes 9889 dnat ip to 192.168.26.1:53 comment "!fw4: Intercept-DNS-IPv4-Dmz"
        }

        chain srcnat_dmz {
                ip daddr 192.168.26.1 tcp dport 53 counter packets 0 bytes 0 masquerade comment "!fw4: Intercept-DNS-IPv4-Dmz-Masq"
                ip daddr 192.168.26.1 udp dport 53 counter packets 0 bytes 0 masquerade comment "!fw4: Intercept-DNS-IPv4-Dmz-Masq"
        }

```

Spamfast, 2023-10-16:
On the 23.05.1 the rules are not seeing any counter hits by the way.

brada4, 2023-10-16:
Do you see anything different in jump chain leading to your rules?

Spamfast, 2023-10-16:
> Do you see anything different in jump chain leading to your rules?

The hijack rule is in chain `dstnat_dmz` which is referenced by chain `dstnat` in rule `iifname "br-dmz" jump dstnat_dmz comment "!fw4: Handle dmz IPv4/IPv6 dstnat traffic"` which is the same in both.

Chain `dstnat` has `type nat hook prerouting priority dstnat; policy accept;` on both as well.

Spamfast, 2023-10-16:
I just tried turning off, in turn, packet steering, h/w flow offloading and s/w flow offloading but that doesn't help either. :-(

raenye, 2023-10-16:
Does it happen with 23.05 release candidates too? (-rc1, -rc2, -rc3, -rc4)

Spamfast, 2023-10-16:
> Does it happen with 23.05 release candidates too? (-rc1, -rc2, -rc3, -rc4)

Sorry, I didn't try with the RCs. They're both production installations so I'm loathe to roll back even if I had the time.

I'd attach `nft --stateless list ruleset` output for them but I'd have to do a lot of redaction. I'm kdiffing that at the moment though.

dhewg, 2023-10-16:
I have a rule to archive the same, but it's `src_ip` and not `src_dip`. And I have `option proto 'tcp udp'`, but that may not matter.
AFAICS that works for me

Spamfast, 2023-10-16:
> I have a rule to archive the same, but it's `src_ip` and not `src_dip`. And I have `option proto 'tcp udp'`, but that may not matter. AFAICS that works for me

Well, the two rules were from an old OpenWrt wiki article on hijacking but https://openwrt.org/docs/guide-user/firewall/fw3_configurations/intercept_dns doesn't work for me either. :-(

dhewg, 2023-10-16:
The wiki is often outdated, it even has a `fw3` in the URL there, but we're at fw4.
Try luci instead, that should be in sync, and I think it's what I've been using to create that rule.

Spamfast, 2023-10-16:
> v23.05 adds 2nd gigabit interface between switch and CPU on most 7621 devices, the name of 'wan' port is usually as printed label like 'internet' on your router, while dsa home ports are something like lan1@eth0 etc Please check carefully if eth0 is dsa cpu port and in this case it does not belong to any firewall zone at all.

I still only have eth0 for the onboard hardware and it is not configured into any OpenWrt `config device` or `config interface` in /etc/config/network.

Doesn't DSA hide all the details of the CPU-MAC to switch stuff, other than showing the eth0 in ifconfig -a with a MTU of 1504?

FYI

```
root@charybdis:~# brctl show
bridge name     bridge id               STP enabled     interfaces
br-network              7fff.02110000eeee       yes             lan4
br-secure               7fff.021100000064       yes             phy0-ap1
                                                        phy1-ap1
                                                        phy0-ap2
                                                        lan4.100
                                                        phy1-ap0
br-dmz          7fff.0211000000c8       yes             lan4.200
                                                        phy0-ap0
                                                        phy0-ap3
br-test         7fff.021100000190       yes             lan4.400
br-dsl          7fff.0211000001f4       yes             lan4.500
                                                        wan
```

Spamfast, 2023-10-16:
> The wiki is often outdated, it even has a `fw3` in the URL there, but we're at fw4. Try luci instead, that should be in sync, and I think it's what I've been using to create that rule.

I am using luci. The question is still why it works with 22.03.5 but not with 22.05.0?

brada4, 2023-10-16:
`ip link` please , you can cover MAC addresses  like 02:24:46:XX:XX:BC

brada4, 2023-10-16:
> 4: lan4@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master br-network state UP qlen 1000
    link/ether 9c:c9:eb:51:7d:1d brd ff:ff:ff:ff:ff:ff
5: lan3@eth0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN qlen 1000
    link/ether 9c:c9:eb:51:7d:1d brd ff:ff:ff:ff:ff:ff
6: lan2@eth0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN qlen 1000
    link/ether 9c:c9:eb:51:7d:1d brd ff:ff:ff:ff:ff:ff
7: lan1@eth0: <BROADCAST,MULTICAST> mtu 1500 qdisc noqueue state DOWN qlen 1000
    link/ether 9c:c9:eb:51:7d:1d brd ff:ff:ff:ff:ff:ff

brand new DSA "second port" eth0 does not belong to any firewall zones
check with 'fw4 print' (it is in variables on top, the offload section is generated from supported devices)

Spamfast, 2023-10-16:
It was using DSA on 22.03.5 and had eth0, lan1/2/3/4 and wan interfaces exactly as it does now. So 'eth0' is not 'brand new'.

```
$ ssh charybdis

BusyBox v1.36.1 (2023-10-08 12:14:50 UTC) built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt 23.05.0, r23497-6637af95aa
 -----------------------------------------------------
```

brada4, 2023-10-16:
It should not be a bridge or zone member on its own.
>  it's now { "eth0", "br-dsl" }

Spamfast, 2023-10-16:
Ah. Sorry. I was leading you astray. It was eth1 which is an RNDIS LTE dongle that is in the { br-dsl, eth1 }.

dave14305, 2023-10-16:
Have you run tcpdump to ensure the queries are arriving as you expect them to?
`tcpdump -i br-dmz -nv port 53 and host 1.1.1.1`

brada4, 2023-10-16:
try to scroll through `fw4 print` with less, the variables of interfaces in the zone are presented in variables on top of the output.

dave14305, 2023-10-16:
You can use nftables trace to see what’s being evaluated.
```
nft add chain inet fw4 mytrace '{ type filter hook prerouting priority -1; policy accept; }'
nft add rule inet fw4 mytrace udp dport 53 ip daddr 1.1.1.1 meta nftrace set 1
nft monitor trace
```
Then run your dig to 1.1.1.1 and watch the output of nft monitor. Once you’re done, cleanup:
```
nft delete chain inet fw4 mytrace
```

Spamfast, 2023-10-17:
I've rolled back to 22.03.5 because I was getting other instabilities.

And (egg on face) it's still not working on this router. The fact that the same rules on the other router do work must be down to some other configuration difference.

Apologies for the waste of time but thanks for the ideas regarding tracing the packets.

I'm closing this now.

closed 2023-10-17 by Spamfast