openwrt/openwrt #13713: zram-swap failed to setup on ath79/generic

InsaneKnight, 2023-10-17:
### Describe the bug

On ath79/generic zram-swap failed unable to initialise lzo-rle compressing backend, while it works on ath79/nand.

### OpenWrt version

r23497-6637af95aa

### OpenWrt target/subtarget

ath79/generic

### Device

TP-Link Archer C7 v2

### Image kind

Self-built image

### Steps to reproduce

Install zram-swap

### Actual behaviour

On ath79/generic zram-swap failed with the following log:
# dmesg|grep -E 'zram|swap'
```
[   43.593202] zram: Added device: zram0
[   53.101306] zram: Cannot initialise lzo-rle compressing backend
[   53.224266] Unable to find swap-space signature
```

### Expected behaviour

while it works on ath79/nand
```
[   12.329518] zram: Added device: zram0
[   21.479994] zram0: detected capacity change from 0 to 120832
[   21.530665] Adding 60412k swap on /dev/zram0.  Priority:100 extents:1 across:60412k SSDsc
```

### Additional info

PACKAGES="uci luci-app-firewall luci-app-opkg luci-app-ddns luci-app-upnp luci-mod-admin-full luci-mod-network luci-mod-status luci-mod-system luci-proto-ipv6 luci-proto-ppp luci-theme-bootstrap rpcd rpcd-mod-file rpcd-mod-iwinfo rpcd-mod-luci rpcd-mod-rrdns uhttpd curl wget-ssl stubby zram-swap drill nano avahi-nodbus-daemon tcpproxy sslh kmod-ath9k wpad-mbedtls pciutils kmod-mtd-rw -wpad-basic-mbedtls -kmod-ath10k-ct -ath10k-firmware-qca988x-ct -kmod-mwlwifi -mwlwifi-firmware-88w8864"

### Diffconfig

```text
can't find file Config.in
can't open file 'tmp/.diffconfig.stage1' at ./scripts/kconfig.pl line 32.
can't find file Config.in
can't open file 'tmp/.diffconfig.stage2' at ./scripts/kconfig.pl line 32.
CONFIG_TARGET_ath79=y
CONFIG_TARGET_ath79_generic=y
CONFIG_TARGET_MULTI_PROFILE=y
CONFIG_TARGET_DEVICE_ath79_generic_DEVICE_8dev_carambola2=y
CONFIG_TARGET_DEVICE_PACKAGES_ath79_generic_DEVICE_8dev_carambola2=""
CONFIG_TARGET_DEVICE_ath79_generic_DEVICE_8dev_lima=y
CONFIG_TARGET_DEVICE_PACKAGES_ath79_generic_DEVICE_8dev_lima=""
CONFIG_TARGET_DEVICE_ath79_generic_DEVICE_adtran_bsap1800-v2=y
CONFIG_TARGET_DEVICE_PACKAGES_ath79_generic_DEVICE_adtran_bsap1800-v2=""
CONFIG_TARGET_DEVICE_ath79_generic_DEVICE_adtran_bsap1840=y
CONFIG_TARGET_DEVICE_PACKAGES_ath79_generic_DEVICE_adtran_bsap1840=""
CONFIG_TARGET_DEVICE_ath79_generic_DEVICE_airtight_c-75=y
CONFIG_TARGET_DEVICE_PACKAGES_ath79_generic_DEVICE_airtight_c-75=""
CONFIG_TARGET_DEVICE_ath79_generic_DEVICE_alcatel_hh40v=y
CONFIG_TARGET_DEVICE_PACKAGES_ath79_generic_DEVICE_alcatel_hh40v=""
CONFIG_TARGET_DEVICE_ath79_generic_DEVICE_alfa-network_ap121f=y
[... 665 lines cut ...]
CONFIG_TARGET_DEVICE_ath79_generic_DEVICE_yuncore_xd3200=y
CONFIG_TARGET_DEVICE_PACKAGES_ath79_generic_DEVICE_yuncore_xd3200=""
CONFIG_TARGET_DEVICE_ath79_generic_DEVICE_yuncore_xd4200=y
CONFIG_TARGET_DEVICE_PACKAGES_ath79_generic_DEVICE_yuncore_xd4200=""
CONFIG_TARGET_DEVICE_ath79_generic_DEVICE_zbtlink_zbt-wd323=y
CONFIG_TARGET_DEVICE_PACKAGES_ath79_generic_DEVICE_zbtlink_zbt-wd323=""
CONFIG_TARGET_DEVICE_ath79_generic_DEVICE_zyxel_nbg6616=y
CONFIG_TARGET_DEVICE_PACKAGES_ath79_generic_DEVICE_zyxel_nbg6616=""
CONFIG_TARGET_DEVICE_ath79_generic_DEVICE_zyxel_nwa1100-nh=y
CONFIG_TARGET_DEVICE_PACKAGES_ath79_generic_DEVICE_zyxel_nwa1100-nh=""
CONFIG_TARGET_DEVICE_ath79_generic_DEVICE_zyxel_nwa1121-ni=y
CONFIG_TARGET_DEVICE_PACKAGES_ath79_generic_DEVICE_zyxel_nwa1121-ni=""
CONFIG_TARGET_DEVICE_ath79_generic_DEVICE_zyxel_nwa1123-ac=y
CONFIG_TARGET_DEVICE_PACKAGES_ath79_generic_DEVICE_zyxel_nwa1123-ac=""
CONFIG_TARGET_DEVICE_ath79_generic_DEVICE_zyxel_nwa1123-ni=y
CONFIG_TARGET_DEVICE_PACKAGES_ath79_generic_DEVICE_zyxel_nwa1123-ni=""
CONFIG_ALL_KMODS=y
CONFIG_ALL_NONSHARED=y
CONFIG_DEVEL=y
CONFIG_TARGET_PER_DEVICE_ROOTFS=y
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

M95D, 2023-10-22:
I don't have lzo-rle as a compressor available in Luci (on mvebu/cortex-a9). Did you reset the configuration, or did you import/keep your old one? lzo-rle may have been removed.
- Select another compressor and see if it works.
- `make kernel_menuconfig` and add lzo-rle (but don't know if Luci is compatible with that).

InsaneKnight, 2023-10-22:
> I don't have lzo-rle as a compressor available in Luci (on mvebu/cortex-a9). Did you reset the configuration, or did you import/keep your old one? lzo-rle may have been removed.
> - Select another [...]
All of my devices is running an image built from imagebuilder, with the same packages. The issue is that zram works on ath79/nand but not on ath79/generic.

M95D, 2023-10-22:
You didn't answer my question. Are you trying to say that zram doesn't work at all, or that it doesn't work with lzo-rle?

InsaneKnight, 2023-11-01:
@M95D kmod-lib-lzo is missing, so this is actually a bug of imagebuilder-23.05.0-ath79-generic, for I did include zram-swap, which depends on kmod-zram, which should depend on kmod-lib-lzo. Imagebuilder does include kmod-zram, but not kmod-lib-lzo.

InsaneKnight, 2023-11-01:
kmod-zram used to depend on kmod-lib-lzo, but not currently, while on ath79/nand, ubifs includes kmod-lib-lzo now.