openwrt/openwrt #13714: DHCP server does not work if LAN interface (br-lan) interface is changed from the default 192.168.1.1/24

vochong, 2023-10-17:
### Describe the bug

With the latest master snapshot, if I change the LAN interface's IP address from the default 192.168.1.1/24 to something else (e.g. 192.168.10.1/24. or 172.30.0.0/24 etc.), client devices connected to the switch ports or WIFI will not get any DHCP IP address.  As soon as I change the IP address back to 192.168.1.1/24,  client devices will get DHCP IP address properly.

Tested on IPQ4019-based routers.

### OpenWrt version

Latest master snapshot (Oct 15 2023)

### OpenWrt target/subtarget

IPQ4019

### Device

Google WIFI

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

vochong, 2024-03-08:
Removing the mesh11sd module fixed the issue.

closed 2024-03-08 by vochong

schinckel, 2025-02-08:
I'm pretty confident I'm seeing this behaviour on OpenWrt 24.10.0 r28427-6df0e3d02a, and I don't have that package install.d