openwrt/openwrt #13716: Netgear WNDR4700 not booting after update to OpenWRT version 23.05.0

kisgezenguz, 2023-10-17:
### Describe the bug

Netgear WNDR4700 switch to TFTP firmware upgrade mode after sysupgrade from 22.03.5 to 23.05.0.

### OpenWrt version

23.05.0

### OpenWrt target/subtarget

apm821xx/nand

### Device

Netgear WNDR4700

### Image kind

Official downloaded image

### Actual behaviour

After sysupgrade router switch to TFTP firmware update mode.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

chunkeey, 2023-10-20:
Yes, looks like something also broke with the factory.img + sysupgrade:

    Bad Magic Number,it is forbidden to be written to flash!!
 
This is coming from the uboot. I think I found a bug in my commit 5dece2d9355a.
The WNDR4700's DTB need to be embedded into an uboot header. But the image
generation scripts don't use the generated file and instead just stick with the "raw"
dtb.

chunkeey, 2023-10-20:
Issue should be addressed in -main by:
https://github.com/openwrt/openwrt/commit/d6a11833ad67c33ad10dadf396f6c30bb44ef30f

if -snapshot builds prove to be good, I'll backport it to openwrt-23.05 and close this bug.,

kisgezenguz, 2023-10-23:
> Issue should be addressed in -main by: [d6a1183](https://github.com/openwrt/openwrt/commit/d6a11833ad67c33ad10dadf396f6c30bb44ef30f)
> 
> if -snapshot builds prove to be good, I'll backport it to openwrt-23.05 and close this bug.,

I try the latest snap, and I could confirm the fixes are working. My WNDR4700 starting well. 
Thank you for the support.

`## Booting kernel from Legacy Image at 00b00000 ...
   Image Name:   POWERPC OpenWrt Linux-5.15.135
   Image Type:   PowerPC Linux Kernel Image (lzma compressed)
   Data Size:    3247343 Bytes =  3.1 MB
   Load Address: 00000000
   Entry Point:  00000000
   Verifying Checksum ... OK
## Flattened Device Tree blob at 00ae0040
   Booting using the fdt blob at 0xae0040
   Uncompressing Kernel Image ... OK
[    0.000000] Activating Kernel Userspace Execution Prevention
[    0.000000] Linux version 5.15.135 (builder@buildhost) (powerpc-openwrt-linux-musl-gcc (OpenWrt GCC 12.3.0 r24195-bb629a630f) 12.3.0, GNU ld (GNU Binutils) 2.40.0) #0 Mon Oct 23 09:01:03 2023

********

BusyBox v1.36.1 (2023-10-23 09:01:03 UTC) built-in shell (ash)

/bin/ash: can't access tty; job control turned off
  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt SNAPSHOT, r24195-bb629a630f
 -----------------------------------------------------`

commit 03cb2d63 (2023-10-21): apm821xx: WNDR4700: fix broken sysupgrade, factory images

prepend-dtb got extended to handle the Meraki devices too,
the problem here was that the Netgear WNDR4700 expects an
u-boot header in front of the DTB, whereas Meraki devices
don't.

Since the header was dropped, the WNDR4700's uboot started
to complain:
  Bad Magic Number,it is forbidden to be written to flash!!

when flashing the factory.img since it expects an u-boot
header there.

Fixes: 5dece2d9355a ("apm821xx: switch over from DTB_SIZE to DEVICE_DTC_FLAGS")
Fixes: #13716
Reported-by: @kisgezenguz
Reported-by: Tamas Szabo

(cherry [... cut]
closed 2024-03-08 by chunkeey