openwrt/openwrt #13717: IPQ807x/ATH11k 5ghz vht80 not working channels 52-64 , 132-144 in RU zone

lmahmutov, 2023-10-17:
### Describe the bug

Setup from luci show all OK. Hostapd started and AP viewed on wifi scanner. But cannot connect to AP.
[image]

### OpenWrt version

23.05

### OpenWrt target/subtarget

IPQ807x

### Device

ax3600

### Image kind

Official downloaded image

### Steps to reproduce

Setup any channel from diapasone 52-64, 132-144 with vht80 and Reg zone RU

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

lmahmutov, 2023-10-17:
IW LIST **https://pastebin.com/w7iCU3Lf**
cat/etc/config/wireless **https://pastebin.com/aQJ25i8s**

brada4, 2023-10-17:
You must select country for all 3 adapters, the 2 in pic seem OP.
https://git.kernel.org/pub/scm/linux/kernel/git/sforshee/wireless-regdb.git/tree/db.txt?id=HEAD#n1547

lmahmutov, 2023-10-18:
screen croped ssory ) setuped for all adapters
@robimarko need your help

robimarko, 2023-10-18:
What does logread say when you try to use those channels?

lmahmutov, 2023-10-18:
> What does logread say when you try to use those channels?

nothing.... if i enable -d on hostapd i see only beacon

kirdesde, 2023-10-18:
Don't run hostapd via cli.

wifi up

logread|grep hostapd

And then we will ses the hostapd related logging

brada4, 2023-10-18:
Radio0 does not have country, so it is in no-ir mode at best, i.e needing other transmitter in channel before bringing up transmission.

lmahmutov, 2023-10-18:
another AP with ipq8074

`config wifi-device 'radio0'
	option type 'mac80211'
	option path 'platform/soc/c000000.wifi'
	option channel '60'
	option band '5g'
	option htmode 'HE80'
	option he_bss_color '8'
	option hwmode '11a'
	option rssi_reject_assoc_rssi '-96'
	option he_su_beamformee '1'
	option country 'RU'
	option cell_density '0'

config wifi-device 'radio1'
	option type 'mac80211'
	option path 'platform/soc/c000000.wifi+1'
	option channel 'auto'
	option band '2g'
	option htmode 'HT20'
	option noscan '0'
	option hwmode '11g'
	option txpower '30'
	option rssi_reject_assoc_rssi '-96'
	option country 'RU'

config wifi-iface 'wifinet0'
	option device 'radio0'
	option mode 'ap'
	option ssid 'OpenWrt'
	option encryption 'none'
	option network 'lan'

`

lmahmutov, 2023-10-18:
> Don't run hostapd via cli.

hostapd runnig via wpad 
i changed /etc/init.d/wpad and add -d for hostapd. And changed loglevel in menuconfig. 
In logread full debug info

lmahmutov, 2023-10-18:
issue is only in dfs zone rule 2 and rule 4 ...

kirdesde, 2023-10-18:
I can't replicate your issue on a DL-WRX36.

Wifi AP with country RU and channel 60 is working fine.
According to https://en.wikipedia.org/wiki/List_of_WLAN_channels channel 60 is not a DFS channel, so this isn't related to DFS.

Please revert your change in /etc/init.d/wpad and post a new log.

lmahmutov, 2023-10-18:
> I can't replicate your issue on a DL-WRX36.
> 
Normally connected ?

brada4, 2023-10-18:
https://git.kernel.org/pub/scm/linux/kernel/git/sforshee/wireless-regdb.git/tree/db.txt?id=HEAD#n1547
the dfs scan starting in log is absolutely unnecessary.

kirdesde, 2023-10-18:
> https://git.kernel.org/pub/scm/linux/kernel/git/sforshee/wireless-regdb.git/tree/db.txt?id=HEAD#n1547 the dfs scan starting in log is absolutely unnecessary.

What DFS scan are you referring? There is no "DFS-CAC-START" in the provided logfile.

Only this line: DFS 0 channels required radar detection

Also the provided screenshot from iw reg get shows no signs of DFS for the 5 Ghz radio.

lmahmutov, 2023-10-31:
> I can't replicate your issue on a DL-WRX36.
> 
> Wifi AP with country RU and channel 60 is working fine.
> 
> According to https://en.wikipedia.org/wiki/List_of_WLAN_channels channel 60 is not a DFS channel in RU, so this isn't related to DFS
> 
> Please revert your change in /etc/init.d/wpad and post a new log.

can you connect any device to AP at this channel?

kirdesde, 2023-10-31:
I can connect wifi devices

lmahmutov, 2023-11-01:
> I can connect wifi devices

can you paste logs and 
iw reg get