openwrt/openwrt #13720: "route6" option "source" is broken

notgood, 2023-10-17:
### Describe the bug

I want to specify preferred source IPv6 adress for static route, but there is no way to do it at the moment.

"route6" option "source" according to Luci help: "Specifies the preferred source address when sending to destinations covered by the target"

"route6" option "source" according to wiki: "The route source address in source-address dependent routes. It's called “from” in the ip command. "

It was fixed for IPv4 years ago, but kept broken for IPv6 for backwards compatibility purposes.
https://github.com/openwrt/netifd/commit/25289c946427db676d41c0fa4e6c01ade2f43b66

### OpenWrt version

23.05

### OpenWrt target/subtarget

all

### Device

all

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-10-18:
Whar 'ip -6 route` command you want to get out of LuCI

notgood, 2023-10-18:
> Whar 'ip -6 route` command you want to get out of LuCI

ip -6 route add default dev [DEV] src [IPV6 ADDR]

raenye, 2023-10-20:
Is this achievable via manual editing of `/etc/config/network`? (and if so, should be reported as a LuCI bug)

notgood, 2023-10-20:
> Is this achievable via manual editing of `/etc/config/network`? (and if so, should be reported as a LuCI bug)

No, it's not.

Routes are set by netifd, simple search confirms that netifd only supports setting RTA_PREFSRC for IPv4 routes.
https://github.com/search?type=code&q=RTA_PREFSRC+repo%3Aopenwrt%2Fnetifd

brada4, 2023-10-20:
in the else {} branch it sets 'src' option for !af_inet

notgood, 2023-10-20:
> in the else {} branch it sets 'src' option for !af_inet

...and it sets completely different flag, RTA_SRC != RTA_PREFSRC.

Please read commit message linked in first post.

uvok, 2025-06-17:
I would be interested in contributing to help fix this. Since the `source` config option behavior needs to be kept for backwards-compatible behavior, I think a new option needs to be added? (maybe `prefsrc` or `pref_source`? - which could also be used for IPv4 routes). I can imagine the procedure for adding new options may involve some discussion and consensus, though?