openwrt/openwrt #13724: Redmi AX6000 is not able to connect to tftp for recovery

ram108, 2023-10-18:
### Describe the bug

We have many reports on the openwrt forum, as well as on the 4pda forum about not being able to connect to the tftp server to download recovery.

The issue concerns latest RCs and release. 

There is a report of successful connection to tftp on snapshots before RC2.

Source: https://forum.openwrt.org/t/add-openwrt-support-for-xiaomi-redmi-ax6000/125008/1039?u=ram108

### OpenWrt version

r23497-6637af95aa

### OpenWrt target/subtarget

mediatek/filogic

### Device

Xiaomi Redmi Router AX6000 (OpenWrt U-Boot layout)

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

solomonricky, 2023-10-18:
Same here. The router UART can be access just the network is unreachable at all. Even changed to other `ipaddr` and `serverip` still can't ping from both side. I tried every single Ethernet port including WAN still unreachable.

AmirulAndalib, 2023-10-18:
same for me too, tftp server is broken, tried various ip configurations. I have attached the tftp uart logs below:

[image]

note: this is openwrt ubootmod layout. Tftp for stock uboot is working fine...

AmirulAndalib, 2023-10-18:
so for the old version users. How are we gonna be fixing this bricked router? Any suggestions would be helpful. Thanks

ram108, 2023-10-19:
So, on the 4pda forum today we have confirmation that tftp works on bootloader 23.05.0-rc2.

The script below downloads and installs the bootloader from firmware 23.05.0-rc2 (r23228-cd17d8df2a).

The script will be valid until the issue of broken tftp will be fixed. After that you will need to re-flash the bootloader from the latest firmware version.

Warning! You must understand what you are doing. This code below can turn your router into the ultimate brick.

```
opkg update && opkg install kmod-mtd-rw
insmod /lib/modules/$(uname -r)/mtd-rw.ko i_want_a_brick=1

wget https://downloads.openwrt.org/releases/23.05.0-rc2/targets/mediatek/filogic/openwrt-23.05.0-rc2-mediatek-filogic-xiaomi_redmi-router-ax6000-ubootmod-preloader.bin -O /tmp/ax6000-ubootmod-preloader.bin

wget https://downloads.openwrt.org/releases/23.05.0-rc2/targets/mediatek/filogic/openwrt-23.05.0-rc2-mediatek-filogic-xiaomi_redmi-router-ax6000-ubootmod-bl31-uboot.fip -O /tmp/ax6000-ubootmod-bl31-uboot.fip

mtd write /tmp/ax6000-ubootmod-preloader.bin BL2
mtd write /tmp/ax6000-ubootmod-bl31-uboot.fip FIP

rm /tmp/ax6000-ubootmod-preloader.bin /tmp/ax6000-ubootmod-bl31-uboot.fip
```

Reboot.

AmirulAndalib, 2023-10-19:
> So, on the 4pda forum today we have confirmation that tftp works on bootloader 23.05.0-rc2.
> 
> The script below downloads and installs the bootloader from firmware 23.05.0-rc2 (r23228-cd17d8df2a).
> 
> The script will be valid until the issue of broken tftp will be fixed. After that you will need to re-flash the bootloader from the latest firmware version.
> 
> Warning! You must understand what you are doing. This code below can turn your router into the ultimate brick.
> 
> ```
> opkg update && opkg install wget kmod-mtd-rw
> insmod /lib/modules/$(uname -r)/mtd-rw.ko i_want_a_brick=1
> 
> wget https://downloads.openwrt.org/releases/23.05.0-rc2/targets/mediatek/filogic/openwrt-23.05.0-rc2-mediatek-filogic-xiaomi_redmi-router-ax6000-ubootmod-preloader.bin -O /tmp/ax6000-ubootmod-preloader.bin
> wget https://downloads.openwrt.org/releases/23.05.0-rc2/targets/mediatek/filogic/openwrt-23.05.0-rc2-mediatek-filogic-xiaomi_redmi-router-ax6000-ubootmod-bl31-uboot.fip -O /tmp/ax6000-ubootmod-bl31-uboot.fip
> 
> mtd write /tmp/ax6000-ubootmod-preloader.bin BL2
> mtd write /tmp/ax6000-ubootmod-bl31-uboot.fip FIP
> 
> rm /tmp/ax6000-ubootmod-preloader.bin /tmp/ax6000-ubootmod-bl31-uboot.fip
> ```
> 
> Reboot and test tftp.

my device is currently bricked and no leds are on ever since. But tftp and uboot are active. I am on an old version of this os and my tftp is broken like the others. So i need a bit of advice and help on how to recover from this situation. Or is it not possible?

AmirulAndalib, 2023-10-21:
any reply?

ram108, 2023-10-21:
> any reply?

If your router is no longer booting and there is no way to boot via tftp, then the only option left is to manually program the chip using uart. There are not many detailed instructions. I am not an expert in this field.

We have several reports on the Russian forum 4pda on manual flashing of the chip.

HZFrodo, 2023-10-22:
@dangowrt 
Please check this issue.
23.05.0-rc2 with U-Boot 2023.04, Redmi AX6000 U-Boot TFTP works well
23.05.0-rc3 with U-Boot 2023.07.02, ping and TFTP do not work at all.

HZFrodo, 2023-10-27:
Fixed in #13809

ram108, 2023-10-28:
I can confirm that this fix solves the TFTP issue. We have built a bootloader with the fix and tested its work. We were able to boot via TFTP. Details (in Russian): https://4pda.to/forum/index.php?showtopic=1045443&st=1940#entry125873692

closed 2023-10-28 by ram108

hauke, 2023-10-28:
The xiaomi mi router wr30u probably has the same bug, see:  https://github.com/openwrt/openwrt/blob/main/package/boot/uboot-mediatek/patches/434-add-xiaomi_mi-router-wr30u.patch#L270

AmirulAndalib, 2023-11-30:
Chill

-- 
-------------------------------------------------------------------------------------------------------------------
*Official email communication will always be from an address ending with
@gmail.com <http://gmail.com/> Please confirm this before responding to an
email. Email communication is not secure and therefore it can be
intercepted in transit by hackers. Please do not share any confidential
information over the email to avoid any type of breach of privacy and
financial loss.*

*Note: To protect against computer viruses, e-mail programs may
preventsending or receiving certain types of file attachments. Check your
e-mailsecurity settings to determine how attachments are handled.*

*You are therefore kindly requested to verify the attachment/mails (if
sent) properly.*

*Thank for your kind cooperation.*

*Note: This is an automatic safety check signature from the sender.*
-------------------------------------------------------------------------------------------------------------------

commit 03987d2d (2023-10-28): uboot-mediatek: Sync phy-mode for Xiaomi Redmi Router AX6000

Commit 572ea6807053 ("uboot-mediatek: add patches for MT7988 and
builds for RFB") renamed HSGMII to 2500basex, but forgot to update
the dts of Redmi Router AX6000, makes the network unusable.
This patch makes the network usable again.

Fixes: #13724
Fixes: 572ea6807053 ("uboot-mediatek: add patches for MT7988 and builds for RFB")
commit 620721f6 (2023-10-30): uboot-mediatek: Sync phy-mode for Xiaomi Redmi Router AX6000

Commit 572ea6807053 ("uboot-mediatek: add patches for MT7988 and
builds for RFB") renamed HSGMII to 2500basex, but forgot to update
the dts of Redmi Router AX6000, makes the network unusable.
This patch makes the network usable again.

Fixes: #13724
Fixes: 572ea6807053 ("uboot-mediatek: add patches for MT7988 and builds for RFB")

(cherry picked from commit 03987d2d11c4954e3e8afa3d44ba6213e774c742)