openwrt/openwrt #13726: ath11k module crashes on xiaomi ax3600 (IPQ8071A)

piontec, 2023-10-18:
### Describe the bug

After roughly 1 day uptime, the wi-fi driver crashes and renders radio interface unusable. Other than that, the router is working and accessible over eth connection.

### OpenWrt version

r23497-6637af95aa

### OpenWrt target/subtarget

ipq807x/generic

### Device

Xiaomi AX3600

### Image kind

Official downloaded image

### Steps to reproduce

I can't point to anything triggering the bug. Router runs, then suddenly wi-fi disappears.

### Actual behaviour

Wi-fi driver crashes

### Expected behaviour

It doesn't

### Additional info

Found the following in `dmesg`:

```
[114743.198424] qcom-q6v5-wcss-pil cd00000.q6v5_wcss: fatal error received: 
[114743.198424] QC Image Version: QC_IMAGE_VERSION_STRING=WLAN.HK.2.9.0.1-01385-QCAHKSWPL_SILICONZ-1
[114743.198424] Image Variant : IMAGE_VARIANT_STRING=8074.wlanfw.eval_v2Q
[114743.198424] 
[114743.198424] wal_peer_control.c:2870 Assertion is_graceful_to_handle failedparam0 :zero, param1 :zero, param2 :zero.
[114743.198424] Thread ID      : 0x00000069  Thread name    : WLAN RT0  Process ID     : 0
[114743.198424] Register:
[114743.198424] SP : 0x4bfb9230
[114743.198424] FP : 0x4bfb9238
[114743.198424] PC : 0x4b1080c4
[114743.198424] SSR : 0x00000008
[114743.198424] BADVA : 0x00020000
[114743.198424] LR : 0x4b107860
[114743.198424] 
[114743.198424] Stack Dump
[114743.198424] from : 0x4bfb9230
[114743.198424] to   : 0x4bfb9aa8
[114743.198424] 
[114743.249251] remoteproc remoteproc0: crash detected in cd00000.q6v5_wcss: type fatal error
[114743.271450] remoteproc remoteproc0: handling crash #1 in cd00000.q6v5_wcss
[114743.279463] remoteproc remoteproc0: recovering cd00000.q6v5_wcss
[114743.312391] remoteproc remoteproc0: stopped remote processor cd00000.q6v5_wcss
[114743.312957] ath11k c000000.wifi: failed to transmit frame -108
[114743.329211] ath11k c000000.wifi: failed to transmit frame -108
[114743.595680] ath11k c000000.wifi: failed to find peer 30:03:c8:4b:fb:97 on vdev 3 after creation
[114743.595736] ath11k c000000.wifi: failed to find peer vdev_id 3 addr 30:03:c8:4b:fb:97 in delete
[114743.603574] ath11k c000000.wifi: failed peer 30:03:c8:4b:fb:97 delete vdev_id 3 fallback ret -22
[114743.612357] ath11k c000000.wifi: Failed to add peer: 30:03:c8:4b:fb:97 for VDEV: 3
[114743.621095] ath11k c000000.wifi: Failed to add station: 30:03:c8:4b:fb:97 for VDEV: 3
[114743.685887] ath11k c000000.wifi: failed to update rx tid queue, tid 0 (-108)
[114743.685934] ath11k c000000.wifi: failed to update reo for rx tid 0: -108
[114743.692067] phy1-ap0: HW problem - can not stop rx aggregation for 30:03:c8:4b:fb:97 tid 0
[114743.698837] ath11k c000000.wifi: failed to update rx tid queue, tid 0 (-108)
[114743.707004] phy1-ap0: HW problem - can not stop rx aggregation for 30:03:c8:4b:fb:97 tid 7
[114748.575672] qcom-q6v5-wcss-pil cd00000.q6v5_wcss: start timed out
[114748.575718] remoteproc remoteproc0: can't start rproc cd00000.q6v5_wcss: -110
[114748.815481] ath11k_warn: 1 callbacks suppressed
[114748.815498] ath11k c000000.wifi: failed to flush transmit queue, data pkts pending 5
[114748.819255] ath11k c000000.wifi: failed to send WMI_PEER_DELETE cmd
[114748.826936] ath11k c000000.wifi: failed to delete peer vdev_id 0 addr 30:03:c8:4b:fb:97 ret -108
[114748.833363] ath11k c000000.wifi: Failed to delete peer: 30:03:c8:4b:fb:97 for VDEV: 0
[114748.842136] ath11k c000000.wifi: Found peer entry 9c:9d:7e:75:0a:d9 n vdev 0 after it was supposedly removed
[114748.850010] ------------[ cut here ]------------
[114748.859890] WARNING: CPU: 0 PID: 1960 at 0xffffffc000bcba70 [mac80211@00000000b88c6988+0x85000]
[114748.864586] Modules linked in: pppoe ppp_async nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet ath11k_ahb ath11k ath10k_pci ath10k_core ath wireguard pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mac80211 libchacha20poly1305 iptable_mangle iptable_filter ipt_REJECT ipt_ECN ip_tables chacha_neon cfg80211 xt_time xt_tcpudp xt_tcpmss xt_statistic xt_multiport xt_mark xt_mac xt_limit xt_length xt_hl xt_ecn xt_dscp xt_comment xt_TCPMSS xt_LOG xt_HL xt_DSCP xt_CLASSIFY x_tables slhc sch_cake qrtr_smd qrtr qmi_helpers poly1305_neon nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 libcurve25519_generic libcrc32c libchacha crc_ccitt compat sch_tbf sch_ingress sch_htb sch_hfsc em_u32 cls_u32
[114748.864838]  cls_route cls_matchall cls_fw cls_flow cls_basic act_skbedit act_mirred act_gact ifb ip6_udp_tunnel udp_tunnel sha512_generic seqiv jitterentropy_rng drbg michael_mic kpp hmac cmac leds_gpio xhci_plat_hcd xhci_pci xhci_hcd dwc3 dwc3_qcom qca_nss_dp qca_ssdk gpio_button_hotplug ext4 mbcache jbd2 aquantia hwmon crc32c_generic
[114748.967432] CPU: 0 PID: 1960 Comm: hostapd Not tainted 5.15.134 #0
[114748.989667] Hardware name: Xiaomi AX3600 (DT)
[114748.995915] pstate: 60400005 (nZCv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[114749.000433] pc : 0xffffffc000bcba70 [mac80211@00000000b88c6988+0x85000]
[114749.007637] lr : 0xffffffc000bcba6c [mac80211@00000000b88c6988+0x85000]
[114749.014321] sp : ffffffc00d423840
[114749.020999] x29: ffffffc00d423840 x28: ffffff8001f8d240 x27: ffffffc00d423dd0
[114749.024217] x26: ffffff8001c70080 x25: ffffffc008b760c0 x24: ffffffc000b8e000
[114749.031423] x23: ffffffc00d423a38 x22: ffffff8009b48900 x21: ffffff80077b3aa8
[114749.038628] x20: ffffff8005d60880 x19: ffffff80077b3000 x18: ffffffc008a77ba0
[114749.045832] x17: 302076656476206e x16: 2039643a61303a35 x15: 000000000000081c
[114749.053037] x14: 0000000000000004 x13: 0000000000000004 x12: 00000000881ddb71
[114749.060242] x11: ffffffc008b62158 x10: 0000000000000002 x9 : 0000000000000000
[114749.067446] x8 : ffffffc00d423420 x7 : 00000000de49cf23 x6 : 0000000000008e20
[114749.074652] x5 : 0000000000000000 x4 : 0000000000000000 x3 : ffffff8001f8d240
[114749.081856] x2 : 0000000000000000 x1 : ffffff8001f8d240 x0 : 00000000ffffff94
[114749.089061] Call trace:
[114749.096258]  0xffffffc000bcba70 [mac80211@00000000b88c6988+0x85000]
[114749.098955]  0xffffffc000bcbd34 [mac80211@00000000b88c6988+0x85000]
[114749.105293]  0xffffffc000be423c [mac80211@00000000b88c6988+0x85000]
[114749.111629]  0xffffffc000b5e1a8 [cfg80211@00000000b10ed3da+0x45000]
[114749.117965]  0xffffffc00863fb9c
[114749.124296]  0xffffffc008640028
[114749.127508]  0xffffffc00863ec3c
[114749.130720]  0xffffffc00863f4a8
[114749.133933]  0xffffffc00863e3d8
[114749.137143]  0xffffffc00863e674
[114749.140355]  0xffffffc00859be68
[114749.143567]  0xffffffc00859e848
[114749.146779]  0xffffffc00859e998
[114749.149991]  0xffffffc00859ea24
[114749.153203]  0xffffffc0080230ac
[114749.156415]  0xffffffc0080231c0
[114749.159627]  0xffffffc0087927b8
[114749.162838]  0xffffffc0087934a8
[114749.166049]  0xffffffc0080115f4
[114749.169261] ---[ end trace 2fa789978170a047 ]---
[114749.174433] ath11k c000000.wifi: failed to send WMI_PDEV_BSS_CHAN_INFO_REQUEST cmd
[114749.177154] ath11k c000000.wifi: failed to send pdev bss chan info request
[114749.184757] ath11k c000000.wifi: failed to send WMI_PDEV_SET_PARAM cmd
[114749.191515] ath11k c000000.wifi: Failed to set beacon mode for VDEV: 4
[114749.198118] ath11k c000000.wifi: failed to send WMI_BCN_TMPL_CMDID
[114752.575458] br-lan: port 8(phy1-ap1.sta1) neighbor 7fff.50:d2:f5:25:b2:20 lost
[114752.575530] br-lan: topology change detected, propagating
[114755.208213] ath11k_warn: 113 callbacks suppressed
[114755.208237] ath11k c000000.wifi: failed to send WMI_PDEV_BSS_CHAN_INFO_REQUEST cmd
[114755.211928] ath11k c000000.wifi: failed to send pdev bss chan info request
[114755.220045] ath11k c000000.wifi: failed to send WMI_PDEV_SET_PARAM cmd
[114755.226461] ath11k c000000.wifi: Failed to set beacon mode for VDEV: 4
[114755.233123] ath11k c000000.wifi: failed to send WMI_BCN_TMPL_CMDID
[114755.239650] ath11k c000000.wifi: failed to submit beacon template command: -108
[114755.245886] ath11k c000000.wifi: failed to update bcn template: -108
[114755.253507] ath11k c000000.wifi: failed to send WMI_VDEV_SET_PARAM_CMDID
[114755.259853] ath11k c000000.wifi: failed to set BA BUFFER SIZE 256 for vdev: 4
[114755.266623] ath11k c000000.wifi: failed to send WMI_VDEV_SET_PARAM_CMDID
[114761.280412] ath11k_warn: 108 callbacks suppressed
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-10-18:
Disable firewall flow table acceleration and measure again. It puts packets in output queue and may race with sta queue flush?

piontec, 2023-10-18:
Which option do you mean exactly? Is it `Software flow offloading`?

brada4, 2023-10-19:
Yes, that  one.

piontec, 2023-10-23:
This seems to help, no crash for 3 days. But transfer speed is lower now, so I guess it's either higher speed or SQM :|

brada4, 2023-10-24:
May not crash eithermway though..
Firewall flow offload bypasses qdisc, including one configured by sqm. Same must happen with default qdisc coupled  with offload alike. 
If you can accomodate radio downtime switch back pn the offload and try to capture logread events leading to crash.

I suspect following state flow:
- sta disassociates
- driver cleans pending packets to sta
-  offload writes one on queue in the meantime anyway
-  driver finds 'should never happen' packet left in queue and crashes

piontec, 2023-10-25:
After enabling it back, wi-fi becomes unstable - I'm also getting transient periods when nothing works (no connectivity), but my laptop is reconnected (so the driver doesn't crash permanently). 

`dmesg` in this situation has this:

```
[539781.225242] nss-dp 3a001400.dp3 lan1: PHY Link is down
[539781.225553] br-lan: port 1(lan1) entered disabled state
[575286.847668] nss-dp 3a001400.dp3 lan1: PHY Link up speed: 1000
[575286.847755] br-lan: port 1(lan1) entered blocking state
[575286.852412] br-lan: port 1(lan1) entered listening state
[575295.007316] br-lan: port 1(lan1) entered learning state
[575303.327319] br-lan: port 1(lan1) entered forwarding state
[575303.327367] br-lan: topology change detected, sending tcn bpdu
[575348.207640] nss-dp 3a001400.dp3 lan1: PHY Link is down
[575348.207953] br-lan: port 1(lan1) entered disabled state
[575354.447738] nss-dp 3a001400.dp3 lan1: PHY Link up speed: 1000
[575354.447827] br-lan: port 1(lan1) entered blocking state
[575354.452482] br-lan: port 1(lan1) entered listening state
[575362.847369] br-lan: port 1(lan1) entered learning state
[575371.167370] br-lan: port 1(lan1) entered forwarding state
[575371.167418] br-lan: topology change detected, sending tcn bpdu
[575937.647802] ath11k c000000.wifi: failed to flush transmit queue, data pkts pending 2
[576249.257997] br-lan: port 8(phy1-ap1.sta1) neighbor 7fff.50:d2:f5:25:b2:20 lost
[576249.258062] br-lan: topology change detected, propagating
[576281.888023] br-lan: port 8(phy1-ap1.sta1) neighbor 7fff.50:d2:f5:25:b2:20 lost
[576281.888104] br-lan: topology change detected, propagating
[576443.728166] ath11k c000000.wifi: failed to flush transmit queue, data pkts pending 1
[576895.168498] ath11k c000000.wifi: failed to flush transmit queue, data pkts pending 2
[578880.689972] ath11k c000000.wifi: failed to flush transmit queue, data pkts pending 1
[591156.220726] nss-dp 3a001400.dp3 lan1: PHY Link is down
[591156.221214] br-lan: port 1(lan1) entered disabled state
[592628.781600] br-lan: port 8(phy1-ap1.sta1) neighbor 7fff.50:d2:f5:25:b2:20 lost
[592628.781684] br-lan: topology change detected, propagating
[593463.982655] nss-dp 3a001400.dp3 lan1: PHY Link up speed: 1000
[593463.982743] br-lan: port 1(lan1) entered blocking state
[593463.987402] br-lan: port 1(lan1) entered listening state
[593465.022638] nss-dp 3a001400.dp3 lan1: PHY Link is down
[593465.023131] br-lan: port 1(lan1) entered disabled state
[593467.102650] nss-dp 3a001400.dp3 lan1: PHY Link up speed: 1000
[593467.102739] br-lan: port 1(lan1) entered blocking state
[593467.107397] br-lan: port 1(lan1) entered listening state
[593475.502301] br-lan: port 1(lan1) entered learning state
[593483.822305] br-lan: port 1(lan1) entered forwarding state
[593483.822363] br-lan: topology change detected, sending tcn bpdu
[595558.784019] ath11k c000000.wifi: failed to flush transmit queue, data pkts pending 1
``` 

and `logread` (filtered by my MAC)

```
ed Oct 25 14:28:45 2023 daemon.info hostapd: phy1-ap3: STA 7c:b2:7d:47:22:23 WPA: pairwise key handshake completed (RSN)
Wed Oct 25 14:28:45 2023 daemon.notice hostapd: phy1-ap3: EAPOL-4WAY-HS-COMPLETED 7c:b2:7d:47:22:23
Wed Oct 25 14:28:45 2023 daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan) 192.168.3.200 7c:b2:7d:47:22:23
Wed Oct 25 14:28:45 2023 daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan) 192.168.3.200 7c:b2:7d:47:22:23 t495
Wed Oct 25 14:29:08 2023 daemon.notice hostapd: phy1-ap3: AP-STA-DISCONNECTED 7c:b2:7d:47:22:23
Wed Oct 25 14:29:14 2023 daemon.info hostapd: phy1-ap3: STA 7c:b2:7d:47:22:23 IEEE 802.11: authenticated
Wed Oct 25 14:29:14 2023 daemon.info hostapd: phy1-ap3: STA 7c:b2:7d:47:22:23 IEEE 802.11: associated (aid 1)
Wed Oct 25 14:29:14 2023 daemon.notice hostapd: phy1-ap3: AP-STA-CONNECTED 7c:b2:7d:47:22:23 auth_alg=open
Wed Oct 25 14:29:14 2023 daemon.info hostapd: phy1-ap3: STA 7c:b2:7d:47:22:23 WPA: pairwise key handshake completed (RSN)
Wed Oct 25 14:29:14 2023 daemon.notice hostapd: phy1-ap3: EAPOL-4WAY-HS-COMPLETED 7c:b2:7d:47:22:23
Wed Oct 25 14:29:15 2023 daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan) 192.168.3.200 7c:b2:7d:47:22:23
Wed Oct 25 14:29:15 2023 daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan) 192.168.3.200 7c:b2:7d:47:22:23 t495
Wed Oct 25 14:29:23 2023 daemon.notice hostapd: phy1-ap3: AP-STA-DISCONNECTED 7c:b2:7d:47:22:23
Wed Oct 25 14:29:29 2023 daemon.info hostapd: phy1-ap3: STA 7c:b2:7d:47:22:23 IEEE 802.11: authenticated
Wed Oct 25 14:29:29 2023 daemon.info hostapd: phy1-ap3: STA 7c:b2:7d:47:22:23 IEEE 802.11: associated (aid 1)
Wed Oct 25 14:29:29 2023 daemon.notice hostapd: phy1-ap3: AP-STA-CONNECTED 7c:b2:7d:47:22:23 auth_alg=open
Wed Oct 25 14:29:29 2023 daemon.info hostapd: phy1-ap3: STA 7c:b2:7d:47:22:23 WPA: pairwise key handshake completed (RSN)
Wed Oct 25 14:29:29 2023 daemon.notice hostapd: phy1-ap3: EAPOL-4WAY-HS-COMPLETED 7c:b2:7d:47:22:23
Wed Oct 25 14:29:29 2023 daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan) 192.168.3.200 7c:b2:7d:47:22:23
Wed Oct 25 14:29:29 2023 daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan) 192.168.3.200 7c:b2:7d:47:22:23 t495
Wed Oct 25 14:29:53 2023 daemon.notice hostapd: phy1-ap3: AP-STA-DISCONNECTED 7c:b2:7d:47:22:23
Wed Oct 25 14:30:07 2023 daemon.info hostapd: phy1-ap3: STA 7c:b2:7d:47:22:23 IEEE 802.11: authenticated
Wed Oct 25 14:30:07 2023 daemon.info hostapd: phy1-ap3: STA 7c:b2:7d:47:22:23 IEEE 802.11: associated (aid 1)
[... cut ...]
Wed Oct 25 14:30:53 2023 daemon.notice hostapd: phy1-ap3: AP-STA-DISCONNECTED 7c:b2:7d:47:22:23
Wed Oct 25 14:30:59 2023 daemon.info hostapd: phy1-ap3: STA 7c:b2:7d:47:22:23 IEEE 802.11: authenticated
Wed Oct 25 14:30:59 2023 daemon.info hostapd: phy1-ap3: STA 7c:b2:7d:47:22:23 IEEE 802.11: associated (aid 1)
Wed Oct 25 14:30:59 2023 daemon.notice hostapd: phy1-ap3: AP-STA-CONNECTED 7c:b2:7d:47:22:23 auth_alg=open
Wed Oct 25 14:30:59 2023 daemon.info hostapd: phy1-ap3: STA 7c:b2:7d:47:22:23 WPA: pairwise key handshake completed (RSN)
```

brada4, 2023-10-25:
You can check offload interfaces in `fw4 print | less` and sill use hw offload before no offload at all likely.
How dramatic is the slowdown? 10% is expected with sqm unless offload bypasses scheduler discipline (and arp and route etc)

uklatt, 2023-10-27:
I have the same problem with **Software Flow Offloading** disabled and SNAPSHOT r24194, firmware WLAN.HK.2.9.0.1-01890-QCAHKSWPL_SILIONZ-1.
Also older Openwrt versions crash.
It seems to be a firmware problem of the Qualcom firmware image.
See also: https://forum.openwrt.org/t/xiaomi-ax3600-ath11k-firmware-crash-qcom-q6v5-wcss-pil-cd00000-q6v5-wcss-fatal-error-received

piontec, 2023-10-28:
Yeah, I think it's something between 10-20%, I would have to measure exactly. Should I close this and just keep offloading disabled?

brada4, 2023-10-28:
You can try hardware offliad ensuring it is not enabled on wifi, but yes, better without offload if it is found unstable.

URBANsUNITED, 2023-11-01:
Hi!

I am having the same problem.
Nearly every day the wifi crashes and I have to reboot the router.
In my settings was "Software flow offloading" already disabled! Firewall, DHCP and DNS are disabled on my side.

This is part of my log as it repeats after the wifi crash.

> Sat Oct 28 17:49:51 2023 daemon.err hostapd: Failed to set beacon parameters
Sat Oct 28 17:49:51 2023 kern.warn kernel: [78392.971394] ath11k_warn: 83 callbacks suppressed
Sat Oct 28 17:49:51 2023 kern.warn kernel: [78392.971417] ath11k c000000.wifi: failed to flush transmit queue, data pkts pending 2
Sat Oct 28 17:49:51 2023 kern.warn kernel: [78392.976139] ath11k c000000.wifi: failed to send WMI_REQUEST_STATS cmd
Sat Oct 28 17:49:51 2023 kern.warn kernel: [78392.982900] ath11k c000000.wifi: could not request fw stats (-108)
Sat Oct 28 17:49:51 2023 kern.warn kernel: [78392.989170] ath11k c000000.wifi: failed to request fw pdev stats: -108
Sat Oct 28 17:49:51 2023 kern.warn kernel: [78392.996012] ath11k c000000.wifi: failed to send WMI_REQUEST_STATS cmd
Sat Oct 28 17:49:51 2023 kern.warn kernel: [78393.001790] ath11k c000000.wifi: could not request fw stats (-108)
Sat Oct 28 17:49:51 2023 kern.warn kernel: [78393.008266] ath11k c000000.wifi: failed to request fw pdev stats: -108
Sat Oct 28 17:49:51 2023 kern.warn kernel: [78393.015583] ath11k c000000.wifi: failed to send WMI_REQUEST_STATS cmd
Sat Oct 28 17:49:51 2023 kern.warn kernel: [78393.020857] ath11k c000000.wifi: could not request fw stats (-108)
Sat Oct 28 17:49:51 2023 kern.warn kernel: [78393.027413] ath11k c000000.wifi: failed to request fw pdev stats: -108

Before the stable 23.05 release I didn't had this issues.

Cheers

URBANsUNITED

stixif, 2023-11-20:
here the same!
with this Version - OpenWrt SNAPSHOT r0-9062e5f 
on AX3600 and Dynalink DL-WRX36

GitPuffy, 2023-11-30:
Similar issue here. 
* Dynalink DL-WRX36

Seems to work **fine** with a SNAPSHOT I had been running from October 23 (`r24198-1998027d7c`) but once I switched to the SNAPSHOT from November 20 (`r24433-2634d3f855`) [ as well as November 27, and November 29] it becomes unstable.  In desperation I am now running stable 23.05.2 because I have no way to install the handful of additional packages (umdns, wireguard, pbr, ddns) I use.

```
Tue Nov 28 22:26:41 2023 kern.err kernel: [ 1614.547305] qcom-q6v5-wcss-pil cd00000.q6v5_wcss: fatal error received:
Tue Nov 28 22:26:41 2023 kern.err kernel: [ 1614.547305] QC Image Version: QC_IMAGE_VERSION_STRING=WLAN.HK.2.9.0.1-01890-QCAHKSWPL_SILICONZ-1
Tue Nov 28 22:26:41 2023 kern.err kernel: [ 1614.547305] Image Variant : IMAGE_VARIANT_STRING=8074.wlanfw.eval_v2Q
Tue Nov 28 22:26:41 2023 kern.err kernel: [ 1614.547305]
Tue Nov 28 22:26:41 2023 kern.err kernel: [ 1614.547305] wal_peer_control.c:2904 Assertion is_graceful_to_handle failedparam0 :zero, param1 :zero, param2 :zero.
Tue Nov 28 22:26:41 2023 kern.err kernel: [ 1614.547305] Thread ID      : 0x00000069  Thread name    : WLAN RT0  Process ID     : 0
Tue Nov 28 22:26:41 2023 kern.err kernel: [ 1614.547305] Register:
Tue Nov 28 22:26:41 2023 kern.err kernel: [ 1614.547305] SP : 0x4bfb9340
Tue Nov 28 22:26:41 2023 kern.err kernel: [ 1614.547305] FP : 0x4bfb9348
Tue Nov 28 22:26:41 2023 kern.err kernel: [ 1614.547305] PC : 0x4b1080c4
Tue Nov 28 22:26:41 2023 kern.err kernel: [ 1614.547305] SSR : 0x00000008
Tue Nov 28 22:26:41 2023 kern.err kernel: [ 1614.547305] BADVA : 0x00020000
Tue Nov 28 22:26:41 2023 kern.err kernel: [ 1614.547305] LR : 0x4b107860
Tue Nov 28 22:26:41 2023 kern.err kernel: [ 1614.547305]
Tue Nov 28 22:26:41 2023 kern.err kernel: [ 1614.547305] Stack Dump
Tue Nov 28 22:26:41 2023 kern.err kernel: [ 1614.547305] from : 0x4bfb9340
Tue Nov 28 22:26:41 2023 kern.err kernel: [ 1614.547305] to   : 0x4bfb9ba0
Tue Nov 28 22:26:41 2023 kern.err kernel: [ 1614.547305]
Tue Nov 28 22:26:41 2023 kern.err kernel: [ 1614.596323] remoteproc remoteproc0: crash detected in cd00000.q6v5_wcss: type fatal error
Tue Nov 28 22:26:41 2023 kern.err kernel: [ 1614.618535] remoteproc remoteproc0: handling crash #1 in cd00000.q6v5_wcss
Tue Nov 28 22:26:41 2023 kern.err kernel: [ 1614.626757] remoteproc remoteproc0: recovering cd00000.q6v5_wcss
Tue Nov 28 22:26:41 2023 kern.info kernel: [ 1614.659415] remoteproc remoteproc0: stopped remote processor cd00000.q6v5_wcss
Tue Nov 28 22:26:42 2023 kern.warn kernel: [ 1614.922477] ath11k c000000.wifi: failed to transmit frame -108
Tue Nov 28 22:26:42 2023 kern.warn kernel: [ 1614.925234] ath11k c000000.wifi: failed to find peer 50:bc:96:9f:83:86 on vdev 2 after creation
Tue Nov 28 22:26:42 2023 kern.warn kernel: [ 1614.927212] ath11k c000000.wifi: failed to find peer vdev_id 2 addr 50:bc:96:9f:83:86 in delete
Tue Nov 28 22:26:42 2023 kern.warn kernel: [ 1614.935824] ath11k c000000.wifi: failed peer 50:bc:96:9f:83:86 delete vdev_id 2 fallback ret -22
Tue Nov 28 22:26:42 2023 kern.warn kernel: [ 1614.944484] ath11k c000000.wifi: Failed to add peer: 50:bc:96:9f:83:86 for VDEV: 2
Tue Nov 28 22:26:42 2023 kern.warn kernel: [ 1614.953548] ath11k c000000.wifi: Failed to add station: 50:bc:96:9f:83:86 for VDEV: 2
Tue Nov 28 22:26:42 2023 daemon.notice hostapd: phy0-ap0: STA 50:bc:96:9f:83:86 IEEE 802.11: Could not add STA to kernel driver
Tue Nov 28 22:26:42 2023 kern.warn kernel: [ 1615.005379] ath11k c000000.wifi: failed to update rx tid queue, tid 0 (-108)
Tue Nov 28 22:26:42 2023 kern.warn kernel: [ 1615.005407] ath11k c000000.wifi: failed to update reo for rx tid 0: -108
Tue Nov 28 22:26:42 2023 kern.info kernel: [ 1615.011487] phy0-ap1: HW problem - can not stop rx aggregation for 50:bc:96:9f:83:86 tid 0
Tue Nov 28 22:26:42 2023 kern.warn kernel: [ 1615.018196] ath11k c000000.wifi: failed to update rx tid queue, tid 0 (-108)
Tue Nov 28 22:26:42 2023 kern.info kernel: [ 1615.026253] phy0-ap1: HW problem - can not stop rx aggregation for 50:bc:96:9f:83:86 tid 1
Tue Nov 28 22:26:47 2023 kern.err kernel: [ 1619.925096] qcom-q6v5-wcss-pil cd00000.q6v5_wcss: start timed out
Tue Nov 28 22:26:47 2023 kern.err kernel: [ 1619.925144] remoteproc remoteproc0: can't start rproc cd00000.q6v5_wcss: -110
Tue Nov 28 22:26:47 2023 kern.warn kernel: [ 1620.165086] ath11k_warn: 1 callbacks suppressed
Tue Nov 28 22:26:47 2023 kern.warn kernel: [ 1620.165099] ath11k c000000.wifi: failed to flush transmit queue, data pkts pending 20
Tue Nov 28 22:26:47 2023 kern.warn kernel: [ 1620.168486] ath11k c000000.wifi: failed to send WMI_PEER_DELETE cmd
Tue Nov 28 22:26:47 2023 kern.warn kernel: [ 1620.176443] ath11k c000000.wifi: failed to delete peer vdev_id 3 addr 50:bc:96:9f:83:86 ret -108
Tue Nov 28 22:26:47 2023 kern.warn kernel: [ 1620.182503] ath11k c000000.wifi: Failed to delete peer: 50:bc:96:9f:83:86 for VDEV: 3
Tue Nov 28 22:26:47 2023 kern.warn kernel: [ 1620.191538] ath11k c000000.wifi: Found peer entry a6:97:33:df:b3:64 n vdev 3 after it was supposedly removed
Tue Nov 28 22:26:47 2023 kern.warn kernel: [ 1620.199308] ------------[ cut here ]------------
Tue Nov 28 22:26:47 2023 kern.warn kernel: [ 1620.209142] WARNING: CPU: 2 PID: 2684 at sta_set_sinfo+0xcb4/0xd30 [mac80211]
Tue Nov 28 22:26:47 2023 kern.warn kernel: [ 1620.213747] Modules linked in: pppoe ppp_async nft_fib_inet nf_flow_table_inet ath11k_ahb ath11k wireguard pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_ [... cut ...]
Tue Nov 28 22:26:47 2023 kern.warn kernel: [ 1620.499786] ath11k c000000.wifi: Failed to set beacon mode for VDEV: 0
Tue Nov 28 22:26:47 2023 daemon.info hostapd: phy0-ap1: STA 50:bc:96:9f:83:86 IEEE 802.11: deauthenticated due to local deauth request
Tue Nov 28 22:26:47 2023 kern.warn kernel: [ 1620.506302] ath11k c000000.wifi: failed to send WMI_BCN_TMPL_CMDID
Tue Nov 28 22:26:52 2023 daemon.warn dnsmasq[1]: possible DNS-rebind attack detected: 192-168-1-103.abcdefghijklmnopqrstuvwxyz012345.plex.direct
Tue Nov 28 22:26:53 2023 kern.warn kernel: [ 1626.514440] ath11k_warn: 83 callbacks suppressed
```

PPP01, 2023-12-02:
Same Issue here with two Xiaomi AX3600. Both are totally unstable and have to be rebooted daily.
Both are on OpenWrt 23.05.2 (r23630-842932a63d)

brada4, 2023-12-02:
Can you confirm trigger is enabling firewall offload?

GitPuffy, 2023-12-03:
> Can you confirm trigger is enabling firewall offload?

How would I do this?  What should I check or look at?

brada4, 2023-12-03:
@robimarko may know better.

robimarko, 2023-12-04:
Its not a red herring, most of the people with crashes reported the same scenario.

PPP01, 2023-12-10:
> The IRQ-Balance Script seems to work with my system. Uptime now more than 20 days.

Which IRQ-Balance Script do you mean? I have not updated since OpenWrt 23.05.2 (r23630-842932a63d) because I feared more trouble

uklatt, 2023-12-10:
I use this:
https://github.com/openwrt/openwrt/issues/13562#issuecomment-1817573570

PPP01, 2023-12-10:
> I use this: [#13562 (comment)](https://github.com/openwrt/openwrt/issues/13562#issuecomment-1817573570)

Thank you! I will try. I updated one of my Xiaomis to r24600-9291f6025f now. 
If I understand it correctly, the script is executed once at boot time. Did you place it into the rc.local

uklatt, 2023-12-10:
Yes, you can pplace it in rc.local (luci: System - Startup - Local Startup)

PPP01, 2023-12-11:
I guess I did too many changes at once. I removed all 5Ghz + 2.4 Ghz with the same SSID (to only 5 Ghz and only 2,4Ghz) and upgraded to the newest snapshot. The result is an extremely unstable 5Ghz WLAN.

I have no ath11k_warn'ings anymore, but a lot of ``nl80211: kernel reports: key addition failed``

`
Sun Dec 10 23:34:41 2023 daemon.notice hostapd: phy1-ap0: AP-STA-DISCONNECTED 1a:7a:4b:e5:28:a2
Sun Dec 10 23:34:46 2023 daemon.notice hostapd: handle_assoc_cb: STA 1a:7a:4b:e5:28:a2 not found
Sun Dec 10 23:34:46 2023 kern.warn kernel: [ 2789.033741] ath11k c000000.wifi: failed to flush transmit queue, data pkts pending 46
Sun Dec 10 23:34:46 2023 daemon.info hostapd: phy1-ap0: STA 1a:7a:4b:e5:28:a2 IEEE 802.11: authenticated
Sun Dec 10 23:34:46 2023 daemon.info hostapd: phy1-ap0: STA 1a:7a:4b:e5:28:a2 IEEE 802.11: authenticated
Sun Dec 10 23:34:51 2023 daemon.notice hostapd: phy1-ap0: STA 1a:7a:4b:e5:28:a2 IEEE 802.11: did not acknowledge authentication response
Sun Dec 10 23:34:51 2023 daemon.notice hostapd: phy1-ap0: STA 1a:7a:4b:e5:28:a2 IEEE 802.11: did not acknowledge authentication response
Sun Dec 10 23:34:52 2023 daemon.notice hostapd: phy1-ap0: STA 1a:7a:4b:e5:28:a2 IEEE 802.11: did not acknowledge authentication response
Sun Dec 10 23:34:52 2023 daemon.notice hostapd: phy1-ap0: STA 1a:7a:4b:e5:28:a2 IEEE 802.11: did not acknowledge authentication response
Sun Dec 10 23:34:52 2023 daemon.notice hostapd: phy1-ap0: STA 1a:7a:4b:e5:28:a2 IEEE 802.11: did not acknowledge authentication response
Sun Dec 10 23:34:52 2023 daemon.notice hostapd: phy1-ap0: STA 1a:7a:4b:e5:28:a2 IEEE 802.11: did not acknowledge authentication response
Sun Dec 10 23:34:52 2023 daemon.notice hostapd: phy1-ap0: STA 1a:7a:4b:e5:28:a2 IEEE 802.11: did not acknowledge authentication response
Sun Dec 10 23:34:52 2023 daemon.notice hostapd: phy1-ap0: STA 1a:7a:4b:e5:28:a2 IEEE 802.11: did not acknowledge authentication response
Sun Dec 10 23:34:52 2023 daemon.notice hostapd: phy1-ap0: STA 1a:7a:4b:e5:28:a2 IEEE 802.11: did not acknowledge authentication response
Sun Dec 10 23:34:52 2023 daemon.notice hostapd: phy1-ap0: STA 1a:7a:4b:e5:28:a2 IEEE 802.11: did not acknowledge authentication response
Sun Dec 10 23:34:57 2023 daemon.info hostapd: phy1-ap0: STA 1a:7a:4b:e5:28:a2 IEEE 802.11: authenticated
Sun Dec 10 23:34:57 2023 daemon.info hostapd: phy1-ap0: STA 1a:7a:4b:e5:28:a2 IEEE 802.11: associated (aid 1)
Sun Dec 10 23:34:57 2023 daemon.notice hostapd: phy1-ap0: AP-STA-CONNECTED 1a:7a:4b:e5:28:a2 auth_alg=open
Sun Dec 10 23:34:57 2023 daemon.info hostapd: phy1-ap0: STA 1a:7a:4b:e5:28:a2 WPA: pairwise key handshake completed (RSN)
Sun Dec 10 23:34:57 2023 daemon.notice hostapd: phy1-ap0: EAPOL-4WAY-HS-COMPLETED 1a:7a:4b:e5:28:a2
Sun Dec 10 23:37:33 2023 daemon.info hostapd: phy2-ap0: STA a2:d4:dc:6c:a3:54 IEEE 802.11: authenticated
Sun Dec 10 23:37:33 2023 daemon.info hostapd: phy2-ap0: STA a2:d4:dc:6c:a3:54 IEEE 802.11: associated (aid 3)
Sun Dec 10 23:37:33 2023 daemon.notice hostapd: phy2-ap0: AP-STA-CONNECTED a2:d4:dc:6c:a3:54 auth_alg=open
Sun Dec 10 23:37:33 2023 daemon.info hostapd: phy2-ap0: STA a2:d4:dc:6c:a3:54 WPA: pairwise key handshake completed (RSN)
Sun Dec 10 23:37:33 2023 daemon.notice hostapd: phy2-ap0: EAPOL-4WAY-HS-COMPLETED a2:d4:dc:6c:a3:54
Sun Dec 10 23:37:33 2023 daemon.notice hostapd: phy2-ap0: AP-STA-DISCONNECTED a2:d4:dc:6c:a3:54
Sun Dec 10 23:37:33 2023 daemon.info hostapd: phy2-ap0: STA a2:d4:dc:6c:a3:54 IEEE 802.11: disassociated
Sun Dec 10 23:37:34 2023 daemon.info hostapd: phy2-ap0: STA a2:d4:dc:6c:a3:54 IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Sun Dec 10 23:50:20 2023 daemon.notice netifd: safe_dhcp (3652): udhcpc: sending renew to server 192.168.110.253
Sun Dec 10 23:50:20 2023 daemon.notice netifd: safe_dhcp (3652): udhcpc: lease of 192.168.110.250 obtained from 192.168.110.253, lease time 7200
Sun Dec 10 23:50:24 2023 daemon.info hostapd: phy1-ap1: STA aa:53:6f:9d:d5:68 IEEE 802.11: authenticated
Sun Dec 10 23:50:24 2023 daemon.info hostapd: phy1-ap1: STA aa:53:6f:9d:d5:68 IEEE 802.11: associated (aid 2)
Sun Dec 10 23:50:24 2023 daemon.notice hostapd: phy1-ap1: AP-STA-CONNECTED aa:53:6f:9d:d5:68 auth_alg=open
Sun Dec 10 23:50:24 2023 daemon.info hostapd: phy1-ap1: STA aa:53:6f:9d:d5:68 WPA: pairwise key handshake completed (RSN)
Sun Dec 10 23:50:24 2023 daemon.notice hostapd: phy1-ap1: EAPOL-4WAY-HS-COMPLETED aa:53:6f:9d:d5:68
Mon Dec 11 00:00:40 2023 daemon.notice hostapd: phy2-ap1: AP-STA-DISCONNECTED 7c:49:eb:88:f5:69
Mon Dec 11 00:00:52 2023 daemon.info hostapd: phy2-ap1: STA 7c:49:eb:88:f5:69 IEEE 802.11: authenticated
Mon Dec 11 00:00:52 2023 daemon.info hostapd: phy2-ap1: STA 7c:49:eb:88:f5:69 IEEE 802.11: associated (aid 3)
Mon Dec 11 00:00:52 2023 daemon.notice hostapd: phy2-ap1: AP-STA-CONNECTED 7c:49:eb:88:f5:69 auth_alg=open
Mon Dec 11 00:00:52 2023 daemon.info hostapd: phy2-ap1: STA 7c:49:eb:88:f5:69 WPA: pairwise key handshake completed (RSN)
Mon Dec 11 00:00:52 2023 daemon.notice hostapd: phy2-ap1: EAPOL-4WAY-HS-COMPLETED 7c:49:eb:88:f5:69
Mon Dec 11 00:01:55 2023 daemon.notice hostapd: phy2-ap1: AP-STA-DISCONNECTED 7c:49:eb:88:f5:69
Mon Dec 11 00:01:55 2023 daemon.info hostapd: phy2-ap1: STA 7c:49:eb:88:f5:69 IEEE 802.11: authenticated
Mon Dec 11 00:01:55 2023 daemon.info hostapd: phy2-ap1: STA 7c:49:eb:88:f5:69 IEEE 802.11: associated (aid 3)
[... cut ...]
Mon Dec 11 07:08:39 2023 daemon.err hostapd: nl80211: kernel reports: key addition failed
Mon Dec 11 07:08:42 2023 daemon.err hostapd: nl80211: kernel reports: key addition failed
Mon Dec 11 07:08:45 2023 daemon.err hostapd: nl80211: kernel reports: key addition failed
Mon Dec 11 07:08:55 2023 daemon.err hostapd: nl80211: kernel reports: key addition failed
Mon Dec 11 07:08:59 2023 daemon.err hostapd: nl80211: kernel reports: key addition failed
Mon Dec 11 07:08:59 2023 daemon.err hostapd: nl80211: kernel reports: key addition failed
`

tperalta82, 2024-01-18:
Seems to be the same issue somehow as #9555 , reverted the blob from snapshot to the one on 23.05 on my build, works a charm and has been for days/weeks now

brada4, 2024-01-18:
There is no "crash" , just FT failed and full 4way did fine after. See https://forum.openwrt.org/t/daemon-err-hostapd-nl80211-kernel-reports-key-addition-failed-is-this-a-problem/100370/7

tperalta82, 2024-01-18:
> There is no "crash" , just FT failed and full 4way did fine after. See https://forum.openwrt.org/t/daemon-err-hostapd-nl80211-kernel-reports-key-addition-failed-is-this-a-problem/100370/7

Yeah, I deleted the comment, but the reference still stayed on the other issue! Sorry ;)

maurerle, 2025-10-04:
For ath11k, there is a potential patch to the "failed to flush transmit queue" with https://github.com/openwrt/openwrt/pull/20293
Please test and report back in the PR.