openwrt/openwrt #13727: usb not work on ath79? (gl-inet 6416)

pexcn, 2023-10-18:
### Describe the bug

ath79 seems to have a problem with usb support. I seem to have seen similar problems at the following link:
https://github.com/openwrt/openwrt/issues/10904
https://forum.openwrt.org/t/gl-inet-6416-no-usb-with-21-02-0-rc1/95206
https://forum.gl-inet.com/t/gl-ar300m16-openwrt-22-03-0-rc5-usb-port-power-off-by-default/23199

### OpenWrt log
```sh
root@OpenWrt:~# dmesg | grep -i usb
[    3.980640] usbcore: registered new interface driver usbfs
[    3.984852] usbcore: registered new interface driver hub
[    3.990244] usbcore: registered new device driver usb
[    4.010637] ehci_hcd: USB 2.0 'Enhanced' Host Controller (EHCI) Driver
root@OpenWrt:~# lsusb
unable to initialize libusb: -99
```

### OpenWrt version

openwrt 23.05-rc4

### OpenWrt target/subtarget

ath79/generic

### Device

GL.iNet 6416

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-10-18:
Why not release?

pexcn, 2023-10-18:
> Why not release?

I tested it many days ago, when the latest version was rc4.

brada4, 2023-10-18:
Install 23.05.0 https://firmware-selector.openwrt.org/?version=23.05.0&target=ath79%2Fgeneric&id=glinet_6416

No idea why ypur image does not support chipidea usb

pexcn, 2023-10-19:
> Install 23.05.0 https://firmware-selector.openwrt.org/?version=23.05.0&target=ath79%2Fgeneric&id=glinet_6416
> 
> No idea why ypur image does not support chipidea usb

same problem, chipidea packages was installed, but still not work.
```sh
root@OpenWrt:~# opkg list-installed | grep usb
kmod-usb-chipidea - 5.15.132-1
kmod-usb-chipidea2 - 5.15.132-1
kmod-usb-core - 5.15.132-1
kmod-usb-ehci - 5.15.132-1
kmod-usb-gadget - 5.15.132-1
kmod-usb-phy-nop - 5.15.132-1
kmod-usb-roles - 5.15.132-1
libusb-1.0-0 - 1.0.26-3
usbutils - 014-1
```

brada4, 2023-10-19:
Please install 23.05.0.

pexcn, 2023-10-19:
There should be the same issue, this issue has been existing since ar71xx was migrated to ath79.
As far as I know it's been around since at least openwrt-19.07.

brada4, 2023-10-19:
strangely enough both your refetences are to -rc versions. Obviously -rc snapshot will not be changed,while release will get updates for a year or more. rc-s are not sabotaged, but please try release.

pexcn, 2023-10-19:
Because my gl-inet 6416 only has one ethernet port and needs to make some necessary changes, I used my fork version.
In fact, it is not fundamentally different from the original openwrt.

source: https://github.com/openwrt-dev/openwrt/commits/openwrt-23.05

raenye, 2023-10-20:
@brada4 AFAIK rc4 is identical to the release...

pexcn, 2023-10-20:
> You mean this is device-specific or general problem that No device works? [openwrt-dev@4fe902d](https://github.com/openwrt-dev/openwrt/commit/4fe902d65c066b3326fc390f1eadf2ccf52f4411)

Maybe it's a problem with ath79 devices with USB port, but currently I only have gl-inet 6416 to test.

pexcn, 2023-10-20:
Additionally, I've seen similar issues on another ath79 devices, see https://github.com/openwrt/openwrt/pull/9720

brada4, 2023-10-20:
@raenye that does not make rc4 supported release in any way.

brada4, 2023-10-22:
There is no problem with ath79 tplink archer c7v5
you will need to repeat perceived problem with unmodded hardware and clean openwrt release. Your fork and implying hw mod is exactly what is not helped here.