openwrt/openwrt #13730: ASU Fails because of kmod-crypto-gf128

milouk, 2023-10-18:
### Describe the bug

Im trying to perform a ASU for Xiaomi AX3600 and I get the following error:

`Error: Impossible package selection: kmod-crypto-gf128 version not as requested: 6.1.56-1 vs. 6.1.57-1`

[image]

### OpenWrt version

SNAPSHOT r24154-6f31941d38

### OpenWrt target/subtarget

qualcommax/ipq807x

### Device

Xiaomi AX3600

### Image kind

Official downloaded image

### Steps to reproduce

Try to update` kmod-crypto-gf128` to 6.1.57-1

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-10-19:
Any reason to hide the command you used?

milouk, 2023-10-19:
> Any reason to hide the command you used?

Not really, Its just a huge log and couldn't fit in a single screenshot. I run `auc` followed by a `y` regarding the package upgrade

brada4, 2023-10-19:
Sometimes snapshot downloads are not completely in sync. Usually that goes away next day. Also luci attended sysupgrade permits editing out (this time transiently) broken packages.

closed 2023-10-25 by milouk