openwrt/openwrt #13731: [23.05] WRT3200ACM Wifi disconnects

buggsdummy, 2023-10-19:
### Describe the bug

The wifi will just die in about 6-8hours of usage and so I was able to capture some logs after installing a USB on it. Is anyone else having this issue or is it just me? I want to try to reproduce this on another router, not sure if it's old hardware or something else. Thanks!

### OpenWrt version

r23497-6637af95aa

### OpenWrt target/subtarget

mvebu/cortexa9

### Device

Linksys WRT3200ACM

### Image kind

Official downloaded image

### Steps to reproduce

Leave it run for a few hours

### Actual behaviour

Wifi disconnects

### Expected behaviour

Work like normal and be stable

### Additional info

Logging

Wed Oct 18 16:41:25 2023 user.notice firewall: Reloading firewall due to ifupdate of wan (wan)
Wed Oct 18 16:56:43 2023 kern.info kernel: [24111.026229] ieee80211 phy0: radar detected by firmware
Wed Oct 18 16:56:43 2023 daemon.notice hostapd: phy0-ap0: DFS-RADAR-DETECTED freq=5260 ht_enabled=0 chan_offset=0 chan_width=3 cf1=5290 cf2=0
Wed Oct 18 16:56:43 2023 daemon.notice hostapd: dfs_downgrade_bandwidth: no DFS channels left, waiting for NOP to finishWed Oct 18 16:56:43 2023 daemon.notice hostapd: phy0-ap0: AP-DISABLED
Wed Oct 18 16:56:43 2023 daemon.notice hostapd: phy0-ap0: AP-STA-DISCONNECTED de:a1:71:6e:ab:23
Wed Oct 18 16:56:43 2023 daemon.notice hostapd: phy0-ap0: AP-STA-DISCONNECTED 1c:bf:ce:bc:8a:2a
Wed Oct 18 16:56:43 2023 daemon.notice hostapd: phy0-ap0: AP-STA-DISCONNECTED a4:38:cc:9c:fa:c3
Wed Oct 18 16:56:43 2023 daemon.err hostapd: 20/40 MHz: center segment 0 (=58) and center freq 1 (=5270) not in sync
Wed Oct 18 16:56:43 2023 daemon.notice hostapd: phy0-ap0: AP-STA-DISCONNECTED 22:95:9c:3e:c2:16
Wed Oct 18 16:56:43 2023 daemon.notice hostapd: phy0-ap0: AP-STA-DISCONNECTED 82:0e:c1:80:89:3e
Wed Oct 18 16:56:43 2023 daemon.notice hostapd: phy0-ap0: AP-STA-DISCONNECTED d4:f5:47:5c:14:f3
Wed Oct 18 16:56:43 2023 daemon.notice hostapd: phy0-ap0: AP-STA-DISCONNECTED 14:c1:4e:40:ee:55
Wed Oct 18 16:56:43 2023 daemon.err hostapd: 20/40 MHz: center segment 0 (=58) and center freq 1 (=5270) not in sync
Wed Oct 18 16:56:43 2023 daemon.notice hostapd: nl80211: deinit ifname=phy0-ap0 disabled_11b_rates=0
Wed Oct 18 16:56:43 2023 daemon.notice netifd: Network device 'phy0-ap0' link is down
Wed Oct 18 16:56:43 2023 kern.info kernel: [24111.042040] br-lan: port 5(phy0-ap0) entered disabled state
Wed Oct 18 16:56:43 2023 kern.debug kernel: [24111.475960] ieee80211 phy0: staid 1 deleted
Wed Oct 18 16:56:44 2023 kern.debug kernel: [24111.535988] ieee80211 phy0: staid 7 deleted
Wed Oct 18 16:56:44 2023 kern.debug kernel: [24111.596016] ieee80211 phy0: staid 9 deleted
Wed Oct 18 16:56:44 2023 kern.debug kernel: [24111.656022] ieee80211 phy0: staid 4 deleted
Wed Oct 18 16:56:44 2023 kern.debug kernel: [24111.716051] ieee80211 phy0: staid 8 deleted
Wed Oct 18 16:56:44 2023 kern.debug kernel: [24111.774167] ieee80211 phy0: staid 5 deleted
Wed Oct 18 16:56:44 2023 kern.debug kernel: [24111.826115] ieee80211 phy0: staid 3 deleted
Wed Oct 18 16:56:44 2023 kern.debug kernel: [24111.844294] ieee80211 phy0: change: 0x40
Wed Oct 18 16:56:44 2023 kern.debug kernel: [24111.941965] ieee80211 phy0: change: 0x100
Wed Oct 18 16:56:44 2023 daemon.notice hostapd: phy0-ap0: interface state ENABLED->DISABLED
Wed Oct 18 16:56:44 2023 kern.info kernel: [24112.005747] br-lan: port 5(phy0-ap0) entered disabled state
Wed Oct 18 16:56:44 2023 kern.debug kernel: [24112.174110] ieee80211 phy0: change: 0xffffffff
Wed Oct 18 16:56:44 2023 daemon.notice netifd: bridge 'br-lan' link is down
Wed Oct 18 16:56:44 2023 daemon.notice netifd: Interface 'lan' has link connectivity loss
Wed Oct 18 16:56:44 2023 daemon.notice hostapd: phy0-ap0: interface state DISABLED->COUNTRY_UPDATE
Wed Oct 18 16:56:44 2023 daemon.warn odhcpd[1873]: No default route present, overriding ra_lifetime!
Wed Oct 18 16:56:44 2023 daemon.notice hostapd: phy0-ap0: interface state COUNTRY_UPDATE->HT_SCAN
Wed Oct 18 16:56:44 2023 kern.debug kernel: [24112.284021] ieee80211 phy0: change: 0x100
Wed Oct 18 16:56:44 2023 kern.debug kernel: [24112.297269] ieee80211 phy0: change: 0x40
Wed Oct 18 16:56:45 2023 kern.debug kernel: [24112.507160] ieee80211 phy0: change: 0x40
Wed Oct 18 16:56:45 2023 daemon.err hostapd: could not get valid channel
Wed Oct 18 16:56:45 2023 daemon.notice hostapd: phy0-ap0: interface state HT_SCAN->DFS
Wed Oct 18 16:56:45 2023 kern.debug kernel: [24112.757245] ieee80211 phy0: change: 0x40
Wed Oct 18 16:56:45 2023 kern.debug kernel: [24112.853293] ieee80211 phy0: change: 0x100
Wed Oct 18 17:11:25 2023 user.notice firewall: Reloading firewall due to ifupdate of wan (wan)
Wed Oct 18 17:13:00 2023 daemon.info dnsmasq[1]: read /etc/hosts - 12 names
Wed Oct 18 17:13:00 2023 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 2 names
Wed Oct 18 17:13:00 2023 daemon.info dnsmasq[1]: read /tmp/hosts/odhcpd - 0 names
Wed Oct 18 17:13:00 2023 daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses
Wed Oct 18 17:27:21 2023 daemon.notice hostapd: phy0-ap0: DFS-NOP-FINISHED freq=5260 ht_enabled=0 chan_offset=0 chan_width=0 cf1=5260 cf2=0
Wed Oct 18 17:27:21 2023 daemon.err hostapd: could not get valid channel
Wed Oct 18 17:27:21 2023 daemon.notice hostapd: phy0-ap0: interface state DFS->DFS
Wed Oct 18 17:27:21 2023 daemon.notice hostapd: phy0-ap0: DFS-NOP-FINISHED freq=5280 ht_enabled=0 chan_offset=0 chan_width=0 cf1=5280 cf2=0
Wed Oct 18 17:27:21 2023 daemon.notice hostapd: phy0-ap0: interface state DFS->DFS
Wed Oct 18 17:27:21 2023 daemon.notice hostapd: phy0-ap0: DFS-CAC-START freq=5260 chan=52 sec_chan=1, width=0, seg0=58, seg1=0, cac_time=60s
Wed Oct 18 17:27:21 2023 daemon.err hostapd: 20/40 MHz: center segment 0 (=58) and center freq 1 (=5270) not in sync
Wed Oct 18 17:27:21 2023 daemon.err hostapd: Can't set freq params
Wed Oct 18 17:27:21 2023 daemon.err hostapd: DFS start_dfs_cac() failed, -1
Wed Oct 18 17:27:21 2023 daemon.notice hostapd: phy0-ap0: DFS-NOP-FINISHED freq=5300 ht_enabled=0 chan_offset=0 chan_width=0 cf1=5300 cf2=0
Wed Oct 18 17:27:21 2023 daemon.notice hostapd: phy0-ap0: interface state DFS->DFS
Wed Oct 18 17:27:21 2023 daemon.notice hostapd: phy0-ap0: DFS-CAC-START freq=5260 chan=52 sec_chan=1, width=0, seg0=58, seg1=0, cac_time=60s
Wed Oct 18 17:27:21 2023 daemon.err hostapd: 20/40 MHz: center segment 0 (=58) and center freq 1 (=5270) not in sync
Wed Oct 18 17:27:21 2023 daemon.err hostapd: Can't set freq params
Wed Oct 18 17:27:21 2023 daemon.err hostapd: DFS start_dfs_cac() failed, -1
Wed Oct 18 17:27:21 2023 daemon.notice hostapd: phy0-ap0: DFS-NOP-FINISHED freq=5320 ht_enabled=0 chan_offset=0 chan_width=0 cf1=5320 cf2=0
Wed Oct 18 17:27:21 2023 daemon.notice hostapd: phy0-ap0: interface state DFS->DFS
Wed Oct 18 17:27:21 2023 daemon.notice hostapd: phy0-ap0: DFS-CAC-START freq=5260 chan=52 sec_chan=1, width=0, seg0=58, seg1=0, cac_time=60s
Wed Oct 18 17:27:21 2023 daemon.err hostapd: 20/40 MHz: center segment 0 (=58) and center freq 1 (=5270) not in sync
Wed Oct 18 17:27:21 2023 daemon.err hostapd: Can't set freq params
Wed Oct 18 17:27:21 2023 daemon.err hostapd: DFS start_dfs_cac() failed, -1

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-10-19:
It works just fine. You might want to provide more 5ghz channels to jump out of radar's way though. Like setting it to auto channel.

buggsdummy, 2023-10-19:
I switched it to auto, will update as I test throughout the day.

brada4, 2023-10-19:
Let  it qualify for another day and check `logread -e RARAR` after.

buggsdummy, 2023-10-22:
This is working fine now after switching to auto, no further issues. Thanks!

closed 2023-10-22 by buggsdummy