openwrt/openwrt #13732: dnsmasq: bug? per-DHCP server config item `dhcp_option` might be improperly handled

Lanchon, 2023-10-19:
### Describe the bug

in **Interfaces > Edit "my-interface" > DHCP Server > Advanced settings**, field **DHCP-Options** seems to be incorrectly processed.

if i add value `15,my.domain` to that field, the following line is added to `/etc/config/dhcp`, which is fine:

```
config dhcp 'my-interface'
	...
	list dhcp_option '15,my.domain'
```

then this dubious line is added to `/tmp/etc/dnsmasq.conf.*`:

```
dhcp-option=my-interface,15,my.domain
```

i believe the correct line should be:

```
dhcp-option=tag:my-interface,15,my.domain
```

as per [the manual](https://thekelleys.org.uk/dnsmasq/docs/dnsmasq-man.html), the valid syntax are:

```
--dhcp-option=[tag:<tag>,[tag:<tag>,]][encap:<opt>,][vi-encap:<enterprise>,][vendor:[<vendor-class>],][<opt>|option:<opt-name>|option6:<opt>|option6:<opt-name>],[<value>[,<value>]]
--dhcp-option-force=[tag:<tag>,[tag:<tag>,]][encap:<opt>,][vi-encap:<enterprise>,][vendor:[<vendor-class>],]<opt>,[<value>[,<value>]]
```

and both require the literal `tag:` prefix for tags. (however, i see no errors in the log.)

in the source, [this line](https://github.com/openwrt/openwrt/blob/v23.05.0/package/network/services/dnsmasq/files/dnsmasq.init#L473) seems ok, but apparently `dhcp_tag_add()` is not being called.

instead [this line](https://github.com/openwrt/openwrt/blob/v23.05.0/package/network/services/dnsmasq/files/dnsmasq.init#L664) seems to be used. maybe this fix is needed there?

```
	xappend "--dhcp-option${force:+-force}=${networkid:+$networkid,}$option"
```
to:
```
	xappend "--dhcp-option${force:+-force}=${networkid:+tag:$networkid,}$option"
```

### OpenWrt version

r23497-6637af95aa

### OpenWrt target/subtarget

ipq806x/generic

### Device

Askey RT4230W REV6

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-10-19:
Please close the issue adter pfofoundly reading manual you linked. Namely parts about default tag: being option:.

Lanchon, 2023-10-19:
> Please close the issue adter pfofoundly reading manual you linked. Namely parts about default tag: being option:.

what do you mean being optional? the tag? or the `tag:` prefix?

i know the tag is optional, but i think the tag prefix **is not optional** if a tag is included.

the issue is that i think a tag **is being included**, but without the `tag:` prefix. the current software is producing `dhcp-option=my-interface,15,my.domain`. isn't `my-interface` a tag? why is then included without the `tag:` prefix?

please take a moment to clarify your statement.

thanks

Lanchon, 2023-10-19:
anybody knows who is the maintainer of [dnsmasq.init](https://github.com/openwrt/openwrt/blob/main/package/network/services/dnsmasq/files/dnsmasq.init)?

i need to discuss several extensions/fixes.

brada4, 2023-10-19:
Make pull request(s) for improvements.

exact string `tag:` is not expexted ever, instead dnsmasq assumes `option:`, it is written in manual you kindly linked to.

Lanchon, 2023-10-19:
@brada4,

thanks

> exact string `tag:` is not expexted ever, instead dnsmasq assumes `option:`, it is written in manual you kindly linked to.

what do you mean by "not expected ever"? it is legal for dnsmasq as shown in its manual, so `tag:` is expected.

form the manual:

```
--dhcp-option=[tag:<tag>,[tag:<tag>,]][encap:<opt>,][vi-encap:<enterprise>,][vendor:[<vendor-class>],][<opt>|option:<opt-name>|option6:<opt>|option6:<opt-name>],[<value>[,<value>]]
```

you can clearly see the `tag:<tag>` constructs, so `tag:` is expected.

and now, this is the line produced by openwrt:

```
dhcp-option=my-interface,15,my.domain
```

how do you think that line is parsed by dnsmasq and why?

what is `my-interface` parsed as?
what is `15` parsed as?
what is `my.domain` parsed as?

thanks.

Lanchon, 2023-10-19:
> Make pull request(s) for improvements.

i cannot make pull requests in this case. there are design issues that need to be discussed with the maintainer(s) before any code is written. again, i need to talk to the maintainer(s).

Lanchon, 2023-10-19:
> what is `my-interface` parsed as?
> what is `15` parsed as?
> what is `my.domain` parsed as?

i'll answer my own question:

```
--dhcp-option=
    [tag:<tag>,[tag:<tag>,]]
    [encap:<opt>,]
    [vi-encap:<enterprise>,]
    [vendor:[<vendor-class>],]
    [<opt>|option:<opt-name>|option6:<opt>|option6:<opt-name>],[<value>[,<value>]]
```

```
dhcp-option=my-interface,15,my.domain
```
is parsed as:
- `my-interface` does not match any literal prefix (tag:, encap:, etc)
- so `my-interface` is parsed as `<opt>`
- `15` as value
- and `my.domain` as value

and this is wrong:

- `my-interface` is not the option, it is a tag!
- `15` is not a value, it is the option!

brada4, 2023-10-19:
historivally [filter],option,value.

Lanchon, 2023-10-19:
@brada4 you are not a member of the openwrt team and you are not making any sense, please do not troll this issue any longer

dave14305, 2023-10-19:
In dnsmasq, the label is optional.

https://thekelleys.org.uk/gitweb/?p=dnsmasq.git;a=blob;f=src/option.c;hb=HEAD#l1509

Lanchon, 2023-10-19:
well i'll assume you are right and that the dnsmasq docs are wrong/outdated.

thanks.

closed 2023-10-19 by Lanchon