openwrt/openwrt #13734: github release: is github release info for 23.05.0 incorrect?

Lanchon, 2023-10-19:
### Describe the bug

hi @ynezz

you just released https://github.com/openwrt/openwrt/releases/tag/v23.05.0

the banner there says:

```
  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt 23.05.0, r20134-5f15225c1e
 -----------------------------------------------------
```

but isn't `r20134-5f15225c1e` from https://openwrt.org/releases/22.03/notes-22.03.5 ?

i think the right release info is:

```
 OpenWrt 23.05.0, r23497-6637af95aa
```

thanks

### OpenWrt version

r23497-6637af95aa

### OpenWrt target/subtarget

-

### Device

-

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

raenye, 2023-10-20:
I see this (on x64):

```
  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt 23.05.0, r23497-6637af95aa
 -----------------------------------------------------
```

Lanchon, 2023-10-20:
> I see this (on x64):
>
> ```
>   _______                     ________        __
>  |       |.-----.-----.-----.|  |  |  |.----.|  |_
>  |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
>  |_______|| [...]

yep, all archs come from the same commit.

ynezz, 2023-10-20:
Thanks for the report! Should be fixed.

closed 2023-10-20 by ynezz