openwrt/openwrt #13736: TP-Link Archer C6 v3

kabal-pl, 2023-10-19:
### Describe the bug

Hello,
I would like to report a different "gpio" entry for the "wps" button in the Archer C6(EU) router ver:3.20
Default entry in this moment is gpio-508:
```
cat /sys/kernel/debug/gpio
gpiochip2: GPIOs 416-447, parent: platform/1e000600.gpio, 1e000600.gpio-bank2:

gpiochip1: GPIOs 448-479, parent: platform/1e000600.gpio, 1e000600.gpio-bank1:
 gpio-461 (                    |green:wan           ) out hi ACTIVE LOW

gpiochip0: GPIOs 480-511, parent: platform/1e000600.gpio, 1e000600.gpio-bank0:
 gpio-484 (                    |green:power         ) out lo ACTIVE LOW
 gpio-488 (                    |reset               ) in  hi IRQ ACTIVE LOW
 gpio-494 (                    |orange:wan          ) out hi ACTIVE LOW
 gpio-495 (                    |green:lan           ) out lo ACTIVE LOW
 gpio-496 (                    |green:wifi5g        ) out hi ACTIVE LOW
 gpio-498 (                    |green:wifi2g        ) out hi ACTIVE LOW
 gpio-499 (                    |reset               ) out hi ACTIVE LOW
 gpio-508 (                    |wps                 ) in  hi IRQ ACTIVE LOW 
```

In my case (my router motherboard) WPS button exist in gpio-490 entry. It looks like this:
Button pressed:
gpio-490 (                    |sysfs               ) in  lo 

Button released:
gpio-490 (                    |sysfs               ) in  hi 

### OpenWrt version

22.03-SNAPSHOT r20220-92a0dd2447

### OpenWrt target/subtarget

ramips/mt7621

### Device

TP-Link Archer C6 v3

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

raenye, 2023-10-19:
@amishv @vrpatil 
Did you test wps button? Are you sure the correct GPIO for it is 28 rather than 10?

kabal-pl, 2023-10-20:
_@ Did you test wps button?_
Yes, I tested it. Physically button works properly.

   

_@ Are you sure the correct GPIO for it is 28 rather than 10?_
Yup, today I tested it also on Openwrt 23.05 SNAPSHOOT and its the same. So in my case GPIO entry for wps button is the same like in Openwrt 22.03 - which is gpio-490 (offset 10 like You wrote).
Maybe my router have different motherboard revision? Is there any possibility to read detailed revision of motherboard from Openwrt? If yes, please give a hint how to check this.

raenye, 2023-10-20:
Sorry, I aimed these questions at the people who authored the A6 and C6 support.
One thing you could do is to download GPL sources for your device (from TP-Link website) and verify the GPIO assignment for the original firmware.

kabal-pl, 2023-10-20:
_@ One thing you could do is to download GPL sources for your device (from TP-Link website) and verify the GPIO assignment for the original firmware._
Like You suggested Ive downloaded "GPL_Archer A6&C6.tar" and I did some research.. 
I found file `ralink_gpio_C6V3.c` in folder `(\GPL_Archer A6&C6.tar\GPL_Archer A6&C6\c6u-gpl\mtk_ApSoC_5050\Uboot\board\rt2880\ralink_gpio_C6V3.c)`
and at the beginning (top) of this file I found this info:
```
{    
    GPIO_LED_POWER = 4,
    GPIO_LED_2G = 18,
    GPIO_LED_5G = 16,
    GPIO_LED_INTERNET_GREEN = 45,
    GPIO_LED_INTERNET_ORANGE = 14,
    GPIO_LED_LAN = 15,
    GPIO_LED_USB = 13,
    GPIO_LED_WPS = 4,
    GPIO_POWER_USB = 3,      
    GPIO_BUTTON_WIFI = 10,
    GPIO_BUTTON_RESET = 8,   
    GPIO_BUTTON_WPS = 10,
}
```

So, it looks like all entries (besides WIFI/WPS button) match to:
```
root@OpenWrt:~# cat /sys/kernel/debug/gpio
gpiochip2: GPIOs 416-447, parent: platform/1e000600.gpio, 1e000600.gpio-bank2:

gpiochip1: GPIOs 448-479, parent: platform/1e000600.gpio, 1e000600.gpio-bank1:
 gpio-461 (                    |green:wan           ) out hi ACTIVE LOW

gpiochip0: GPIOs 480-511, parent: platform/1e000600.gpio, 1e000600.gpio-bank0:
 gpio-484 (                    |green:power         ) out lo ACTIVE LOW
 gpio-488 (                    |reset               ) in  hi IRQ ACTIVE LOW
 gpio-494 (                    |orange:wan          ) out hi ACTIVE LOW
 gpio-495 (                    |green:lan           ) out lo ACTIVE LOW
 gpio-496 (                    |green:wifi5g        ) out hi ACTIVE LOW
 gpio-498 (                    |green:wifi2g        ) out hi ACTIVE LOW
 gpio-499 (                    |reset               ) out hi ACTIVE LOW
 gpio-508 (                    |wps                 ) in  hi IRQ ACTIVE LOW
```

What You think about this?

raenye, 2023-10-20:
Thank you for verifying this, I'll prepare a pull request.

I see there's also a USB LED in the GPL sources; is there one on the device? is it indeed controlled by GPIO 13 ("gpio-503")?

kabal-pl, 2023-10-20:
_@ Thank you for verifying this, I'll prepare a pull request._
Your welcome :)

_@ I see there's also a USB LED in the GPL sources; is there one on the device? is it indeed controlled by GPIO 13 ("gpio-503")?_
Nope, this model (Archer C6 v3) don't have external USB port.

raenye, 2023-10-20:
Done, now it needs to be reviewed.
Do you want your name mentioned in the commit? ("_Reported-by: Real Name \<email@domain\>_")

kabal-pl, 2023-10-20:
_@ Done, now it needs to be reviewed._
Sound great, thank You!

_@ Do you want your name mentioned in the commit? ("Reported-by: Real Name <email@domain>")_
If there is such a possibility - sure! This will be my first contribution to the OpenWrt community :)
`("Reported-by: Waldemar Czabaj <kaball@wp.pl>")`

Thank You!

closed 2023-10-22 by openwrt-bot

raenye, 2023-10-22:
Dziękuję bardzo @kabal-pl for the very informative feedback, which, as you can see, led to this getting fixed _very_ quickly.
It will probably take time for this to reach stable OpenWrt versions, but the next snapshot build will have it.

kabal-pl, 2023-10-22:
:) Proszę bardzo @raenye 
I'm glad I could help on this topic.
Till next time ;)
Best regards!

commit fe5e4987 (2023-10-22): ramips: TP-link archer A6/C6 device tree updates

Set correct GPIO (10) for the WPS button. This matches GPIO settings in
vendor GPL sources. Note that GPL sources also mention a USB indicator
LED (GPIO 13) but the device has neither an external USB port nor a USB LED.

In addition, prefixes (button-, led-) are added to relevant DT entries,
as well as color and function specifications for LEDs.

Closes: #13736

Reported-by: Waldemar Czabaj <kaball@wp.pl>

(added led mitigations for wifi leds)
commit 75fe50de (2023-10-25): ramips: TP-link archer A6/C6 device tree updates

Set correct GPIO (10) for the WPS button. This matches GPIO settings in
vendor GPL sources. Note that GPL sources also mention a USB indicator
LED (GPIO 13) but the device has neither an external USB port nor a USB LED.

In addition, prefixes (button-, led-) are added to relevant DT entries,
as well as color and function specifications for LEDs.

Closes: #13736

Reported-by: Waldemar Czabaj <kaball@wp.pl>

(added led mitigations for wifi leds)

(cherry picked from commit fe5e4987776ef66c6788f70251dcbc0ca80a1c5f)