openwrt/openwrt #13739: problems with Xiaomi AX3200 router

Mining77, 2023-10-19:
### Describe the bug

- Misidentification of router model by OpenWrt
- No AX band on 2G
- download speed problems
- Speed on 2G reaching 107 Mbps on a 1 Gbps link

### OpenWrt version

r23497-6637af95aa

### OpenWrt target/subtarget

mediatek/mt7622

### Device

Xiaomi Redmi Router AX6S

### Image kind

Official downloaded image

### Actual behaviour

- OpenWrt shows that my router is Xiaomi Redmi Router AX6S, this is not true. My router is a Xiaomi AX3200.
-On the 2G radio I do not have the ability to turn on the AX band, I am very much asking you to create the ability to turn on the AX band on the Xiaomi AX3200.
- On my 1 Gbps symmetrical fiber internet speed I have a download of 830 Mbps on an upload of 940 Mbps. On my old ZTE router the download speeds were 943 Mbps, upload 940 Mbps.
- The speed on the 2G band is only 107 Mbps, I ask you very much to improve the speed on the 2G band.

### Expected behaviour

OpenWrt stability and performance improvement on Xiaomi AX3200 router

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

raenye, 2023-10-20:
> OpenWrt shows that my router is Xiaomi Redmi Router AX6S

AX3200 is the global name for AX6S.

> No AX band on 2G

Is this a regression? meaning, was it better for you with an older version of OpenWrt?

> I have a download of 830 Mbps on an upload of 940 Mbps

Did you enable HW offloading (in the firewall webpage)?

Mining77, 2023-10-20:
> > No AX band on 2G
> 
> Is this a regression? meaning, was it better for you with an older version of OpenWrt?
> >you mean I would install an older version of OpenWrt 22.03.5 on my Xiaomi AX3200? I did not have an older version of OpenWrt on the Xiaomi AX3200. This is my new router and on it I installed 23.05.0 right away.

> > I have a download of 830 Mbps on an upload of 940 Mbps
> 
> Did you enable HW offloading (in the firewall webpage)?
>My flow offloading is disable 

tell me more please why on the WiFi 2G band I don't have more speed like 100 Mbp/s if my connection is 1 Gbps?

castiel652, 2023-10-20:
> * No AX band on 2G
It's not a bug. Hardware just doesn't have the capability to do AX on 2.4G

raenye, 2023-10-20:
> My flow offloading is disable

So enable both software and hardware offloading. You should also enable WED.
See more here: https://openwrt.org/toh/xiaomi/ax3200

> > * No AX band on 2G

>   It's not a bug. Hardware just doesn't have the capability to do AX on 2.4G

Indeed, the maximum you could officially get on 2.4 GHz for a 802.11n-compliant 2x2 client is 300Mbps signaling, or something like ~250 actual. Since you're seeing ~100, I suspect you don't have 40MHz active; you could force it at _Device Configuration->Advanced Settings_.
You can see the current signaling rate of each client at _Network->Wireless->Associated Stations->RX Rate / TX Rate_.
[image]
As you can see here, this 1x1 client is only getting 72Mbps signaling, while it could get 144 with 40MHz.

> Maybe this #12594 could raise speeds on 2.4GHz on compatible devices, but they would be still lower than on Wifi6 devices.

This should give you 400Mbps signaling. The most recent version of VHT on 2.4 GHz is https://github.com/openwrt/openwrt/pull/12904 AFAIK.
But this is not yet merged (so only on non-official builds), but I can attest it works fine on AX6S.

Mining77, 2023-10-20:
Could you please expand the abbreviation WED?

I forced 40 MHz and the speed is still 100 Mbps. It did not help to increase the width to 40 MHz

I should add that I checked the speed on the spedtest app on iPhone 14 Pro hardware

You can see the current signaling rate of each client at Network->Wireless->Associated Stations->
RX Rate / TX Rate. RX/ TX is still showing me 20 MHz despite the inclusion of 40 MHz. And the speed is 24 Mbps, 144 Mbps

lukasz1992, 2023-10-20:
Wireless Ethernet Dispatch - offloading CPU for lan->wlan traffic and sometimes wlan->lan too.

This case WED should be needed - cpu is fast enough to do it without offload.
I guess device will not use 40MHz anyway on 2.4GHz. Even if you force 40MHz on AP, device can possibly still want to use 20MHz channel width.

brada4, 2023-10-20:
Link rate does not mean transfer rate on shared air or network. Especially 40MHz width will not gain anything in very likely presence of other signal in 2.4GHz. 2x2 radio on your phone should allow you 144Mbps in 20MHz 2.4GHz band, which in practice means 1/n of all people in channel, 100Mbps achieved plays video well and means your neighbours are fairly idle.

brada4, 2023-10-20:
WED is capability of AX mt76 radios:
```
ls -l /sys/module/*/parameters/wed*
cat /sys/module/*/parameters/wed*
```
Which means firewall hardware offload will go easier on CPU also to wifi devices. It comes with a drawback that roaming will drop TCP/UDP connections when roaming between them. Will not change the ait though.

Mining77, 2023-10-20:
So I have a request to our wise OpenWrt developers, please overclock the Xiaomi AX3200 router as much as possible :)

It would be nice if you would create an OpenWrt error reporting function. The reports would be sent automatically to "headquarters "and OpenWrt :)

does what you write here also go to OpenWrt programmers?

brada4, 2023-10-20:
There is no bug reported so far. Please close the non-issue.

closed 2023-10-20 by Mining77

raenye, 2023-10-20:
@Mining77 I think you have a wrong view of how open source projects work.
If you want something done, you could do it yourself and submit a contribution, or try to interest other people to do so.

Sometimes what you want exists, but is not well-documented and you need to read forum threads and figure out how to do it.

Writter77, 2023-10-22:
There is a problem with the speed on LAN, symmetrical connection 1 GB / s, Xiaomi AX3200 router download 888 Mbps, upload 945 Mbps.
Old router, manufacturer's software 944 download, 940 upload.

WiFi 2G without a speed problem. Admittedly, the speed on the 2G radio is only 108 Mbps, but it is symmetrical. WiFi 5G similar problem for LAN, download 650 Mbps, upload 720 Mbps.