openwrt/openwrt #13743: FritzBox 7520 all ethernet ports down after upgrade to 23.05.0

bofhus2, 2023-10-20:
### Describe the bug

FritzBox 7520 "new look" model. Works perfectly with 22.03.5 (openwrt-22.03.5-ipq40xx-generic-avm_fritzbox-7530-squashfs-sysupgrade.bin), after upgrade to 23.05.0 (openwrt-23.05.0-ipq40xx-generic-avm_fritzbox-7530-squashfs-sysupgrade.bin) all ethernet ports are down (there is no link on it, device connected to box doesn't show link). I was able to connect via WiFi and downgrade. When I go back to 22.03.5 all ethernet ports works again.

### OpenWrt version

r23497-6637af95aa

### OpenWrt target/subtarget

ipq40xx/generic

### Device

AVM FRITZ!Box 7530

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

robimarko, 2023-10-20:
Did you preserve the config during the update(Shouldnt be even possible) as the target switched to DSA

bofhus2, 2023-10-20:
I did the upgrade via LuCI. I left the "preserve conf" checkbox checked. There was a huge red warning, that preserving may not work. I've checked "force" and start upgrade. I assumed, that after upgrade I'll get default configuration. But after upgrade all ethernet links go down. But it was able to connect via WiFi - it didn't reset to default config, it was "my" SID and password. And after downgrade router was fully configured, restoring configuration was unnecessary.

robimarko, 2023-10-20:
Well, may this be the learning lesson that you are not supposed to force things.

That warning was added intentionally to prevent users from preserving the config as the target was migrated to DSA and the networking config is not compatible, it should even print that as the reason.

closed 2023-10-20 by mkresin