openwrt/openwrt #13753: SNMPD memory consumption

perachemarc, 2023-10-20:
### Describe the bug

It seems that snmpd suffers from a memory leak. The memory consumption keeps growing from a few Kb to multiple Mb after a few days. I have this issue on 3 different devices: 
- ramips/mt7621: Zbtlink ZBT-WE1326
- ipq40xx/generic: GL.iNet GL-B1300
- x86/64: BESSTAR TECH LIMITED GK41

### OpenWrt version

r23497-6637af95aa

### OpenWrt target/subtarget

ipq40xx/generic

### Device

GL.iNet GL-B1300

### Image kind

Official downloaded image

### Steps to reproduce

Just leave snmpd up and running

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-10-20:
https://github.com/openwrt/packages/issues/22428
TL; DR: - fixed v5.9.1-7 will fan out soon, you can close the issue and let time work for you

closed 2023-10-20 by perachemarc

mhoran, 2023-10-23:
Unfortunately I still see a memory leak with 5.9.7-1 on an EdgeRouter-X.

[image]