openwrt/openwrt #13756: 23.05.0: IPQ40xx: When a mac address was seen on LAN, this mac address can't communicate on bridged devices

StifflersMom, 2023-10-21:
### Describe the bug

On my D-LINK DAP-2610 devices a Wifi Station can't communicate (eg. get an ip adress from a dhcp server on lan network), if the mac address was seen on the LAN port before.
I described this in https://github.com/openwrt/openwrt/issues/13218 before, but the steps to reproduce this were not clear.

### OpenWrt version

r23497-6637af95aa

### OpenWrt target/subtarget

ipq40xx/generic

### Device

D-Link DAP 2610

### Image kind

Official downloaded image

### Steps to reproduce

Configure 2 APs as bridged to LAN, with DHCP server on the LAN network.
Let a wifi device (eg smartphone) talk to the network (eg using a ssid on AP 2)
Now switch to AP1 (the DAP-2610)

### Actual behaviour

Packets don't reach the LAN for ~ 300 seconds, if the station was seen on LAN before

### Expected behaviour

Packets should be always bridged.

### Additional info

May be the issue starts with changed VLAN handling between 22.03 and 23.05 ...

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-10-21:
How the things are connected? ie like dhcp on 3rd party switch and openwrt's as basic antennas in workstations' places?

StifflersMom, 2023-10-21:
The dhcp is connected to a 24 port POE+ switch, also both APs are connected to the 24port POE switch.
The same configuration works with 22.03.
If the smartphone doing the dhcp discover broadcast, you cant see the broadcast outside the AP. It doesn't reach the LAN.

brada4, 2023-10-21:
`cat /sys/class/net/br-lan/bridge/ageing_time` - it will be in 100ths
`brctl setageing br-lan 5` - set expiry interval smaller, too small means more frames will be broadcast to everybody around.

StifflersMom, 2023-10-21:
This does not change anything. For notice: I have an ubiquiti mesh-ac with  exactly the same configuration, there is no problem.
On the DAP-2610 the problem appears with removing of swconfig in 23.05

brada4, 2023-10-21:
What did you get from sysfs and did brctl change the value? DSA offloads are expected to follow it to the point.
`brctl showmacs br-lan` - "non-local" should have rolling expire timers
(i am just trying to figure out if kernel on CPU or DSA switch config is to blame)

StifflersMom, 2023-10-21:
```
root@DAP-UNTEN:/proc# cat /sys/class/net/br-lan/bridge/ageing_time
500

root@DAP-UNTEN:/proc# brctl showmacs br-lan
port no mac addr                is local?       ageing timer
  1     00:25:22:a6:ee:d8       no                 0.09
  1     04:d9:f5:cf:2c:ea       no                 0.54
  2     8c:7a:3d:c4:cb:0a       no                 4.41
  1     ec:ad:e0:79:98:40       yes                0.00
  1     ec:ad:e0:79:98:40       yes                0.00
  2     ec:ad:e0:79:98:48       yes                0.00
  2     ec:ad:e0:79:98:48       yes                0.00
  4     ee:ad:e0:79:98:40       yes                0.00
  4     ee:ad:e0:79:98:40       yes                0.00

```

The mac address of my mobile device doing the dhcp discover is not at the mac table.
If I do a tcpdump on the wifi:

```
tcpdump -n -i phy0-ap0 'ether host 8c:7a:3d:c4:cb:0a'
18:35:42.835968 8c:7a:3d:c4:cb:0a > ff:ff:ff:ff:ff:ff Null Unnumbered, xid, Flags [Response], length 6: 01 00
18:35:42.997264 IP6 :: > ff02::16: HBH ICMP6, multicast listener report v2, 4 group record(s), length 88
18:35:43.042253 IP 0.0.0.0.68 > 255.255.255.255.67: BOOTP/DHCP, Request from 8c:7a:3d:c4:cb:0a, length 310

```
8c:7a:3d:c4:cb:0a is the mac address of my mobile device.

xback, 2023-10-21:
Could you test again by building the master branch?
It contains this commit: [link](https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=b0048b314620cc99f155c8c5cd4541af5e99e578)

robimarko, 2023-10-22:
Assisted learning has been enabled on ipq40xx from the start because of roaming issues like these.

That patch wont help as ipq40xx has a modified qca8k copy in 23.05(In master we are reusing generic qca8k finally).

Kind of only idea I have is that maybe we need to disable learning on all ports by default and then let DSA enable it via bridge port attributes(this is the preffered way)

StifflersMom, 2023-10-22:
I rebuilt my build environment (sounds funny) and after serval hours of warming up my home with compiling, I managed to build the firmwar from the master branch. **I did not patch anything** and:
**It works without any problems with the master branch.**

raenye, 2023-10-22:
This is the same as #11650, right? not particular to `ipq40xx`.

siwind, 2023-10-29:
I build the openwrt-master branch using in my ipq40x device,  the problem is still there.

StifflersMom, 2023-10-29:
> I build the openwrt-master branch using in my ipq40x device, the problem is still there.

I have installed an actual snapshot on my DAP-2610 and the problem is not present. Actually, it is only at the 23.0.5 branch.
What is your device model?

siwind, 2023-11-17:
My router runs in my home with latest build openwrt main branch,  and  an AP connect to it's lan port without  dns service.
My mobile devices(such as phone) cann't get IP address from AP device, but can get IP address from my router.

The router is:
```root@OpenWrt:~# opkg list-installed | grep ath
ath10k-board-qca4019 - 20230804-1
ath10k-firmware-qca4019-ct - 2020-11-08-1
kmod-ath - 6.1.62+6.5-1
kmod-ath10k-ct - 6.1.62+2023-06-05-fadd0768-1
ucode-mod-math - 2023-11-07-a6e75e02-1
root@OpenWrt:~# uname -a
Linux OpenWrt 6.1.62 #0 SMP Fri Nov 17 12:12:09 2023 armv7l GNU/Linux

```

The AP device reports these error message:

```
[57953.642035] mt7530-mdio mdio-bus:1f: port 1 failed to delete 60:67:20:8f:bc:e4 vid 0 from fdb: -2
[57953.659874] mt7530-mdio mdio-bus:1f: port 1 failed to delete 60:67:20:8f:bc:e4 vid 0 from fdb: -2
[57953.677653] mt7530-mdio mdio-bus:1f: port 0 failed to delete 60:67:20:8f:bc:e4 vid 0 from fdb: -2
[57953.695417] mt7530-mdio mdio-bus:1f: port 0 failed to delete 60:67:20:8f:bc:e4 vid 0 from fdb: -2
[57966.400706] br-lan: received packet on lan3 with own address as source address (addr:e4:ea:83:af:98:8f, vlan:0)
[57978.908513] mt7530-mdio mdio-bus:1f: port 2 failed to delete 60:67:20:8f:bc:e4 vid 0 from fdb: -2
[57978.926500] mt7530-mdio mdio-bus:1f: port 0 failed to delete 60:67:20:8f:bc:e4 vid 0 from fdb: -2
[57978.944387] mt7530-mdio mdio-bus:1f: port 1 failed to delete 60:67:20:8f:bc:e4 vid 0 from fdb: -2
[57978.962189] mt7530-mdio mdio-bus:1f: port 0 failed to delete 60:67:20:8f:bc:e4 vid 0 from fdb: -2
[58023.541319] mt7530-mdio mdio-bus:1f: port 1 failed to delete 60:67:20:8f:bc:e4 vid 0 from fdb: -2
[58023.559104] mt7530-mdio mdio-bus:1f: port 0 failed to delete 60:67:20:8f:bc:e4 vid 0 from fdb: -2
[58023.576924] mt7530-mdio mdio-bus:1f: port 1 failed to delete 60:67:20:8f:bc:e4 vid 0 from fdb: -2
[58023.594927] mt7530-mdio mdio-bus:1f: port 0 failed to delete 60:67:20:8f:bc:e4 vid 0 from fdb: -2
[58029.249282] mt7530-mdio mdio-bus:1f: port 1 failed to delete 60:67:20:8f:bc:e4 vid 0 from fdb: -2
```

=========
But if my router(IPQ40xx) runs  previous openwrt version (the kenel is 5.xx), then any thing goes well.

raenye, 2023-11-17:
Did you try the script from #11650?

webknjaz, 2024-11-18:
Not sure if related but I'm seeing something similar on TOTOLINK X5000R. There's no VLAN but I use bonding. So `lan1` and `lan2` make up `bond0`. Then, `br-lan` is made of `bond0`, `lan3` and `lan4`. `lan3` and `lan4` don't have cables connected. `lan1` and `lan2` are connected to a managed QNAP switch using LACP.

At some point, the hotspot connectivity over ethernet is lost (devices attempting to connect don't get any DHCP traffic and I can't connect to the device via its known IP).
When I look into the QNAP device, it shows individual ports as having links but LACP shows as disabled or something (I didn't make a screenshot at the time).
In the past replugging the cables into the hotspot made it reachable again. This time I tried something new — I clicked edit on the LAG group in QNAP and saved. It then shown in the log that LAG was deleted and recreated. Not sure what it did under the hood.
And it started showing the LAG link speed as 100 Mbps. But it should be 2Gbps.
The AP became reachable w/o reboot so I lurked into the logs. And there's not that many clues but here's what I saw:

```console
[356264.580996] mt7530-mdio mdio-bus:1f: port 3 failed to delete xx:xx:xx:xx:xx:xx vid 0 from fdb: -2
[356264.590083] mt7530-mdio mdio-bus:1f: port 3 failed to delete xx:xx:xx:xx:xx:xx vid 0 from fdb: -2
[713319.589314] mt7530-mdio mdio-bus:1f lan1: Link is Down
[713319.589800] bond0: (slave lan1): speed changed to 0 on port 1
[713319.686944] bond0: (slave lan1): link status definitely down, disabling slave
[713319.694219] bond0: active interface up!
[713323.236861] mt7530-mdio mdio-bus:1f lan1: Link is Up - 1Gbps/Full - flow control off
[713323.336678] bond0: (slave lan1): link status definitely up, 1000 Mbps full duplex
[713324.012565] mt7530-mdio mdio-bus:1f lan1: Link is Down
[713324.013049] bond0: (slave lan1): speed changed to 0 on port 1
[713324.106678] bond0: (slave lan1): link status definitely down, disabling slave
[713326.573909] mt7530-mdio mdio-bus:1f lan1: Link is Up - 1Gbps/Full - flow control off
[713326.656695] bond0: (slave lan1): link status definitely up, 1000 Mbps full duplex
[713327.172462] mt7530-mdio mdio-bus:1f lan1: Link is Down
[713327.172942] bond0: (slave lan1): speed changed to 0 on port 1
[713327.206631] bond0: (slave lan1): link status definitely down, disabling slave
[713329.591776] mt7530-mdio mdio-bus:1f lan1: Link is Up - 1Gbps/Full - flow control off
[713329.646616] bond0: (slave lan1): link status definitely up, 1000 Mbps full duplex
[713330.065582] mt7530-mdio mdio-bus:1f lan1: Link is Down
[713330.066069] bond0: (slave lan1): speed changed to 0 on port 1
[713330.086622] bond0: (slave lan1): link status definitely down, disabling slave
[713332.595334] mt7530-mdio mdio-bus:1f lan1: Link is Up - 1Gbps/Full - flow control off
[713332.626613] bond0: (slave lan1): link status definitely up, 1000 Mbps full duplex
[713333.275188] mt7530-mdio mdio-bus:1f lan1: Link is Down
[713333.281087] bond0: (slave lan1): speed changed to 0 on port 1
[713333.306575] bond0: (slave lan1): link status definitely down, disabling slave
[713335.670496] mt7530-mdio mdio-bus:1f lan1: Link is Up - 1Gbps/Full - flow control off
[713335.726610] bond0: (slave lan1): link status definitely up, 1000 Mbps full duplex
[842877.422729] mt7530-mdio mdio-bus:1f lan1: Link is Down
[842877.428727] bond0: (slave lan1): speed changed to 0 on port 1
[842877.473049] bond0: (slave lan1): link status definitely down, disabling slave
[842879.751943] mt7530-mdio mdio-bus:1f lan1: Link is Up - 1Gbps/Full - flow control off
[842879.783063] bond0: (slave lan1): link status definitely up, 1000 Mbps full duplex
[948834.312417] mt7530-mdio mdio-bus:1f: port 3 failed to delete [5c:3e:1b:dd:fd:1c](xx:xx:xx:xx:xx:xx) vid 0 from fdb: -2
[948834.321412] mt7530-mdio mdio-bus:1f: port 3 failed to delete xx:xx:xx:xx:xx:xx vid 0 from fdb: -2
```

After all this, the DHCP traffic started flowing too.

I'm not sure where to dig at the moment so I'm just documenting this to see if other people think that the symptoms match enough to assume that the root cause is the same or entirely different. Can it be a bad cable? A driver bug? A combination of the two?

StifflersMom, 2024-12-08:
I checked this with 24.10.0-rc2 r28161-ea17e958b9 today - It's working **without any problems**.

The bug is present in all 23.05 versions (because the fix was not backported I guess).

closed 2024-12-08 by StifflersMom