openwrt/openwrt #13758: building image from imagebuilder fails at "Enabling dnsmasq" - missing /lib/functions.sh

risa2000, 2023-10-21:
### Describe the bug

I am building an ImageBuilder from the trunk/master and then use the IB to build the FW image for TP-Link Archer AX23.
Using the recent master and package feeds produces an error in FW image build:
```
Finalizing root filesystem...
/home/risa/openwrt/openwrt-imagebuilder-ramips-mt7621.Linux-x86_64/build_dir/target-mipsel_24kc_musl/root-ramips/etc/init.d/dnsmasq: line 4: /lib/functions.sh: No such file or directory
Enabling boot
Enabling bootcount
Enabling cron
Enabling ddns
./etc/init.d/dnsmasq: line 4: /lib/functions.sh: No such file or directory
Enabling dnsmasq
Enabling done
Enabling dropbear
Enabling firewall
Enabling gpio_switch
Enabling led
Enabling log
<... cut ...>
```
The previous built I did was from commit `r23436-8d06bc1751` on `2023-09-15`  was fine. So I assume something has changed either in `dnsmasq` package or some the dependency, between this date and now.

The full log of the build is attached.
[make_risa-wrt-ax23.log](https://github.com/openwrt/openwrt/files/13060844/make_risa-wrt-ax23.log)

### OpenWrt version

bb8fd41f9a6775734882fe9707f7ad5e5edfbf9e

### OpenWrt target/subtarget

ramips/mt7621

### Device

TP-Link Archer AX23 v1

### Image kind

Self-built image

### Diffconfig

```text
CONFIG_TARGET_ramips=y
CONFIG_TARGET_ramips_mt7621=y
CONFIG_TARGET_ramips_mt7621_DEVICE_tplink_archer-ax23-v1=y
CONFIG_DEVEL=y
CONFIG_TOOLCHAINOPTS=y
CONFIG_BUSYBOX_CUSTOM=y
# CONFIG_BINUTILS_USE_VERSION_2_40 is not set
CONFIG_BINUTILS_USE_VERSION_2_41=y
CONFIG_BINUTILS_VERSION="2.41"
CONFIG_BINUTILS_VERSION_2_41=y
CONFIG_BUSYBOX_CONFIG_DIFF=y
CONFIG_BUSYBOX_CONFIG_FEATURE_DIFF_DIR=y
CONFIG_BUSYBOX_CONFIG_FEATURE_DIFF_LONG_OPTIONS=y
CONFIG_BUSYBOX_CONFIG_FEATURE_NSLOOKUP_LONG_OPTIONS=y
# CONFIG_BUSYBOX_CONFIG_NTPD is not set
CONFIG_BUSYBOX_CONFIG_PATCH=y
CONFIG_DROPBEAR_ECC=y
# CONFIG_GCC_USE_VERSION_12 is not set
CONFIG_GCC_USE_VERSION_13=y
CONFIG_GCC_VERSION="13.2.0"
CONFIG_GCC_VERSION_13=y
# CONFIG_GDB is not set
CONFIG_IB=y
CONFIG_IB_STANDALONE=y
CONFIG_OPENSSL_ENGINE=y
CONFIG_OPENSSL_PREFER_CHACHA_OVER_GCM=y
CONFIG_OPENSSL_WITH_ASM=y
CONFIG_OPENSSL_WITH_CHACHA_POLY1305=y
CONFIG_OPENSSL_WITH_CMS=y
CONFIG_OPENSSL_WITH_DEPRECATED=y
CONFIG_OPENSSL_WITH_ERROR_MESSAGES=y
CONFIG_OPENSSL_WITH_IDEA=y
CONFIG_OPENSSL_WITH_MDC2=y
CONFIG_OPENSSL_WITH_PSK=y
CONFIG_OPENSSL_WITH_SEED=y
CONFIG_OPENSSL_WITH_SRP=y
CONFIG_OPENSSL_WITH_TLS13=y
CONFIG_OPENSSL_WITH_WHIRLPOOL=y
CONFIG_PACKAGE_atop=m
CONFIG_PACKAGE_ddns-scripts=m
CONFIG_PACKAGE_ddns-scripts-services=m
CONFIG_PACKAGE_dnsmasq=m
CONFIG_PACKAGE_hostapd-basic-openssl=m
CONFIG_PACKAGE_hostapd-openssl=m
CONFIG_PACKAGE_hostapd-utils=m
CONFIG_PACKAGE_htop=m
CONFIG_PACKAGE_iftop=m
CONFIG_PACKAGE_ip-bridge=m
CONFIG_PACKAGE_ip-full=m
CONFIG_PACKAGE_iperf3=m
CONFIG_PACKAGE_ipset=m
CONFIG_PACKAGE_iwcap=m
CONFIG_PACKAGE_kmod-crypto-kpp=m
CONFIG_PACKAGE_kmod-crypto-lib-chacha20=m
CONFIG_PACKAGE_kmod-crypto-lib-chacha20poly1305=m
CONFIG_PACKAGE_kmod-crypto-lib-curve25519=m
CONFIG_PACKAGE_kmod-crypto-lib-poly1305=m
CONFIG_PACKAGE_kmod-fs-exfat=m
CONFIG_PACKAGE_kmod-fs-vfat=m
CONFIG_PACKAGE_kmod-ipt-core=m
CONFIG_PACKAGE_kmod-ipt-ipset=m
CONFIG_PACKAGE_kmod-mii=m
CONFIG_PACKAGE_kmod-nf-ipt=m
CONFIG_PACKAGE_kmod-nls-base=y
CONFIG_PACKAGE_kmod-nls-cp437=m
CONFIG_PACKAGE_kmod-nls-iso8859-1=m
CONFIG_PACKAGE_kmod-nls-utf8=m
CONFIG_PACKAGE_kmod-ppp=m
CONFIG_PACKAGE_kmod-pppoe=m
CONFIG_PACKAGE_kmod-pppox=m
CONFIG_PACKAGE_kmod-scsi-core=m
CONFIG_PACKAGE_kmod-slhc=m
CONFIG_PACKAGE_kmod-udptunnel4=m
CONFIG_PACKAGE_kmod-udptunnel6=m
CONFIG_PACKAGE_kmod-usb-core=y
CONFIG_PACKAGE_kmod-usb-ehci=y
CONFIG_PACKAGE_kmod-usb-ledtrig-usbport=y
CONFIG_PACKAGE_kmod-usb-net=m
CONFIG_PACKAGE_kmod-usb-net-cdc-ether=m
CONFIG_PACKAGE_kmod-usb-net-rndis=m
CONFIG_PACKAGE_kmod-usb-ohci=m
CONFIG_PACKAGE_kmod-usb-printer=m
CONFIG_PACKAGE_kmod-usb-storage=m
CONFIG_PACKAGE_kmod-usb-uhci=m
CONFIG_PACKAGE_kmod-usb2=y
CONFIG_PACKAGE_kmod-wireguard=m
CONFIG_PACKAGE_libatomic=m
CONFIG_PACKAGE_libbpf=m
CONFIG_PACKAGE_libcap=m
CONFIG_PACKAGE_libelf=m
CONFIG_PACKAGE_libevent2-core=m
CONFIG_PACKAGE_libevent2-pthreads=m
CONFIG_PACKAGE_libiperf3=m
CONFIG_PACKAGE_libipset=m
# CONFIG_PACKAGE_libmbedtls is not set
CONFIG_PACKAGE_libncurses=m
CONFIG_PACKAGE_libopenssl=m
CONFIG_PACKAGE_libopenssl-conf=m
CONFIG_PACKAGE_libopenssl-legacy=m
CONFIG_PACKAGE_libpcap=m
CONFIG_PACKAGE_libstdcpp=m
# CONFIG_PACKAGE_libustream-mbedtls is not set
CONFIG_PACKAGE_libustream-openssl=m
CONFIG_PACKAGE_mcproxy=m
CONFIG_PACKAGE_nstat=m
CONFIG_PACKAGE_ntp-utils=m
CONFIG_PACKAGE_ntpd=m
CONFIG_PACKAGE_odhcp6c=m
CONFIG_PACKAGE_odhcpd-ipv6only=m
CONFIG_PACKAGE_p910nd=m
# CONFIG_PACKAGE_ppp is not set
CONFIG_PACKAGE_tcpdump=m
CONFIG_PACKAGE_terminfo=m
CONFIG_PACKAGE_ucode-mod-nl80211=m
CONFIG_PACKAGE_ucode-mod-rtnl=m
CONFIG_PACKAGE_ucode-mod-uloop=m
CONFIG_PACKAGE_vim=m
CONFIG_PACKAGE_vim-full=m
CONFIG_PACKAGE_vim-fuller=m
CONFIG_PACKAGE_vim-runtime=m
CONFIG_PACKAGE_wireguard-tools=m
CONFIG_PACKAGE_wpa-cli=m
# CONFIG_PACKAGE_wpad-basic-mbedtls is not set
CONFIG_PACKAGE_wpad-basic-openssl=m
CONFIG_PACKAGE_wpad-openssl=m
CONFIG_PACKAGE_zlib=m
CONFIG_PACKAGE_kmod-lib-crc-ccitt=y
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-10-21:
Add explicit base-files, otherwise whatever you cached long ago gets passed on to image.

risa2000, 2023-10-21:
@brada4
I have found several references in the IB to "base-files". Not sure what exactly do you mean:
There is a package folder for `ramips/mt7621` with the following content:
```
risa@gentoo-vb ~/openwrt $ ls -R openwrt-imagebuilder-ramips-mt7621.Linux-x86_64/target/linux/ramips/mt7621/base-files
openwrt-imagebuilder-ramips-mt7621.Linux-x86_64/target/linux/ramips/mt7621/base-files:
etc  lib  sbin

openwrt-imagebuilder-ramips-mt7621.Linux-x86_64/target/linux/ramips/mt7621/base-files/etc:
board.d  hotplug.d  init.d  rc.button  uci-defaults

openwrt-imagebuilder-ramips-mt7621.Linux-x86_64/target/linux/ramips/mt7621/base-files/etc/board.d:
01_leds  02_network  03_gpio_switches  05_compat-version

openwrt-imagebuilder-ramips-mt7621.Linux-x86_64/target/linux/ramips/mt7621/base-files/etc/hotplug.d:
firmware  ieee80211

openwrt-imagebuilder-ramips-mt7621.Linux-x86_64/target/linux/ramips/mt7621/base-files/etc/hotplug.d/firmware:
10-ath9k-eeprom

openwrt-imagebuilder-ramips-mt7621.Linux-x86_64/target/linux/ramips/mt7621/base-files/etc/hotplug.d/ieee80211:
05-wifi-migrate  10_fix_wifi_mac

openwrt-imagebuilder-ramips-mt7621.Linux-x86_64/target/linux/ramips/mt7621/base-files/etc/init.d:
bootcount

openwrt-imagebuilder-ramips-mt7621.Linux-x86_64/target/linux/ramips/mt7621/base-files/etc/rc.button:
lights_toggle

openwrt-imagebuilder-ramips-mt7621.Linux-x86_64/target/linux/ramips/mt7621/base-files/etc/uci-defaults:
01_enable_packet_steering

openwrt-imagebuilder-ramips-mt7621.Linux-x86_64/target/linux/ramips/mt7621/base-files/lib:
upgrade

openwrt-imagebuilder-ramips-mt7621.Linux-x86_64/target/linux/ramips/mt7621/base-files/lib/upgrade:
iodata.sh  platform.sh  ubnt.sh  zyxel.sh

openwrt-imagebuilder-ramips-mt7621.Linux-x86_64/target/linux/ramips/mt7621/base-files/sbin:
fixup-mac-address
```
There is also a package:
```
openwrt-imagebuilder-ramips-mt7621.Linux-x86_64/packages/base-files_1544-r24187-bb8fd41f9a_mipsel_24kc.ipk
```

brada4, 2023-10-21:
base-files provide /lib/functions.sh (along with all filesystem default mount points), it is assumed to be omnipresent.

risa2000, 2023-10-21:
Ok, what do you want me to post here exactly?

brada4, 2023-10-21:
I want you to add base-files explicitly to your image.

risa2000, 2023-10-21:
Ah, ok, I did not get it. The package `base-files` is pulled as a dependency into the image build as you can see in the log I posted in the first post.

I have added `base-files` explicitly to the image build script, but it does not change anything. The same package is pulled in, the build output just says at the explicit pull in:
```
Package base-files (1544-r24187-bb8fd41f9a) installed in root is up to date.
```
and the `dnsmasq` "enabling" fails the same way.

For the info here is the build script I have been using so far, adding `base-files` to `PACKAGE_LIST` does not help:
```bash
#!/bin/bash
ALIAS=risa-wrt-ax23
TARGET=ramips
SUBTARGET=mt7621
OPENWRT_ROOT=~/openwrt
IB_ROOT=${OPENWRT_ROOT}/openwrt-imagebuilder-${TARGET}-${SUBTARGET}.Linux-x86_64

PACKAGES_LIST=" \
    -libustream-wolfssl \
    -libustream-mbedtls \
    libustream-openssl \
    atop \
    ddns-scripts \
    hostapd-basic-openssl \
    hostapd-utils \
    htop \
    iftop \
    ip-full \
    ip-bridge \
    iperf3 \
    ipset \
    iw \
    mcproxy \
    nstat \
    ntpd \
    -ppp \
    -ppp-mod-pppoe \
    tcpdump \
    vim-runtime \
    vim-full \
    wireguard-tools \
    -wpa-cli \
    -wpad-basic-wolfssl \
    -wpad-basic-mbedtls \
    -wpad-basic-openssl \
    "

DISABLED="mcproxy"

CFG_FILES=${OPENWRT_ROOT}/files/${ALIAS}
rm $(find ${CFG_FILES} -name "\.*\.un~")

cd ${IB_ROOT}
make clean
make image PROFILE=tplink_archer-ax23-v1 FILES=${CFG_FILES} PACKAGES="${PACKAGES_LIST}" DISABLED_SERVICES="${DISABLED}" 2>&1 | tee make_${ALIAS}.log

[[ -e "${OPENWRT_ROOT}/images/${ALIAS}" ]] && rm -rf "${OPENWRT_ROOT}/images/${ALIAS}"
cp -rv ${IB_ROOT}/bin/targets/${TARGET}/${SUBTARGET} ${OPENWRT_ROOT}/images/${ALIAS} | tee -a make_${ALIAS}.log
cp -v make_${ALIAS}.log ${OPENWRT_ROOT}/images/${ALIAS} | tee -a make_${ALIAS}.log
mv -v make_${ALIAS}.log ${OPENWRT_ROOT}/admin
```

Maybe the package itself is not built correctly? I am normally doing a clean build from the master branch, everything updated (as mentioned in the first post).

brada4, 2023-10-21:
just add base-files to package list? There is no maybe, package is absent from final image and it is not a dependency of any package.

risa2000, 2023-10-21:
@brada4
`base-file` package is in my build with the `PACKAGE_LIST` I posted right above already pulled in as a dependency. See the build log in the first post. Adding it to the `PACKAGE_LIST` explicitly, did not change a thing.
Here is a complete log with `base-files` stated explicitly in `PACKAGE_LIST`: 
[make_risa-wrt-ax23.log](https://github.com/openwrt/openwrt/files/13061847/make_risa-wrt-ax23.log)

brada4, 2023-10-21:
Building image succeeds. It just does not emit default config from 2 packages?

dave14305, 2023-10-22:
This is a new addition since 6b23836071b171499f3983286a37a8d5c15eeab9. Maybe it wasn’t tested with image builder.

Ping @yogo1212 and @pprindeville

pprindeville, 2023-10-22:
Sorry, what is the actual issue?

dave14305, 2023-10-22:
> Sorry, what is the actual issue?

The recent commit with the new ipcalc function manually sources /lib/functions.sh at the top of dnsmasq.init. In an ImageBuilder build, the init script is run on the build host in order to enable the service in the build, but cannot find /lib/functions.sh on the build host. rc.common sources /lib/functions.sh itself, so I doubt if dnsmasq.init also needs to source it explicitly in order to use ipcalc().

A cheap hack workaround would be to source the script only from within the start() function so it isn't invoked during enable().

At least that's my understanding of the problem.

pprindeville, 2023-10-22:
Okay. I don't want to have too many fingers in the pie, so I'll let @yogo1212 address that one... Plus as `isc-dhcp` maintainer I don't actually use `dnsmasq` so testing would be a problem.

pprindeville, 2023-10-22:
Well, I couldn't stop myself.  Here's the trivial PR, but I've not tested it.

risa2000, 2023-10-22:
@pprindeville
This fixes the image builder build all right.
On the target filesystem `/lib/functions.sh` is instantiated correctly, so if it is correctly pulled by `/etc/rc.common` I assume this fixes it.
```bash
risa@gentoo-vb ~/openwrt/openwrt-imagebuilder-ramips-mt7621.Linux-x86_64/build_dir/target-mipsel_24kc_musl/root-ramips $ ls lib
config        ld-musl-mipsel-sf.so.1       libc.so                     libpreload-seccomp.so  libubus.so.20230605  modules  upgrade
firmware      libatomic.so.1               libfstools.so               libpreload-trace.so    libuci.so            netifd   wifi
functions     libatomic.so.1.2.0           libgcc_s.so.1               libsetlbf.so           libustream-ssl.so    network
functions.sh  libblobmsg_json.so.20230523  libjson_script.so.20230523  libubox.so.20230523    libvalidate.so       preinit
```

closed 2023-11-02 by openwrt-bot

pprindeville, 2023-11-05:
@risa2000 Can this be closed out?

risa2000, 2023-11-05:
> @risa2000 Can this be closed out?

It seems to be closed already. For what my problem concerns, yes, it fixes this (as I mentioned earlier).

commit ac199c57 (2023-11-02): dnsmasq: don't source functions.sh twice

It's already pulled in from /etc/rc.common.

Fixes: #13758

Fixes: 6b23836071b1 ("package: avoid the use of eval to parse ipcalc.sh output")
PR #13766 (2023-11-02): dnsmasq: don't source functions.sh twice
It's already pulled in from /etc/rc.common.

Fixes #13758.

cc: @dave14305 @yogo1212