openwrt/openwrt #13759: 23.05.0: Orange Pi R1 Plus LTS: eth0 not working

jmlzr, 2023-10-21:
### Describe the bug

No matter how configured eth0 is not passing (neither sending nor receiving) traffic. Link state is shown correctly though.

I was previously using the Xulong fork (https://github.com/orangepi-xunlong/openwrt) where both interfaces work flawlessly.

### OpenWrt version

r23497-6637af95aa

### OpenWrt target/subtarget

rockchip/armv8

### Device

Xunlong Orange Pi R1 Plus LTS

### Image kind

Official downloaded image

### Steps to reproduce

Flash the official 23.05 squashfs image.

Configure eth0 as static LAN interface. For example with that network config (to have both devices in bridged mode as lan):
```
config interface 'loopback'
	option device 'lo'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'

config globals 'globals'
	option ula_prefix 'fd14:eac3:1e2c::/48'

config device
	option name 'br-lan'
	option type 'bridge'
	list ports 'eth0'
	list ports 'eth1'

config interface 'lan'
	option device 'br-lan'
	option proto 'static'
	option ipaddr '192.168.1.1'
	option netmask '255.255.255.0'
	option ip6assign '60'
```
(I also tried various other setups like other subnet/IP for eth0, only configuring eth0, my working config from the xulong fork, ...)

All other configuration untouched.

Connect a computer to eth0. Set a static IP on the computer since DHCP will not work.

Try to ping in either direction.

### Additional info

Full serial log:

```
U-Boot TPL 2021.07-OpenWrt-r23497-6637af95aa (Oct 09 2023 - 21:45:35)
LPDDR3, 333MHz
BW=32 Col=10 Bk=8 CS0 Row=14 CS1 Row=14 CS=2 Die BW=16 Size=1024MB
Trying to boot from BOOTROM
Returning to boot ROM...

U-Boot SPL 2021.07-OpenWrt-r23497-6637af95aa (Oct 09 2023 - 21:45:35 +0000)
[... 383 lines cut ...]
[    8.671210] br-lan: port 1(eth0) entered blocking state
[    8.671741] br-lan: port 1(eth0) entered disabled state
[    8.672577] device eth0 entered promiscuous mode
[    8.675257] br-lan: port 1(eth0) entered blocking state
[    8.675786] br-lan: port 1(eth0) entered forwarding state
[    8.692896] br-lan: port 2(eth1) entered blocking state
[    8.693430] br-lan: port 2(eth1) entered disabled state
[    8.694285] device eth1 entered promiscuous mode
[    8.695716] br-lan: port 2(eth1) entered blocking state
[    8.696251] br-lan: port 2(eth1) entered forwarding state
[    9.606985] br-lan: port 1(eth0) entered disabled state
[    9.607808] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[    9.609182] br-lan: port 2(eth1) entered disabled state
[   11.767495] rk_gmac-dwmac ff540000.ethernet eth0: Link is Up - 1Gbps/Full - flow control rx/tx
[   11.768412] br-lan: port 1(eth0) entered blocking state
[   11.768909] br-lan: port 1(eth0) entered forwarding state
[   12.115206] r8152 4-1:1.0 eth1: Promiscuous mode enabled
[   12.116025] r8152 4-1:1.0 eth1: carrier on
[   12.117190] br-lan: port 2(eth1) entered blocking state
[   12.117691] br-lan: port 2(eth1) entered forwarding state
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

regae, 2023-10-21:
try this

```
--- a/arch/arm64/boot/dts/rockchip/rk3328-orangepi-r1-plus-lts.dts
+++ b/arch/arm64/boot/dts/rockchip/rk3328-orangepi-r1-plus-lts.dts
@@ -68,6 +68,10 @@
 	};
 };
 
+&io_domains {
+	vccio4-supply = <&vcc_18>;
+};
+
 &pinctrl {
 	button {
 		reset_button_pin: reset-button-pin {
```

jmlzr, 2023-10-21:
@regae What am I supposed to do with that patch?

Importing it with quilt doesn't work and it also won't apply to `./build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.134/arch/arm64/boot/dts/rockchip/rk3328-orangepi-r1-plus-lts.dts`.

I have the v23.05.0 tag checked out and built an image for this device.

regae, 2023-10-22:
sorry, post/patch edited
just follow the idea
```
+&io_domains {
+	vccio4-supply = <&vcc_18>;
+};
```

raenye, 2023-10-22:
@jmlzr kernel patches can be placed at `target/linux/rockchip/patches-5.15/`. These are applied to the `build_dir` tree during `make`.

kristof-dragon, 2023-10-24:
This seems to be the same issue that was also reported in #13117  ( https://github.com/openwrt/openwrt/issues/13117 )

jmlzr, 2023-10-25:
Following @regae suggestion i put the following patch into `target/linux/rockchip/patches-5.15/`:

```
--- a/arch/arm64/boot/dts/rockchip/rk3328-orangepi-r1-plus-lts.dts
+++ b/arch/arm64/boot/dts/rockchip/rk3328-orangepi-r1-plus-lts.dts
@@ -38,3 +38,7 @@
 		};
 	};
 };
+
+&io_domains {
+	vccio4-supply = <&vcc_18>;
+};
```

But this did not fix the issue.

NHellFire, 2023-10-28:
> Following @regae suggestion i put the following patch into `target/linux/rockchip/patches-5.15/`:
> But this did not fix the issue.

Are you sure the kernel was rebuilt with it applied? I've just built from v23.05.0 with that applied and now have eth0 working. Previously I could ping and nothing more, now HTTP is working and iperf is managing ~800Mb/s

Kuzya-2023, 2023-10-29:
Good day
I created a file in target/linux/rockchip/patches-5.15/ with custom content
--- a/arch/arm64/boot/dts/rockchip/rk3328-orangepi-r1-plus-lts.dts
+++ b/arch/arm64/boot/dts/rockchip/rk3328-orangepi-r1-plus-lts.dts
@@ -38,3 +38,7 @@
  };
  };
  };
+
+&io_domains {
+ vccio4-supply = <&vcc_18>;
+};
But plowed the next mistake:
pplying /home/kuzya/openwrt/target/linux/rockchip/patches-5.15/rk3328-orangepi-r1-plus-lts.dts.patch using plaintext:
patching file arch/arm64/boot/dts/rockchip/rk3328-orangepi-r1-plus-lts.dts
patch unexpectedly ends in the middle of the line
patch: **** malformed patch at line 9:

Patch failed! Please fix /home/kuzya/openwrt/target/linux/rockchip/patches-5.15/rk3328-orangepi-r1-plus-lts.dts.patch!
make[4]: *** [Makefile:32: /home/kuzya/openwrt/build_dir/target-aarch64_generic_musl/linux-rockchip_armv8/linux-5.15.134/.prepared_6f603c91aef8fd38ebcd562374f29e0a] Error 1
make[4]: Leaving directory '/home/kuzya/openwrt/target/linux/rockchip'
make[3]: *** [Makefile:11: compile] Error 2
make[3]: Leaving directory '/home/kuzya/openwrt/target/linux'
time: target/linux/compile#13.94#11.46#24.44
     ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:30: target/linux/compile] Error 1
make[2]: Leaving directory '/home/kuzya/openwrt'
make[1]: *** [target/Makefile:23: /home/kuzya/openwrt/staging_dir/target-aarch64_generic_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/kuzya/openwrt'
make: *** [/home/kuzya/openwrt/include/toplevel.mk:232: world]
Tell me what I'm doing wrong
Sorry for my English!!!!

NHellFire, 2023-10-29:
@Kuzya-2023 add a blank line to the end of the patch.

jmlzr, 2023-10-30:
>Are you sure the kernel was rebuilt with it applied? I've just built from v23.05.0 with that applied and now have eth0 working. Previously I could ping and nothing more, now HTTP is working and iperf is managing ~800Mb/s

Yes, the Kernel was rebuilt and I confirmed the patch was applied by looking at the dts file in the build_dir. If you had ping working then you are having a different issue! This bug is about eth0 is not transceiving any kind of trafic.
Do you have the the LTS version of the Orange Pi R1 Plus?

Kuzya-2023, 2023-10-30:
Thanks, everything worked for me after applying the patch

NHellFire, 2023-10-30:
> If you had ping working then you are having a different issue! This bug is about eth0 not transceiving any kind of traffic. Do you have the the LTS version of the Orange Pi R1 Plus?

Yes, I have the LTS. Ping was the only thing that worked after assigning a static IP, any bigger packets just disappeared. Which was resolved after building v23.05.0 with the device tree patch as the only change.

Device info:
```
root@OpenWrt:~# grep . /proc/device-tree/compatible
xunlong,orangepi-r1-plus-lts
rockchip,rk3328
root@OpenWrt:~# cat /etc/os-release
NAME="OpenWrt"
VERSION="23.05.0"
ID="openwrt"
ID_LIKE="lede openwrt"
PRETTY_NAME="OpenWrt 23.05.0"
VERSION_ID="23.05.0"
HOME_URL="https://openwrt.org/"
BUG_URL="https://bugs.openwrt.org/"
SUPPORT_URL="https://forum.openwrt.org/"
BUILD_ID="r23497-6637af95aa"
OPENWRT_BOARD="rockchip/armv8"
OPENWRT_ARCH="aarch64_generic"
OPENWRT_TAINTS="no-all"
OPENWRT_DEVICE_MANUFACTURER="OpenWrt"
OPENWRT_DEVICE_MANUFACTURER_URL="https://openwrt.org/"
OPENWRT_DEVICE_PRODUCT="Generic"
OPENWRT_DEVICE_REVISION="v0"
OPENWRT_RELEASE="OpenWrt 23.05.0 r23497-6637af95aa"
```

And source:
```
nathan:~/source/openwrt/orangepi-r1/openwrt$ git describe --tags --exact-match
v23.05.0
nathan:~/source/openwrt/orangepi-r1/openwrt$ git status
HEAD detached at v23.05.0
Untracked files:
  (use "git add <file>..." to include in what will be committed)
        target/linux/rockchip/patches-5.15/999-orangepi-r1-plus-lts-regulators.patch

nothing added to commit but untracked files present (use "git add" to track)
```

kulichs, 2023-11-01:
I confirm that the patch is functional. Thanks

jmlzr, 2023-11-04:
I retested and it turns out that the example network config from my initial report where both interfaces are bonded to the lan bridge has eth0 not working even with the patch my production config actually works as expected with the patched device tree.
So either there is a separate bug with bridging or my provided config has some issue...

Anyhow I can confirm the patch works now as well.

hauke, 2023-11-05:
@regae Could you please create a pull request with this change. Please create it as a separate patch. It would be nice if you could also send this to the upstream Linux kernel. 
Do you or anyone else have access to the schematics of this device or the none LTS version? If you have access could you please check if the other power domains are correctly described.
Is this only a problem for the LTS variant or do we have this problem also on the none LTS variant?

regae, 2023-11-06:
> @regae Could you please create a pull request with this change. Please create it as a separate patch. It would be nice if you could also send this to the upstream Linux kernel. Do you or anyone else [...]

i am sorry, unable to do so, i wish someone will, based on this patch.
i do not have the schematics, i only tinker with devices in my possession, on this case LTS version.
been running R1 LTS with this patch since r1 LTS is merged upstream.

maybe wait for feedback from other users before someone submit a PR.

NHellFire, 2023-11-06:
The schematics are published:
http://www.orangepi.org/html/hardWare/computerAndMicrocontrollers/service-and-support/Orange-Pi-R1-Plus-LTS.html

jmlzr, 2023-11-06:
Schematics are openly available although they are of limited quality and comments are often in Chinese.

Orange Pi R1 Plus: http://www.orangepi.org/html/hardWare/computerAndMicrocontrollers/service-and-support/Orange-Pi-R1-Plus.html
Orange Pi R1 Plus LTS: http://www.orangepi.org/html/hardWare/computerAndMicrocontrollers/service-and-support/Orange-Pi-R1-Plus-LTS.html

As far as I can see the power domains are the same for both versions. The manufacturer supplied device tree (see the Xulong openwrt fork) also uses the same `io_domain` settings (with `vccio4-supply = <&vcc_18>;`) for both versions. So this should probably be fixed in the non-LTS version. I only have the LTS hardware so I can't test this though.

When comparing the device trees i noticed two issues with the LTS:
1. The manufacturer disables some DMC operating points for the LTS. Since the device tree's here use a different include chain than in the fork it's hard to tell if they are enabled here. Might lead to some possible issues...
2. Unlike the non-LTS version the LTS version supports UHS cards. Unlike the fork, official openwrt does not enable that though.

See: https://github.com/orangepi-xunlong/openwrt/blob/master/target/linux/rockchip/patches-5.10/206-Add-support-for-OrangePi-R1-Plus-LTS.patch

I'll open a new issues for at least the second point.

jmlzr, 2023-11-06:
I had a closer look at the schematic, the part regarding VCCIO4 is identical for both. On page 4 it shows a "Option" to either have VCC_IO or VCC_18 connected to VCCIO4. On page 10 it shows VCCIO4 connected to VCC_IO. That must be an error though.

The only difference in power domains is VCCIO3 which for the LTS will later need to be configured to the sdmmc regulator to support UHS operation.

@hauke I would be willing to create a PR and upstream to get this fix moving forward. Shall I apply the patch to the non-LTS version considering the schematic is the same and the manufacturer does so in their device tree but me not having hardware to actually test?

kulichs, 2023-11-07:
The patch is also functional for kernel 6.1.60. Tested on LTS

teqlee, 2023-11-09:
Patch worked with R1plus LTS kernel 6.1.

1715173329, 2023-11-15:
> I had a closer look at the schematic, the part regarding VCCIO4 is identical for both. On page 4 it shows a "Option" to either have VCC_IO or VCC_18 connected to VCCIO4. On page 10 it shows VCCIO4 connected to VCC_IO. That must be an error though.

On page 4:

[image]
They made 3v3 the default option, I guess this is the reason why dts sets `VCCIO4` to `vcc_io`.

> The manufacturer supplied device tree (see the Xulong openwrt fork) also uses the same io_domain settings (with `vccio4-supply `= `<&vcc_18>;`) for both versions. 

They just simply reused NanoPi R2S's dts.

> The manufacturer disables some DMC operating points for the LTS. Since the device tree's here use a different include chain than in the fork it's hard to tell if they are enabled here. Might lead to some possible issues...

DMC (Dynamic Memory Controller) requires additional driver and closed-source blobs (miniloader), which is not supported here.
They limited the RAM frequency to 786 MHz, but without the miniloader we can only get 333 MHz... So, it's "safe".

> Unlike the non-LTS version the LTS version supports UHS cards. But official openwrt does not enable that.

Due to the hardware bug on rockchip SoCs, we don't enable UHS mode anyway.
You can see they added some hacks to the mmc subsystem to work around (re)boot issues.

cantalupo555, 2023-11-16:
Will we have this fix in OpenWrt 23.05.2?

hauke, 2023-11-26:
I do not know if the schematics are matching the real hardware. I had some Orange Pi devices where it did not match.

I also do not know so much about the SoC. We could also try to force the rgmii signals to 3.3V on the PHY.

hauke, 2023-11-26:
I think you can operate the YT8531C PHY in 1.8V mode, see
https://datasheet.lcsc.com/lcsc/2302091130_Motorcomm-YT8531C_C5357044.pdf

The RTL8111F datasheet does not mention 1.8Volt mode:
https://datasheet.lcsc.com/lcsc/1811141436_Realtek-Semicon-RTL8111F-CG_C220087.pdf

Maybe we somehow configure the YT8531C PHY into 1.8 mode, but do not configure the MAC correctly. Maybe this works when we do not have any PHY driver.

hauke, 2023-11-27:
I looked more closely into the schematics.

This is the pin strapping configuration of the PHY:
[image]

CFG_EXT is pulled up:
[image]

CFG_LDO0 and CFG_LDO1 are pulled down:
[image]

I assume the PHY is configured into 3.3 volts mode using the PIN strapping.

1715173329, 2023-12-13:
> btw this change looks interesting
> [torvalds/linux@79e71d9](https://github.com/torvalds/linux/commit/79e71d9569959b15ec7b150d0c43f993c2f6e32e)
> [torvalds/linux@7a561e9](https://github.com/torvalds/linux/commit/7a561e9351ae7e3fb1f08584d40b49c1e55dde60)

```patch
diff --git a/arch/arm64/boot/dts/rockchip/rk3328-orangepi-r1-plus-lts.dts b/arch/arm64/boot/dts/rockchip/rk3328-orangepi-r1-plus-lts.dts
index 5d7d567283e5..4237f2ee8fee 100644
--- a/arch/arm64/boot/dts/rockchip/rk3328-orangepi-r1-plus-lts.dts
+++ b/arch/arm64/boot/dts/rockchip/rk3328-orangepi-r1-plus-lts.dts
@@ -26,9 +26,11 @@ yt8531c: ethernet-phy@0 {
 			compatible = "ethernet-phy-ieee802.3-c22";
 			reg = <0>;
 
+			motorcomm,auto-sleep-disabled;
 			motorcomm,clk-out-frequency-hz = <125000000>;
 			motorcomm,keep-pll-enabled;
-			motorcomm,auto-sleep-disabled;
+			motorcomm,rx-clk-drv-microamp = <5020>;
+			motorcomm,rx-data-drv-microamp = <5020>;
 
 			pinctrl-0 = <&eth_phy_reset_pin>;
 			pinctrl-names = "default";
```

now works fine

1715173329, 2023-12-13:
@regae could you test this please?

cantalupo555, 2023-12-13:
@1715173329 Was this bug correction accepted in the mainline kernel?
Is the correction available on Immortalwrt?

regae, 2023-12-13:
i will test later.
how did you get the value of drv-microamp _(5020)_ ?

> @regae could you test this please?
> 
> fyi patch for motorcomm: https://github.com/immortalwrt/immortalwrt/blob/22616d37fecd150c2525689118a113069b928276/target/linux/generic/backport-6.1/791-v6.6-11-net-phy-motorcomm-Add-pad-drive-strength-cfg-support.patch

1715173329, 2023-12-13:
> @1715173329 Was this bug correction accepted in the mainline kernel? Is the correction available on Immortalwrt?

Not yet. Needs someone to comfirm it. If this is the correct fix I will send it to upstream kernel.

> how did you get the value of drv-microamp (5020) ?

The easiest way, check each of them until find a working one.

1715173329, 2023-12-13:
There's a new datasheet https://www.electronshik.ru/pdf/y/YT8531H.pdf which contains more description (see 6.1.15), but it's still not clear enough to me.

cantalupo555, 2023-12-13:
> Not yet. Needs someone to comfirm it. If this is the correct fix I will send it to upstream kernel.

Thank you.

regae, 2023-12-13:
with https://github.com/torvalds/linux/commit/7a561e9351ae7e3fb1f08584d40b49c1e55dde60 applied,
https://github.com/openwrt/openwrt/issues/13759#issuecomment-1853852694 applied,
below patch removed.
```
&io_domains {
	vccio4-supply = <&vcc_18>;
};
```
so far..
pppoe established, connection seems working fine.
i will keep monitoring.

tanjilbhuiyan, 2023-12-14:
if need to be tested, i would like to help. though will need some guideline. just bough orange pi r1 plus lts 1 week ago.

matey1116, 2023-12-15:
> with [torvalds/linux@7a561e9](https://github.com/torvalds/linux/commit/7a561e9351ae7e3fb1f08584d40b49c1e55dde60) applied, [#13759 (comment)](https://github.com/openwrt/openwrt/issues/13759#issuecomment-1853852694) applied, below patch removed.
> 
> ```
> &io_domains {
> 	vccio4-supply = <&vcc_18>;
> };
> ```
> 
> so far.. pppoe established, connection seems working fine. i will keep monitoring.

This worked for me, set it up couple hours ago, seems to be stable, got max speeds on the interface and before it was maxing ~10mbps

ohabu, 2023-12-17:
Tested on kernel 5.15(23.05.2 tag) and kernel 6.1(main branch), multiple VLAN setup in a noisy environent. Most of the time it ran on kernel 5.15. I had no issues.

1715173329, 2023-12-17:
Thank you all for test! I have sent this patch to upstream kernel:
https://lore.kernel.org/linux-rockchip/20231216040723.17864-1-cnsztl@gmail.com/T/#u

commit 3645ac8a (2024-01-04): rockchip: configure eth pad driver strength for orangepi r1 plus lts

The default strength is not enough to provide stable connection
under 3.3v LDO voltage.

Fixes: 32d5921b8b55 ("rockchip: add Orange Pi R1 Plus LTS support")
Fixes: #13117
Fixes: #13759
PR #14276 (2024-01-04): rockchip: configure eth pad driver strength for orangepi r1 plus lts
The default strength is not enough to provide stable connection under 3.3v LDO voltage.

Fixes: 32d5921b8b55 ("rockchip: add Orange Pi R1 Plus LTS support")
Fixes: #13117
Fixes: #13759
commit 80ef582d (2024-01-05): rockchip: configure eth pad driver strength for orangepi r1 plus lts

The default strength is not enough to provide stable connection
under 3.3v LDO voltage.

Fixes: 3f3586a06d27 ("rockchip: add Orange Pi R1 Plus LTS support")
Fixes: #13117
Fixes: #13759

(cherry picked from commit 3645ac8a10d9abb1451343beaf7d65b53eeecffd)
[rebased onto openwrt-23.05 branch]
closed 2024-01-04 by openwrt-bot