openwrt/openwrt #13760: 23.05.0 SDK build fails at prereq

yokrysty, 2023-10-21:
### Describe the bug

host OS: ubuntu 22.04

i can successfully build my package with the following SDKs:
openwrt-sdk-19.07.10-mvebu-cortexa9_gcc-7.5.0_musl_eabi.Linux-x86_64
openwrt-sdk-21.02.7-mvebu-cortexa9_gcc-8.4.0_musl_eabi.Linux-x86_64
openwrt-sdk-22.03.2-mvebu-cortexa9_gcc-11.2.0_musl_eabi.Linux-x86_64

but using SDK openwrt-sdk-23.05.0-mvebu-cortexa9_gcc-12.3.0_musl_eabi.Linux-x86_64 i get the following errors when checking prereq (even if I have all of them):
Build dependency: Please install GNU 'tar'
Build dependency: Please install GNU 'find'
Build dependency: Please install 'xargs' that supports '-r/--no-run-if-empty'
Build dependency: Please install GNU 'patch'
Build dependency: Please install 'bzip2'

after some messing around and comparing files between different SDKs, I suspected the function SetupHostCommand in include/prereq.mk. I replaced the implementation with the one from the previous SDK and then my build was successful. This function is way to ugly and complicated for me to understand it and figure what is wrong.

openwrt-sdk-23.05.0-mvebu-cortexa9_gcc-12.3.0_musl_eabi.Linux-x86_64
include/prereq.mk SetupHostCommand
```bash
define SetupHostCommand
  define Require/$(1)
	mkdir -p "$(STAGING_DIR_HOST)/bin"; \
	for cmd in $(call QuoteHostCommand,$(3)) $(call QuoteHostCommand,$(4)) \
	           $(call QuoteHostCommand,$(5)) $(call QuoteHostCommand,$(6)) \
	           $(call QuoteHostCommand,$(7)) $(call QuoteHostCommand,$(8)) \
	           $(call QuoteHostCommand,$(9)) $(call QuoteHostCommand,$(10)) \
	           $(call QuoteHostCommand,$(11)) $(call QuoteHostCommand,$(12)); do \
		if [ -n "$$$$$$$$cmd" ]; then \
			bin="$$$$$$$$(command -v "$$$$$$$${cmd%% *}")"; \
			if [ -x "$$$$$$$$bin" ] && eval "$$$$$$$$cmd" >/dev/null 2>/dev/null; then \
				case "$$$$$$$$(ls -dl -- $(STAGING_DIR_HOST)/bin/$(strip $(1)))" in \
					*" -> $$$$$$$$bin"*) \
						[ -x "$(STAGING_DIR_HOST)/bin/$(strip $(1))" ] && exit 0 \
						;; \
					"-"*) \
						find "$(STAGING_DIR_HOST)/stamp" | grep $(strip $(1)) && \
						[ -x "$(STAGING_DIR_HOST)/bin/$(strip $(1))" ] && exit 0 \
						;; \
					*" -> /"*) \
						;; \
					*" -> "*) \
						[ -x "$(STAGING_DIR_HOST)/bin/$(strip $(1))" ] && exit 0 \
						;; \
				esac; \
				ln -sf "$$$$$$$$bin" "$(STAGING_DIR_HOST)/bin/$(strip $(1))"; \
				exit 0; \
			fi; \
		fi; \
	done; \
	exit 1
  endef

  $$(eval $$(call Require,$(1),$(if $(2),$(2),Missing $(1) command)))
endef
```

openwrt-sdk-22.03.2-mvebu-cortexa9_gcc-11.2.0_musl_eabi.Linux-x86_64
```bash
define SetupHostCommand
  define Require/$(1)
	[ -f "$(STAGING_DIR_HOST)/bin/$(strip $(1))" ] && exit 0; \
	for cmd in $(call QuoteHostCommand,$(3)) $(call QuoteHostCommand,$(4)) \
	           $(call QuoteHostCommand,$(5)) $(call QuoteHostCommand,$(6)) \
	           $(call QuoteHostCommand,$(7)) $(call QuoteHostCommand,$(8)) \
	           $(call QuoteHostCommand,$(9)) $(call QuoteHostCommand,$(10)) \
	           $(call QuoteHostCommand,$(11)) $(call QuoteHostCommand,$(12)); do \
		if [ -n "$$$$$$$$cmd" ]; then \
			bin="$$$$$$$$(PATH="$(subst $(space),:,$(filter-out $(STAGING_DIR_HOST)/%,$(subst :,$(space),$(PATH))))" \
				command -v "$$$$$$$${cmd%% *}")"; \
			if [ -x "$$$$$$$$bin" ] && eval "$$$$$$$$cmd" >/dev/null 2>/dev/null; then \
				mkdir -p "$(STAGING_DIR_HOST)/bin"; \
				ln -sf "$$$$$$$$bin" "$(STAGING_DIR_HOST)/bin/$(strip $(1))"; \
				exit 0; \
			fi; \
		fi; \
	done; \
	exit 1
  endef

  $$(eval $$(call Require,$(1),$(if $(2),$(2),Missing $(1) command)))
endef
```

### OpenWrt version

r23497-6637af95aa

### OpenWrt target/subtarget

mvebu/cortexa9

### Device

Linksys WRT3200ACM

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

jow-, 2023-10-25:
Please check if picking the changes from c1ef10c8d873254ce7c1f3019d821c4a87227474 and 3210aa8e0a113cc4354628b08b608c5c8f792941 fixes it. If so, we just need to cherry pick those fix commits to openwrt-23.05

yokrysty, 2023-10-26:
> Please check if picking the changes from [c1ef10c](https://github.com/openwrt/openwrt/commit/c1ef10c8d873254ce7c1f3019d821c4a87227474) and [3210aa8](https://github.com/openwrt/openwrt/commit/3210aa8e0a113cc4354628b08b608c5c8f792941) fixes it. If so, we just need to cherry pick those fix commits to openwrt-23.05

unfortunately it doesn't work, i get even more errors:
```
Build dependency: Please install the GNU C++ Compiler (g++) 6 or later
Build dependency: Please install Git (git-core) >= 1.7.12.2
Build dependency: Please install 'rsync'
Build dependency: Please install GNU 'bash'
Build dependency: Please install GNU diffutils
Build dependency: Please install GNU fileutils
Build dependency: Please install seq
Build dependency: Please install GNU 'grep'
Build dependency: Please install GNU 'grep'
Build dependency: Please install an extended getopt version that supports --long
Build dependency: Please install a 'realpath' utility
Build dependency: Cannot find a file stat utility
Build dependency: Please install 'gzip'
Build dependency: Please install 'unzip'
Build dependency: Please install GNU 'wget'
Build dependency: Please install GNU 'install'
Build dependency: Please install Perl 5.x
Build dependency: Please install the 'file' package
```

like i said above it works fine with the code for SetupHostCommand function from openwrt-sdk-22.03.2-mvebu-cortexa9_gcc-11.2.0_musl_eabi.Linux-x86_64

sateuwdie, 2025-02-25:
on alpine Linux latest sdk give me this error
```

Build dependency: Please install 'unzip'
Build dependency: Please install GNU 'install'

Prerequisite check failed. Use FORCE=1 to override.
make[2]: *** [/root/openwrt-sdk-lantiq-xrx200_gcc-13.3.0_musl.Linux-x86_64/include/toplevel.mk:183: /root/openwrt-sdk-lantiq-xrx200_gcc-13.3.0_musl.Linux-x86_64/staging_dir/host/.prereq-build] Error 1
make[1]: *** [/root/openwrt-sdk-lantiq-xrx200_gcc-13.3.0_musl.Linux-x86_64/include/toplevel.mk:79: prepare-tmpinfo] Error 2
make: *** [/root/openwrt-sdk-lantiq-xrx200_gcc-13.3.0_musl.Linux-x86_64/include/toplevel.mk:224: world] Error 2
```

of course install and unzip are installed

apk info unzip
unzip-6.0-r15 description:
Extract PKZIP-compatible .zip files

unzip-6.0-r15 webpage:
http://www.info-zip.org/UnZip.html

unzip-6.0-r15 installed size:
240 KiB

yokrysty, 2025-02-25:
same problem on `openwrt-sdk-24.10.0-mvebu-cortexa9_gcc-13.3.0_musl_eabi.Linux-x86_64` and the fix that i proposed to use the function `SetupHostCommand` in file `include/prereq.mk` from `openwrt-sdk-22.03.2-mvebu-cortexa9_gcc-11.2.0_musl_eabi.Linux-x86_64` also works

fake errors on openwrt-sdk-24.10.0-mvebu-cortexa9_gcc-13.3.0_musl_eabi.Linux-x86_64:
```
Build dependency: Please install the GNU C++ Compiler (g++) 8 or later
Build dependency: Please reinstall the GNU C++ Compiler (8 or later) - it appears to be broken
Build dependency: Please install Git (git-core) >= 1.7.12.2
Build dependency: Please install 'rsync'
Build dependency: Please install GNU 'bash'
Build dependency: Please install GNU diffutils
Build dependency: Please install GNU fileutils
Build dependency: Please install seq
Build dependency: Please install GNU 'grep'
Build dependency: Please install GNU 'grep'
Build dependency: Please install an extended getopt version that supports --long
Build dependency: Please install a 'realpath' utility
Build dependency: Cannot find a file stat utility
Build dependency: Please install 'gzip'
Build dependency: Please install 'unzip'
Build dependency: Please install GNU 'wget'
Build dependency: Please install GNU 'install'
Build dependency: Please install Perl 5.x
Build dependency: Please install the 'file' package
```

simontretter, 2025-04-28:
Same issue on the imagebuilder snapshot / x86:
`Build dependency: Please install GNU 'install'`