openwrt/openwrt #13761: Mt798x-wmac 18000000.wifi: Message xxxxxxxx (seq 5) timeout

philS1, 2023-10-22:
### Describe the bug

I'm on a zyxel nwa50ax pro and every few minutes my client get's disconnected. 

All I see is this error:

`Sat Oct 21 07:35:49 2023 daemon.notice hostapd: phy1-ap0: BEACON-REQ-TX-STATUS da:ef:5e:a8:df:a4 2 ack=1
Sat Oct 21 07:35:49 2023 daemon.notice hostapd: phy0-ap0: BEACON-REQ-TX-STATUS 76:f1:11:e8:b7:2b 27 ack=1
Sat Oct 21 07:35:49 2023 daemon.notice hostapd: phy1-ap0: BEACON-RESP-RX da:ef:5e:a8:df:a4 2 00 802c00000000000000006400008400f44d5c5a7ec7009a7bf00301d868a973d5070000006400311000064275636b656e01088c129824b048606c03012c0712444520240417340414640b1a90010095070d301c0100000fac040100000fac040300000fac02000fac04000fac06cc000b050100050000460572000000003603fdea013b0280002d1aef0917ffff0000000000000000000001000000000000000000003d162c0504000000000000000000000000000000000000007f0a04100a0a010001400040451102e7070a14160e2b000000000000000000bf0cf6798933faff0000faff0020c005012a00fcffc304022e2e2e
Sat Oct 21 07:35:49 2023 daemon.notice hostapd: phy0-ap0: BEACON-RESP-RX 76:f1:11:e8:b7:2b 27 00 802c00000000000000006400007c00f44d5c5a7ec70097f1540001de475073d5070000006400311000064275636b656e01088c129824b048606c03012c0504010200000712444520240417340414640b1a90010095070d301c0100000fac040100000fac040300000fac02000fac04000fac06cc000b050100050000460572000000003603fdea013b0280002d1aef0917ffff0000000000000000000001000000000000000000003d162c0504000000000000000000000000000000000000007f0a04100a0a010001400040451102e7070a14160e2b000000000000000000bf0cf6798933faff0000faff0020c005012a00fcffc304022e2e2e
Sat Oct 21 07:36:01 2023 daemon.warn odhcpd[1693]: No default route present, overriding ra_lifetime!
Sat Oct 21 07:36:05 2023 daemon.warn odhcpd[1693]: No default route present, overriding ra_lifetime!
Sat Oct 21 07:36:10 2023 kern.err kernel: [33667.431778] mt798x-wmac 18000000.wifi: Message 00005aed (seq 5) timeout
Sat Oct 21 07:36:31 2023 kern.err kernel: [33687.906575] mt798x-wmac 18000000.wifi: Message 000026ed (seq 6) timeout
Sat Oct 21 07:36:31 2023 daemon.notice hostapd: phy0-ap0: BEACON-REQ-TX-STATUS 76:f1:11:e8:b7:2b 28 ack=0
Sat Oct 21 07:36:31 2023 daemon.notice hostapd: phy1-ap0: BEACON-REQ-TX-STATUS da:ef:5e:a8:df:a4 3 ack=0
Sat Oct 21 07:36:34 2023 daemon.notice hostapd: phy1-ap0: BEACON-REQ-TX-STATUS da:ef:5e:a8:df:a4 4 ack=0
Sat Oct 21 07:36:34 2023 daemon.notice hostapd: phy0-ap0: BEACON-REQ-TX-STATUS 76:f1:11:e8:b7:2b 29 ack=0
Sat Oct 21 07:36:38 2023 daemon.notice hostapd: phy1-ap0: BEACON-REQ-TX-STATUS da:ef:5e:a8:df:a4 5 ack=0`

here is my wireless config
`config wifi-device 'radio0'
	option type 'mac80211'
	option path 'platform/18000000.wifi'
	option channel '6'
	option band '2g'
	option htmode 'HE20'
	option cell_density '0'
	option country 'DE'

config wifi-iface 'default_radio0'
	option device 'radio0'
	option network 'lan'
	option mode 'ap'
	option ssid 'Bucken'
	option encryption 'psk2+ccmp'
	option multicast_to_unicast_all '1'
	option key 'xxxx'
	option ieee80211r '1'
	option mobility_domain 'fdea'
	option reassociation_deadline '20000'
	option ft_over_ds '1'
	option ft_psk_generate_local '1'
	option ieee80211w '2'
	option wpa_disable_eapol_key_retries '1'
	option ieee80211k '1'
	option time_advertisement '2'
	option wnm_sleep_mode '1'
	option wnm_sleep_mode_no_keys '1'
	option bss_transition '1'
	option proxy_arp '1'

config wifi-device 'radio1'
	option type 'mac80211'
	option path 'platform/18000000.wifi+1'
	option channel '44'
	option band '5g'
	option htmode 'HE80'
	option cell_density '0'
	option country 'DE'

config wifi-iface 'default_radio1'
	option device 'radio1'
	option network 'lan'
	option mode 'ap'
	option ssid 'Bucken'
	option encryption 'psk2+ccmp'
	option multicast_to_unicast_all '1'
	option key 'xxxx'
	option ieee80211r '1'
	option mobility_domain 'fdea'
	option reassociation_deadline '20000'
	option ft_over_ds '1'
	option ft_psk_generate_local '1'
	option ieee80211w '2'
	option wpa_disable_eapol_key_retries '1'
	option ieee80211k '1'
	option time_advertisement '2'
	option wnm_sleep_mode '1'
	option wnm_sleep_mode_no_keys '1'
	option bss_transition '1'
	option proxy_arp '1'`

### OpenWrt version

r23497-6637af95aa

### OpenWrt target/subtarget

mediatek/filogic

### Device

ZyXEL NWA50AX Pro

### Image kind

Official downloaded image

### Steps to reproduce

Just connect a client to the AP

### Actual behaviour

Client gets disconnected every few minutes

### Expected behaviour

Client stays connected

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-10-22:
Thge most important part would be a minute leading to error in logread....

philS1, 2023-10-23:
ok, here's another try:
`Mon Oct 23 05:33:10 2023 daemon.notice hostapd: phy1-ap0: BEACON-REQ-TX-STATUS da:ef:5e:a8:df:a4 221 ack=1
Mon Oct 23 05:33:10 2023 daemon.notice hostapd: phy1-ap0: BEACON-RESP-RX da:ef:5e:a8:df:a4 221 04 0000000000000000000000008000000000000000000000000000
Mon Oct 23 05:33:13 2023 daemon.notice hostapd: phy1-ap0: BEACON-REQ-TX-STATUS da:ef:5e:a8:df:a4 222 ack=1
Mon Oct 23 05:33:13 2023 daemon.notice hostapd: phy1-ap0: BEACON-RESP-RX da:ef:5e:a8:df:a4 222 00 510600000000000000006400027a00f44d5c5a7ec60010bb2d0101df2848825a2e0000006400311400064275636b656e01088c129824b048606c0301060706444520010d142a0100301c0100000fac040100000fac040300000fac02000fac04000fac06cc000b050100290000460572000000003603fdea013b0251002d1aed0917ffff0000000000000000000001000000000000000000003d16060004000000000000000000000000000000000000007f0a04100a0a010001400040451102e7070a16142713000000000000000004ff1c230500081a44100220ce926f09afc8000c00fafffaff391cc7711c07ff0724f43f001efcffff022703
Mon Oct 23 05:33:16 2023 daemon.notice hostapd: phy1-ap0: BEACON-REQ-TX-STATUS da:ef:5e:a8:df:a4 223 ack=1
Mon Oct 23 05:33:16 2023 daemon.notice hostapd: phy1-ap0: BEACON-RESP-RX da:ef:5e:a8:df:a4 223 00 742c00000000000000006400008200f44d5c5a7ec7000bbf2d0101dd5e20b65a2e0000006400311000064275636b656e01088c129824b048606c03012c0504020300020712444520240417340414640b1a90010095070d301c0100000fac040100000fac040300000fac02000fac04000fac06cc000b050100020000460572000000003603fdea013b0274002d1aef0917ffff0000000000000000000001000000000000000000003d162c0504000000000000000000000000000000000000007f0a04100a0a010001400040451102e7070a16142e0b000000000000000000bf0cf6798933faff0000faff0020c005002e00fcffc303012e2e
Mon Oct 23 05:33:20 2023 daemon.notice hostapd: phy1-ap0: BEACON-REQ-TX-STATUS da:ef:5e:a8:df:a4 224 ack=1
Mon Oct 23 05:33:20 2023 daemon.notice hostapd: phy1-ap0: BEACON-RESP-RX da:ef:5e:a8:df:a4 224 00 510100000000000000006400023600f44d5c5a804300fbc22d0101dfcb766c6d0e0000006400311400064275636b656e01088c129824b048606c0301010706444520010d142a0100301c0100000fac040100000fac040300000fac02000fac04000fac06cc000b050000310000460572000000003603fdea013b0251002d1aed0917ffff0000000000000000000001000000000000000000003d16010000000000000000000000000000000000000000007f0a04100a0a010001400040451102e7070a16142b08000000000000000002ff1c230500081a44100220ce926f09afc8000c00fafffaff391cc7711c07ff0724f43f0005fcffff022703
Mon Oct 23 05:33:23 2023 daemon.notice hostapd: phy1-ap0: BEACON-REQ-TX-STATUS da:ef:5e:a8:df:a4 225 ack=1
Mon Oct 23 05:33:23 2023 daemon.notice hostapd: phy1-ap0: BEACON-RESP-RX da:ef:5e:a8:df:a4 225 00 7400000000000000000000008000000000000000000000000000
Mon Oct 23 05:33:26 2023 daemon.notice hostapd: phy1-ap0: BEACON-REQ-TX-STATUS da:ef:5e:a8:df:a4 226 ack=1
Mon Oct 23 05:33:30 2023 daemon.notice hostapd: phy1-ap0: BEACON-REQ-TX-STATUS da:ef:5e:a8:df:a4 227 ack=1
Mon Oct 23 05:33:30 2023 daemon.notice hostapd: phy1-ap0: BEACON-RESP-RX da:ef:5e:a8:df:a4 227 04 0000000000000000000000008000000000000000000000000000
Mon Oct 23 05:33:30 2023 daemon.notice hostapd: phy1-ap0: BEACON-RESP-RX da:ef:5e:a8:df:a4 226 00 002c00000000000000006400008000f44d5c5a7ec70020cf2d0101dd4e80875b2e0000006400311000064275636b656e01088c129824b048606c03012c0504000300000712444520240417340414640b1a90010095070d301c0100000fac040100000fac040300000fac02000fac04000fac06cc000b050100060000460572000000003603fdea013b0274002d1aef0917ffff0000000000000000000001000000000000000000003d162c0504000000000000000000000000000000000000007f0a04100a0a010001400040451102e7070a16142e0b000000000000000000bf0cf6798933faff0000faff0020c005002e00fcffc303012e2e
Mon Oct 23 05:33:33 2023 daemon.notice hostapd: phy1-ap0: BEACON-REQ-TX-STATUS da:ef:5e:a8:df:a4 228 ack=1
Mon Oct 23 05:33:33 2023 daemon.notice hostapd: phy1-ap0: BEACON-RESP-RX da:ef:5e:a8:df:a4 228 04 0000000000000000000000008000000000000000000000000000
Mon Oct 23 05:33:35 2023 daemon.notice hostapd: phy0-ap0: BEACON-REQ-TX-STATUS 76:f1:11:e8:b7:2b 190 ack=1
Mon Oct 23 05:33:35 2023 daemon.notice hostapd: phy0-ap0: BEACON-REQ-TX-STATUS 76:f1:11:e8:b7:2b 191 ack=1
Mon Oct 23 05:33:35 2023 daemon.notice hostapd: phy0-ap0: BEACON-REQ-TX-STATUS 76:f1:11:e8:b7:2b 192 ack=1
Mon Oct 23 05:33:35 2023 daemon.notice hostapd: phy0-ap0: BEACON-REQ-TX-STATUS 76:f1:11:e8:b7:2b 193 ack=1
Mon Oct 23 05:33:35 2023 daemon.notice hostapd: phy0-ap0: BEACON-REQ-TX-STATUS 76:f1:11:e8:b7:2b 194 ack=1
Mon Oct 23 05:33:35 2023 daemon.notice hostapd: phy0-ap0: BEACON-REQ-TX-STATUS 76:f1:11:e8:b7:2b 195 ack=1
Mon Oct 23 05:33:35 2023 daemon.notice hostapd: phy0-ap0: BEACON-REQ-TX-STATUS 76:f1:11:e8:b7:2b 196 ack=1
Mon Oct 23 05:33:35 2023 daemon.notice hostapd: phy0-ap0: BEACON-REQ-TX-STATUS 76:f1:11:e8:b7:2b 197 ack=1
Mon Oct 23 05:33:35 2023 daemon.notice hostapd: phy0-ap0: BEACON-REQ-TX-STATUS 76:f1:11:e8:b7:2b 198 ack=1
Mon Oct 23 05:33:55 2023 daemon.warn odhcpd[1693]: No default route present, overriding ra_lifetime!
Mon Oct 23 05:33:56 2023 kern.err kernel: [199090.654980] mt798x-wmac 18000000.wifi: Message 000026ed (seq 7) timeout
Mon Oct 23 05:34:16 2023 kern.err kernel: [199111.129757] mt798x-wmac 18000000.wifi: Message 00005aed (seq 8) timeout
Mon Oct 23 05:34:37 2023 kern.err kernel: [199131.604468] mt798x-wmac 18000000.wifi: Message 000026ed (seq 9) timeout
`

lucize, 2023-10-25:
@philS1 I had the same issue since filogic platform was converted to nvmem cells, I fixed like this

tried also stable and snapshot but please see when you boot if the firmware can be loaded, you should have one error
bad:
```
[   11.889113] mt798x-wmac 18000000.wifi: eeprom load fail, use default bin
[   11.895873] mt798x-wmac 18000000.wifi: Direct firmware load for mediatek/mt7986_eeprom_mt7976_dbdc.bin failed with error -2
[   11.906981] mt798x-wmac 18000000.wifi: Falling back to sysfs fallback for: mediatek/mt7986_eeprom_mt7976_dbdc.bin
```
good:
`[   11.889113] mt798x-wmac 18000000.wifi: eeprom load fail, use default bin
`

I've replaced the packed firmware files with the ones from https://anduin.linuxfromscratch.org/sources/linux-firmware/mediatek/ didn't had the time to verify how are the oficial openwrt firmware files, but the sha256sum was different from the ones in the link
then I saw that the country was set to standard and the wifi power was bad, just select your region and all was fine for me

philS1, 2023-10-25:
Unfortunately the files from the link and under /lib/firmware/mediatek don't differ here and I also set country to DE in /etc/config/wireless for both 2.4ghz and 5ghz
`ffbc07a53fcacc5d18ebb5cecf3f9f78  mt7981_rom_patch.bin
18d7188e84cf39350813a1dc1a6e1632  mt7981_wa.bin
9f0a8590d661afa1978775efeb260ea7  mt7981_wm.bin
e72ed57e4e98068c92d5e04b51b7d90d  mt7981_wo.bin`

dmesg looks quite okay to me
[    8.242971] mt798x-wmac 18000000.wifi: HW/SW Version: 0x8a108a10, Build Time: 20221208201745a
[    8.242971] 
[    8.454171] mt798x-wmac 18000000.wifi: WM Firmware Version: ____000000, Build Time: 20221208201806
[    8.557343] mt798x-wmac 18000000.wifi: WA Firmware Version: DEV_000000, Build Time: 20221208202048
[    8.563200] mtk_soc_eth 15100000.ethernet eth0: Link is Up - 1Gbps/Full - flow control off

philS1, 2023-10-29:
Sometimes I'm even getting a trace:
`Sun Oct 29 11:04:52 2023 daemon.info hostapd: phy1-ap0: STA 84:ab:1a:a2:e2:17 IEEE 802.11: authenticated
Sun Oct 29 11:04:53 2023 daemon.notice hostapd: phy0-ap0: BEACON-REQ-TX-STATUS da:ef:5e:a8:df:a4 183 ack=1
Sun Oct 29 11:04:54 2023 daemon.info hostapd: phy1-ap0: STA 84:ab:1a:a2:e2:17 IEEE 802.11: authenticated
Sun Oct 29 11:04:54 2023 daemon.notice hostapd: phy1-ap0: STA-OPMODE-N_SS-CHANGED 84:ab:1a:a2:e2:17 1
Sun Oct 29 11:04:54 2023 daemon.info hostapd: phy1-ap0: STA 84:ab:1a:a2:e2:17 IEEE 802.11: associated (aid 2)
Sun Oct 29 11:04:54 2023 daemon.info hostapd: phy1-ap0: STA 84:ab:1a:a2:e2:17 RADIUS: starting accounting session 3716B8C28B645D65
Sun Oct 29 11:04:54 2023 daemon.info hostapd: phy1-ap0: STA 84:ab:1a:a2:e2:17 WPA: pairwise key handshake completed (RSN)
Sun Oct 29 11:04:54 2023 daemon.notice hostapd: phy1-ap0: EAPOL-4WAY-HS-COMPLETED 84:ab:1a:a2:e2:17
Sun Oct 29 11:04:59 2023 daemon.notice hostapd: phy0-ap0: BEACON-REQ-TX-STATUS da:ef:5e:a8:df:a4 184 ack=1
Sun Oct 29 11:04:59 2023 daemon.notice hostapd: phy0-ap0: BEACON-RESP-RX da:ef:5e:a8:df:a4 184 00 000100000000000000006400023e00f44d5c5a8043004432240001d8403031c28b0000006400311400064275636b656e01088c129824b048606c0301010504020300040706444520010d142a0100301c0100000fac040100000fac040300000fac02000fac04000fac06cc000b0502000e0000460572000000003603fdea003b0251002d1aed0917ffff0000000000000000000001000000000000000000003d16010004000000000000000000000000000000000000007f0a04100a0a010001400040451102e7070a1d0a0b1c000000000000000005ff1c230500081a44100220ce926f09afc8000c00fafffaff391cc7711c07
Sun Oct 29 11:05:06 2023 daemon.notice hostapd: phy0-ap0: BEACON-REQ-TX-STATUS da:ef:5e:a8:df:a4 185 ack=1
Sun Oct 29 11:05:06 2023 daemon.notice hostapd: phy0-ap0: BEACON-RESP-RX da:ef:5e:a8:df:a4 185 00 000600000000000000006400028c00f44d5c5a7ec600f03a240001d851d01270480000006400311400064275636b656e01088c129824b048606c0301060504010300000706444520010d142a0100301c0100000fac040100000fac040300000fac02000fac04000fac06cc000b0501007c0000460572000000003603fdea003b0251002d1aed0917ffff0000000000000000000001000000000000000000003d16060004000000000000000000000000000000000000007f0a04100a0a010001400040451102e7070a1d0a0a32000000000000000003ff1c230500081a44100220ce926f09afc8000c00fafffaff391cc7711c07
Sun Oct 29 11:05:12 2023 daemon.notice hostapd: phy0-ap0: BEACON-REQ-TX-STATUS da:ef:5e:a8:df:a4 186 ack=1
Sun Oct 29 11:05:12 2023 daemon.info hostapd: phy1-ap0: STA 84:ab:1a:a2:e2:17 IEEE 802.11: authenticated
Sun Oct 29 11:05:12 2023 daemon.notice hostapd: phy0-ap0: BEACON-RESP-RX da:ef:5e:a8:df:a4 186 00 000100000000000000006400023600f44d5c5a804300dd41240001d84f50fcc28b0000006400311400064275636b656e01088c129824b048606c0301010504010300040706444520010d142a0100301c0100000fac040100000fac040300000fac02000fac04000fac06cc000b050200060000460572000000003603fdea003b0251002d1aed0917ffff0000000000000000000001000000000000000000003d16010004000000000000000000000000000000000000007f0a04100a0a010001400040451102e7070a1d0a0b1c000000000000000005ff1c230500081a44100220ce926f09afc8000c00fafffaff391cc7711c07
Sun Oct 29 11:05:13 2023 daemon.notice hostapd: phy0-ap0: BEACON-RESP-RX da:ef:5e:a8:df:a4 186 00 000600000000000000006400027a00f44d5c5a7ec6009e42240001df27d27670480000006400311400064275636b656e01088c129824b048606c0301060706444520010d142a0100301c0100000fac040100000fac040300000fac02000fac04000fac06cc000b050100690000460572000000003603fdea003b0251002d1aed0917ffff0000000000000000000001000000000000000000003d16060004000000000000000000000000000000000000007f0a04100a0a010001400040451102e7070a1d0a0a32000000000000000003ff1c230500081a44100220ce926f09afc8000c00fafffaff391cc7711c07ff0724f43f001ffcffff022703
Sun Oct 29 11:05:14 2023 daemon.info hostapd: phy1-ap0: STA 84:ab:1a:a2:e2:17 IEEE 802.11: authenticated
Sun Oct 29 11:05:14 2023 daemon.notice hostapd: phy1-ap0: STA-OPMODE-N_SS-CHANGED 84:ab:1a:a2:e2:17 1
Sun Oct 29 11:05:14 2023 daemon.info hostapd: phy1-ap0: STA 84:ab:1a:a2:e2:17 IEEE 802.11: associated (aid 2)
Sun Oct 29 11:05:14 2023 daemon.info hostapd: phy1-ap0: STA 84:ab:1a:a2:e2:17 RADIUS: starting accounting session 3716B8C28B645D65
Sun Oct 29 11:05:14 2023 daemon.info hostapd: phy1-ap0: STA 84:ab:1a:a2:e2:17 WPA: pairwise key handshake completed (RSN)
Sun Oct 29 11:05:14 2023 daemon.notice hostapd: phy1-ap0: EAPOL-4WAY-HS-COMPLETED 84:ab:1a:a2:e2:17
Sun Oct 29 11:05:14 2023 daemon.warn odhcpd[1688]: No default route present, overriding ra_lifetime!
Sun Oct 29 11:05:19 2023 daemon.notice hostapd: phy0-ap0: BEACON-REQ-TX-STATUS da:ef:5e:a8:df:a4 187 ack=1
Sun Oct 29 11:05:19 2023 daemon.notice hostapd: phy0-ap0: BEACON-RESP-RX da:ef:5e:a8:df:a4 187 04 0000000000000000000000008000000000000000000000000000
Sun Oct 29 11:05:26 2023 daemon.notice hostapd: phy0-ap0: BEACON-REQ-TX-STATUS da:ef:5e:a8:df:a4 188 ack=1
Sun Oct 29 11:05:26 2023 daemon.notice hostapd: phy0-ap0: BEACON-RESP-RX da:ef:5e:a8:df:a4 188 00 000600000000000000006400028c00f44d5c5a7ec6003752240001d8d3f04171480000006400311400064275636b656e01088c129824b048606c0301060504020300020706444520010d142a0100301c0100000fac040100000fac040300000fac02000fac04000fac06cc000b050100500000460572000000003603fdea003b0251002d1aed0917ffff0000000000000000000001000000000000000000003d16060004000000000000000000000000000000000000007f0a04100a0a010001400040451102e7070a1d0a0a32000000000000000003ff1c230500081a44100220ce926f09afc8000c00fafffaff391cc7711c07
[... cut ...]
Sun Oct 29 11:07:53 2023 kern.debug kernel: [311220.571774]  0xffffffc00092ad58 [mac80211@0000000030893f08+0x83000]
Sun Oct 29 11:07:53 2023 kern.debug kernel: [311220.571783]  0xffffffc008050910
Sun Oct 29 11:07:53 2023 kern.debug kernel: [311220.571786]  0xffffffc008050c20
Sun Oct 29 11:07:53 2023 kern.debug kernel: [311220.571790]  0xffffffc008058e1c
Sun Oct 29 11:07:53 2023 kern.debug kernel: [311220.571794]  0xffffffc00801581c
Sun Oct 29 11:07:53 2023 kern.warn kernel: [311220.571798] ---[ end trace 51c83589df13e422 ]---

brada4, 2023-10-29:
There is firewall offload between pppoe components. Could you try disabling that for few crach cycles?

philS1, 2023-10-29:
@brada4  you mean disable firewalld on this openwrt device? Or on my ISPs router??? Actually I have cable internet. I thought pppoe is only active for DSL?

brada4, 2023-10-29:
Luci/Network/Firewall/ somewhere mid-screen uncheck offload checkbox, then tap apply below.

philS1, 2023-10-30:
Software flow offloading is already unchecked

badguy001, 2023-11-23:
TP-Link TL-XDR6088, OpenWrt 23.05.2 r23630-842932a63d, as a dumb ap, similar error in kernel log.
<details><summary>kernel log</summary>
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]<br>
[    0.000000] Linux version 5.15.137 (builder@buildhost) (aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 12.3.0 r23630-842932a63d) 12.3.0, GNU ld (GNU Binutils) 2.40.0) #0 SMP Tue Nov 14 13:38:11 2023<br>
[    0.000000] Machine model: TP-Link TL-XDR6088<br>
[    0.000000] Zone ranges:<br>
[    0.000000]   DMA      [mem 0x0000000040000000-0x000000005fffffff]<br>
[    0.000000]   DMA32    empty<br>
[    0.000000]   Normal   empty<br>
[    0.000000] Movable zone start for each node<br>
[    0.000000] Early memory node ranges<br>
[    0.000000]   node   0: [mem 0x0000000040000000-0x0000000042ffffff]<br>
[    0.000000]   node   0: [mem 0x0000000043000000-0x000000004302ffff]<br>
[    0.000000]   node   0: [mem 0x0000000043030000-0x000000004fbfffff]<br>
[    0.000000]   node   0: [mem 0x000000004fc00000-0x000000004ffbffff]<br>
[    0.000000]   node   0: [mem 0x000000004ffc0000-0x000000005fffffff]<br>
[    0.000000] Initmem setup node 0 [mem 0x0000000040000000-0x000000005fffffff]<br>
[    0.000000] psci: probing for conduit method from DT.<br>
[    0.000000] psci: PSCIv1.1 detected in firmware.<br>
[    0.000000] psci: Using standard PSCI v0.2 function IDs<br>
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.<br>
[    0.000000] psci: SMC Calling Convention v1.4<br>
[    0.000000] percpu: Embedded 17 pages/cpu s30040 r8192 d31400 u69632<br>
[    0.000000] pcpu-alloc: s30040 r8192 d31400 u69632 alloc=17*4096<br>
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 <br>
[    0.000000] Detected VIPT I-cache on CPU0<br>
[    0.000000] CPU features: detected: GIC system register CPU interface<br>
[    0.000000] CPU features: kernel page table isolation disabled by kernel configuration<br>
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 129024<br>
[    0.000000] Kernel command line: <br>
[    0.000000] Dentry cache hash table entries: 65536 (order: 7, 524288 bytes, linear)<br>
[    0.000000] Inode-cache hash table entries: 32768 (order: 6, 262144 bytes, linear)<br>
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off<br>
[    0.000000] Memory: 498088K/524288K available (8384K kernel code, 912K rwdata, 1472K rodata, 448K init, 303K bss, 26200K reserved, 0K cma-reserved)<br>
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1<br>
[    0.000000] rcu: Hierarchical RCU implementation.<br>
[    0.000000] 	Tracing variant of Tasks RCU enabled.<br>
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.<br>
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0<br>
[    0.000000] GICv3: GIC: Using split EOI/Deactivate mode<br>
[    0.000000] GICv3: 640 SPIs implemented<br>
[    0.000000] GICv3: 0 Extended SPIs implemented<br>
[    0.000000] GICv3: Distributor has no Range Selector support<br>
[    0.000000] Root IRQ handler: 0xffffffc008422e34<br>
[    0.000000] GICv3: 16 PPIs implemented<br>
[    0.000000] GICv3: CPU0: found redistributor 0 region 0:0x000000000c080000<br>
[    0.000000] arch_timer: cp15 timer(s) running at 13.00MHz (phys).<br>
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x2ff89eacb, max_idle_ns: 440795202429 ns<br>
[    0.000000] sched_clock: 56 bits at 13MHz, resolution 76ns, wraps every 4398046511101ns<br>
[    0.000109] Calibrating delay loop (skipped), value calculated using timer frequency.. 26.00 BogoMIPS (lpj=130000)<br>
[    0.000116] pid_max: default: 32768 minimum: 301<br>
[    0.000315] Mount-cache hash table entries: 1024 (order: 1, 8192 bytes, linear)<br>
[    0.000324] Mountpoint-cache hash table entries: 1024 (order: 1, 8192 bytes, linear)<br>
[    0.001332] rcu: Hierarchical SRCU implementation.<br>
[    0.001605] smp: Bringing up secondary CPUs ...<br>
[    0.001865] Detected VIPT I-cache on CPU1<br>
[    0.001886] GICv3: CPU1: found redistributor 1 region 0:0x000000000c0a0000<br>
[    0.001911] CPU1: Booted secondary processor 0x0000000001 [0x410fd034]<br>
[    0.002169] Detected VIPT I-cache on CPU2<br>
[    0.002181] GICv3: CPU2: found redistributor 2 region 0:0x000000000c0c0000<br>
[    0.002192] CPU2: Booted secondary processor 0x0000000002 [0x410fd034]<br>
[    0.002429] Detected VIPT I-cache on CPU3<br>
[    0.002439] GICv3: CPU3: found redistributor 3 region 0:0x000000000c0e0000<br>
[    0.002448] CPU3: Booted secondary processor 0x0000000003 [0x410fd034]<br>
[    0.002472] smp: Brought up 1 node, 4 CPUs<br>
[    0.002485] SMP: Total of 4 processors activated.<br>
[    0.002488] CPU features: detected: 32-bit EL0 Support<br>
[    0.002492] CPU features: detected: CRC32 instructions<br>
[    0.002513] CPU features: emulated: Privileged Access Never (PAN) using TTBR0_EL1 switching<br>
[    0.002564] CPU: All CPU(s) started at EL2<br>
[... 338 lines cut ...]
[207763.737139]  0xffffffc008050aa0<br>
[207763.740350]  0xffffffc008058cac<br>
[207763.743562]  0xffffffc00801581c<br>
[207763.746774] ---[ end trace 21a9e8fa4b9c1651 ]---<br>
[291398.890719] mt7530-mdio mdio-bus:1f lan3: Link is Up - 100Mbps/Full - flow control off<br>
[291398.898765] br-lan: port 3(lan3) entered blocking state<br>
[291398.904081] br-lan: port 3(lan3) entered forwarding state<br>
[... cut ...]
[291407.617374] br-lan: port 4(lan4) entered disabled state<br>
[291409.197091] mt7530-mdio mdio-bus:1f lan4: Link is Up - 100Mbps/Full - flow control off<br>
[291409.205127] br-lan: port 4(lan4) entered blocking state<br>
[291409.210426] br-lan: port 4(lan4) entered forwarding state<br>
[291847.109235] mt7530-mdio mdio-bus:1f lan4: Link is Down<br>
[291847.114525] br-lan: port 4(lan4) entered disabled state<br>
[291851.338815] mt7530-mdio mdio-bus:1f lan3: Link is Down<br>
[291851.344102] br-lan: port 3(lan3) entered disabled state<br>
</details>

philS1, 2023-11-27:
@badguy001 Have you tried setting these parameters?
option max_inactivity '86400'           
option disassoc_low_ack '0'             
option wpa_group_rekey '86400'

brada4, 2023-11-27:
It has to handle inactivity+reassociation and rekey properly....

philS1, 2023-11-29:
@brada4 100% agree

In the openwrt forum user retuor confirmed
If I stress the AP with iperf3 on WiFi connected devices I can reproduce the messages (25 connected devices in total):

```
Wed Nov 29 16:32:23 2023 kern.err kernel: [ 2802.378439] mt798x-wmac 18000000.wifi: Message 00005aed (seq 5) timeout
Wed Nov 29 16:32:44 2023 kern.err kernel: [ 2822.853228] mt798x-wmac 18000000.wifi: Message 000026ed (seq 6) timeout
Wed Nov 29 16:32:44 2023 kern.err kernel: [ 2822.859863] mt798x-wmac 18000000.wifi: Message 000026ed (seq 7) timeout
Wed Nov 29 16:32:55 2023 kern.err kernel: [ 2834.311039] mt798x-wmac 18000000.wifi: Message 00005aed (seq 4) timeout
Wed Nov 29 16:33:14 2023 kern.err kernel: [ 2853.095811] mt798x-wmac 18000000.wifi: Message 00005aed (seq 4) timeout
Wed Nov 29 16:33:32 2023 kern.err kernel: [ 2871.141260] mt798x-wmac 18000000.wifi: Message 00005aed (seq 2) timeout
Wed Nov 29 16:33:44 2023 kern.err kernel: [ 2882.977593] mt798x-wmac 18000000.wifi: Message 00005aed (seq 2) timeout
Wed Nov 29 16:34:20 2023 kern.err kernel: [ 2918.927495] mt798x-wmac 18000000.wifi: Message 00005aed (seq 4) timeout
Wed Nov 29 16:34:21 2023 kern.err kernel: [ 2920.357468] mt798x-wmac 18000000.wifi: Message 00005aed (seq 3) timeout
Wed Nov 29 16:35:27 2023 kern.err kernel: [ 2986.499628] mt798x-wmac 18000000.wifi: Message 00005aed (seq 14) timeout
Wed Nov 29 16:36:12 2023 kern.err kernel: [ 3030.927934] mt798x-wmac 18000000.wifi: Message 00005aed (seq 14) timeout
Wed Nov 29 16:36:27 2023 kern.err kernel: [ 3045.723855] mt798x-wmac 18000000.wifi: Message 00005aed (seq 9) timeout
```

zhaojh329, 2023-11-30:
Latest mt76 has not this issue.

philS1, 2023-11-30:
@zhaojh329 so is it enough to switch to snapshot build? There was already an kernel update. Or do I need newer firmware files?

brada4, 2023-11-30:
@zhaojh329 the question is if you had problem with 23.05.2 and it disappeared with snapshot.

philS1, 2024-10-24:
and it happened again on 23.05.5:
`[ 1877.827516] Ignoring NSS change in VHT Operating Mode Notification from 34:1c:f0:3b:4c:f6 with invalid nss 2
[ 2040.433295] Ignoring NSS change in VHT Operating Mode Notification from 34:1c:f0:3b:4c:f6 with invalid nss 2
[ 2071.307491] Ignoring NSS change in VHT Operating Mode Notification from 34:1c:f0:3b:4c:f6 with invalid nss 2
[ 2101.715970] ------------[ cut here ]------------
[ 2101.715984] Timeout waiting for MCU reset state 10
[ 2101.716028] WARNING: CPU: 0 PID: 2940 at 0xffffffc0009e2d38 [mt7915e@00000000d8f400e1+0x22000]
[ 2101.716053] Modules linked in: pppoe ppp_async nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mt7915e mt76_connac_lib mt76 mac80211 cfg80211 slhc nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c compat crypto_safexcel sha1_generic seqiv md5 des_generic libdes authencesn authenc leds_gpio gpio_button_hotplug aquantia crc_ccitt
[ 2101.716183] CPU: 0 PID: 2940 Comm: kworker/u4:1 Not tainted 5.15.167 #0
[ 2101.716190] Hardware name: ZyXEL NWA50AX Pro (DT)
[ 2101.716196] Workqueue: mt76 0xffffffc0009e4df0 [mt7915e@00000000d8f400e1+0x22000]
[ 2101.716207] pstate: 80400005 (Nzcv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[ 2101.716214] pc : 0xffffffc0009e2d38 [mt7915e@00000000d8f400e1+0x22000]
[ 2101.716222] lr : 0xffffffc0009e2d38 [mt7915e@00000000d8f400e1+0x22000]
[ 2101.716229] sp : ffffffc00923bc40
[ 2101.716233] x29: ffffffc00923bc40 x28: ffffff8003482ff8 x27: ffffff800348a920
[ 2101.716243] x26: ffffff8003482920 x25: ffffff80034868e8 x24: ffffff8003482920
[ 2101.716253] x23: 0000000000000000 x22: ffffff800348e920 x21: 0000000000000010
[ 2101.716262] x20: ffffff80034903e0 x19: 0000000000000000 x18: ffffffc008a5a2a8
[ 2101.716272] x17: 313a3433206d6f72 x16: 66206e6f69746163 x15: 00000000000002d0
[ 2101.716281] x14: 00000000000000f0 x13: ffffffc00923b978 x12: ffffffc008ab22a8
[ 2101.716291] x11: fffffffffffe3ef8 x10: ffffffc008ab22a8 x9 : 0000000000000000
[ 2101.716300] x8 : ffffffc008a5a258 x7 : ffffffc008a5a2a8 x6 : 0000000000000001
[ 2101.716309] x5 : ffffff801fead6f8 x4 : 0000000000000000 x3 : 0000000000000027
[ 2101.716319] x2 : 0000000000000027 x1 : 0000000000000023 x0 : 0000000000000026
[ 2101.716328] Call trace:
[ 2101.716332]  0xffffffc0009e2d38 [mt7915e@00000000d8f400e1+0x22000]
[ 2101.716340]  0xffffffc0009e5b44 [mt7915e@00000000d8f400e1+0x22000]
[ 2101.716348]  0xffffffc0080508e0
[ 2101.716351]  0xffffffc0080516d0
[ 2101.716355]  0xffffffc008058e6c
[ 2101.716359]  0xffffffc00801581c
[ 2101.716363] ---[ end trace 850ac5883a9fec64 ]---
[ 2131.788765] ------------[ cut here ]------------
[ 2131.788779] Timeout waiting for MCU reset state 20
[ 2131.788825] WARNING: CPU: 0 PID: 2940 at 0xffffffc0009e2d38 [mt7915e@00000000d8f400e1+0x22000]
[ 2131.788849] Modules linked in: pppoe ppp_async nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet pppox ppp_generic nft_reject_ipv6 nft_reject_ipv4 nft_reject_inet nft_reject nft_redir nft_quota nft_objref nft_numgen nft_nat nft_masq nft_log nft_limit nft_hash nft_flow_offload nft_fib_ipv6 nft_fib_ipv4 nft_fib nft_ct nft_counter nft_chain_nat nf_tables nf_nat nf_flow_table nf_conntrack mt7915e mt76_connac_lib mt76 mac80211 cfg80211 slhc nfnetlink nf_reject_ipv6 nf_reject_ipv4 nf_log_syslog nf_defrag_ipv6 nf_defrag_ipv4 libcrc32c compat crypto_safexcel sha1_generic seqiv md5 des_generic libdes authencesn authenc leds_gpio gpio_button_hotplug aquantia crc_ccitt
[ 2131.788979] CPU: 0 PID: 2940 Comm: kworker/u4:1 Tainted: G        W         5.15.167 #0
[ 2131.788988] Hardware name: ZyXEL NWA50AX Pro (DT)
[ 2131.788993] Workqueue: mt76 0xffffffc0009e4df0 [mt7915e@00000000d8f400e1+0x22000]
[ 2131.789005] pstate: 80400005 (Nzcv daif +PAN -UAO -TCO -DIT -SSBS BTYPE=--)
[ 2131.789012] pc : 0xffffffc0009e2d38 [mt7915e@00000000d8f400e1+0x22000]
[ 2131.789020] lr : 0xffffffc0009e2d38 [mt7915e@00000000d8f400e1+0x22000]
[ 2131.789027] sp : ffffffc00923bc40
[ 2131.789030] x29: ffffffc00923bc40 x28: ffffff8003482ff8 x27: ffffff800348a920
[ 2131.789040] x26: ffffff8003482920 x25: ffffff80034868e8 x24: ffffff8003482920
[ 2131.789050] x23: 0000000000000000 x22: ffffff800348e920 x21: 0000000000000020
[ 2131.789060] x20: ffffff80034903e0 x19: 0000000000000000 x18: ffffffc008a5a2a8
[ 2131.789070] x17: 00000000981c214e x16: 0000000090a2d5c8 x15: 0000000000000327
[ 2131.789079] x14: 000000000000010d x13: ffffffc00923b978 x12: ffffffc008ab22a8
[ 2131.789088] x11: fffffffffffe48a0 x10: ffffffc008ab22a8 x9 : 0000000000000000
[ 2131.789098] x8 : ffffffc008a5a258 x7 : ffffffc008a5a2a8 x6 : 0000000000000001
[ 2131.789107] x5 : ffffff801fead6f8 x4 : 0000000000000000 x3 : 0000000000000027
[ 2131.789116] x2 : 0000000000000027 x1 : 0000000000000023 x0 : 0000000000000026
[ 2131.789125] Call trace:
[ 2131.789129]  0xffffffc0009e2d38 [mt7915e@00000000d8f400e1+0x22000]
[ 2131.789137]  0xffffffc0009e5028 [mt7915e@00000000d8f400e1+0x22000]
[ 2131.789144]  0xffffffc0080508e0
[ 2131.789148]  0xffffffc0080516d0
[ 2131.789152]  0xffffffc008058e6c
[ 2131.789155]  0xffffffc00801581c
[ 2131.789160] ---[ end trace 850ac5883a9fec65 ]---
[ 2152.903683] mt798x-wmac 18000000.wifi: Message 00005aed (seq 10) timeout
[ 2173.378554] mt798x-wmac 18000000.wifi: Message 00005aed (seq 11) timeout
[ 2193.853300] mt798x-wmac 18000000.wifi: Message 00005aed (seq 12) timeout`

pktpls, 2024-12-20:
This has been fixed a while ago in the 24.10 and snapshot branches