openwrt/openwrt #13762: EdgeRouter E200 series MMC support is broken

532910, 2023-10-22:
### Describe the bug

lede-17.01.7 was the latest working release

Starting from 18.06.0 till 19.07.9 `cat /dev/mmcblk0 > /dev/null` produces many errors:
```
mmcblk0: response CRC error sending r/w cmd command, card status 0x900
print_req_error: I/O error, dev mmcblk0, sector 2056
```

Starting from 21.02.0 till 23.05.0 it gives different results without errors:
```
# cat /dev/mmcblk0p1 | md5sum 
3722c6b59b1d46d9c0be08f40ea347ce  -
# cat /dev/mmcblk0p1 | md5sum 
555ce388342ba6b40983ec747b45037a  -
# cat /dev/mmcblk0p1 | md5sum 
162cea5bf3f299ba412bbc6653477e38  -
# cat /dev/mmcblk0p1 | md5sum 
5f2436c6af801fa20d5308a6431d455a  -
```

mmcblk0 is the main storage so it's not possible to install OpenWrt on this devices

### OpenWrt version

18.06.0

### OpenWrt target/subtarget

octeon/generic

### Device

UBNT_E200 (CN6120p1.1-800-NSP)

### Image kind

Official downloaded image

### Steps to reproduce

Just install openwrt.

### Actual behaviour

a lots of errors on boot up
squashfs errors (decompression failed, data probably corrupt)

### Expected behaviour

should work

### Additional info

Another user report:
https://www.mail-archive.com/openwrt-devel@lists.openwrt.org/msg62623.html
Another user try to fix this issue:
https://forum.openwrt.org/t/edgerouter-pro-doesnt-work-with-the-latest-release/126154
My call for help:
https://forum.openwrt.org/t/edgerouter-er-8-mmc-support-is-broken/175154

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

532910, 2023-10-22:
I have tried to look into this issue and can confirm that it is due to an incorrect `max-frequency` value as mmc works fine with my device tree. I'd happy to PR, but I can't figure out how it broke and how it should be fixed.

What exactly not clear:
- Neither 15.05 nor 23.05 has device tree for E200 series. I have not found it the code and early boot says "Using passed Device Tree", that means it's passed from uboot and hasn't been changed, right?
- `150-mmc-octeon-add-host-driver-for-octeon-mmc-controller.patch` don't differ between v17.01 and v18.06.0 (first release when mmc support was broken)
- all openwrt
- `soc@0/mmc@1180000002000/mmc-slot@0/spi-max-frequency` is 52000000 on all versions of openwrt I tested and original firmware.

As I see there are to ways to to fix this: find what was broken and fix it without creating device tree for this board or create device tree.

532910, 2023-10-22:
```
Octeon ubnt_e200# mmcinfo 
Device: Octeon MMC/SD0
Manufacturer ID: 45
OEM: 100
Name: SEM04 
Tran Speed: 26000000
Rd Block Len: 512
MMC version 4.0
High Capacity: Yes
Capacity: 3.7 GiB
Bus Width: 8-bit
```

`26000000` is the correct value that should be used for `max-frequency` or `spi-max-frequency`
BTW has it something common with the nearest quartz that has 50.000 inscription:
[image]
[image]

532910, 2023-10-22:
These dts were dumped from working openwrt official releases `/proc/device-tree`:
[17.01.7.dts](https://github.com/openwrt/openwrt/files/13063193/17.01.7.dts.txt)
[23.05.0.dts](https://github.com/openwrt/openwrt/files/13063192/23.05.0.dts.txt)

532910, 2023-10-22:
Above dts looks similar to cn71xx.dtsi, so I based my own on it:

```
// SPDX-License-Identifier: GPL-2.0-or-later OR MIT

#include "cn71xx.dtsi"

#include <dt-bindings/gpio/gpio.h>
#include <dt-bindings/input/input.h>

/ {
        compatible = "ubnt,e200";
        model = "Ubiquiti EdgeRouter ER-8";

        aliases {
                smi0 = "/soc@0/mdio@1180000001800";
                smi1 = "/soc@0/mdio@1180000001900";
                uart0 = "/soc@0/serial@1180000000800";
                uart1 = "/soc@0/serial@1180000000c00";
                twsi0 = "/soc@0/i2c@1180000001000";
                twsi1 = "/soc@0/i2c@1180000001200";
                pip = "/soc@0/pip@11800a0000000";
                flash0 = "/soc@0/bootbus@1180000000000/nor@0,0";
        };
        soc {
                //usb0: uctl@118006f000000 {
                uctl@118006f000000 {
                        compatible = "cavium,octeon-6335-uctl";
                        reg = <0x11800 0x6f000000 0x0 0x100>;
                        ranges; /* Direct mapping */
                        #address-cells = <2>;
                        #size-cells = <2>;
                        /* 12MHz, 24MHz and 48MHz allowed */
                        refclk-frequency = <12000000>;
                        /* Either "crystal" or "external" */
                        refclk-type = "crystal";

                        ehci@16f0000000000 {
                                compatible = "cavium,octeon-6335-ehci","usb-ehci";
                                reg = <0x16f00 0x00000000 0x0 0x100>;
                                interrupts = <3 56>;
                                big-endian-regs;
                        };
                        ohci@16f0000000400 {
                                compatible = "cavium,octeon-6335-ohci","usb-ohci";
                                reg = <0x16f00 0x00000400 0x0 0x100>;
                                interrupts = <3 56>;
                                big-endian-regs;
                        };
                };
        };
};

&uart0 {
        status = "okay";
        clock-frequency = <600000000>;
};

//&usb0 {
//      status = "okay";
//};

//&xhci0 {
//      status = "okay";
//      dr_mode = "host";
//};

&mmc {
        status = "okay";

[... 125 lines cut ...]
                ethernet@2 {
                        label = "eth6";
                        status = "okay";
                        phy-handle = <&phy6>;
                        nvmem-cells = <&macaddr_eeprom_0>;
                        nvmem-cell-names = "mac-address";
                        mac-address-increment = <(6)>;
                };
                ethernet@3 {
                        label = "eth7";
                        status = "okay";
                        phy-handle = <&phy7>;
                        nvmem-cells = <&macaddr_eeprom_0>;
                        nvmem-cell-names = "mac-address";
                        mac-address-increment = <(7)>;
                };
        };
};
```

The issues are:
1. USB doesn't work
2. MACs are wrong
3. interfaces has write lables (eth0 in linux corresponds eth0 port on the router) but the order of interface in `ip l` is strange.
4. The last, but not least: E200 is CN61XX based, not CN71xx. It looks like `cn71xx.dtsi` should be reworked and both cn71xx.dtsi cn61xx.dtsi should inherit common parent.

532910, 2023-10-22:
CN61XX CN71XX product briefs:
https://datasheet.octopart.com/CN6130-110SV-G-Cavium-Networks-datasheet-25352746.pdf
https://www.marvell.com/content/dam/marvell/en/public-collateral/embedded-processors/marvell-infrastructure-processors-octeon-iii-cn70xx-71xx-ap-product-brief-2019.pdf

raenye, 2023-10-30:
Maybe the device trees are present in the upstream kernel?

532910, 2023-11-02:
1. This doesn't explain why it worked fine before device trees were included in the upstream kernel.
2. 17.01.7 uses 4.4 and 18.06.0 uses 4.14 and `arch/mips/boot/dts/cavium-octeon` were changed between them, but there is nothing about `mmc` or `1180000002000`
https://github.com/torvalds/linux/tree/v4.4/arch/mips/boot/dts/cavium-octeon
https://github.com/torvalds/linux/tree/v4.14/arch/mips/boot/dts/cavium-octeon
3. All openwrt I've tested say `Using passed Device Tree` on early boot, so it means all uses device tree from uboot, right?

532910, 2024-03-29:
dts from the latest edgeos, via sshfs:
```
sshfs ubnt@192.168.1.1: ~/mnt
dtc -I fs -O dts -o v2.0.9-hotfix.7.dts ~/mnt/sys/firmware/devicetree/base
```
[v2.0.9-hotfix.7.dts.txt](https://github.com/openwrt/openwrt/files/14802685/v2.0.9-hotfix.7.dts.txt)

532910, 2024-03-29:
[v2.0.9-hotfix.7.dmesg.txt](https://github.com/openwrt/openwrt/files/14802831/v2.0.9-hotfix.7.dmesg.txt)

tucnak, 2024-08-15:
So this thread exists!

I've been trying to get ERPro-8 working as it seemed like the most capable rackmount router with SFP capability that is also supported by OpenWrt. I also wonder if it's possible to patch the kernel to try and perhaps work around [the hardware backdoor](https://news.ycombinator.com/item?id=37570407) NSA had planted in Cavium cpu's. (I have some ideas as far as Huawei-made GPON SFP firmware is considered!) I'd really like to have observability, and perhaps some counter-measures for my Internet-facing router, and currently it doesn't seem like there's a better option (esp. considering the backdoor, the enemy you know is better than... and so on!) This is quite a rabbit-hole: although I think bypassing octeon_rng would be a good place to start... I digress.

The eMMC issue is something that stopped me in my tracks pretty early on, I had initially assumed it was a hardware failure but further reading suggested otherwise. I had also tried to boot from USB, however that didn't work: either the openwrt kernel build, or U-boot seems to ignore bootargs I had put in place. So far I have not been able to boot from USB. **To somebody who's not knowledgable in kernel development and OpenWrt build system per se, how hard is it to work around this issue?** I don't mind using the USB stick, really, and forget that eMMC exists. I was also thinking, maybe if LUKS were included in initramfs environment, it could potentially allow booting rootfs from encrypted partition, but honestly at this stage I'd be happy just to just move ahead!

P.S. EdgeOS is simply ridiculous; I cannot imagine how people use it, even the most basic of IPv6 tasks are completely intractable, and the shell is maddening. However, at least the GPON SFP's I have lying around worked there, and for whatever reason none of my SFP devices could acquire a link on OpenWrt builds. The `ethtool -i` had suggested `octeon_ethernet` driver is in use, however `lsmod` didn't show none, unlike EdgeOS where it comes up nicely with a bunch of other stuff. Does it mean that it's built as a kernel module in EdgeOS, and as a kernel driver in OpenWrt, or what?

RaylynnKnight, 2024-08-18:
I've been running 19.07.4 on an EdgeRouter 8 for some time.  In fact the https://openwrt.org/toh/ubiquiti/edgerouter page was created as I installed that version on my EdgeRouter.  See the bootlog on the toh page https://openwrt.org/toh/ubiquiti/edgerouter#openwrt_bootlog.  Perhaps there is some difference in the U-Boot installed on the 8MB SPI flash on your device?

tucnak, 2024-08-18:
@RaylynnKnight I did follow https://openwrt.org/toh/ubiquiti/edgerouter_pro initially which is up-to-date (and broken.) I'm not sure if it's wise to expose a four-year old release OS to the Internet.

To my knowledge, the bootloader for 8/8pro is the same: `e200_002_80eda` here.

I can boot in initramfs environment manually, but the dmesg shows KERNEL CMDLINE different to bootargs as saved in U-boot. I had also tried putting rootfs on the USB /dev/sda2, point to it via bootargs, and boot the kernel.bin from USB /dev/sda1 but that didn't work either. I'd love to skip eMMC completely, but no luck :-(

532910, 2024-08-19:
I "fixed" this by manually modifying the device-tree in the ROM (changed mmc frequency: found 52000000 and replaced with it with 26000000 in hexeditor). Main bootloader becomes broken after this action, but the second one work fine and mmc works correctly.

tucnak, 2024-08-19:
Can you provide some details, i.e. how I would be editting device-tree ROM and what do you mean by _main_ and _second_ bootloader? Image1/Image2 or? I'm not sure I can follow & wouldn't want to brick my device at this time. Also: have you had the chance to look into SFP support, octeon_ethernet loading, or EEPROM read fails (although these may be present on EdgeOS, too)?

532910, 2024-08-19:
> editting device-tree ROM

do not remember exact tools I used, but
dump mtd device, hexedit, rwite

>  _main_ and _second_ bootloader?

boot0 and boot1 mtd devices

sur5r, 2024-08-24:
I dug around a little and found the following:

- The MMC frequency can be changed at runtime via sysfs. This makes the md5sum test pass.
- The eMMC reports 25 MHz in its CSD, I need to find out why the kernel seems to ignore it. The code to set the correct frequency is there.
- The DTB from the ROM matches https://github.com/jcowgill/u-boot-ubnt-e200/blob/master/board/octeon/ubnt_e200/ubnt_e200.dts, so maybe this can be included in OpenWRT in the future if the CSD stuff can't be figured out.

532910, 2024-08-24:
> The MMC frequency can be changed at runtime via sysfs. This makes the md5sum test pass.

how exactly?

> The eMMC reports 25 MHz in its CSD, I need to find out why the kernel seems to ignore it. The code to set the correct frequency is there.

possibly due to the Dual Data Rate

sur5r, 2024-08-25:
> > The MMC frequency can be changed at runtime via sysfs. This makes the md5sum test pass.
> 
> how exactly?

echo 26000000 > /sys/kernel/debug/mmc0/clock

> > The eMMC reports 25 MHz in its CSD, I need to find out why the kernel seems to ignore it. The code to set the correct frequency is there.
> 
> possibly due to the Dual Data Rate

Wouldn't a misinterpretation of DDR do the exact opposite?

sur5r, 2024-08-27:
Fruther research reveals: The 25 MHz from the CSD (or rather 26, the decoder I used must be wrong here) is only for standard mode. High Speed mode, which the SDIN7DP2-4G should support as it reports itself as eMMC Version 4.3, is different. The details of HS mode are encoded in EXT_CSD, which is rather complex.
I haven't understood how exactly the switch to HS happens, yet. I have built a kernel with more debugging features enabled. Let's hope this gives some insight.

tucnak, 2024-09-07:
Sorry to chime in all clueless again, like... I have been trying to replicate what @532910 had done earlier since I'm no longer able to simply operate the router from initramfs environment... to re-trace my steps for _posterity_ and hopefully so that it makes sense.

```
# dd if=/dev/mtdblock0 of=/mnt/mtd0.bin
... after hexedit
# diff <(xxd pre/mtd0.bin) <(xxd post/mtd0.bin)
27220c27220
< 0006a530: 0000 0107 0319 7500 0000 0003 0000 0004  ......u.........
---
> 0006a530: 0000 0107 018c ba80 0000 0003 0000 0004  ................
# dd if=post/mtd0.bin of=/dev/mtdblock0
dd: error writing '/dev/mtdblock0': Operation not permitted
1+0 records in
0+0 records out
```

I've had the exact same outcome with `/dev/mtd0` and `mtd` tool, I imagine these partitions are readonly?

The diff itself I think suggests that I had patched the frequency constant correctly: from 0x3197500 (52M) to 0x18cba80 (26M) and the alignment looks right, too. That is to say, I'm pretty comfortable flashing this even though I feel like there's a chance I'll brick this thing. It's a shame I could not! So could I do this from U-boot environment instead: to read the updated images from USB not unlike I would boot? I'm just not sure how to pick the address for `fatload` and whether I wish to use `nand` tool or whatnot. It looks quite intimidating and my brick-radar is turning on!

```
usb start
fatload usb 0:1 0x20000000 initramfs-kernel.bin
bootoctlinux 0
```

alexxx113, 2024-09-15:
have you launched version 23.05.4?
how do I do this ?

alukacs03, 2024-09-27:
Has anyone had any success?
I've tried compiling with 52MHz, 26MHz, zstd compression, ext4 and the combination of all of them, but couldn't get it to work. The SQUASHFS errors start after a few minutes.

sur5r, 2024-09-27:
It's very odd. For me, the issue is just gone. Initially, I could easily reproduce it with `md5sum` as seen in the first post. I retried it yesterday with 23.05.5 and current SNAPSHOT and it didn't fail. My next step is to retry with 23.05.4, 23.05.5 and SNAPSHOT while also checking for differences in mmc extcsd.

beufanet, 2024-10-18:
Hello,

I've the same issue. I can install easily 23.05.5, but have squashfs issues when rebooting it. 

How can I fix this ?

532910, 2024-10-18:
@tucnak, have you tried `mtd unlock`? I do not remember what exactly I did to flash `boot0` and `boot1`, but it was done from openwrt, not uboot.

patching `boot0` and `boot1` that way I specified will break their checksums, so boot process will skip broken `boot0` and continue with `boot1`, it will show checksum errors for both of them, but `boot1` will successfully boot

Do this at your own risk of course and backup before flash.

sur5r, 2024-10-18:
@beufanet That won't help by itself. Either you need to build a new DTB and store it alongside u-boot as @532910 wrote. This is tricky and includes the risk of bricking the device.
If this issue persists with other devices (as I said, it's gone for me, so I'm a bit los) the better way forward would be to get a proper DTS for ER-8 (e200) and change the openwrt build to include it and ignore the one from u-boot.

532910, 2024-10-18:
`cavium-octeon/cn71xx.dtsi` is not used by this platform

beufanet, 2024-10-18:
Ok, not sure to understand every word you mentioned @sur5r .

On my side I moved from EdgeOS to OpenWRT using https://openwrt.org/toh/ubiquiti/edgerouter_pro#easy_installation_via_serial_and_usb

What should I do after that if I cannot fix the openwrt image ? Is it the bootloader that needs to be fixed ? 

Or should I prefer to run OpenWRT from /dev/sda1 usb key with a custom image build, so I can bypassed the eMMC freq issue ?

532910, 2024-10-18:
> https://openwrt.org/toh/ubiquiti/edgerouter_pro

page updated, thank you!

> Is it the bootloader that needs to be fixed ?

this is one one of the possible solutions

> Or should I prefer to run OpenWRT from /dev/sda1

this is another

and the third one is to provide device tree from the linux boot process

tucnak, 2024-10-18:
@532910 I _did_ try `mtd unlock` to no avail. In all honesty, I'm not sure it can be unlocked with stock tools... perhaps somebody could figure out a way to do this at initramfs time? I never got around to building a OpenWrt image but would happily use this opportunity as I'm currently stuck in initramfs and would be forced to restore the whole system from backup, if necessary.

Thankfully, the UPS does its job so it has been a relatively low risk all things considered.

tucnak, 2024-10-18:
@beufanet
> Or should I prefer to run OpenWRT from /dev/sda1 usb key with a custom image build, so I can bypassed the eMMC freq issue ?

I wish you could do this but no, OpenWrt ignores `bootargs` from U-boot and always proceeds with a hard-coded `CMDLINE` so you cannot even boot from USB properly! It's a bit ridiculous, and closes the door on encrypted rootfs of any kind (even though there's cryptsetup/dm-crypt in the initramfs, I think?)

beufanet, 2024-10-19:
> 
> > Is it the bootloader that needs to be fixed ?
> 
> this is one one of the possible solutions
> 

Is this related ? https://github.com/jcowgill/u-boot-ubnt-e200/tree/master

sur5r, 2024-10-19:
I think so. At least the DTS is the same IIRC. I haven't tried building it, yet. My idea was to take the DTS from there and switch OpenWRT to use its own DTB.

Having everyone update u-boot on their devices might be too complex for some users. With a DT shipped with OpenWRT itself, that would not be an issue plus it would make future DT fixes a lot easier, should the need arise.

532910, 2024-10-21:
@tucnak:
```
opkg update
opkg install kmod-mtd-rw
insmod /lib/modules/5.15.167/mtd-rw.ko i_want_a_brick=yes
mtd unlock boot0
```

532910, 2024-10-21:
> I wish you could do this but no, OpenWrt ignores `bootargs` from U-boot and always proceeds with a hard-coded `CMDLINE` so you cannot even boot from USB properly!

this is not true:

```
setenv bootcmd 'usb start;fatload usb 0:1 $(loadaddr) vmlinux.64;bootoctlinux $(loadaddr) numcores=2 endbootargs mem=0 root=/dev/sda2 rootdelay=10 rw rootsqimg=squashfs.img rootsqwdir=w mtdparts=phys_mapped_flash:640k(boot0),640k(boot1),64k(eeprom)'
saveenv
reset
```

take care to save original `bootcmd` before

sur5r, 2024-10-21:
How is this supposed to work with a stock openwrt build? The kernel has an embedded cmdline starting with a hyphen which means it will replace any bootloader supplied cmdline (see `octeon_use_image_cmdline()`). This also matches my observations while working on openwrt/openwrt#15992.

532910, 2024-10-21:
ah, of course it will not work with a stock openwrt build, i was confused by the original question:

> run OpenWRT from /dev/sda1 usb key with a custom image build

moreover imagebuilder has `patch-cmdline` so you don't even need to rebuild the kernel

beufanet, 2024-12-05:
I'm still confused about how I can help fixing this issue ... No idea if it's uboot related or openwrt issue. 

I really would like to have ER8Pro working as my ER4. Because Ubiquity EdgeOS has so many bugs and no supported anymore...

532910, 2024-12-05:
> I'm still confused about how I can help fixing this issue

write DTS so well so it could be accepted to to openwrt or even linux kernel

beufanet, 2024-12-06:
I need to write a new one ? Or should I update an existing one if platform already exists ?

nwf, 2025-01-11:
Just to throw further confusion into the mix, I am happily running a custom build of OpenWRT, based on e74ead2249 from July 2024 (about 1200 commits before what would become v24.10.0-rc1), on an er8pro.

The system is using the device tree provided by the bootloader, naturally enough, and it reports
```
root@dengw:/tmp# cat /sys/kernel/debug/mmc0/clock
52000000
```
Despite that, I can repeatedly `md5sum` the `mmcblk0p1` device with no reported errors and no differences in results, suggesting that the SPI bus is working just fine at this clock rate.

Given this issue and the dire warning on the device wiki page, however, I've been wary of upgrading to a newer snapshot, and so have a 2nd instance of the hardware in the mail.  I'll test on that one, see if I can reproduce this, and report back findings.

sur5r, 2025-01-16:
@nwf When testing er8pro, please be aware of https://github.com/openwrt/openwrt/pull/16238 and https://github.com/openwrt/openwrt/pull/15992. The commit you mentioned doesn't have them, v24.10-rc* does.

That being said I think it's still worthwile to move the DTS to openwrt. That way, changes can be applied much quicker, should the need arise. I will try and draft a PR tomorrow morning.

nwf, 2025-01-18:
The device has arrived in the mail and preliminary investigation suggests...

- Indeed, the LEDE 17.01 initramfs image can access the MMC just fine; double-checking all reads and writes indicates data has landed stably and can be read back intact.
- A byte-for-byte copy of /dev/mmcblockp1 I have on the working device (which reads stably), when transplanted to this new device, fails to boot because of squashfs errors.

This is very strange.  Investigating with `mmc-utils` suggests that the only difference in the MMC devices is that the working one was manufactured in 2005/May while the not working one is from 2001/Apr, but that doesn't seem plausible as a cause.  I'm loathe to crack open the case of the working one to see if it has a different crystal frequency from the not-working one.

The bootloaders (/dev/mtdblock[01]), and so I presume the DTBs, are identical across the two devices.

nwf, 2025-01-19:
I have done a few more hours of investigating and... I have no idea why my one device works and the other does not.  Further investigation reveals that 18.06.0 indeed does not work without halving the clock rate.  It's worth noting, perhaps, that 17.01 used a driver that was patched in from OpenWRT, while 18.06 used a kernel that had its own `drivers/mmc/host/cavium-octeon.c` which is slightly different in the details (especially `octeon_mmc_set_ios` which looks relevant, in particular in its handling of DDR, which might be germane?).

In any case, yes, surely the right answer is to use (y)our own DTB rather than the one provided by the bootloader.  OpenWRT already sets `CONFIG_MIPS_ELF_APPENDED_DTB` and will prefer an appended DTB to the kernel to the one passed from the bootloader.

532910, 2025-01-19:
> one device works and the other does not

could they have different emmc chips?

nwf, 2025-01-19:
@532910 As I wrote earlier,
> Investigating with mmc-utils suggests that the only difference in the MMC devices is that the working one was manufactured in 2005/May while the not working one is from 2001/Apr, but that doesn't seem plausible as a cause.

nwf, 2025-01-26:
OK, I _think_ the most plausible way forward for these devices is to...

1. Import https://github.com/jcowgill/u-boot-ubnt-e200 into the tree, enabling the `CONFIG_FDT_CMD` along the way,
2. Set the native u-boot environment to chainload into that,
3. Set its environment to modify the FDT's `spi-max-requency` value, and
4. Load OpenWRT from there.

I've done some of this by hand rather than properly and will continue to experiment.

tucnak, 2025-04-05:
@nwf I take it that this sequence on serial has to be performed on every power cycle? Not being familiar with neither OpenWrt internals, nor U-boot, I'm finding it hard to follow your instructions. I've booted mine a few months back from USB, and thankfully it hasn't crashed or required reboot since but I'm still looking for a real solution.

Also: you did mention that #16224 is probably related to this, I take it that your approach never resolved this? I know for a fact that the official fw works fine. Can you try and explain, from first principles, if you will, how the two issues are related, and is a permanent solution eventually possible here?

Or perhaps it's not worth the hassle at this point and there's better rack-mount alternative that would also support SFP?

nwf, 2025-04-05:
@tucnak 

I apologize; I haven't the time to explain things "from first principles", but I will try to give a higher level summary of my thinking so far.

From observational data, it appears that there are (at least) two variants of the E200 board floating around: one which requires a 25MHz SPI clock, and one which tolerates a 50MHz clock.  You and I seem to be lucky and have the latter flavor in production, and I also seem to own one of the former flavor in my test lab.

The way that Ubiquiti seems to have dealt with this -- and this is _complete speculation_, since I do not have their sources in my possession -- is by having some combination of U-boot and the very early Linux kernel boot code rewrite parts of the FDT.  That is also the connection with #16224: the in-FLASH FDT seems to contain more Ethernet port descriptors than the hardware has, along with annotations for which should get removed during the boot chain, before the kernel drivers come up.

I think that we will want to do something similar -- and yes, on every boot -- but I have not had time to do more than some initial experiments.  Because my in-production unit works well enough for me, and I have documented above an unpleasant but acceptable workaround for the test unit, should it need to be pressed into service, I have not undertaken either the effort required to figure out exactly what edits must be done for SFP (or, indeed, if edits to the FDT are sufficient) nor the effort required to automate the act of editing the FDT (which will require chaining through a 2nd u-boot, because the in-FLASH one Ubiquiti shipped does not have the general `fdt` manipulation command exposed to the U-boot command interpreter).

It could be possible to solve the problem per-board differently, by rewriting the FDT contained within the bootloader FLASH, but as I am not equipped to JTAG these boards back to life, I have not wanted to go down that path.

tucnak, 2025-04-05:
Thanks so much for taking the time to explain this so thoroughly. I'm also really glad that you took the time to experiment with what must seem like ancient hardware at this point; it doesn't seem to interest anybody, it seems. It may sound silly, but I'm still determined to learn my way around it, if only to see it to the end, & have it part of my dream home-lab rack with as many open components as possible—including a OpenPOWER server, & some FPGA stuff for root-of-trust. I can see that you're a security researcher; maybe you would be open to a consulting engagement that would serve as learning opportunity for me? I was really intrigued having learned that the CPU in it was backdoored, and dealing with that, but haven't got far on my own.

Likewise, I would also appreciate it if your advice were that my energies are better spent elsewhere!

nwf, 2025-04-06:
Almost surely our times are better spent elsewhere, rather than reverse engineering old equipment.  However, because mysteries are fun and I can't sleep...

Using the thing that calls itself `U-Boot 2012.04.01 (UBNT Build ID: 4670715-g7c4b1d0) (Build time: May 27 2014 - 11:19:05)` with `md5sum` `a3451a9aa05006d19ff57ed015d40d46`, `binwalk` says

> 426176        0x680C0         Flattened device tree, size: 9251 bytes, version: 17

So `dd bs=1 if=loader iseek=426176 count=9251 | dtc --sort -O dts -o loader.dts -I dtb` will extract the in-FLASH DTB and render it for human consumption.  First, note that many of the `ethernet-phy` nodes have properties like `cavium,qlm-trim = "0,nosfp"`.

(Curiously, and tangentially, this differs, even after reformatting, from the DTS available in the GPL sources at https://github.com/jcowgill/u-boot-ubnt-e200/blob/8169e56b7f9f748161dd43a4c7d72ffdfd24c128/board/octeon/ubnt_e200/ubnt_e200.dts not just in the use of aliases to refer to nodes, but the latter omits several `linux,phandle` properties and apparently changes the exact structure, but not leaf bytes, of some values (notably mmc's `reg` and interrupt properties).  Anyway, moving right along...)

We can obtain the DTB used by the kernel, as modulated by the bootloader, from a running system.  Here, we'll take the 50MHz system currently running OpenWRT fairly near HEAD.  `dtc --sort -o running-50MHz-owrt.dts -O dts -I fs /sys/firmware/devicetree/base`.  This differs from the DTS extracted above from the loader in several key ways:

- There's a `memory` block describing the shape of DRAM in physical address space.
- Four `ethernet-phy` nodes (those with "cavium,qlm-trim" values of "0,nosfp") are deleted.
- All `qlm-trim` properties are deleted.
- All remaining `ethernet` nodes have their `local-mac-address` and `mac-address` properties populated.
- The two "octeon-3860-uart" `serial` nodes have their `clock-frequency` values populated (with 0x23c34600, perhaps meaning 600 MHz).

We can find the code responsible for the `qlm-trim` processing and the MAC address population at
https://github.com/jcowgill/u-boot-ubnt-e200/blob/8169e56b7f9f748161dd43a4c7d72ffdfd24c128/board/octeon/ubnt_e200/ubnt_e200_board.c#L44-L98 with further ethernet processing, the memory node, and the UART node updates being in the generic function it overrides and calls, https://github.com/jcowgill/u-boot-ubnt-e200/blob/8169e56b7f9f748161dd43a4c7d72ffdfd24c128/arch/mips/cpu/octeon/octeon_fdt.c#L204-L319

While we can't find any evidence of `spi-max-frequency` twiddling in U-boot, extracting the kernel from a Ubiquiti firmware dump and hitting with `strings` sure gives some suspicious results: 

> usbn
> 6UBNT board DTS pruning ... board %d major %d
> /soc/mmc/mmc-slot@2
> /soc/mmc/mmc-slot@0
> spi-max-frequency
> 3Error: (%s) Invalid irq trigger specification: %x

If someone wants to hit that with Ghidra, we may be able to figure out what they're keying on determining which boards need work.

I have my suspicions, tho'.  I'd be very curious what the output of `xxd /dev/mtdblock2 | head -n2` (or `hexdump` instead of `xxd`) on a **non-52MHz-capable** device says.  Note that this will include your MAC address, which might be sensitive.  In particular, bytes 0xc through 0xf (0-indexing and inclusive) on my **52MHz-capable** device are `ee110777`, indicating that this is an E200 board.  I wonder if the other boards have this set to `ee1307777`, the mysterious and elusive "E202" board; I'll test this hypothesis if nobody beats me to it before I retrieve my test machine from storage.

Vexingly, the bootloader (and who knows what else in early machine bringup) gets to look at parts of flash that aren't exposed by the kernel to userspace in the default configuration.  Specifically, the "GD" and uboot environment are well past the end of the "eeprom" mtd partition.  I'll play with that, too, when I have my test device available again.

hecatae, 2025-05-20:
I've just received one of these, if there's anything you want me to compare please let me know.

DependableUser, 2025-06-10:
Hi all, is 17.01.7 still the last official stable realease working on the E200? or is 24.10 OK?
I have an 8Pro I would like to move to Openwrt but not with 6 year old firmware and as it is for production I cannot go for a beta or non offcial
TIA

RaylynnKnight, 2025-06-11:
> Hi all, is 17.01.7 still the last official stable realease working on the E200? or is 24.10 OK? I have an 8Pro I would like to move to Openwrt but not with 6 year old firmware and as it is for production I cannot go for a beta or non offcial TIA

See my post from August 18, 2024 above.  At least 19.07.4 was stable for me.

nwf, 2025-06-11:
I _believe_ the state of play is that if you have "the right" sub-kind of E200 board, all recent releases will work for you, but if you are unlucky, _something_ has changed since 17.01 and it's either always broken or almost always broken.  See the, mm, journey of discovery I've taken in this thread.  I have yet to reverse engineer out the magic bits the Ubiquiti kernel is looking at, and nobody (including them) has been able or willing to hand me their kernel sources.

FWIW, the `initramfs` images should *always* work, since they don't touch the MMC, so if you're comfortable with building your own images and baking in configuration (`files/` in the build tree) and with any changes reverting on reboot, that is, technically, an option for these devices.  (You _might_ even be able to take that approach and the ext-root support and a fair bit of elbow grease and force and get something that persists configuration to a suitably reconfigured MMC, too.)

hecatae, 2025-06-12:
@nwf I have an e220 on it's way to me, aka the USG Pro 4, so I'll be able to confirm that against my Edgerouter when it arrives.

nwf, 2025-06-12:
@sur5r :O :O :O  Good find!  Yes, it looks like I suspected: the MMC clock speed is incorrectly specified in the DTB provided by the boot loader and they fixed it in the kernel.  ARGH!  I'm not quite sure what the right way forward for OpenWRT is, but that's still fantastic, thank you!

DependableUser, 2025-06-13:
I know nothing about compiling, I have to wait until you who know release the goods.
Is there an easy way to know whether my board is a "good" one ? like a list of compatible chips I can check against the ones in my 8P? 
Cheers

sur5r, 2025-06-13:
@DependableUser There's no risk of bricking it. It's "just" the risk of instability and data corruption. You will always be able to reinstall or `firstboot` it. But given this issue has been open for more than a year there's no point in rushig stuff now.

@nwf Just to make sure we don't duplicate effort here: Do you want to prepare the patch or should I?

sur5r, 2025-06-18:
More stuff:
https://github.com/sur5r/linux contains branches for each OpenWRT release with the patch applied.
https://github.com/sur5r/openwrt contains branches for each OpenWRT release with the patch included.

I will do a snapshot build of all 3 (23.05, 24.10, main) and put them online for everyone to test.

[19 more replies]