openwrt/openwrt #13763: Build all core packages for selected target / Build Packages with external toolchain (x86, 64) / Build x86/64 error

InBomBi, 2023-10-22:
### Describe the bug

`Run make target/compile -j$(nproc) BUILD_LOG=1 || ret=$? .github/workflows/scripts/show_build_failures.sh
  make target/compile -j$(nproc) BUILD_LOG=1 || ret=$? .github/workflows/scripts/show_build_failures.sh
  shell: su buildbot -c "sh -e {0}"
WARNING: Makefile 'package/utils/busybox/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/busybox/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/busybox/Makefile' has a build dependency on 'libpam', which does not exist
WARNING: Makefile 'package/boot/kexec-tools/Makefile' has a dependency on 'liblzma', which does not exist
WARNING: Makefile 'package/network/services/lldpd/Makefile' has a dependency on 'libnetsnmp', which does not exist
WARNING: Makefile 'package/utils/policycoreutils/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/policycoreutils/Makefile' has a dependency on 'libpam', which does not exist
WARNING: Makefile 'package/utils/policycoreutils/Makefile' has a build dependency on 'libpam', which does not exist
make[[2](https://github.com/InBomBi/openwrt-build/actions/runs/6603794618/job/17937452904#step:31:2)]: Entering directory '/__w/openwrt-build/openwrt-build/openwrt/scripts/config'
make[2]: 'conf' is up to date.
make[2]: Leaving directory '/__w/openwrt-build/openwrt-build/openwrt/scripts/config'
 make[1] target/compile
 make[2] -C target/linux compile
    ERROR: target/linux failed to build.
make: *** [/__w/openwrt-build/openwrt-build/openwrt/include/toplevel.mk:2[3](https://github.com/InBomBi/openwrt-build/actions/runs/6603794618/job/17937452904#step:31:3)2: target/compile] Error 1
make -r target/compile: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
====== Make errors from logs/target/linux/compile.txt ======
./scripts/pahole-flags.sh: 12: [: Illegal number: 
./scripts/pahole-flags.sh: 16: [: Illegal number: 
./scripts/pahole-flags.sh: 20: [: Illegal number: 
pahole: error while loading shared libraries: libdwarves_emit.so.1: cannot open shared object file: No such file or directory
./scripts/pahole-flags.sh: 12: [: Illegal number: 
./scripts/pahole-flags.sh: 16: [: Illegal number: 
./scripts/pahole-flags.sh: 20: [: Illegal number: 
pahole: error while loading shared libraries: libdwarves_emit.so.1: cannot open shared object file: No such file or directory
init/Kconfig:97: syntax error
init/Kconfig:96: invalid statement
make[6]: *** [scripts/kconfig/Makefile:77: syncconfig] Error 1
make[5]: *** [Makefile:628: syncconfig] Error 2
make[[4](https://github.com/InBomBi/openwrt-build/actions/runs/6603794618/job/17937452904#step:31:5)]: *** [Makefile:748: include/config/auto.conf.cmd] Error 2
make[3]: *** [Makefile:21: /__w/openwrt-build/openwrt-build/openwrt/build_dir/target-mips-openwrt-linux-musl_musl/linux-ath79_generic/linux-[5](https://github.com/InBomBi/openwrt-build/actions/runs/6603794618/job/17937452904#step:31:6).15.135/.modules] Error 2
make[2]: *** [Makefile:11: compile] Error 2
time: target/linux/compile#25.[6](https://github.com/InBomBi/openwrt-build/actions/runs/6603794618/job/17937452904#step:31:7)6#[7](https://github.com/InBomBi/openwrt-build/actions/runs/6603794618/job/17937452904#step:31:8).62#2[9](https://github.com/InBomBi/openwrt-build/actions/runs/6603794618/job/17937452904#step:31:10).91
Error: Process completed with exit code 2.`

### OpenWrt version

Main Branch 23.05

### OpenWrt target/subtarget

All target kernel

### Device

All

### Image kind

Official downloaded image

### Steps to reproduce

I have got the error when i run sync folk, and workfollow running
Build all core packages for selected target / Build Packages with external toolchain (x86, 64) / Build x86/64

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-10-22:
> Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on

brada4, 2023-10-22:
actually not... it is general problem thats functions is often called twice

InBomBi, 2023-10-23:
> actually not... it is general problem thats functions is often called twice

So , will i be wait for fix from support nor what will i do.
This is action following of git.
Thanks

brada4, 2023-10-23:
First  you need to say what linux you use for action. Then make diffconfig. Then run diagnostic build. You need to attach  long log with cause, not consequence of error.

InBomBi, 2023-10-23:
https://github.com/InBomBi/openwrt-build/blob/main/.github/workflows/kernel.yml

This action auto running. I clone repo from openwrt git. 
The linux they run on ubuntu.
And when i sync folk from openwrt, its runs the action., every check action is ok, then build kernel runs with error as i posted.

closed 2024-03-08 by InBomBi