openwrt/openwrt #13767: odhcpd doesn't log its own leases - LuCi UI indicates expired when the leases are not expired

a-spicer, 2023-10-23:
### Describe the bug

OpenWRT 23.05, odhcpd 2023-06-24-52112643-1, luci git-23.051.66410-a505bb1

odhcpd assigns addresses ok. It does not update its /tmp/hosts/odhcpd leases file. It sometimes seems to get into "ubus call dhcp ipv6leases".

And because of this the Luci seems unable to keep up in:

"/cgi-bin/luci/admin/status/overview" "Active DHCPV6 Leases".

With the "Lease time remaining" showing as "Expired".

So in the UI and in the command line ... it's not getting updated. In the old version it also seemed a bit sluggish updating the leases ... but a least the Luci UI show valid leases and time remaining. (22.03?)

### OpenWrt version

23.05.0

### OpenWrt target/subtarget

mediatek/mt7622

### Device

Linksys E8450 (UBI)

### Image kind

Official downloaded image

### Steps to reproduce

Just run the router normally. Renew IPv6 clients to see if it changes anything. Sometimes RENEW6'ing a Windows box gets that one renewed in the leases file (odhcpd) and in Luci UI.

### Actual behaviour

Actually after some hours Luci UI shows "expired" for DHCPv6 leases. The leases file for odhcpd does not update.

### Expected behaviour

Expect the leases file to reflect the actual lease renewals that get done. Expect the LuCi UI to indicate the leases are ok (to reflect reality.)

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

a904055262, 2023-10-24:
the same here x86 23.05

a-spicer, 2023-10-29:
There were problems back in 2019

https://forum.openwrt.org/t/solved-19-07-regression-luci-not-showing-the-2nd-ipv6-addresses/48271/5

with rpcd-mod-luci and luci-mod-status but I doubt those issues still exist. This seems to point to where the IPV6 DHCP Leases info on the Status page comes from. The JS is thick as molasses and hard to decipher. But I think it still comes down to LuCi is reading /etc/hosts/odhcpd file and it's not being updated. If they were using "ubus" call

ubus call dhcp ipv6leases

for dhcpv6 leases they would get it just fine. I don't even know how that leases file is containing the lease remaining time. Unless Luci is taking the time stamp of the file and counting down the 12 hours lease time based on that. (It has the lease time in the uci dhcp config file.)

ldir-EDB0, 2023-11-01:
I believe this should be fixed by https://git.openwrt.org/?p=project/odhcpd.git;a=commit;h=4bbc6e74248feeb756bd03dc500fb4f446ccfc49

a-spicer, 2023-11-02:
I believe this should be fixed by LuCi UI wherever that is. Either that or ODHCPD maintainer ought to fix whatever is broken that LuCi UI is reading to get the IPv6 leases data. ODHCPD doesn't seem to keep its /tmp/hosts/odhcpd leases file updated. It goes past the 12 hour default lease time (and actually performs the renewal with clients just fine) without updating that file.

root@OpenWrt:~# ll /tmp/hosts/odhcpd
-rw-r--r--    1 root     root           341 Oct 31 11:17 /tmp/hosts/odhcpd
root@OpenWrt:~# date
Wed Nov  1 22:03:18 EDT 2023
-------------------------------------------------------------------------------------------------------------------------------------

ubus call dhcp ipv6leases seems to be kept up with the current leases status ... and it's pretty trivial to calculate the time remaining. The LuCi UI (status page) says expired for all 4 of the following. (Note: I installed "bc" to do the math.)

root@OpenWrt:~# cat ./ubus-hours-left-dhcp-ipv6leases.sh

#!/bin/sh

for i in `./ubus-call-dhcp-ipv6leases.sh | grep address | awk '{print $2}' | cut -d\" -f2`

do

echo "$i"

done

for i in `./ubus-call-dhcp-ipv6leases.sh | grep valid-l | awk '{print $2}'`

do

echo "Time - Seconds: $i"

echo -n "Time - Hours: "

echo "$i/3600" | bc

done

root@OpenWrt:~# cat ./ubus-call-dhcp-ipv6leases.sh

#!/bin/sh

ubus call dhcp ipv6leases

root@OpenWrt:~#

-----------------------------------------------------------------------------------------------------------

XXXX:XXXX:XXXX:a64a::9e5

XXXX:XXXX:XXXX:a648::9e5

XXXX:XXXX:XXXX:a64a::e81

XXXX:XXXX:XXXX:a648::e81

Time - Seconds: 28936

Time - Hours: 8

Time - Seconds: 28936

Time - Hours: 8

Time - Seconds: 41032

Time - Hours: 11

Time - Seconds: 41032

Time - Hours: 11

----------------------------------------------------------------------------------------------------------

Slightly improve hours count script:

#!/bin/sh

count=0

for i in `./ubus-call-dhcp-ipv6leases.sh | grep address | awk '{print $2}' | cut -d\" -f2`

do

count=$((count+1))

echo "IpV6 Lease IP # $count: $i"

done

count=0

for i in `./ubus-call-dhcp-ipv6leases.sh | grep valid-l | awk '{print $2}'`

do

count=$((count+1))

echo "Lease IP # $count: Time - Seconds: $i"

echo -n "Lease IP # $count:Time - Hours: "

echo "$i/3600" | bc

done

----------------------------------------------------------------------------------------------------------
root@OpenWrt:~# ./ubus-hours-left-dhcp-ipv6leases.sh
IpV6 Lease IP # 1: XXXX:XXXX:XXXX:a64a::9e5
IpV6 Lease IP # 2: XXXX:XXXX:XXXX:a648::9e5
IpV6 Lease IP # 3: XXXX:XXXX:XXXX:a64a::e81
IpV6 Lease IP # 4: XXXX:XXXX:XXXX:a648::e81
Lease IP # 1: Time - Seconds: 26699
Lease IP # 1:Time - Hours: 7
Lease IP # 2: Time - Seconds: 26699
Lease IP # 2:Time - Hours: 7
Lease IP # 3: Time - Seconds: 38795
Lease IP # 3:Time - Hours: 10
Lease IP # 4: Time - Seconds: 38795
Lease IP # 4:Time - Hours: 10
root@OpenWrt:~#

a-spicer, 2023-11-02:
ubus still being kept up to date either, this morning. Two leases for Windows not even in there. iPhone leases is.

a-spicer, 2023-11-08:
This issue seems to have been resolved. Their was a 10-24-23 odhcpd-ipv6only update - which I installed. As well as an update to Luci Status UI module. Among several updates for OpenWRT 23.05 for this platform. Updates seem to occur almost instantly to the leases file mention above and to ubus readout. The LuCi Status page now gets that leases are updated and updates its displayed information. Thank you to those that fixed this.

a-spicer, 2024-08-31:
I hate to revive an old post, but I just upgraded to OpenWrt 23.05.4 from 23.05.0 and DHCPv6 (odhcpd) seems to be screwy all over again. It's now renewing to 50 some minutes instead of what it should be - like 1 day and 12 hours. The same is also NOW reflecting in the UBUS call in seconds ... which should be like 36 hours not 1 hour or less.

a-spicer, 2024-08-31:
root@OpenWrt:~# service odhcpd info | grep thresh
                                        "threshold": 3600,
Why is the threshold 3600, which is 1 hour?

a-spicer, 2024-09-01:
Closing this issue as new issue has been created:
https://github.com/openwrt/openwrt/issues/16305

closed 2024-09-01 by a-spicer