openwrt/openwrt #13769: [IPQ40xx] spi-nand spi0.1: SPI transfer failed: -110

stanislav-brabec, 2023-10-23:
### Describe the bug

I am experiencing strange erors on My Asus RT-AC58U, OpenWRT 23.05.0.

Mon Oct 23 02:24:28 2023 kern.err kernel: [198676.024985] spi-nand spi0.1: SPI transfer failed: -110
Mon Oct 23 02:24:28 2023 kern.err kernel: [198676.025616] spi_master spi0: failed to transfer one message from queue
Mon Oct 23 02:24:28 2023 kern.warn kernel: [198676.029446] ubi0 warning: 0xc0708a50: error -110 while reading 1556 bytes from PEB 142:14000, read only 336 bytes, retry

or elsewhere:

Mon Oct 23 02:30:35 2023 kern.err kernel: [199040.020515] spi-nand spi0.1: SPI transfer failed: -110
Mon Oct 23 02:30:38 2023 kern.err kernel: [199040.020610] spi_master spi0: failed to transfer one message from queue
Mon Oct 23 02:30:39 2023 kern.warn kernel: [199040.024601] ubi0 warning: 0xc0708a50: error -110 while reading 1483 bytes from PEB 186:43592, read only 1464 bytes, retry

and elsewhere:

Mon Oct 23 03:11:41 2023 kern.err kernel: [201508.876641] spi-nand spi0.1: SPI transfer failed: -110
Mon Oct 23 03:11:41 2023 kern.err kernel: [201508.877288] spi_master spi0: failed to transfer one message from queue
Mon Oct 23 03:11:41 2023 kern.warn kernel: [201508.880991] ubi0 warning: 0xc0708a50: error -110 while reading 4096 bytes from PEB 597:118784, read only 0 bytes, retry
Mon Oct 23 03:11:41 2023 kern.err kernel: [201508.979730] SQUASHFS error: xz decompression failed, data probably corrupt
Mon Oct 23 03:11:41 2023 kern.err kernel: [201508.979808] SQUASHFS error: Failed to read block 0x2a68b6: -5
Mon Oct 23 03:11:41 2023 kern.err kernel: [201508.985524] SQUASHFS error: Unable to read fragment cache entry [2a68b6]
Mon Oct 23 03:11:41 2023 kern.err kernel: [201508.991472] SQUASHFS error: Unable to read page, block 2a68b6, size f68c
Mon Oct 23 03:11:41 2023 kern.err kernel: [201508.998327] SQUASHFS error: Unable to read fragment cache entry [2a68b6]
Mon Oct 23 03:11:41 2023 kern.err kernel: [201509.005056] SQUASHFS error: Unable to read page, block 2a68b6, size f68c
Mon Oct 23 03:11:41 2023 kern.err kernel: [201509.011880] SQUASHFS error: Unable to read fragment cache entry [2a68b6]
Mon Oct 23 03:11:41 2023 kern.err kernel: [201509.018625] SQUASHFS error: Unable to read page, block 2a68b6, size f68c
Mon Oct 23 03:11:41 2023 kern.err kernel: [201509.025386] SQUASHFS error: Unable to read fragment cache entry [2a68b6]
Mon Oct 23 03:11:41 2023 kern.err kernel: [201509.032167] SQUASHFS error: Unable to read page, block 2a68b6, size f68c
Mon Oct 23 03:11:41 2023 kern.err kernel: [201509.038982] SQUASHFS error: Unable to read fragment cache entry [2a68b6]
Mon Oct 23 03:11:41 2023 kern.err kernel: [201509.045683] SQUASHFS error: Unable to read page, block 2a68b6, size f68c
Mon Oct 23 03:11:41 2023 kern.err kernel: [201509.052505] SQUASHFS error: Unable to read fragment cache entry [2a68b6]
Mon Oct 23 03:11:41 2023 kern.err kernel: [201509.059248] SQUASHFS error: Unable to read page, block 2a68b6, size f68c

followed by disconnection from all networks for several minutes

### OpenWrt version

r23497-6637af95aa

### OpenWrt target/subtarget

ipq40xx/generic

### Device

ASUS RT-AC58U

### Image kind

Official downloaded image

### Steps to reproduce

The bug appears randomly about once or twice daily.

### Actual behaviour

After that error, the device stops responding for several minutes. No access from LAN nor WAN. Then it again starts to communicate (at lease sometimes).

### Additional info

I think it is a regression. Up to probably 22.03.3 it was rock stable. Starting with 22.03.4 or 22.03.5, I started to experience random network disconnections. But I do not have a log from these disconnections. This is a first time I decided do not hard reboot immediately, but wait and try to get the log.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

jlpapple, 2023-10-29:
Could it be related to this recent commit?  https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=98d325aaf8bef992cc92e94feb14fe271d370dc0

christf, 2025-08-11:
I am seeing this too on an AVM 4040

Djfe, 2025-08-19:
feel free to retest
3ff8a3dca8bc24296f501b3c74214eb5005354bf was just merged and will be backported soon by #19796

zhangsd20, 2025-10-17:
ipq4019 E2600AC c1
openwrt22.03.4

I have also tested other versions such as 21.02, 22.03, and 23.05, but only versions 19.07 and earlier work properly.