openwrt/openwrt #13771: Xiaomi AX3200

Writter77, 2023-10-23:
### Describe the bug

Stability and speed

### OpenWrt version

r23497-6637af95aa

### OpenWrt target/subtarget

mediatek/mt7622

### Device

Xiaomi AX3200

### Image kind

Official downloaded image

### Actual behaviour

There is a problem with the speed on LAN, symmetrical connection 1 GB / s, Xiaomi AX3200 router download 888 Mbps, upload 945 Mbps.
Old router, manufacturer's software 944 download, 940 upload.

WiFi 2G without a speed problem. Admittedly, the speed on the 2G radio is only 108 Mbps, but it is symmetrical. WiFi 5G similar problem for LAN, download 650 Mbps, upload 720 Mbps.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

khanjui, 2023-10-23:
Turn on hardware flow offloading from firewall settings.

Writter77, 2023-10-23:
I had offloading and hardware support enabled before. The problem continues to occur

raenye, 2023-10-23:
This is a continuation of #13739.

danpawlik, 2023-10-23:
Did you try to enable packet steering?

Writter77, 2023-10-25:
> Did you try to enable packet steering?

No. Could you tell me how to turn it on in Luci?

danpawlik, 2023-10-26:
> > Did you try to enable packet steering?
> 
> No. Could you tell me how to turn it on in Luci?

Via luci:
Network -> Interfaces -> Global network options -> Packet Steering

Via ssh:
```
# in /etc/config/network

config globals 'globals'
        option packet_steering '1'
```

You can try also to install irqbalance and remember to turn it on in `/etc/config/irqbalance` it is: `option enabled '1'`

Writter77, 2023-10-29:
> > > Did you try to enable packet steering?
> > 
> > 
> > No. Could you tell me how to turn it on in Luci?
> 
> Via luci: Network -> Interfaces -> Global network options -> Packet Steering
> 
> Via ssh:
> 
> ```
> # in /etc/config/network
> 
> config globals 'globals'
>         option packet_steering '1'
> ```
> 
> You can try also to install irqbalance and remember to turn it on in `/etc/config/irqbalance` it is: `option enabled '1'`
> 
> EDIT: you can also change `governor` to `performance`:
> 
> * add into /etc/rc.local
> 
> ```
> echo performance > /sys/devices/system/cpu/cpufreq/policy0/scaling_governor
> ```

I have a symmetrical LAN on 1Gb download 940 and upload 840 sometimes 500?? There was no problem on the router from the operator. 
I turned it on everything you were talking about and nothing helped

danpawlik, 2023-10-29:
> I have a symmetrical LAN on 1Gb download 940 and upload 840 sometimes 500?? There was no problem on the router from the operator. I turned it on everything you were talking about and nothing helped

That's odd. I got same bandwidth at home and right now when I do test results are: 933/935 (I can even share screenshot:) ).

Is your ISP require PPPoE?

Writter77, 2023-10-29:
Well, on other routers with openwrt this problem did not occur once.
On the current version of openwrt 23.05 I have problems with sending. The same problem is on 5G, download 550, sending 640

There is something wrong with OpenWrt. This is a mistake.

Writter77, 2023-10-29:
I turned off hardware support in offloading and it seems to be oki now. Speeds are symmetrical again on LAN and on 5G. So my Xiaomi ax3200 does not support hardware support.

danpawlik, 2023-10-30:
> I turned off hardware support in offloading and it seems to be oki now. Speeds are symmetrical again on LAN and on 5G. So my Xiaomi ax3200 does not support hardware support.

it supports, but somewhere is a bug.
Could you just confirm, that you are using PPPoE auth for wan connection? If other, please share it here.