openwrt/openwrt #13772: [BananaPi BPI-R3] Thermal not correctly working with 3-pin pwm fan

cla-azzarello, 2023-10-23:
### Describe the bug

Hi community,

I think temp trip points for Banana BPI-R3 is too high and they are not set correctly.

I changed temperature trip points in /sys/class/thermal/thermal_zone0.

The original points in the firmware are:
125000 <-- trip_point_0_temp (critical)
120000 <-- trip_point_1_temp (hot)
115000 <-- trip_point_2_temp (active)
85000 <-- trip_point_3_temp (active)
60000 <-- trip_point_4_temp (active)

I changed my points to:
100000 <-- trip_point_0_temp (critical)
90000 <-- trip_point_1_temp (hot)
80000 <-- trip_point_2_temp (active)
60000 <-- trip_point_3_temp (active)
45000 <-- trip_point_4_temp (active)

I'm running around 50 degrees (the connection is quite idle).
```
root@openwrt:/sys/class/thermal/thermal_zone0# cat temp 
50029
```

But the fan is not starting
```
root@openwrt:/sys/class/thermal/thermal_zone0# cat ../cooling_device0/type 
pwm-fan
root@openwrt:/sys/class/thermal/thermal_zone0# cat ../cooling_device0/cur_state 
0
```

Looking at the cooling devices available on my system
```
root@openwrt:/sys/class/thermal/thermal_zone0# cat ../cooling_device0/type 
pwm-fan
root@openwrt:/sys/class/thermal/thermal_zone0# cat ../cooling_device1/type 
mt7915_phy0
root@openwrt:/sys/class/thermal/thermal_zone0# cat ../cooling_device2/type 
mt7915_phy1
```
As you can see, cdev 1 and 2 is bound to the mt7915, that are the wifi chipsets.
Why a wifi chipset is seen as a cooling device?

Let's see at the trip points bound to these cooling devices
```
root@openwrt:/sys/class/thermal/thermal_zone0# ls cdev*_trip_point
cdev0_trip_point  cdev1_trip_point  cdev2_trip_point
root@openwrt:/sys/class/thermal/thermal_zone0# cat cdev0_trip_point
2
root@openwrt:/sys/class/thermal/thermal_zone0# cat cdev1_trip_point
3
root@openwrt:/sys/class/thermal/thermal_zone0# cat cdev2_trip_point
4
```

I see the cooling device 0 (my pwm fan) is bound only to trip point 2.

Let's change trip point 2 to a lower temperature
```
root@openwrt:/sys/class/thermal/thermal_zone0# echo 45000 > trip_point_2_temp
root@openwrt:/sys/class/thermal/thermal_zone0# cat trip_point_2_temp
45000
root@openwrt:/sys/class/thermal/thermal_zone0# cat temp
45197
root@openwrt:/sys/class/thermal/thermal_zone0# cat ../cooling_device0/cur_state 
2
root@openwrt:/sys/class/thermal/thermal_zone0# cat temp
44895
root@openwrt:/sys/class/thermal/thermal_zone0# cat ../cooling_device0/cur_state 
0
root@openwrt:/sys/class/thermal/thermal_zone0# cat temp
45046
root@openwrt:/sys/class/thermal/thermal_zone0# cat ../cooling_device0/cur_state 
2
root@openwrt:/sys/class/thermal/thermal_zone0# cat temp
44895
root@openwrt:/sys/class/thermal/thermal_zone0# cat ../cooling_device0/cur_state 
0
```
The fan is up, but it's continuously spin up and down due to the limit temperature choosen.

At the moment, I set trip point 2 to 31 degrees (I think I never reach this temperature, so the fan is always up at max speed until I find a definitive solution)
```
root@openwrt:/sys/class/thermal/thermal_zone0# echo 31000 > trip_point_2_temp
root@openwrt:/sys/class/thermal/thermal_zone0# cat trip_point_2_temp
31000
```

My temp is stable around 41 degrees at the moment, but I don't understand how to manage speed 1 and 2.

I think there is something wrong in the firmware.

I don't know very much about thermal subsystem, so it's not clear for me.

### OpenWrt version

OpenWrt 23.05.0, r23497-6637af95aa

### OpenWrt target/subtarget

mediatek/filogic

### Device

Bananapi BPI-R3

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

antnyfls, 2023-10-23:
Comparing bpi-r3 dts form the bpi source _`cooling-levels = <255 96 52 0>`_; vs openwrt _`cooling-levels = <255 96 0>;`_
the missing _`52`_ could be  the issue.
bpi source https://github.com/BPI-SINOVOIP/BPI-R3-bsp-5.15/blob/783d667801f406299d2b478602d087e72a2e5a0a/linux-mt/arch/arm64/boot/dts/mediatek/mt7986a-bananapi-bpi-r3-emmc-snand.dts#L477
openwrt https://github.com/openwrt/openwrt/blob/03cb2d63f487a5d716711c2693f126c913e599cf/target/linux/mediatek/files-5.15/arch/arm64/boot/dts/mediatek/mt7986a-bananapi-bpi-r3.dts#L45

raenye, 2023-10-23:
@dangowrt

lukasz1992, 2023-10-25:
@https://github.com/openwrt/openwrt/issues/13259 I have similar observations with mt7981 device

skramstad, 2023-10-25:
Just want to add my experience from snapshot 2023-10-24 with kernel 6.x. 
My 3 pin Noctua fan is spinning all the time.

dangowrt, 2023-10-25:
To me it almost looks like PWM *is not* inverted when using the 3-pin connector, and it *is* inverted when using the 2-pin connector. So we may need yet another device tree overlay to configure that...

skramstad, 2023-10-25:
@dangowrt You might be right there. If I remember correctly my 2 pin fan, did spin up and down a little while doing some testing. I can verify this later if you want me to.

ChaingTsung, 2023-10-31:
Same trouble. And I installed two fans, one is 3pin and another is 2pin(near next to 3pin), and all of them Only run crazily in the moment when the power is turned on. And then fans stops and do not work anymore. Kernel ver is 5.15.136. Anyone knows how to solve?

mikeg-de, 2024-01-11:
I experience the same situation that fans spinning like crazy upon boot. While exchanging with frank-w in the banana pi forum he noted the following (just a summary):

- PWM on kernel 6.1 works for him
- Using kernel 5.15 the clock driver seems different requiring a different pwm config in dts

https://forum.banana-pi.org/t/banana-pi-r3-fan-with-pwm/14406/112
https://patchwork.kernel.org/project/linux-mediatek/list/?series=777712

Does this help?

cla-azzarello, 2024-01-16:
@mikeg-de I think the discussion on the BananaPI forum is on the right path.

I see @dangowrt is on the thread.

Maybe these patches can be optimized and merged on the current stable?

mikeg-de, 2024-01-16:
I believe so too. Unfortunately the possible solution provided by frank-w is, because of time and priorities, out of scope for me to test. Modifying the trip points, regardless of the values I chose, had no effect, though. If the patches could get incorporated, that would be fantastic.

cla-azzarello, 2024-01-17:
> I believe so too. Unfortunately the possible solution provided by frank-w is, because of time and priorities, out of scope for me to test. Modifying the trip points, regardless of the values I chose, had no effect, though. If the patches could get incorporated, that would be fantastic.

Changing the trip points had no effect for me as well.
If you see in my original description of this bug, I pointed out that it seems that only trip_point_2 is bound to the PWM fan.
Indeed, changing that one turns the fan on.

komlevv, 2024-01-30:
same issue here on BPI-R3 with standard 3pin heatsink fan and openwrt-23.05.2-mediatek-filogic-bananapi_bpi-r3-sdcard.img

using a temp workaround setting `echo 40000 > /sys/class/thermal/thermal_zone0/trip_point_2_temp` in `/etc/rc.local`

ByJacob, 2024-03-11:
I'll leave it here, as if someone hits with a similar problem. :) 

[image]

Source: https://forum.banana-pi.org/t/banana-pi-r3-fan-with-pwm/14406/193

mkshevetskiy, 2024-11-15:
found a dts bug which affects cooler rotation

https://github.com/openwrt/openwrt/pull/16962

cla-azzarello, 2025-02-12:
Someone had some updates on that?

I'll do some test on 24.10 when I'll have the time.

I've seen that [PR16974](https://github.com/openwrt/openwrt/pull/16974) was merged on main, but it was reverted immediately.

mkshevetskiy, 2025-02-16:
I post a patch to linux kernel, but seems it was missed. I plan to resend a patch in a few coming days

mkshevetskiy, 2025-02-16:
just now you can use a recipe from the wiki  https://openwrt.org/toh/sinovoip/bananapi_bpi-r3#pwm_fan_rotation_speed

mkshevetskiy, 2026-02-12:
The pwm fan settings was fixed in openwrt main, 25.12, 24.10, but not in 23.05. See
* https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=2916275a3d6cdda8bbc9decb5fd4df5e89a425a1
* https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=095151b2354752be6b4be81e6b22b65227c59746

So may safely tweak trip point limits. If you use older openwrt without a fix you may use recipe from https://openwrt.org/toh/sinovoip/bananapi_bpi-r3#pwm_fan_rotation_speed