openwrt/openwrt #13773: How to switch back to wolfSSL or choose OpenSSL？

ilxp, 2023-10-23:
### Describe the bug

the 23.05.0 release：
Switch from wolfssl to mbedtls as default
OpenWrt has transitioned its default cryptographic library from wolfSSL to Mbed TLS. This shift brings several changes and implications:
Size Efficiency: Mbed TLS is considerably smaller, making it an optimal choice for systems where storage space is paramount.
LTS and ABI Stability: Mbed TLS consistently provides updates via its Long Term Support (LTS) branch, ensuring both security and a stable application binary interface (ABI). In contrast, wolfSSL does not offer an LTS release, and its stable ABI is limited to a specific set of functions.
TLS 1.3 Support: Users should be aware that Mbed TLS 2.28 no longer supports TLS 1.3.
While Mbed TLS is now the default, users who have specific needs or preferences can still manually switch back to wolfSSL or choose OpenSSL.。

but，i  want to use  wolfSSL or choose OpenSSL. what firles need to switch back。 thanks！

### OpenWrt version

23.05.0

### OpenWrt target/subtarget

/etc/openwrt_release

### Device

x86

### Image kind

Self-built image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

raenye, 2023-10-23:
This sounds like a forum question, not an issue.

jow-, 2023-10-24:
Please use forum, mailing lists or IRC for support questions.

closed 2023-10-24 by jow-

bam80, 2024-10-07:
I have the same question, if somebody asked on the forum (or elsewhere) already, please share the link.
Thanks.