openwrt/openwrt #13774: <mbedtls> No more support of P-521 starting from openwrt 23

farfade, 2023-10-23:
### Describe the bug

I use P-521 based CA and intermediate CA.

I face a regression starting from version 23, compared to version 22, because mbedtls does not support it, contrary to wolfssl.

### OpenWrt version

23

### OpenWrt target/subtarget

ramips/mt7621

### Device

D-Team Newifi D2

### Image kind

Official downloaded image

### Steps to reproduce

Use CA and intermediate CA based on P-521 in uhttpd : it won't start.

### Actual behaviour

Have the ability to Use CA and intermediate CA based on P-521

### Expected behaviour

The expected behaviour was the one working in version 22.

### Additional info

Looking at mbedtls repository, I see references to P-521 [here](https://github.com/Mbed-TLS/mbedtls/blob/2bf0870e25fceb93f4b00e0b9142e26e3280168f/include/mbedtls/ecp.h#L528) and [there](https://github.com/Mbed-TLS/mbedtls/blob/2bf0870e25fceb93f4b00e0b9142e26e3280168f/tests/scripts/generate_tls13_compat_tests.py#L147) (in a file of tests for TLS1.3)

It will probably need enablement at compile time.

### Diffconfig

```text
No change in config between 22 and 23.
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

farfade, 2023-10-23:
Rereformulated from #13728

hauke, 2023-10-25:
Do you see this with a normal public CA? 
If you have a website which is affected please post the URL then I can easily reproduce it.

farfade, 2023-10-26:
Thanks @hauke for your interest.

I've no public website (openwrt listens only on an admin network) but the certs chain is [there](https://github.com/openwrt/openwrt/files/13060541/cert.txt)

xback, 2023-10-31:
@farfade 
I think you are referring to symbol `MBEDTLS_ECP_DP_SECP521R1_ENABLED` ? (which is currently disabled)
regarding [this curve](https://neuromancer.sk/std/secg/secp521r1).

Any chance of building a custom firmware enabling this one to test?

brada4, 2023-11-01:
> confirm that the RSA Public Key is at least 2048 bits or that one of the following ECC curves is used: P-256, P-384, or P-521

https://cabforum.org/wp-content/uploads/CA-Browser-Forum-BR-v2.0.1-redlined.pdf

Shine-, 2023-11-02:
@farfade I just compiled 23.05-SNAPSHOT with `MBEDTLS_ECP_DP_SECP521R1_ENABLED=y`, now your CA chain as attached in #13728 works.

farfade, 2023-11-03:
Thanks a lot @Shine- !

@hauke, @xback, @brada4 : I see two options now : 
- compile mbedtls with _MBEDTLS_ECP_DP_SECP521R1_ENABLED=y_, but it goes against your willingness of image size control (?)
- let it as it is, but document a little more how to switch to openssl with the custom image builder. It should also probably help to workaround a lot of other limitations, and it seems to me that a lot of cheap devices do not really worry about the image size nowadays. But on the other hand, is it clear that openwrt will always support openssl alternative ?

closed 2023-11-06 by openwrt-bot

brada4, 2023-11-06:
@farfade i just backed you with industry standard, welcome.

farfade, 2023-11-08:
Thanks a lot. And bravo, openwrt just rocks !

commit 3c17cdbc (2023-11-06): mbedtls: Activate secp521r1 curve by default

Activate the secp521r1 ecliptic curve by default. This curve is allowed
by the CA/Browser forum, see
https://cabforum.org/wp-content/uploads/CA-Browser-Forum-BR-v2.0.1-redlined.pdf#page=110

This increases the size of libmbedtls12_2.28.5-1_aarch64_generic.ipk by
about 400 bytes:
Without:
252,696 libmbedtls12_2.28.5-1_aarch64_generic.ipk
With:
253,088 libmbedtls12_2.28.5-2_aarch64_generic.ipk

Fixes: #13774
Acked-by: Koen Vandeputte <koen.vandeputte@citymesh.com>
PR #13882 (2023-11-06): mbedtls: Activate secp521r1 curve by default
Activate the secp521r1 ecliptic curve by default. This curve is allowed by the CA/Browser forum, see
https://cabforum.org/wp-content/uploads/CA-Browser-Forum-BR-v2.0.1-redlined.pdf#page=110

This increases the size of libmbedtls12_2.28.5-1_aarch64_generic.ipk by about 400 bytes:
Without:
252,696 libmbedtls12_2.28.5-1_aarch64_generic.ipk
With:
253,088 libmbedtls12_2.28.5-1_aarch64_generic.ipk

Fixes: #13774
commit 3223f31f (2023-11-06): mbedtls: Activate secp521r1 curve by default

Activate the secp521r1 ecliptic curve by default. This curve is allowed
by the CA/Browser forum, see
https://cabforum.org/wp-content/uploads/CA-Browser-Forum-BR-v2.0.1-redlined.pdf#page=110

This increases the size of libmbedtls12_2.28.5-1_aarch64_generic.ipk by
about 400 bytes:
Without:
252,696 libmbedtls12_2.28.5-1_aarch64_generic.ipk
With:
253,088 libmbedtls12_2.28.5-2_aarch64_generic.ipk

Fixes: #13774
Acked-by: Koen Vandeputte <koen.vandeputte@citymesh.com>

(cherry picked from commit 3c17cdbc369d89ff6a7911c3acff2e493778f6c1)