openwrt/openwrt #13777: adding IPv6 routing entry to table 52 but went to table main.

awsong, 2023-10-24:
### Describe the bug

I'm trying to add an entry to IPv6 routing table 52, but the entry went to table main. 

When I try to delete the entry, it seems it IS in table 52, which is weird. 

### OpenWrt version

r23497-6637af95aa

### OpenWrt target/subtarget

x86/64

### Device

QEMU Standard PC (Q35 + ICH9, 2009)

### Image kind

Official downloaded image

### Steps to reproduce

```bash
root@OpenWrt:~# ip -6 r s table 52

root@OpenWrt:~# ip -6 r s table main
default from 2408:8206:243f:f03a::/64 via fe80::8e90:d3ff:fe72:8311 dev pppoe-wan  metric 512 
default from 2408:8207:2433:a450::/60 via fe80::8e90:d3ff:fe72:8311 dev pppoe-wan  metric 512 
unreachable 2408:8206:243f:f03a::/64 dev lo  metric 2147483647 
2408:8207:2433:a450::/64 dev br-lan  metric 1024 
unreachable 2408:8207:2433:a450::/60 dev lo  metric 2147483647 
fd9e:1df:e0ad::/64 dev br-lan  metric 1024 
unreachable fd9e:1df:e0ad::/48 dev lo  metric 2147483647 
fe80::8e90:d3ff:fe72:8311 dev pppoe-wan  metric 256 
fe80::c86d:84e8:aeff:112 dev pppoe-wan  metric 256 
fe80::/64 dev eth1  metric 256 
fe80::/64 dev br-lan  metric 256 
anycast 2408:8206:243f:f03a:: dev pppoe-wan  metric 0 
anycast 2408:8207:2433:a450:: dev br-lan  metric 0 
anycast fd9e:1df:e0ad:: dev br-lan  metric 0 
anycast fe80:: dev eth1  metric 0 
anycast fe80:: dev br-lan  metric 0 
multicast ff00::/8 dev br-lan  metric 256 
multicast ff00::/8 dev eth1  metric 256 
multicast ff00::/8 dev link  metric 256 
multicast ff00::/8 dev pppoe-wan  metric 256

root@OpenWrt:~# ip -6 r a 2001:4860:4860::8888 dev pppoe-wan table 52

root@OpenWrt:~# ip -6 r s table 52

root@OpenWrt:~# ip -6 r s table main
2001:4860:4860::8888 dev pppoe-wan  metric 1024 ⬅⬅⬅⬅⬅⬅⬅
default from 2408:8206:243f:f03a::/64 via fe80::8e90:d3ff:fe72:8311 dev pppoe-wan  metric 512 
default from 2408:8207:2433:a450::/60 via fe80::8e90:d3ff:fe72:8311 dev pppoe-wan  metric 512 
unreachable 2408:8206:243f:f03a::/64 dev lo  metric 2147483647 
2408:8207:2433:a450::/64 dev br-lan  metric 1024 
unreachable 2408:8207:2433:a450::/60 dev lo  metric 2147483647 
fd9e:1df:e0ad::/64 dev br-lan  metric 1024 
unreachable fd9e:1df:e0ad::/48 dev lo  metric 2147483647 
fe80::8e90:d3ff:fe72:8311 dev pppoe-wan  metric 256 
fe80::c86d:84e8:aeff:112 dev pppoe-wan  metric 256 
fe80::/64 dev eth1  metric 256 
fe80::/64 dev br-lan  metric 256 
anycast 2408:8206:243f:f03a:: dev pppoe-wan  metric 0 
anycast 2408:8207:2433:a450:: dev br-lan  metric 0 
anycast fd9e:1df:e0ad:: dev br-lan  metric 0 
anycast fe80:: dev eth1  metric 0 
anycast fe80:: dev br-lan  metric 0 
multicast ff00::/8 dev br-lan  metric 256 
multicast ff00::/8 dev eth1  metric 256 
multicast ff00::/8 dev link  metric 256 
multicast ff00::/8 dev pppoe-wan  metric 256 

root@OpenWrt:~# ip -6 r s table 0
2001:4860:4860::8888 dev pppoe-wan table 52  metric 1024 ⬅⬅⬅⬅⬅⬅⬅
default from 2408:8206:243f:f03a::/64 via fe80::8e90:d3ff:fe72:8311 dev pppoe-wan  metric 512 
default from 2408:8207:2433:a450::/60 via fe80::8e90:d3ff:fe72:8311 dev pppoe-wan  metric 512 
unreachable 2408:8206:243f:f03a::/64 dev lo  metric 2147483647 
2408:8207:2433:a450::/64 dev br-lan  metric 1024 
unreachable 2408:8207:2433:a450::/60 dev lo  metric 2147483647 
fd9e:1df:e0ad::/64 dev br-lan  metric 1024 
unreachable fd9e:1df:e0ad::/48 dev lo  metric 2147483647 
fe80::8e90:d3ff:fe72:8311 dev pppoe-wan  metric 256 
fe80::c86d:84e8:aeff:112 dev pppoe-wan  metric 256 
fe80::/64 dev eth1  metric 256 
fe80::/64 dev br-lan  metric 256 
local ::1 dev lo table local  metric 0 
anycast 2408:8206:243f:f03a:: dev pppoe-wan table local  metric 0 
local 2408:8206:243f:f03a:c86d:84e8:aeff:112 dev pppoe-wan table local  metric 0 
anycast 2408:8207:2433:a450:: dev br-lan table local  metric 0 
local 2408:8207:2433:a450::1 dev br-lan table local  metric 0 
anycast fd9e:1df:e0ad:: dev br-lan table local  metric 0 
local fd9e:1df:e0ad::1 dev br-lan table local  metric 0 
anycast fe80:: dev eth1 table local  metric 0 
anycast fe80:: dev br-lan table local  metric 0 
local fe80::3086:45ff:fed8:6f83 dev eth1 table local  metric 0 
local fe80::3408:b0ff:fe17:3ed8 dev br-lan table local  metric 0 
local fe80::c86d:84e8:aeff:112 dev pppoe-wan table local  metric 0 
multicast ff00::/8 dev br-lan table local  metric 256 
multicast ff00::/8 dev eth1 table local  metric 256 
multicast ff00::/8 dev link table local  metric 256 
multicast ff00::/8 dev pppoe-wan table local  metric 256 

root@OpenWrt:~# ip -6 r d 2001:4860:4860::8888 table main
ip: RTNETLINK answers: No such process
root@OpenWrt:~# ip -6 r d 2001:4860:4860::8888 table 52
root@OpenWrt:~# 
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

JuliaZSong, 2023-10-24:
I have another instance of 21.02 which doesn't have this problem.

brada4, 2023-10-25:
Is it ip command from iproute2 or from busybox?

awsong, 2023-10-25:
It's busybox. I installed ip-full, and it works fine. Is it a known issue of busybox ip command?

brada4, 2023-10-25:
Ever since you made it known... yes.
ip-tiny is sufficient to fix your issue.

So complete test, edit in original post if you like:
```
unzme -r
busybox ip --version
ip --version
add route with first
list with both
```
busybox ip is quite limited, one needs to check result of each non-mainstream command like you did.