openwrt/openwrt #13784: 23.05.0 Bricks Netgear R8000 by Rendering Wired Network Inoperative

gordan-bobic, 2023-10-25:
### Describe the bug

At least some variants of R8000 (not R8000P) will not work with current release of 23.05.0. Wired networking ceases to work. Console output shows that networking is up, switch module reports link status when other devices get connected, but it cannot connect out and nothing can connect to the router's IP address. Network comes up very briefly and can be pinged when the bootloader loads, until it starts loading the kernel, and it never returns after that.

Flashing back to 22.03.5 fixes the problem.

Unfortunately, getting a replacement image to flash is rather difficult with no network and no usb-storage driver in the base image, so perhaps the generic usb storage driver should be placed into the base image for the R8000. With a bit of additional effort I was able to paste the working image to file via serial console and flash from that, but this is quite a complication.

### OpenWrt version

23.05.0

### OpenWrt target/subtarget

bcm53xx

### Device

R8000

### Image kind

Official downloaded image

### Steps to reproduce

Reset to factory settings.
Flash R8000 23.05.0 image.

### Actual behaviour

No wired network, router never responds to anything sent to 192.168.1.1

### Expected behaviour

Router should be accessible via 192.168.1.1

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-10-25:
What command / luci options you used to install image?

gordan-bobic, 2023-10-25:
It was on 22.03.5 to begin with. I used the luci web interface "flash firmware" option to upload 23.05.0 for flashing. After that, the wired networking never came back. Once I got a serial console on it, I tried clearing all configuration with firstboot and clearing jffs2reset and rebooting many times, but nothing brought the wired network back. I can see from ifconfig that the interface is set to 192.168.1.1 but I couldn't ping it over the lan and it couldn't ping out to any machine on the physical network on that lan. Plugging and unplugging the network cables results in a kernel message that link is detected or goes away, but IP itself never comes up in a working state on 23.05.0.

Once I flashed 22.03.5 back using `sysupgrade -n /tmp/image.chk`, it rebooted and immediately worked again. So I am moderately confident that there is something specific in 23.05.0 that breaks it.
I know there are at least two different variants of the R8000 (not including R8000P), this is the one that requires 19V power input, the older version requires 12V. I don't know if this affects only the 19V variant.

brada4, 2023-10-25:
Firstboot works best in safe mode ie pressing button during boot cycle when LED blinks fastest. Seems -n or respective checkbox applied to 23.05.0 upload did not work out and you got broken wired configuration?
But it seems there were some changes (fixes according to changelog) in DSA switch. But those should have kept wifi accessible over upgrade without -n.

gordan-bobic, 2023-10-25:
When I originally flashed 22.05.0, wifi did come up because I ticked the "preserved settings" option. It only went away after I long-pressed the reset button to reset it all back to defaults. At that point, the only way to get it back became serial console. Could some wired configuration have survived that?

brada4, 2023-10-25:
Fairly obvious -n to sysupgrade did not do its part at all since old conf returns few config wipes later

gordan-bobic, 2023-10-25:
So it might have fared better if I DIDN'T preserve config on the original upgrade to 23.05.0 ?
OK, I can try that and report back.

raenye, 2023-10-25:
1) Duplicate of #13548;
2) This is documented in the [release notes](https://openwrt.org/releases/23.05/notes-23.05.0#known_issues);
3) Fixed by a912ee74d6ca08020933dcdb9ce791e74244c25b.

raenye, 2023-11-16:
This should be fixed in just released 23.05.2. Please test.

hauke, 2023-11-26:
Fixed in a912ee74d6ca08020933dcdb9ce791e74244c25b and backported in 7bd682bc58fbfab673290e9d150786946f0788fc.

closed 2023-11-26 by hauke