openwrt/openwrt #13785: 'ip route get' returns no route but can still ping through

awsong, 2023-10-25:
### Describe the bug

```bash
root@OpenWrt:~# ip route get 2408:871a:2100:2:0:ff:b09f:237
RTNETLINK answers: Network unreachable
```
But I can still ping this ip address. 

The real problem is that according to my routing table, it should hit the default route. 
here is my routing table
```bash
root@OpenWrt:~# ip -6 route show table 0
default from 2408:8206:2430:2455::/64 via fe80::8e90:d3ff:fe72:8311 dev pppoe-wan proto static metric 512 pref medium
default from 2408:8207:2432:88e0::/60 via fe80::8e90:d3ff:fe72:8311 dev pppoe-wan proto static metric 512 pref medium
unreachable 2408:8206:2430:2455::/64 dev lo proto static metric 2147483647 pref medium
2408:8207:2432:88e0::/64 dev br-lan proto static metric 1024 pref medium
unreachable 2408:8207:2432:88e0::/60 dev lo proto static metric 2147483647 pref medium
fd9e:1df:e0ad::/64 dev br-lan proto static metric 1024 pref medium
unreachable fd9e:1df:e0ad::/48 dev lo proto static metric 2147483647 pref medium
fe80::282f:ff8e:3a95:24a dev pppoe-wan proto kernel metric 256 pref medium
fe80::8e90:d3ff:fe72:8311 dev pppoe-wan proto kernel metric 256 pref medium
fe80::/64 dev eth1 proto kernel metric 256 pref medium
fe80::/64 dev tailscale0 proto kernel metric 256 pref medium
fe80::/64 dev br-lan proto kernel metric 256 pref medium
local ::1 dev lo table local proto kernel metric 0 pref medium
anycast 2408:8206:2430:2455:: dev pppoe-wan table local proto kernel metric 0 pref medium
local 2408:8206:2430:2455:282f:ff8e:3a95:24a dev pppoe-wan table local proto kernel metric 0 pref medium
anycast 2408:8207:2432:88e0:: dev br-lan table local proto kernel metric 0 pref medium
local 2408:8207:2432:88e0::1 dev br-lan table local proto kernel metric 0 pref medium
anycast fd9e:1df:e0ad:: dev br-lan table local proto kernel metric 0 pref medium
local fd9e:1df:e0ad::1 dev br-lan table local proto kernel metric 0 pref medium
anycast fe80:: dev tailscale0 table local proto kernel metric 0 pref medium
anycast fe80:: dev eth1 table local proto kernel metric 0 pref medium
anycast fe80:: dev br-lan table local proto kernel metric 0 pref medium
local fe80::15ef:db86:e56e:bfad dev tailscale0 table local proto kernel metric 0 pref medium
local fe80::282f:ff8e:3a95:24a dev pppoe-wan table local proto kernel metric 0 pref medium
local fe80::3086:45ff:fed8:6f83 dev eth1 table local proto kernel metric 0 pref medium
local fe80::3408:b0ff:fe17:3ed8 dev br-lan table local proto kernel metric 0 pref medium
multicast ff00::/8 dev br-lan table local proto kernel metric 256 pref medium
multicast ff00::/8 dev eth1 table local proto kernel metric 256 pref medium
multicast ff00::/8 dev link table local proto kernel metric 256 pref medium
multicast ff00::/8 dev tailscale0 table local proto kernel metric 256 pref medium
multicast ff00::/8 dev pppoe-wan table local proto kernel metric 256 pref medium
```

I'm using ip command from ip-full package.

### OpenWrt version

r23497-6637af95aa

### OpenWrt target/subtarget

x86/64

### Device

QEMU Standard PC (Q35 + ICH9, 2009)

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

danpawlik, 2023-10-25:
Did you check https://github.com/openwrt/openwrt/issues/13691 ?

awsong, 2023-10-25:
> Did you check https://github.com/openwrt/openwrt/issues/13691 ?

Doesn't look like the same issue.

In my case, IPv4 is fine, only IPv6 has this problem. And my network  actually works, just routing table has some strange behavior.

jow-, 2023-10-25:
OpenWrt uses source routing for IPv6, pass a "from" argument to "ip route get" to select the proper default route.

```
root@er-x:~# ip -6 r l e ::/0
default from 2001:xx:yy:7d4::/64 dev 6in4-wan6 proto static metric 1024 pref medium
default from 2001:xx:zz::/48 dev 6in4-wan6 proto static metric 1024 pref medium
default from fc00::/7 dev 6in4-wan6 proto static metric 1024 pref medium
root@er-x:~# ip route get 2408:871a:2100:2:0:ff:b09f:237
RTNETLINK answers: Network unreachable
root@er-x:~# ip route get 2408:871a:2100:2:0:ff:b09f:237 from 2001:xx:zz::1
2408:871a:2100:2:0:ff:b09f:237 from 2001:xx:zz::1 dev 6in4-wan6 proto static src 2001:xx:yy:7d4::2 metric 1024 pref medium
root@er-x:~# 
```

closed 2023-10-25 by jow-

axet, 2023-10-27:
I do not think solution is accurate. I have similar issue with ipv6 not get resolved. ip route get success with

ip route get 2001:470:1:c84::20 from  2001:470:28:187::1

but failed with

```
ip route get 2001:470:1:c84::20 from  fc00:10:10:5::1
ip: RTNETLINK answers: Network unreachable
```

I expect booth network can be reachable. Since 'ip route get 1' or 'ip route get 1::' should show default route for ipv4 and ipv6

jow-, 2023-11-01:
> I expect booth network can be reachable. Since 'ip route get 1' or 'ip route get 1::' should show default route for ipv4 and ipv6

You don't have a local `::/64` nor a local `fc00:10:10:5::/64` network so it is clear that `ip route get 2001:470:1:c84::20 from  fc00:10:10:5::1` and `ip route get 2001:470:1:c84::20 from  ::1` will not work. There will be no packets originating from LAN or OpenWrt itself having either a `0:0:0:0:*` or a `fc00:10:10:5:*` IP as source address set.

> Based on table above, which IP kernel going to take since I have two ipv6? Keyword 'from' not helping here.

See https://www.rfc-editor.org/rfc/rfc6724.html#section-5 for the source selection algorithm. If you end up with two or more identical candidates, the numerically lower one should be chosen by the kernel.

When OpenWrt itself initiates connections to the outside, it will chose the global IPv6 address bound to the egress interface (see `ip -6 addr show dev 6in4-wan6he to 2000::/7`) as source IP. When OpenWrt forwards a packet from a LAN host, it will chose the default route based on the `from` prefix matching the source IP of the forwarded packet (likely `2001:470:28:187:*`)

axet, 2023-11-01:
@jow can you explain why it works on desktop? It shoudn't work?

```
ip -6 route show
2001:470:28:187::b10 dev enp5s0 proto kernel metric 100 pref medium
2001:470:28:187::/64 dev enp5s0 proto ra metric 100 pref medium
fc00:10:10:5::b10 dev enp5s0 proto kernel metric 100 pref medium
fc00:10:10:5::/64 dev enp5s0 proto ra metric 100 pref medium
fe80::/64 dev enp5s0 proto kernel metric 1024 pref medium
default via fe80::12fe:edff:fe72:4e36 dev enp5s0 proto ra metric 100 pref medium
axet@axet-desktop:~/bin$ ip route get 2001:470:1:c84::20
2001:470:1:c84::20 from :: via fe80::12fe:edff:fe72:4e36 dev enp5s0 proto ra src 2001:470:28:187::b10 metric 100 pref medium
```

jow-, 2023-11-01:
Because your desktop uses only a single unscoped default route (note the missing `from ...` restriction).