openwrt/openwrt #13787: AP-STA-POSSIBLE-PSK-MISMATCH on OpenWRT 23.05

lukasz1992, 2023-10-25:
### Describe the bug

Hello,

I have an not-so-old Canon printer with Wi-Fi. It is not possible to setup it in any way other than WPS/PBC.
Router is WiFi6 with WPA3/WPA2 PSK encrypted network on mt7981 device.

To work with WPS, I have hostapd-openssl installed and with OpenWRT I am able to setup WiFi connection. Behaviour:
- WiFi works rock-solid on Printer no matter if I reboot printer, disconnect and connect power supply
- After rebooting OpenWRT or just restarting network (/etc/init.d/network restart), printer is not able anymore to connect to WiFi. I observe following messages on logread:

```
Wed Oct 25 17:09:39 2023 daemon.info hostapd: phy0-ap0: STA 74:38:b7:xx:xx:xx IEEE 802.11: authenticated
Wed Oct 25 17:09:39 2023 daemon.info hostapd: phy0-ap0: STA 74:38:b7:xx:xx:xx IEEE 802.11: associated (aid 2)
Wed Oct 25 17:09:39 2023 daemon.notice hostapd: phy0-ap0: AP-STA-POSSIBLE-PSK-MISMATCH 74:38:b7:xx:xx:xx
Wed Oct 25 17:09:40 2023 daemon.notice hostapd: phy0-ap0: AP-STA-POSSIBLE-PSK-MISMATCH 74:38:b7:xx:xx:xx
Wed Oct 25 17:09:41 2023 daemon.notice hostapd: phy0-ap0: AP-STA-POSSIBLE-PSK-MISMATCH 74:38:b7:xx:xx:xx
Wed Oct 25 17:09:42 2023 daemon.notice hostapd: phy0-ap0: AP-STA-POSSIBLE-PSK-MISMATCH 74:38:b7:xx:xx:xx
```

Key is correct, pairing WPS again solved the problem
But do not want to do it every reboot.

Have anybody suffered from such issue?

### OpenWrt version

r23497-6637af95aa

### OpenWrt target/subtarget

mediatek/filogic

### Device

GL.iNet GL-MT3000

### Image kind

Official downloaded image

### Steps to reproduce

- buy TS3100 printer
- setup OpenWRT 23.05.0 release and install wpad-openssl
- configure WPS
- pair printer with WPS
- reboot router

### Actual behaviour

Printer can not connect anymore. Unfortunately it is not possible to gather logs from printer.

### Expected behaviour

Previous router just accepted printer

### Additional info

config wifi-device 'radio0'
        option type 'mac80211'
        option path 'platform/18000000.wifi'
        option greenfield '1'
        option channel '11'
        option band '2g'
        option htmode 'HE20'
        option country 'PL'
        option require_mode 'n'
        option txpower '20'
        option disabled '0'
        option cell_density '0'

config wifi-iface 'default_radio0'
        option device 'radio0'
        option network 'lan'
        option mode 'ap'
        option ssid 'ASUS_E8 2G'
        option encryption 'sae-mixed'
        option key '****'
        option wpa_disable_eapol_key_retries '1'
        option wps_pushbutton '1'
        option multicast_to_unicast_all '1'

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

lukasz1992, 2023-10-25:
Logs from WPS PBC configuration
```
Wed Oct 25 17:26:50 2023 daemon.info hostapd: phy0-ap0: STA 74:38:b7:xx:xx:xx IEEE 802.11: authenticated
Wed Oct 25 17:26:50 2023 daemon.info hostapd: phy0-ap0: STA 74:38:b7:xx:xx:xx IEEE 802.11: associated (aid 3)
Wed Oct 25 17:26:50 2023 daemon.notice hostapd: phy0-ap0: AP-STA-POSSIBLE-PSK-MISMATCH 74:38:b7:xx:xx:xx
Wed Oct 25 17:26:51 2023 daemon.notice hostapd: phy0-ap0: AP-STA-POSSIBLE-PSK-MISMATCH 74:38:b7:xx:xx:xx
Wed Oct 25 17:26:52 2023 daemon.notice hostapd: phy0-ap0: AP-STA-POSSIBLE-PSK-MISMATCH 74:38:b7:xx:xx:xx
Wed Oct 25 17:26:53 2023 daemon.notice hostapd: phy0-ap0: AP-STA-POSSIBLE-PSK-MISMATCH 74:38:b7:xx:xx:xx
Wed Oct 25 17:26:59 2023 daemon.info hostapd: phy0-ap0: STA 74:38:b7:xx:xx:xx IEEE 802.11: deauthenticated due to local deauth request
<CONFIGURE WPS>
Wed Oct 25 17:27:06 2023 daemon.info hostapd: phy0-ap0: STA 74:38:b7:xx:xx:xx IEEE 802.11: authenticated
Wed Oct 25 17:27:06 2023 daemon.info hostapd: phy0-ap0: STA 74:38:b7:xx:xx:xx IEEE 802.11: associated (aid 3)
Wed Oct 25 17:27:06 2023 daemon.notice hostapd: phy0-ap0: CTRL-EVENT-EAP-STARTED 74:38:b7:xx:xx:xx
Wed Oct 25 17:27:06 2023 daemon.notice hostapd: phy0-ap0: CTRL-EVENT-EAP-PROPOSED-METHOD vendor=0 method=1
Wed Oct 25 17:27:06 2023 daemon.notice hostapd: phy0-ap0: CTRL-EVENT-EAP-STARTED 74:38:b7:xx:xx:xx
Wed Oct 25 17:27:06 2023 daemon.notice hostapd: phy0-ap0: CTRL-EVENT-EAP-PROPOSED-METHOD vendor=0 method=1
Wed Oct 25 17:27:06 2023 daemon.notice hostapd: phy0-ap0: CTRL-EVENT-EAP-PROPOSED-METHOD vendor=14122 method=254
Wed Oct 25 17:27:06 2023 daemon.notice hostapd: phy0-ap0: WPS-REG-SUCCESS 74:38:b7:xx:xx:xx 00000000-0000-1000-8000-7438b7xxxxxx
Wed Oct 25 17:27:06 2023 daemon.notice hostapd: phy0-ap0: WPS-PBC-DISABLE
Wed Oct 25 17:27:06 2023 daemon.notice hostapd: phy0-ap0: WPS-SUCCESS
Wed Oct 25 17:27:06 2023 daemon.notice hostapd: phy0-ap0: CTRL-EVENT-EAP-FAILURE 74:38:b7:xx:xx:xx
Wed Oct 25 17:27:06 2023 daemon.warn hostapd: phy0-ap0: STA 74:38:b7:xx:xx:xx IEEE 802.1X: authentication failed - EAP type: 0 (unknown)
Wed Oct 25 17:27:06 2023 daemon.info hostapd: phy0-ap0: STA 74:38:b7:xx:xx:xx IEEE 802.1X: Supplicant used different EAP type: 254 (expanded)
Wed Oct 25 17:27:11 2023 daemon.info hostapd: phy0-ap0: STA 74:38:b7:xx:xx:xx IEEE 802.11: authenticated
Wed Oct 25 17:27:11 2023 daemon.info hostapd: phy0-ap0: STA 74:38:b7:xx:xx:xx IEEE 802.11: associated (aid 3)
Wed Oct 25 17:27:11 2023 daemon.notice hostapd: phy0-ap0: AP-STA-CONNECTED 74:38:b7:xx:xx:xx auth_alg=open
Wed Oct 25 17:27:11 2023 daemon.info hostapd: phy0-ap0: STA 74:38:b7:xx:xx:xx RADIUS: starting accounting session 9FCF21CA0A2EEB7E
Wed Oct 25 17:27:11 2023 daemon.info hostapd: phy0-ap0: STA 74:38:b7:xx:xx:xx WPA: pairwise key handshake completed (RSN)
Wed Oct 25 17:27:11 2023 daemon.notice hostapd: phy0-ap0: EAPOL-4WAY-HS-COMPLETED 74:38:b7:xx:xx:xx
Wed Oct 25 17:27:11 2023 daemon.info dnsmasq-dhcp[1]: DHCPREQUEST(br-lan) 192.168.1.230 74:38:b7:xx:xx:xx
Wed Oct 25 17:27:11 2023 daemon.info dnsmasq-dhcp[1]: DHCPACK(br-lan) 192.168.1.230 74:38:b7:xx:xx:xx CANON
<TURN OFF PRINTER>
Wed Oct 25 17:27:18 2023 daemon.info dnsmasq-dhcp[1]: DHCPRELEASE(br-lan) 192.168.1.230 74:38:b7:xx:xx:xx
Wed Oct 25 17:27:19 2023 daemon.notice hostapd: phy0-ap0: AP-STA-DISCONNECTED 74:38:b7:xx:xx:xx
```

brada4, 2023-10-25:
could you add logs from good connection (logread -e '00:11:22:xx:yy:66' ? 
Hevent seen 'greenfield' used in a while. Also clean wpa2 may help.

lukasz1992, 2023-10-26:
Yes, you are right, WPS is supported for WPA2 devices.
I deleted greenfield, reconfigured network to psk2 and increased debug. I have these logs:

WPS PBC:
```
PAIRING

Thu Oct 26 12:30:48 2023 daemon.debug hostapd: phy0-ap0: STA 74:38:b7:xx:xx:xx IEEE 802.11: authentication OK (open system)
Thu Oct 26 12:30:48 2023 daemon.debug hostapd: phy0-ap0: STA 74:38:b7:xx:xx:xx MLME: MLME-AUTHENTICATE.indication(74:38:b7:xx:xx:xx, OPEN_SYSTEM)
Thu Oct 26 12:30:48 2023 daemon.debug hostapd: phy0-ap0: STA 74:38:b7:xx:xx:xx MLME: MLME-DELETEKEYS.request(74:38:b7:xx:xx:xx)
Thu Oct 26 12:30:48 2023 daemon.info hostapd: phy0-ap0: STA 74:38:b7:xx:xx:xx IEEE 802.11: authenticated
Thu Oct 26 12:30:48 2023 daemon.debug hostapd: phy0-ap0: STA 74:38:b7:xx:xx:xx IEEE 802.11: association OK (aid 2)
Thu Oct 26 12:30:48 2023 daemon.info hostapd: phy0-ap0: STA 74:38:b7:xx:xx:xx IEEE 802.11: associated (aid 2)
Thu Oct 26 12:30:48 2023 daemon.debug hostapd: phy0-ap0: STA 74:38:b7:xx:xx:xx MLME: MLME-ASSOCIATE.indication(74:38:b7:xx:xx:xx)
Thu Oct 26 12:30:48 2023 daemon.debug hostapd: phy0-ap0: STA 74:38:b7:xx:xx:xx MLME: MLME-DELETEKEYS.request(74:38:b7:xx:xx:xx)
Thu Oct 26 12:30:48 2023 daemon.debug hostapd: phy0-ap0: STA 74:38:b7:xx:xx:xx IEEE 802.11: binding station to interface 'phy0-ap0'
Thu Oct 26 12:30:48 2023 daemon.debug hostapd: phy0-ap0: STA 74:38:b7:xx:xx:xx IEEE 802.1X: start authentication
Thu Oct 26 12:30:48 2023 daemon.debug hostapd: phy0-ap0: STA 74:38:b7:xx:xx:xx IEEE 802.1X: unauthorizing port
Thu Oct 26 12:30:48 2023 daemon.notice hostapd: phy0-ap0: CTRL-EVENT-EAP-STARTED 74:38:b7:xx:xx:xx
Thu Oct 26 12:30:48 2023 daemon.notice hostapd: phy0-ap0: CTRL-EVENT-EAP-PROPOSED-METHOD vendor=0 method=1
Thu Oct 26 12:30:48 2023 daemon.debug hostapd: phy0-ap0: STA 74:38:b7:xx:xx:xx IEEE 802.1X: Sending EAP Packet (identifier 1)
Thu Oct 26 12:30:48 2023 daemon.debug hostapd: phy0-ap0: STA 74:38:b7:xx:xx:xx IEEE 802.1X: received EAPOL-Start from STA
Thu Oct 26 12:30:48 2023 daemon.debug hostapd: phy0-ap0: STA 74:38:b7:xx:xx:xx IEEE 802.1X: aborting authentication
Thu Oct 26 12:30:48 2023 daemon.notice hostapd: phy0-ap0: CTRL-EVENT-EAP-STARTED 74:38:b7:xx:xx:xx
Thu Oct 26 12:30:48 2023 daemon.notice hostapd: phy0-ap0: CTRL-EVENT-EAP-PROPOSED-METHOD vendor=0 method=1
Thu Oct 26 12:30:48 2023 daemon.debug hostapd: phy0-ap0: STA 74:38:b7:xx:xx:xx IEEE 802.1X: Sending EAP Packet (identifier 22)
Thu Oct 26 12:30:48 2023 daemon.debug hostapd: phy0-ap0: STA 74:38:b7:xx:xx:xx IEEE 802.1X: received EAP packet (code=2 id=1 len=34) from STA: EAP Response-Identity (1)
Thu Oct 26 12:30:48 2023 daemon.debug hostapd: phy0-ap0: STA 74:38:b7:xx:xx:xx IEEE 802.1X: received EAP packet (code=2 id=22 len=34) from STA: EAP Response-Identity (1)
Thu Oct 26 12:30:48 2023 daemon.notice hostapd: phy0-ap0: CTRL-EVENT-EAP-PROPOSED-METHOD vendor=14122 method=254
Thu Oct 26 12:30:48 2023 daemon.debug hostapd: phy0-ap0: STA 74:38:b7:xx:xx:xx IEEE 802.1X: Sending EAP Packet (identifier 23)
Thu Oct 26 12:30:48 2023 daemon.debug hostapd: phy0-ap0: STA 74:38:b7:xx:xx:xx IEEE 802.1X: received EAP packet (code=2 id=23 len=482) from STA: EAP Response-expanded (254)
Thu Oct 26 12:30:48 2023 daemon.debug hostapd: phy0-ap0: STA 74:38:b7:xx:xx:xx IEEE 802.1X: Sending EAP Packet (identifier 24)
Thu Oct 26 12:30:48 2023 daemon.debug hostapd: phy0-ap0: STA 74:38:b7:xx:xx:xx IEEE 802.1X: received EAP packet (code=2 id=24 len=138) from STA: EAP Response-expanded (254)
Thu Oct 26 12:30:48 2023 daemon.debug hostapd: phy0-ap0: STA 74:38:b7:xx:xx:xx IEEE 802.1X: Sending EAP Packet (identifier 25)
Thu Oct 26 12:30:48 2023 daemon.debug hostapd: phy0-ap0: STA 74:38:b7:xx:xx:xx IEEE 802.1X: received EAP packet (code=2 id=25 len=134) from STA: EAP Response-expanded (254)
Thu Oct 26 12:30:48 2023 daemon.debug hostapd: phy0-ap0: STA 74:38:b7:xx:xx:xx IEEE 802.1X: Sending EAP Packet (identifier 26)
Thu Oct 26 12:30:48 2023 daemon.debug hostapd: phy0-ap0: STA 74:38:b7:xx:xx:xx IEEE 802.1X: received EAP packet (code=2 id=26 len=134) from STA: EAP Response-expanded (254)
Thu Oct 26 12:30:48 2023 daemon.debug hostapd: phy0-ap0: STA 74:38:b7:xx:xx:xx IEEE 802.1X: Sending EAP Packet (identifier 27)
Thu Oct 26 12:30:48 2023 daemon.debug hostapd: phy0-ap0: STA 74:38:b7:xx:xx:xx IEEE 802.1X: received EAP packet (code=2 id=27 len=74) from STA: EAP Response-expanded (254)
Thu Oct 26 12:30:48 2023 daemon.notice hostapd: phy0-ap0: WPS-REG-SUCCESS 74:38:b7:xx:xx:xx 00000000-0000-1000-8000-7438b7xxxxxx
Thu Oct 26 12:30:48 2023 daemon.notice hostapd: phy0-ap0: WPS-PBC-DISABLE
Thu Oct 26 12:30:48 2023 daemon.notice hostapd: phy0-ap0: WPS-SUCCESS
Thu Oct 26 12:30:48 2023 daemon.notice hostapd: phy0-ap0: CTRL-EVENT-EAP-FAILURE 74:38:b7:xx:xx:xx
Thu Oct 26 12:30:48 2023 daemon.debug hostapd: phy0-ap0: STA 74:38:b7:xx:xx:xx IEEE 802.1X: Sending EAP Packet (identifier 27)
Thu Oct 26 12:30:48 2023 daemon.debug hostapd: phy0-ap0: STA 74:38:b7:xx:xx:xx IEEE 802.1X: unauthorizing port
Thu Oct 26 12:30:48 2023 daemon.warn hostapd: phy0-ap0: STA 74:38:b7:xx:xx:xx IEEE 802.1X: authentication failed - EAP type: 0 (unknown)
Thu Oct 26 12:30:48 2023 daemon.info hostapd: phy0-ap0: STA 74:38:b7:xx:xx:xx IEEE 802.1X: Supplicant used different EAP type: 254 (expanded)
Thu Oct 26 12:30:48 2023 daemon.debug hostapd: phy0-ap0: STA 74:38:b7:xx:xx:xx IEEE 802.11: deauthenticated
Thu Oct 26 12:30:48 2023 daemon.debug hostapd: phy0-ap0: STA 74:38:b7:xx:xx:xx MLME: MLME-DEAUTHENTICATE.indication(74:38:b7:xx:xx:xx, 3)
[... cut ...]
Thu Oct 26 12:29:21 2023 daemon.debug hostapd: phy0-ap0: STA 74:38:b7:xx:xx:xx WPA: sending 1/4 msg of 4-Way Handshake
Thu Oct 26 12:29:21 2023 daemon.debug hostapd: phy0-ap0: STA 74:38:b7:xx:xx:xx WPA: received EAPOL-Key frame (2/4 Pairwise)
Thu Oct 26 12:29:21 2023 daemon.debug hostapd: phy0-ap0: STA 74:38:b7:xx:xx:xx WPA: invalid MIC in msg 2/4 of 4-Way Handshake
Thu Oct 26 12:29:21 2023 daemon.notice hostapd: phy0-ap0: AP-STA-POSSIBLE-PSK-MISMATCH 74:38:b7:xx:xx:xx
Thu Oct 26 12:29:22 2023 daemon.debug hostapd: phy0-ap0: STA 74:38:b7:xx:xx:xx WPA: EAPOL-Key timeout
```

brada4, 2023-10-26:
>  WPA: EAPOL-Key timeout

Remove eapol_retries line, it opens a proximal vulnerability back to default but allows longer range for initial connection.

lukasz1992, 2023-10-26:
Did not help - the same.

Printer is next to router, so link quality is reported as 100%.

brada4, 2023-10-26:
https://github.com/openwrt/openwrt/issues/9369

MIC is checksum that printer added on the air message.
Reducing/switching crypto options may trick printer into using correct algo. Like WPA2+CCMP or WPA1+TKIP only, obviously you need to partition printer link out of main AP.

lukasz1992, 2023-10-26:
Thank you for this point. Unfortunately bug is reproducible with WPA2+CCMP only (no 5GHz iface).

On previous Asuswrt router it was possible to reboot both devices and both worked.

brada4, 2023-10-26:
Something in beacon confuses printer to add wrong checksum to 2nd packet. 
Here disable negotiation-confusion more:
https://ij.manual.canon/ij/webmanual/NetworkTool/M/4.4/EN/NWT/nwt_settings0500.html
ie set AP and STA to WPA2-only CCMP-only, whis will kick out pre-2010 clients.

lukasz1992, 2023-10-27:
Thank you @brada4. Looks like firmware do not recognize WPA2/WPA3 networks - listing them as Open.
Anyway switching to WPA2, makes it possible configure printer with this tool. I switched than router to WPA2/WPA3 again.

Then after reboot of all devices, printer was able to connect to printer.
So to sum up:
- Even printer does not recognize encryption correctly it is still able to connect to my AP
- WPS connection is somehow broken in OpenWRT, as I was not able to configure printer for WPA2AES only network (after reboot is stops working)

brada4, 2023-10-27:
Glad to hear, if under warranty you may try to convince vendor to fix issue or at least default to wpa2+ccmp. Ot should be safe to assume there are almost no pre-2004 routers in operation (thats g without n)

lukasz1992, 2023-10-27:
No - other router is able to provide config with WPS to the printer (it persist after reboot).

This is OpenWrt-specific problem

brada4, 2023-10-27:
Wrong MIC (checksum) is sent by the printer, likely it is WPA-1 MD5 kind taking place of WPA2 SHA1 just based on printers own configuration ignoring airwaves need the modern one only.

Nice extract: 
https://security.stackexchange.com/questions/244316/how-is-the-mic-message-integrity-code-generated-in-wpa2

Probably you can dump defective wifi frame from the printer and see 4 zeroes as  MIC field is being padded after 128bit md5 to 160bit sha1, with old access point's old hostapd accepting either way.

lukasz1992, 2023-10-30:
@raenye 
In that issue I am complaining that WPS is not able to provide correct PSK.

The problem is reproducible with psk2-only network.

Problem that Printer sees sae-mixed network as open is not relevant - as Canon tool finally manages to succeed (even on sae-mixed network) printer is able to connect.

brada4, 2023-10-30:
Now you say WPS fails? Initially you said its actually reconnection later?

Can you get the management packet the printer sends out?

Or try rising debug level of access point (1-log each management message 0-some raw packets also)
https://openwrt.org/docs/guide-developer/debugging#logging_hostapd_behaviour

lukasz1992, 2023-10-30:
Setting printer via WPS on old router (AsusWRT - worked)
Setting printer via WPS to PSK2/PSK3 mixed on OpenWRT - works until router reboot
Setting printer via WPS to PSK2-only (WiFi 4) on OpenWRT - works until router reboot
Setting printer via Canon util works - even if device does not recognize sae-mixed network during scan, it can connect and setting is persistent

I will close the ticket, because do not want to spend more time on testing.

closed 2023-10-30 by lukasz1992

brada4, 2023-10-30:
I suspect printer falls back to weakest protocol in its own configuration after mis-detecting network as open.