openwrt/openwrt #13788: LLDP & gmac1 & DSA & mt7621

Npeca75, 2023-10-25:
### Describe the bug

if the WAN port of mt7621 device is connected to gmac1
@arinc9 [patch](https://github.com/openwrt/openwrt/pull/10238) then LLDP does not function properly

simptoms are, that OpenWRT "see" managed switch from other side, but managed switch does not "see" LLDP info comming from OpenWRT. Tested on Eltex & Jetsream managed switches

if i change DTS/DTSI (revert patch) and put back WAN port to switch0 (instead of gmac1) everything start to work as expected

my findings are here
https://forum.openwrt.org/t/lldp-or-dsa-bug-on-mt7621/175442/8

### OpenWrt version

23.05 snapshot

### OpenWrt target/subtarget

ramips/mt7621

### Device

suppose any m7621 with wan on gmac1

### Image kind

Official downloaded image

### Steps to reproduce

fire up the LLDPD on WAN port

### Actual behaviour

managed switch does not "see" OpenWRT device

### Expected behaviour

managed switch need to register LLDP packets from OpenWRT device

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-10-25:
What are you trying to achieve? What you get instead? No random forum links please.
What options you see in /etc/config/lldpd?
Do you know anyone at the other end of wan caring about your lldp?

ptpt52, 2023-10-26:
ping @LGA1150

Npeca75, 2023-10-26:
> What are you trying to achieve? What you get instead? No random forum links please. What options you see in /etc/config/lldpd? Do you know anyone at the other end of wan caring about your lldp?

Hi
i want to use LLDPD :)

and yes, we are using LLDP in large scale, all managed switch are mapped and monitored with LibreNMS, snmp+lldpd+few custom stuff

```
cat /etc/lldpd.d/lldpd.conf 
configure lldp portidsubtype ifname
configure system interface pattern wan
configure system description 'OpenWRT_23.05'
configure system hostname 'wr103.wrt.XXXX'
unconfigure system interface description
configure ports wan lldp custom-tlv add oui 00,80,c2 subtype 1 oui-info '00,01'
configure system ip management pattern 169.254.1.103,fd00:1:255::103
```

and, NO, it is NOT random forum post
in this post/topic i write all my findings bout switch0/gmac1 WAN assigment

what i want to achieve i managed switch with dozen of vlans, dozen of DumpAP, all AP ports in vlan filtering bridge + per passphrase WIFI/VLAN

everything is working as expected, both side "see" each other until WAN port is in switch0 node
when WAN port is in gmac1 node, remote side does not see OWRT lldpd packets

again, please look at forum post
as this is OpenWRT bug, OpenWRT forum post are NOT random post 

thank you

brada4, 2023-10-26:
Could you do what I asked - /etc/config/lldpd is rendered into that conf  you show that is generated at every restart.
Forum thread has contradictory statements 
wan port is not in any switch now. you need to add it to /etc/config file via uci or text editor.

Npeca75, 2023-10-26:
>Could you do what I asked - /etc/config/lldpd
it does NOT exist. as i wrote in forum, i am using custom lldpd/snmpd init scripts. Why are you not satisfied with finished /etc/lldpd.d/lldpd.conf ???

>wan port is not in any switch now
o please don't do this ... 
as stated in forum, twice
```
config device
        option type 'bridge'
        option name 'switch'
        list ports 'lan1'
        list ports 'lan2'
        list ports 'lan3'
        list ports 'lan4'
        list ports 'wan'
 ```
 
 >you need to add it to /etc/config file via uci or text editor
 after so many post in owrt forum, helping others
 and now modifying dst/dtsi,compiling, hunting it down why lldp does not shown on other end, you will teach me that i need to use editor to modify files ?
 well, thank you. very constructive

brada4, 2023-10-26:
Olease unpack ipk and check this file is present:
https://github.com/openwrt/openwrt/blob/openwrt-23.05/package/network/services/lldpd/files/lldpd.config

Npeca75, 2023-10-26:
> Please download and unpack ipk and check this file is present: https://github.com/openwrt/openwrt/blob/openwrt-23.05/package/network/services/lldpd/files/lldpd.config Default 'interface' 'lan' needs to be complemented here or in conf.d with another interface called 'wan'

again, last time
i am NOT using UCI and NOT using standard /etc/init.d/lldpd
so there is NO need for /etc/config/lldpd

if you really want to help, please get one device with mt7621 where WAN port is assigned to GMAC1 node in DTS/DTSI and test the LLPD on WAN port
until then, .. thank you for your help

brada4, 2023-10-26:
You must run through all 'lldpctl show' commands to find which interfaces are active and add missing.

ptpt52, 2023-12-05:
@Bartel-C8 
none of these patches merged to latest openwrt code.

ptpt52, 2023-12-05:
[dsa-7530-patch-set.tar.gz](https://github.com/openwrt/openwrt/files/13561632/dsa-7530-patch-set.tar.gz)
list:
```
900-v6.1-0001-net-dsa-mt7530-fix-handling-of-LLDP-frames.patch
900-v6.1-0002-net-dsa-introduce-preferred_default_local_cpu_port-a.patch
900-v6.1-0003-net-dsa-mt7530-update-PCS-driver-to-use-neg_mode.patch
900-v6.1-0004-net-dsa-mt7530-fix-handling-of-802.1X-PAE-frames.patch
```
@Npeca75 @Bartel-C8 
you may try download these patch set and put them to `target/linux/generic/backport-5.15/`
then build the firmware and test it.

Npeca75, 2024-01-08:
Hi @ptpt52 
> @Npeca75 @Bartel-C8 you may try download these patch set and put them to `target/linux/generic/backport-5.15/` then build the firmware and test it.

sorry for late reply
well, i will try to setup lab environment for this test when i have a spare time
thank you

ptpt52, 2024-01-08:
@Npeca75 
update notes: you may drop `900-v6.1-0003-net-dsa-mt7530-update-PCS-driver-to-use-neg_mode.patch`
since it is not related and connt compile success.

Npeca75, 2024-01-08:
> @Npeca75 update notes: you may drop `900-v6.1-0003-net-dsa-mt7530-update-PCS-driver-to-use-neg_mode.patch` since it is not related and connt compile success.

Hi @ptpt52 

cloned fresh SNAPSHOT
copied patch files to 
target/linux/ramips/patches-5.15

```
cat /etc/board.json 
{
        "model": {
                "id": "cudy,wr1300-v2",
                "name": "Cudy WR1300 v2"
        },
```

```
uname -a
Linux wr103.wrt.ele 5.15.146 #0 SMP Mon Jan 8 13:08:06 2024 mips GNU/Linux
```

```
root@wr103.wrt.ele:~# cat /etc/lldpd.d/lldpd.conf 
configure lldp portidsubtype ifname
configure system interface pattern wan
```

```
root@wr103.wrt.ele:~# lldpctl 
-------------------------------------------------------------------------------
LLDP neighbors:
-------------------------------------------------------------------------------
Interface:    wan, via: LLDP, RID: 1, Time: 0 day, 00:25:52
  Chassis:     
    ChassisID:    mac c0:06:c3:a1:3f:46
    SysName:      sw12.netdev.ele
    SysDescr:     JetStream 8-Port Gigabit L2+ Managed Switch with 2 SFP Slots
    Capability:   Bridge, on
    Capability:   Router, on
  Port:        
    PortID:       ifname gigabitEthernet 1/0/7
    PortDescr:    gigabitEthernet 1/0/7
    TTL:          120
    MFS:          9000
    PMD autoneg:  supported: yes, enabled: yes
      Adv:          10Base-T, HD: yes, FD: yes
      Adv:          100Base-TX, HD: yes, FD: yes
      Adv:          1000Base-X, HD: yes, FD: yes
      Adv:          1000Base-T, HD: yes, FD: yes
      MAU oper type: 1000BaseTFD - Four-pair Category 5 UTP, full duplex mode
    MDI Power:    supported: no, enabled: no, pair control: no
      Device type:  PD
      Power pairs:  unknown
      Class:        unknown
      Power type:   2
      Power Source: unknown
      Power Priority: unknown
      PD requested power Value: 0
      PSE allocated power Value: 0
  VLAN:         1, pvid: yes System-VLAN
  PPVID:        supported: yes, enabled: no
-------------------------------------------------------------------------------
```

```
&gmac1 {
        status = "okay";
        label = "wan";
        phy-handle = <&ethphy4>;
                        
        nvmem-cells = <&macaddr_bdinfo_de00 1>;
        nvmem-cell-names = "mac-address";
};
```

so, the OWRT receive LLDP info from switch
but switch on other side does not receive LLDP info

then, i changed lldpd.conf to
`configure system interface pattern switch.1, wan`
(switch.1 is VLAN#1 untagged on WAN)

i get this
```
sw12.netdev.ele#show lldp neighbor-information                                 
 Port      Device ID               Port ID                 management address      Port Description              System Name                   
 ----      ------------            ------------            ------------            --------------                --------------                
 Gi1/0/7   F4:A4:54:81:45:34       wan                     169.254.1.103           wan                           wr103.wrt.ele 
 ```
 
 so the switch receive LLDP from OWRT
 
 so, basically, nothing has changed
 
 did i miss something ? did i put patch files in wrong place ?

arinc9, 2024-01-08:
It's finally a good time for me to investigate this. First of all, LLDP frames are link-local frames, therefore, please do not use any network interfaces other than the ones that achieve a link with its link partner, and the link partner being another networking device. So, do not use bridge, VLAN, or DSA conduit interfaces. In this case, lldpd should listen on wan, lan1, lan2, lan3, and lan4, and not necessarily on all of them. It should not listen on eth0 as eth0 is used as the DSA conduit for the user port interfaces, which are lan1, lan2, lan3, and lan4.

That said, I can't reproduce your issue. I've tested whether both devices on the link can send & receive LLDP frames properly. They do. One device is an MT7621 SoC device with MT7530 switch. Wan port (phy4) is muxed to gmac1.

```
# bridge v
port              vlan-id  
wan               1 PVID Egress Untagged
                  2
                  100
                  200
                  255
lan1              1 PVID Egress Untagged
lan2              1 PVID Egress Untagged
lan3              1 PVID Egress Untagged
lan4              255 PVID Egress Untagged
br0               1 PVID Egress Untagged
# cat /etc/lldpd.d/lldpd.conf 
configure lldp portidsubtype ifname
configure system interface pattern wan
# lldpctl
-------------------------------------------------------------------------------
LLDP neighbors:
-------------------------------------------------------------------------------
Interface:    wan, via: LLDP, RID: 1, Time: 0 day, 00:00:25
  Chassis:     
    ChassisID:    mac (redacted MAC address)
    SysName:      arinc9-PC
    SysDescr:     Ubuntu 23.10 Linux 6.6.8-060608-generic #202312201634 SMP PREEMPT_DYNAMIC Wed Dec 20 16:47:31 UTC 2023 x86_64
    MgmtIP:       10.10.10.3
    MgmtIface:    3
    MgmtIP:       fd07:abc9:3cd8::18b
    MgmtIface:    3
    Capability:   Bridge, off
    Capability:   Router, on
    Capability:   Wlan, off
    Capability:   Station, off
  Port:        
    PortID:       mac (redacted MAC address)
    PortDescr:    enx
    TTL:          120
    PMD autoneg:  supported: yes, enabled: yes
      Adv:          10Base-T, HD: yes, FD: yes
      Adv:          100Base-TX, HD: yes, FD: yes
      MAU oper type: 100BaseTXFD - 2 pair category 5 UTP, full duplex mode
-------------------------------------------------------------------------------
```

The other device is my desktop computer:
```
$ lldpctl
-------------------------------------------------------------------------------
LLDP neighbors:
-------------------------------------------------------------------------------
Interface:    enx, via: LLDP, RID: 4, Time: 0 day, 00:00:41
  Chassis:     
    ChassisID:    mac (redacted MAC address)
    SysName:      buildroot
    SysDescr:     Buildroot 2023.08-rc1 Linux 6.7.0-rc6-next-20231222-00010-g9c585f4c1a2b #295 SMP Mon Jan  8 23:32:56 +03 2024 mips
    MgmtIP:       192.168.2.1
    MgmtIface:    9
    Capability:   Bridge, on
    Capability:   Router, off
    Capability:   Wlan, off
    Capability:   Station, off
  Port:        
    PortID:       ifname wan
    PortDescr:    wan
    TTL:          120
    PMD autoneg:  supported: yes, enabled: yes
      Adv:          10Base-T, HD: yes, FD: yes
      Adv:          100Base-TX, HD: yes, FD: yes
      Adv:          1000Base-T, HD: no, FD: yes
      MAU oper type: 100BaseTXFD - 2 pair category 5 UTP, full duplex mode
  VLAN:         1, pvid: yes vlan1
  VLAN:         2, pvid: no vlan2
  VLAN:         100, pvid: no vlan100
  VLAN:         200, pvid: no vlan200
  VLAN:         255, pvid: no vlan255
-------------------------------------------------------------------------------
```

This was using a recent linux-next kernel with buildroot filesystem. I have it working on OpenWrt as well.

```
$ lldpctl                        
-------------------------------------------------------------------------------
LLDP neighbors:
-------------------------------------------------------------------------------
Interface:    enx, via: LLDP, RID: 5, Time: 0 day, 00:01:20
  Chassis:     
    ChassisID:    mac (redacted MAC address)
    SysName:      OpenWrt
    SysDescr:     OpenWrt 23.05.2 @ OpenWrt
    MgmtIP:       (redacted IPv6 address)
    MgmtIface:    2
    Capability:   Bridge, on
    Capability:   Router, on
    Capability:   Wlan, off
    Capability:   Station, off
  Port:        
    PortID:       ifname wan
    PortDescr:    wan
    TTL:          120
    PMD autoneg:  supported: yes, enabled: yes
      Adv:          10Base-T, HD: yes, FD: yes
      Adv:          100Base-TX, HD: yes, FD: yes
      Adv:          1000Base-T, HD: no, FD: yes
      MAU oper type: 100BaseTXFD - 2 pair category 5 UTP, full duplex mode
  VLAN:         1, pvid: yes vlan1
  VLAN:         2, pvid: no vlan2
  VLAN:         100, pvid: no vlan100
  VLAN:         200, pvid: no vlan200
  VLAN:         255, pvid: no vlan255
  LLDP-MED:    
    Device Type:  Network Connectivity Device
    Capability:   Capabilities, yes
    Capability:   Policy, yes
    Capability:   Location, yes
    Capability:   MDI/PSE, yes
    Capability:   MDI/PD, yes
    Capability:   Inventory, yes
    Inventory:   
      Software Revision: 5.15.137
-------------------------------------------------------------------------------
```

Please run the commands below on 23.05.2 with lldpd, ip-bridge, and ip-full packages installed. Let me know if both devices see each other.
```
service network stop
ip l add l br0 type bridge vlan_filtering 1
ip l set wan master br0
ip l set lan1 master br0
ip l set lan2 master br0
ip l set lan3 master br0
ip l set lan4 master br0

bridge v a vid 2 dev wan
bridge v a vid 100 dev wan
bridge v a vid 200 dev wan
bridge v a vid 255 dev wan

ip l set up br0
ip l set up wan
ip l set up lan1
ip l set up lan2
ip l set up lan3
ip l set up lan4

cat <<EOF > /etc/lldpd.d/lldpd.conf
configure lldp portidsubtype ifname
configure system interface pattern wan
EOF

service lldpd restart
lldpctl
```