openwrt/openwrt #13789: kernel panic after jffs2_build_filesystem

p-m, 2023-10-25:
### Describe the bug

Hi,
I'm testing the latest git version from the openwrt-23.05 branch on my Asus wl-500gp-v1. Unfortunately there is a kernel panic about 63s after the start. It seems to be related to the "jffs2_build_filesystem".

I don't know, if this is related too: it seems, that the overlay is on the ram. Output of "df -h":
overlayfs:/tmp/root      60.8M

With previous versions, the overlay was on the flash:
overlayfs:/overlay        2.7M

How could I build an openwrt-23.05 for my router please?

TIA for any hints, Peter

### OpenWrt version

r23536-263ef841d4

### OpenWrt target/subtarget

bcm47xx/legacy

### Device

Asus WL500GP V1

### Image kind

Self-built image

### Steps to reproduce

Build the latest version from openwrt-23.05 branch and load it to an Asus WL-500GP-V1.

### Actual behaviour

Kernel panic after
`CPU 0 Unable to handle kernel paging request at virtual address 00000000, epc == 8027b5cc, ra == 8027ba00`

### Expected behaviour

No kernel panic.

### Additional info

Please see attached file [console-output.txt](https://github.com/openwrt/openwrt/files/13168594/console-output.txt).

### Diffconfig

```text
tmp/.config-package.in:33662:error: recursive dependency detected!
tmp/.config-package.in:33662:   symbol PACKAGE_python3-pymysql depends on PYTHON3_PYMYSQL_SHA_PASSWORD_SUPPORT
no-backup/feeds/packages/lang/python/pymysql/Config.in:4:       symbol PYTHON3_PYMYSQL_SHA_PASSWORD_SUPPORT depends on PACKAGE_python3-pymysql
For a resolution refer to Documentation/kbuild/kconfig-language.rst
subsection "Kconfig recursive dependency limitations"

tmp/.config-package.in:33662:error: recursive dependency detected!
tmp/.config-package.in:33662:   symbol PACKAGE_python3-pymysql depends on PYTHON3_PYMYSQL_SHA_PASSWORD_SUPPORT
no-backup/feeds/packages/lang/python/pymysql/Config.in:4:       symbol PYTHON3_PYMYSQL_SHA_PASSWORD_SUPPORT depends on PACKAGE_python3-pymysql
For a resolution refer to Documentation/kbuild/kconfig-language.rst
subsection "Kconfig recursive dependency limitations"

CONFIG_TARGET_bcm47xx=y
CONFIG_TARGET_bcm47xx_legacy=y
CONFIG_TARGET_bcm47xx_legacy_DEVICE_asus_wl-500gp-v1=y
CONFIG_BUSYBOX_CUSTOM=y
# CONFIG_BUSYBOX_CONFIG_FEATURE_NSLOOKUP_BIG is not set
# CONFIG_BUSYBOX_CONFIG_PING6 is not set
# CONFIG_DOWNLOAD_CHECK_CERTIFICATE is not set
# CONFIG_FEED_luci is not set
# CONFIG_FEED_packages is not set
# CONFIG_FEED_routing is not set
# CONFIG_FEED_telephony is not set
# CONFIG_KERNEL_IPV6_MROUTE is not set
CONFIG_OPENSSL_ENGINE=y
CONFIG_OPENSSL_PREFER_CHACHA_OVER_GCM=y
CONFIG_OPENSSL_SMALL_FOOTPRINT=y
CONFIG_OPENSSL_WITH_ASM=y
CONFIG_OPENSSL_WITH_CHACHA_POLY1305=y
CONFIG_OPENSSL_WITH_CMS=y
CONFIG_OPENSSL_WITH_DEPRECATED=y
CONFIG_OPENSSL_WITH_PSK=y
CONFIG_OPENSSL_WITH_SRP=y
CONFIG_OPENSSL_WITH_TLS13=y
# CONFIG_PACKAGE_dropbear is not set
CONFIG_PACKAGE_libatomic=y
CONFIG_PACKAGE_libncurses=y
CONFIG_PACKAGE_libopenssl=y
# CONFIG_PACKAGE_odhcp6c is not set
# CONFIG_PACKAGE_odhcpd-ipv6only is not set
CONFIG_PACKAGE_openssh-keygen=y
CONFIG_PACKAGE_openssh-server=y
CONFIG_PACKAGE_openssh-sftp-server=y
# CONFIG_PACKAGE_ppp is not set
CONFIG_PACKAGE_screen=y
CONFIG_PACKAGE_terminfo=y
CONFIG_PACKAGE_zlib=y
# CONFIG_SIGNATURE_CHECK is not set
# CONFIG_SIGNED_PACKAGES is not set
# CONFIG_OPENSSH_LIBFIDO2 is not set
CONFIG_PACKAGE_kmod-lib-crc-ccitt=y
# CONFIG_PACKAGE_kmod-mppe is not set
CONFIG_PACKAGE_kmod-ppp=y
# CONFIG_PACKAGE_kmod-ppp-synctty is not set
# CONFIG_PACKAGE_kmod-pppoa is not set
CONFIG_PACKAGE_kmod-pppoe=y
# CONFIG_PACKAGE_kmod-pppol2tp is not set
CONFIG_PACKAGE_kmod-pppox=y
# CONFIG_PACKAGE_kmod-pptp is not set
CONFIG_PACKAGE_kmod-slhc=y
# CONFIG_PACKAGE_libcbor is not set
# CONFIG_PACKAGE_libevdev is not set
# CONFIG_PACKAGE_libfido2 is not set
# CONFIG_PACKAGE_libudev-zero is not set
CONFIG_PACKAGE_openwrt-keyring=y
CONFIG_PACKAGE_usign=y
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-10-25:
23.05.0 does not support 32MB RAM devices anymore. Install zram-swap on 22.03.5 and enjoy last year of hostapd crashing less often. Wiki warning in device page has a link to image size trimming advice page.

brada4, 2023-10-25:
This is not a bug, you can close and continue in forum

p-m, 2023-10-26:
Ram is 128MB.

brada4, 2023-10-26:
Well, it starts with 32MB and you see consequence, no ram for initial overlay init, 
In v22 it is luci giving up sometimes with same message in log.

Please check configured RAm with v21 / v22 via `free` 
Then at v23 push button when led blinks fastest to avoid overlay alltogether and check same.

Memory lost in v23 -> memory was not conveyed correctly from nvram - bug!!!! attach both dmesg-s like to original post if you like
32MB in both cases - complete your hardware mod in nvram (using v21/22 for obvious reason)
https://openwrt.org/toh/asus/wl-500gp#enabling_all_ram

p-m, 2023-10-26:
Did you see line 133 of console-output.txt?

It seems, that there is enough memory:
`[    0.000000] Memory: 124268K/131072K available (4029K kernel code, 567K rwdata, 564K rodata, 204K init, 268K bss, 6804K reserved, 0K cma-reserved)`

Where do you see, that it starts with 32MB please?

p-m, 2023-10-26:
Is the oops perhaps related to netifd (line 48 of console-output.txt)?

p-m, 2023-10-26:
Ok, I tested it: after removing /sbin/netifd it works, no more oops.
But now, there is no more network...

I guess, the title of this bug should be modified accordingly.

brada4, 2023-10-26:
Try to ifconfig interfaces just UP, that shoould allocate tx/rx buffer or hundred for each
Then add more config , at least on eth0.... Then you can go into safe mode with button, at least some breathing space from then on.
Bug title and text is in your hands, add relevant info so that it covers our discoveries for newcomer ;-)

p-m, 2023-10-26:
Ok, thanks.

How could I help with debugging?

brada4, 2023-10-26:
line 28 is typical on kmalloc return result not being checked and zero pointer being accessed as a consequence.

brada4, 2023-10-26:
So I ask you to gradually add kmalloc allocations until it fails, starting with eth0 which with some luck might be handled by netifd giving comfort to using device at distance. Adding debug options  past default backtraces will not be an option due to small flash, though netbooting such image may help if at all possible.

brada4, 2023-10-26:
Among kmalloc consumers - complex like journaled filesystems, network benchmarks, thousands of `cat` processes forked in backgound. But first we try stable boot till ssh -OK?

p-m, 2023-10-26:
Could you provide some commands please?
Unfortunately I don't know, how to "gradually add kmalloc allocations".

brada4, 2023-10-26:
First you start default reset device woth eth0 in failsafe mode
Then mount_root
Then ir is sort of stable to test furter.
Say b4? driver may need some option, or usb disk robs all wofi juice, pr ntfs aitomount etc

KanjiMonster, 2023-10-26:
Please rebuild the image with `KERNEL_KALLSYMS` enabled, so the kernel stacktrace becomes readable.

```
 Symbol: KERNEL_KALLSYMS [=y]                                                                                                                                                         │  
  │ Type  : bool                                                                                                                                                                         │  
  │ Defined at config/Config-kernel.in:330                                                                                                                                               │  
  │   Prompt: Compile the kernel with symbol table information                                                                                                                           │  
  │   Location:                                                                                                                                                                          │  
  │     -> Global build settings                                                                                                                                                         │  
  │ (5)   -> Kernel build options                  

```

brada4, 2023-10-26:
check it fits.... you have 2.7M extra available for symbols.

p-m, 2023-10-26:
Ok, here is the trace:

```
[   63.636847] b44 ssb0:0 eth0: powering down PHY
[   63.649036] CPU 0 Unable to handle kernel paging request at virtual address 00000000, epc == 8027f250, ra == 8027f684
[   63.659800] Oops[#1]:
[   63.662137] CPU: 0 PID: 1055 Comm: netifd Not tainted 5.15.134 #0
[   63.668338] $ 0   : 00000000 00000001 1f038000 10000000
[   63.673726] $ 4   : 8090fc80 0000042c 00000008 00000000
[   63.679103] $ 8   : 00000001 81aeba88 00000000 00000009
[   63.684478] $12   : 776f7020 81aebd68 7d09090a 6e697265
[   63.689863] $16   : 8090fc80 00000000 00000000 00001000
[   63.695240] $20   : 7f73b964 81aebe37 818565a0 00004000
[   63.700623] $24   : 00000000 8028eb68                  
[   63.706009] $28   : 81aea000 81aebca0 00000000 8027f684
[   63.711393] Hi    : 0044aa20
[   63.714326] Lo    : 0000001a
[   63.717259] epc   : 8027f250 b44_free_rings+0x2c/0xd4
[   63.722436] ra    : 8027f684 b44_init_rings+0x18/0xd4
[   63.727594] Status: 10008c02 KERNEL EXL 
[   63.731624] Cause : 00808008 (ExcCode 02)
[   63.735701] BadVA : 00000000
[   63.738633] PrId  : 00029006 (Broadcom BMIPS3300)
[   63.743416] Modules linked in: pppoe ppp_async nft_fib_inet nf_flow_table_ipv6 nf_flow_table_ipv4 nf_flow_table_inet b43 pppox ppp_generic nft_reject_ipv6 nc
[   63.816270] Process netifd (pid: 1055, threadinfo=13a663b9, task=59ea7703, tls=77eebdfc)
[   63.824503] Stack : 7f73b964 8090fc80 805c2250 00020000 00000903 01000200 8090fc80 8090fc80
[   63.833130]         8090fc80 81aebd68 00000000 8027f684 00000001 8027e2c0 8090f800 8090fc80
[   63.841744]         81aebd68 8090f800 8090fc80 8027f7c4 81aebe3c 802af324 ffffff09 ffff0a00
[   63.850366]         00000048 8090f800 7f73b964 8030326c 81aebdb4 80510000 81aebec0 00000000
[   63.858989]         81aebe5c 804838e8 00000000 801f3dcc 80d1b4d8 81aebe18 80c46b28 00000013
[   63.867603]         ...
[   63.870129] Call Trace:
[   63.872623] [<8027f250>] b44_free_rings+0x2c/0xd4
[   63.877456] [<8027f684>] b44_init_rings+0x18/0xd4
[   63.882284] [<8027f7c4>] b44_set_pauseparam+0x84/0xf4
[   63.887460] [<8030326c>] dev_ethtool+0xb2c/0x2150
[   63.892295] [<802ddc20>] dev_ioctl+0x44c/0x4c8
[   63.896894] [<80295dd8>] sock_ioctl+0x74/0x3f0
[   63.901459] [<80110138>] vfs_ioctl+0x28/0x40
[   63.905886] 
[   63.907429] Code: 24131000  8e110020  02328821 <8e220000> 1040000b  24070002  8e020298  8e250004  8c440008 
[   63.917505] 
[   63.919275] ---[ end trace 49908d76f98b159f ]---
[   63.924022] Kernel panic - not syncing: Fatal exception
```

KanjiMonster, 2023-10-26:
Okay, looking at the code pointed to by the stacktrace, it shows it being an issue in the b44 driver:

`b44_free_rings()` accesses `b44::rx_buffers` (and `::tx_buffers`) unconditionally, but `b44::rx_buffers` is only valid when the device is up (they get allocated in `b44_open()`, and deallocated again in `b44_close()`), any other time these is just a NULL pointers.

So if you try to change the pause params while the network interface is disabled/administratively down, everything explodes (which likely `netifd` tries to do).

This probably needs something similar to what `b44_change_mtu()` does:

```
	if (!netif_running(dev)) {
		/* We'll just catch it later when the
		 * device is up'd.
		 */
		dev->mtu = new_mtu;
		return 0;
	}
```

@rmilecki can you take care of this?

`b44_set_ringparam()` has the same issue, but it does not exist in 6.6 anymore, so not sure if it's worth fixing.

rezad1393, 2023-12-28:
> `b44_set_ringparam()` has the same issue, but it does not exist in 6.6 anymore, so not sure if it's worth fixing.

based on openwrt history and kernel release dates, kernel 6.1 would be used for v24(late? 2024). so 6.6 would be released at 2025?

maybe the next version (24) should rebase on kernel 6.6 and skip 6.1 but I see that from 2023-05-22 kernel 6.1 was used so I don't this that happens as there is no mention of 6.6.

svobodavac, 2024-02-14:
I have been using my Asus WL500GP V1 for many years for testing purposes.
Unfortunately, due to this bug, it does not work with OpenWrt 23.05.2.
So I tried to fix the bug. This patch works for me:
```
--- build_dir/target-mipsel_mips32_musl/linux-bcm47xx_legacy/linux-5.15.137/drivers/net/ethernet/broadcom/b44.c.Orig    2024-02-09 23:57:46.626310052 +0100
+++ build_dir/target-mipsel_mips32_musl/linux-bcm47xx_legacy/linux-5.15.137/drivers/net/ethernet/broadcom/b44.c 2024-02-11 09:50:12.729336820 +0100
@@ -2054,12 +2054,15 @@
                bp->flags |= B44_FLAG_TX_PAUSE;
        else
                bp->flags &= ~B44_FLAG_TX_PAUSE;
-       if (bp->flags & B44_FLAG_PAUSE_AUTO) {
-               b44_halt(bp);
-               b44_init_rings(bp);
-               b44_init_hw(bp, B44_FULL_RESET);
-       } else {
-               __b44_set_flow_ctrl(bp, bp->flags);
+       if (netif_running(dev)) {
+               return 0;
+               if (bp->flags & B44_FLAG_PAUSE_AUTO) {
+                       b44_halt(bp);
+                       b44_init_rings(bp);
+                       b44_init_hw(bp, B44_FULL_RESET);
+               } else {
+                       __b44_set_flow_ctrl(bp, bp->flags);
+               }
        }
        spin_unlock_irq(&bp->lock);
```
Device works fine now:

```
root@as-br:~# cat /etc/os-release | grep PRETTY
PRETTY_NAME="OpenWrt 23.05.2"

root@as-br:~# cat /tmp/sysinfo/model
Asus WL500GP V1

root@as-br:~# cat /proc/meminfo | grep MemTotal
MemTotal:          26532 kB
```

The bug can also be reproduced by executing following commands:
```
  ip link set dev eth0 down
  ethtool -A eth0 rx on tx on autoneg on
  ethtool -A eth0 rx on tx on autoneg off
```
However, it is necessary first to rename (delete, chmod) file /sbin/netifd and use serial console.

svobodavac, 2024-02-14:
This line
`> +               return 0;`
should not be there.

This patch works too:
```
--- build_dir/target-mipsel_mips32_musl/linux-bcm47xx_legacy/linux-5.15.137/drivers/net/ethernet/broadcom/b44.c.Orig    2024-02-09 23:57:46.626310052 +0100
+++ build_dir/target-mipsel_mips32_musl/linux-bcm47xx_legacy/linux-5.15.137/drivers/net/ethernet/broadcom/b44.c 2024-02-14 23:07:50.884315082 +0100
@@ -2054,12 +2054,14 @@
                bp->flags |= B44_FLAG_TX_PAUSE;
        else
                bp->flags &= ~B44_FLAG_TX_PAUSE;
-       if (bp->flags & B44_FLAG_PAUSE_AUTO) {
-               b44_halt(bp);
-               b44_init_rings(bp);
-               b44_init_hw(bp, B44_FULL_RESET);
-       } else {
-               __b44_set_flow_ctrl(bp, bp->flags);
+       if (netif_running(dev)) {
+               if (bp->flags & B44_FLAG_PAUSE_AUTO) {
+                       b44_halt(bp);
+                       b44_init_rings(bp);
+                       b44_init_hw(bp, B44_FULL_RESET);
+               } else {
+                       __b44_set_flow_ctrl(bp, bp->flags);
+               }
        }
        spin_unlock_irq(&bp->lock);
```

p-m, 2024-04-03:
Thanks, it works! Could this be integrated into the git repository of OpenWrt please?

KanjiMonster, 2024-04-03:
Ideally someone should send this upstream, so we don't need to carry yet another patch.

Also I think the call to `b44_enable_ints(bp);` after `spin_unlock_irq(&bp->lock);` should only be done if the netif is running as well.

p-m, 2024-04-03:
Ok, done: https://lore.kernel.org/netdev/878r1ufs9p.fsf@a16n.net/T/

p-m, 2024-04-06:
> +       if (netif_running(dev)) {

Hi,
Could you please describe your patch, so that I can send it to upstream?
Or alternatively could you send it to upstream yourself?
TIA, Peter

svobodavac, 2024-04-06:
> > * ```
> >     if (netif_running(dev)) {
> >   ```
> 
> Hi, Could you please describe your patch, so that I can send it to upstream? Or alternatively could you send it to upstream yourself? TIA, Peter

I'm not sure what exactly is going on with my patch. I just tried to fix the bug somehow.

p-m, 2024-04-07:
> I'm not sure what exactly is going on with my patch. I just tried to fix the bug somehow.

Without some details, I guess, that the patch won’t be accepted.
See here: https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/Documentation/process/submitting-patches.rst
And here: https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/tree/Documentation/process/maintainer-netdev.rst

I can try with just "this patch avoids a kernel panic", but it would certainly be better, if you could explain what the patch does, and why.

p-m, 2024-05-27:
Hi,
It seems to be resolved in the openwrt-23.05 branch.
Peter