openwrt/openwrt #13797: mt7622: AX6S: Failed to get patch semaphore

nshopik, 2023-10-26:
### Describe the bug

For some reason 2ghz radio won't load up on boot producing these messages in dmesg
```
[   27.350770] mt7622-wmac 18000000.wmac: Message 00000010 (seq 1) timeout
[   27.357411] mt7622-wmac 18000000.wmac: Failed to get patch semaphore
[   27.581151] mt7622-wmac 18000000.wmac: Failed to get patch semaphore
[   27.801121] mt7622-wmac 18000000.wmac: Failed to get patch semaphore
[   28.021101] mt7622-wmac 18000000.wmac: Failed to get patch semaphore
[   28.241076] mt7622-wmac 18000000.wmac: Failed to get patch semaphore
[   28.461106] mt7622-wmac 18000000.wmac: Failed to get patch semaphore
[   28.681102] mt7622-wmac 18000000.wmac: Failed to get patch semaphore
[   28.901101] mt7622-wmac 18000000.wmac: Failed to get patch semaphore
[   29.121122] mt7622-wmac 18000000.wmac: Failed to get patch semaphore
[   29.341098] mt7622-wmac 18000000.wmac: Failed to get patch semaphore
[   29.561086] mt7622-wmac 18000000.wmac: Failed to get patch semaphore
```
But strangely enough, I had snmpd which current version had memory leak and after snmpd out of memory and crash/reboot router it property load up firmware and radio will start. If I manually reboot router once again I'll get back to where 2ghz won't boot.

### OpenWrt version

r23497-6637af95aa

### OpenWrt target/subtarget

mediatek/mt7622

### Device

Xiaomi Redmi Router AX6S

### Image kind

Official downloaded image

### Steps to reproduce

Just flash current openwrt no additional steps needed.

### Actual behaviour

2ghz radio are disabled/non responsive 

### Expected behaviour

working 2ghz radio. 

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-10-26:
Nearly duplicate of 
https://github.com/openwrt/mt76/issues/836

nshopik, 2023-10-26:
Yeah looks similar, only difference I'm running on 5.15 kernel

brada4, 2023-10-26:
Try snapshot - maybe yours goes away, otherwise driver is few months off same on both openwrt kernels.
(really try, then roll back)

nshopik, 2023-10-26:
Actually I tried snapshot, before openwrt 23 release, maybe 4 weeks ago and had same result. Tried again like you suggested `r24194-e1aaa1defd` build and its actually working correctly now. Guess I'll stick with snapshot and monitor to any issues.

brada4, 2023-10-26:
make yourself heard in mt76 thread.

nshopik, 2023-10-26:
Which one? One you linked as dupe?

nshopik, 2023-10-26:
https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/diff/?id=v5.15.135&id2=v5.15.134&dt=2 point release diff and 2 files related to eeprom changed

https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/diff/drivers/net/wireless/mediatek/mt76/mt76x02_eeprom.c?id=v5.15.135&id2=v5.15.134
https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/diff/drivers/net/wireless/mediatek/mt76/mt76x2/eeprom.c?id=v5.15.135&id2=v5.15.134

nshopik, 2023-10-29:
Spoke too soon I guess, issue still there but it appears to be less common. It appears to be random in way. Originally I though enabling SQM may have effect on it, but couldn't really reproduce it with 100%.

nshopik, 2023-11-21:
running OpenWrt 23.05.2 now and previously snapshot release with kernel 5.15.137 was pretty stable, no issues when loading firmware.

erpalma, 2024-03-12:
I guess this issue is still there. Tested on 23.05.2 with 5.15.137:
```
[   26.710686] mt7622-wmac 18000000.wmac: Message 00000010 (seq 1) timeout
[   26.717336] mt7622-wmac 18000000.wmac: Failed to get patch semaphore
[   26.942056] mt7622-wmac 18000000.wmac: Failed to get patch semaphore
[   27.161036] mt7622-wmac 18000000.wmac: Failed to get patch semaphore
[   27.380983] mt7622-wmac 18000000.wmac: Failed to get patch semaphore
[   27.600963] mt7622-wmac 18000000.wmac: Failed to get patch semaphore
[   27.821048] mt7622-wmac 18000000.wmac: Failed to get patch semaphore
[   28.041012] mt7622-wmac 18000000.wmac: Failed to get patch semaphore
[   28.260994] mt7622-wmac 18000000.wmac: Failed to get patch semaphore
[   28.480964] mt7622-wmac 18000000.wmac: Failed to get patch semaphore
[   28.700957] mt7622-wmac 18000000.wmac: Failed to get patch semaphore
[   28.921010] mt7622-wmac 18000000.wmac: Failed to get patch semaphore
```
Can't enable Wifi 2.4 as a result. BTW also tested on latest master, same issue.

brada4, 2024-03-12:
Does it work after cold boot?

erpalma, 2024-03-12:
Not at all, 2.4 works on stock but hasn't worked a single time with Openwrt.

brada4, 2024-03-12:
No point jumping onthers' thread.

nshopik, 2025-01-27:
Looks like issue is back with `OpenWrt 24.10.0-rc5, r28304-6dacba30a7`

maurerle, 2025-11-07:
Hi @nshopik, the issue should be fixed in latest openwrt-24.10.4 - can you report back if this works?

nshopik, 2025-11-07:
I'm pretty sure I never seen issue once since moving into 24.10 (non-RC). Currently 24.10.3 works fine