openwrt/openwrt #13801: ipq40xx: Linksys WHW03 V2: all leds are completely non-working

Lanchon, 2023-10-26:
### Describe the bug

the WHW03 V2 port is unable to use its leds at all.

the device has a single RGB led behind a [PCA9633](https://www.nxp.com/docs/en/data-sheet/PCA9633.pdf) led controller/driver.

the kernel driver works but userland is borked. possibly an issue with the DTS?

```
root@router:~# ls -l /sys/class/leds/pca*
lrwxrwxrwx    1 root     root             0 Dec 31  1969 /sys/class/leds/pca963x:blue -> ../../devices/platform/soc/78b7000.i2c/i2c-0/0-0062/leds/pca963x:blue
lrwxrwxrwx    1 root     root             0 Dec 31  1969 /sys/class/leds/pca963x:green -> ../../devices/platform/soc/78b7000.i2c/i2c-0/0-0062/leds/pca963x:green
lrwxrwxrwx    1 root     root             0 Dec 31  1969 /sys/class/leds/pca963x:red -> ../../devices/platform/soc/78b7000.i2c/i2c-0/0-0062/leds/pca963x:red

root@router:~# ls -l /sys/class/leds/pca963x:red/
-rw-r--r--    1 root     root          4096 Oct 26 15:16 brightness
lrwxrwxrwx    1 root     root             0 Oct 26 15:16 device -> ../../../0-0062
-r--r--r--    1 root     root          4096 Oct 25 03:09 max_brightness
lrwxrwxrwx    1 root     root             0 Oct 26 15:16 subsystem -> ../../../../../../../../class/leds
-rw-r--r--    1 root     root             0 Oct 26 15:16 trigger
-rw-r--r--    1 root     root          4096 Oct 26 15:16 uevent

# this works:
root@router:~# echo 255 >/sys/class/leds/pca963x:red/brightness 
```

leds are controlled by [/etc/diag.sh](https://github.com/openwrt/openwrt/blob/main/package/base-files/files/etc/diag.sh) via [/lib/functions/leds.sh](https://github.com/openwrt/openwrt/blob/main/package/base-files/files/lib/functions/leds.sh). led access go through `alias` entries in the DTS. on this device, the code cannot locate any leds using their aliases.

```
root@router:~# cat /proc/device-tree/aliases/led-failsafe; echo
/soc/i2c@78b7000/led-controller/red@0

root@router:~# . /lib/functions/leds.sh
root@router:~# get_dt_led_path failsafe
/proc/device-tree/soc/i2c@78b7000/led-controller/red@0

root@router:~# ls -l $( get_dt_led_path failsafe )
-r--r--r--    1 root     root             4 Oct 26 01:54 color
-r--r--r--    1 root     root            10 Oct 26 01:54 function
-r--r--r--    1 root     root             4 Dec 31  1969 label
-r--r--r--    1 root     root             5 Oct 26 01:54 linux,default-trigger
-r--r--r--    1 root     root             4 Oct 26 01:54 name
-r--r--r--    1 root     root             4 Oct 26 01:54 reg

root@router:~# cat $( get_dt_led_path failsafe )/name; echo
red
root@router:~# cat $( get_dt_led_path failsafe )/label; echo
red

root@router:~# get_dt_led failsafe
red
```

the problem is that `get_dt_led` returns `red` but should return `pca963x:red` (its [implementation](https://github.com/openwrt/openwrt/blob/5a6368e85deb1d751238447ea8b289576e701cb0/package/base-files/files/lib/functions/leds.sh#L14-L24) just returns the led's `label`, if it exists), and this naming mismatch is why openwrt cannot control the leds (led access happens [here](https://github.com/openwrt/openwrt/blob/5a6368e85deb1d751238447ea8b289576e701cb0/package/base-files/files/lib/functions/leds.sh#L27)).

### why?

this might be some kind of regression, as i doubt the person who ported the device did not notice. maybe a driver update during a kernel upgrade? this would need to be answered by someone more familiar with the kernel. this issue might affect other devices too.

### the DTB

https://github.com/openwrt/openwrt/blob/5a6368e85deb1d751238447ea8b289576e701cb0/target/linux/ipq40xx/files/arch/arm/boot/dts/qcom-ipq4019-whw03v2.dts#L13-L18

### fixes?

i don't know, it seems the device prefix `pca963x:` is in line with expectations. maybe change the label entries in the DTB?

### OpenWrt version

r23497-6637af95aa

### OpenWrt target/subtarget

ipq40xx/generic

### Device

Linksys WHW03 V2 (Velop)

### Image kind

Official downloaded image

### Steps to reproduce

no leds work during:
- regular boot
- sysupgrade
- failsafe

### Actual behaviour

leds not working

### Expected behaviour

leds work

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-10-27:
duplicate #13803

Lanchon, 2023-10-27:
closing this issue, somehow it got posted half-way as i was writing it and ended up posted twice.

closed 2023-10-27 by Lanchon