openwrt/openwrt #13803: ipq40xx: Linksys WHW03 V2: none of the leds work

Lanchon, 2023-10-26:
### Describe the bug

the WHW03 V2 port is unable to use the leds on the device.

the device has a single RGB led behind a [PCA9633](https://www.nxp.com/docs/en/data-sheet/PCA9633.pdf) led controller/driver.

the kernel driver works but userland is borked. possibly an issue with the DTS?

```
root@router:~# ls -l /sys/class/leds/pca*
lrwxrwxrwx    1 root     root             0 Dec 31  1969 /sys/class/leds/pca963x:blue -> ../../devices/platform/soc/78b7000.i2c/i2c-0/0-0062/leds/pca963x:blue
lrwxrwxrwx    1 root     root             0 Dec 31  1969 /sys/class/leds/pca963x:green -> ../../devices/platform/soc/78b7000.i2c/i2c-0/0-0062/leds/pca963x:green
lrwxrwxrwx    1 root     root             0 Dec 31  1969 /sys/class/leds/pca963x:red -> ../../devices/platform/soc/78b7000.i2c/i2c-0/0-0062/leds/pca963x:red

root@router:~# ls -l /sys/class/leds/pca963x:red/
-rw-r--r--    1 root     root          4096 Oct 26 15:16 brightness
lrwxrwxrwx    1 root     root             0 Oct 26 15:16 device -> ../../../0-0062
-r--r--r--    1 root     root          4096 Oct 25 03:09 max_brightness
lrwxrwxrwx    1 root     root             0 Oct 26 15:16 subsystem -> ../../../../../../../../class/leds
-rw-r--r--    1 root     root             0 Oct 26 15:16 trigger
-rw-r--r--    1 root     root          4096 Oct 26 15:16 uevent

# this works:
root@router:~# echo 255 >/sys/class/leds/pca963x:red/brightness 
```

leds are controlled by [/etc/diag.sh](https://github.com/openwrt/openwrt/blob/main/package/base-files/files/etc/diag.sh) via [/lib/functions/leds.sh](https://github.com/openwrt/openwrt/blob/main/package/base-files/files/lib/functions/leds.sh). led access go through `alias` entries in the DTS. on this device, the code cannot locate any leds using their aliases.

```
root@router:~# cat /proc/device-tree/aliases/led-failsafe; echo
/soc/i2c@78b7000/led-controller/red@0

root@router:~# . /lib/functions/leds.sh
root@router:~# get_dt_led_path failsafe
/proc/device-tree/soc/i2c@78b7000/led-controller/red@0

root@router:~# ls -l $( get_dt_led_path failsafe )
-r--r--r--    1 root     root             4 Oct 26 01:54 color
-r--r--r--    1 root     root            10 Oct 26 01:54 function
-r--r--r--    1 root     root             4 Dec 31  1969 label
-r--r--r--    1 root     root             5 Oct 26 01:54 linux,default-trigger
-r--r--r--    1 root     root             4 Oct 26 01:54 name
-r--r--r--    1 root     root             4 Oct 26 01:54 reg

root@router:~# cat $( get_dt_led_path failsafe )/name; echo
red
root@router:~# cat $( get_dt_led_path failsafe )/label; echo
red

root@router:~# get_dt_led failsafe
red
```

the problem is that `get_dt_led` returns `red` but should return `pca963x:red`, and this naming mismatch is why openwrt cannot control any of the leds. note that the implementation just returns the led's `label`, if it exists:
https://github.com/openwrt/openwrt/blob/5a6368e85deb1d751238447ea8b289576e701cb0/package/base-files/files/lib/functions/leds.sh#L14-L24

all setting of led attributes happens here, where `$1` is expected to be `pca963x:<color>`, not just `<color>`:
https://github.com/openwrt/openwrt/blob/5a6368e85deb1d751238447ea8b289576e701cb0/package/base-files/files/lib/functions/leds.sh#L26-L28

### why?

this might be some kind of regression, as i doubt the person who ported the device did not notice. maybe a driver update during a kernel upgrade? a question for someone more familiar with the kernel than myself. if it is a regression, then this issue might affect other devices.

### the DTB

https://github.com/openwrt/openwrt/blob/5a6368e85deb1d751238447ea8b289576e701cb0/target/linux/ipq40xx/files/arch/arm/boot/dts/qcom-ipq4019-whw03v2.dts#L13-L18

https://github.com/openwrt/openwrt/blob/5a6368e85deb1d751238447ea8b289576e701cb0/target/linux/ipq40xx/files/arch/arm/boot/dts/qcom-ipq4019-whw03v2.dts#L269-L300

### fixes?

- i don't know, it seems the device prefix `pca963x:` is in line with expectations. maybe change the label entries in the DTB then?
- or `get_dt_led` could check if the led name exists, and if not try looking for `[^:]+[:]<name>` and return that if a single match exists.
- also, i think `LED_FUNCTION_STATUS` should be used instead of `LED_FUNCTION_INDICATOR` for all leds[^1].
- also, naming is supposed to be `<device>:<color>:<usage>`, so maybe we can rename the leds to `pca963x:red:status`. (a breaking change for users who are scripting their leds, but the device is fairly new. and if this was a regression, we already broke that code before.)
- also, this led should actually be exposed as a [multicolor led](https://github.com/torvalds/linux/blob/master/Documentation/leds/leds-class-multicolor.rst).

### a closing color piece

right now the DTB calls for green light when finished booting. if we fix the leds, this would result in:
- a user visible change: from blue to green when normal.
- ugliness: i tested the green led, and it is just plain ugly when lit all the time. (FYI, this is a non-debatable fully objective statement; i even asked God about it.)
- a bug? because of [this](https://github.com/openwrt/openwrt/blob/5a6368e85deb1d751238447ea8b289576e701cb0/target/linux/ipq40xx/files/arch/arm/boot/dts/qcom-ipq4019-whw03v2.dts#L297) and [this](https://github.com/openwrt/openwrt/blob/5a6368e85deb1d751238447ea8b289576e701cb0/package/base-files/files/etc/diag.sh#L41), i believe the blue led would stay lit after booting, so normal would actually be cyan, not green, and pretty bright.
- is this green (or maybe cyan) thing what the router does with OEM firmware? i don't know, but i can say that the sister port WHW01 runs [blue all the time](https://github.com/openwrt/openwrt/blob/7d6314851aa429e837b0fc1072e307179d114675/target/linux/ipq40xx/files/arch/arm/boot/dts/qcom-ipq4018-whw01.dts#L14-L15).

IMHO, i prefer not changing the running color and stick to the current blue. blue does a good enough job of communicating device state through the boot process. and this frees up the other leds for user defined purposes (i, for one, am using them).

[^1]: this is what ChatGPT 4 thinks on the subject:

    Here's a quick breakdown:

    1. **LED_FUNCTION_STATUS**: 
       - This typically represents a general status indication. For instance, a status LED might blink during normal operation or stay solid when a device is powered on.
       - Often used for things like power status, device readiness, or other general indications that the device is operating normally.

    2. **LED_FUNCTION_INDICATOR**: 
       - This is often used for more specific, transient conditions. 
       - Examples might include showing network activity (like on a router), indicating an unread notification on a device, or showing disk activity on a computer or server.

    How to Choose:

    1. **Purpose of the LED**:
       - If the LED is meant to show a specific, often transient, activity or condition, then `LED_FUNCTION_INDICATOR` is likely more appropriate.
       - If the LED shows a general, continuous status (like power-on status), then `LED_FUNCTION_STATUS` would be a better choice.

    2. **User Expectation**:
       - Consider how users of the device might interpret the LED's behavior. If users expect to see the LED as a sign that the device is simply on and working, `LED_FUNCTION_STATUS` is apt. If they're looking for it to show events or specific activities, then `LED_FUNCTION_INDICATOR` is the way to go.

    3. **Device Context**:
       - The nature of the device can guide your choice. For example, on a network switch, an LED showing Ethernet link status might be considered `LED_FUNCTION_STATUS`, while one showing packet activity might be `LED_FUNCTION_INDICATOR`.

    4. **Consistency**:
       - If you're working on a series of devices or a platform, ensure consistency across devices. If one device uses `LED_FUNCTION_STATUS` for power indication, it would be intuitive for users if other devices in the same series follow suit.

### OpenWrt version

r23497-6637af95aa

### OpenWrt target/subtarget

ipq40xx/generic

### Device

Linksys WHW03 V2 (Velop)

### Image kind

Official downloaded image

### Steps to reproduce

no leds work during:
- regular boot
- sysupgrade
- failsafe

### Actual behaviour

leds not working

### Expected behaviour

leds work

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

Lanchon, 2023-10-26:
author of port is: @vtremblay

Lanchon, 2023-10-27:
@brada4, did you just marked this bug as duplicate of itself? and github let you? watch it, you could bring down the interwebs with those shenanigans! lol

robimarko, 2023-10-27:
Thanks for bringing this up, I think this is a more generic issue with external LED controllers as I hit this on a completely different board with a different controller.

Lanchon, 2023-10-27:
@robimarko

thank you

the gpio leds are not affected. so my first guess would be that this might be a change in the driver(s) or the driver framework in the way they generate the led names. this might have been caused by a kernel update.

what platform/led driver is also affected?

Lanchon, 2023-11-01:
@Ansuel @dangowrt 

any idea why this is happening? i suppose this may have happened to many devices, i don't know if an isolated fix for this device is the best course of action.

egorenar, 2023-11-12:
this commit https://github.com/openwrt/openwrt/commit/36b1dd75fd9da1ea13f1ce1ee679c6b3c9c402cc fixed it for me

robimarko, 2023-11-12:
> @robimarko
> 
> thank you
> 
> the gpio leds are not affected. so my first guess would be that this might be a change in the driver(s) or the driver framework in the way they generate the led names. this might have been caused by a kernel update.
> 
> what platform/led driver is also affected?

Its hard to guess what caused it, or if it ever worked.

Lanchon, 2024-04-11:
> Issue is resolved with this commit: [e814acc](https://github.com/openwrt/openwrt/commit/e814acc59948943c776ac319a348f72c0d19f33c)

thanks! sorry, somehow i missed your message.
i can confirm this bug is fixed on current master.

closed 2024-04-11 by Lanchon