openwrt/openwrt #13804: Asus rt-ax53u led support for WAN, LAN1-3 leds (i need to turn off all)

XipCZ1, 2023-10-26:
### Describe the bug

please Can you add support for configuring leds on Asus rt-ax53u? 

### OpenWrt version

r23497-6637af95aa

### OpenWrt target/subtarget

ramips/mt7621

### Device

ASUS RT-AX53U

### Image kind

Official downloaded image

### Steps to reproduce

in luci i cannot configure theese leds

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

lukasz1992, 2023-10-27:
Does it work on OEM software? I do not see those leds in dts - probably they are controlled by switch hardware

rany2, 2023-10-27:
@lukasz1992 it should be possible to control it via GPIO, I'm somewhat sure this was done at some point but there had to be a good reason to remove that; I'm not privy to the reason though.

XipCZ1, 2023-10-27:
@rany2 i have read something in issues... like that (https://github.com/openwrt/openwrt/issues/11068)

now in 23.05, _(manageble: pwr, usb, wifi24, wifi5) (automatic blinks: wan, lan1, lan2, lan3)_, 

but i need to turn off all

rany2, 2023-10-27:
The WLAN and USB LED are already controllable in SNAPSHOT. So it's already done, it's just the switch ports that can't control right now.

brada4, 2023-10-27:
They program LED-s at gpio 480+18=498 and 480+26=506 in vendor firmware
https://openwrt.org/docs/techref/hardware/port.gpio#software
Expose all in the middle except those already used, try to read "value" and if it changes when cable is plugged it is either MII status port or LED port or speed indicator if it is many of them (LED port may not be readable and writing MII port may actually shut it off until poweroff)
GPIO assumed from uniform mount of LED padding, I might be wrong.

Low-tech piece of electricians insulation tape may save the day as well.

(follows from AX53U wiki page) 

```
[   43.035753] led_send_fw_cmd:
[   43.035848] 0205000000240000
[   43.035848] led_0 map to gpio_18...
[   43.035872] 
[   43.035872] led_send_fw_cmd:
[   43.035898] 0205010000340000
[   43.035898] led_1 map to gpio_26...
```

```
/sys/kernel/debug/gpio 
gpiochip2: GPIOs 416-447, parent: platform/1e000600.gpio, 1e000600.gpio-bank2:

gpiochip1: GPIOs 448-479, parent: platform/1e000600.gpio, 1e000600.gpio-bank1:

gpiochip0: GPIOs 480-511, parent: platform/1e000600.gpio, 1e000600.gpio-bank0:
 gpio-493 (                    |blue:power          ) out lo ACTIVE LOW
 gpio-494 (                    |blue:usb            ) out hi ACTIVE LOW
 gpio-495 (                    |wps                 ) in  hi IRQ ACTIVE LOW
 gpio-496 (                    |reset               ) in  hi IRQ ACTIVE LOW
 gpio-499 (                    |reset               ) out hi ACTIVE LOW
```

XipCZ1, 2023-12-02:
> Does it work on OEM software? I do not see those leds in dts - probably they are controlled by switch hardware

Yes, it work in OEM software, you can seee here:  (WPS button behavior = Turn LED off/on)
[image]

and it work good .. 

so i shoud be implemetable

brada4, 2023-12-02:
which setting of all?

brada4, 2023-12-03:
If you SSH to device - do you have gpio debug file there in OEM firmware? Might give some hint which gpio turns off all LED block.

XipCZ1, 2023-12-03:
> If you SSH to device - do you have gpio debug file there in OEM firmware? Might give some hint which gpio turns off all LED block.

where it shoud be please?

XipCZ1, 2023-12-03:
`admin@RT-AX53U-CBD8:/# cat ./sys/kernel/debug/gpio
GPIOs 0-31, platform/1e000600.gpio, 1e000600.gpio:
 gpio-13  (                    |sysfs               ) out lo
 gpio-14  (                    |sysfs               ) out hi
 gpio-15  (                    |sysfs               ) in  hi
 gpio-16  (                    |sysfs               ) in  hi
 gpio-19  (                    |                    ) out hi

GPIOs 32-63, platform/1e000600.gpio, 1e000600.gpio:

GPIOs 64-95, platform/1e000600.gpio, 1e000600.gpio:
`

`admin@RT-AX53U-CBD8:/# find | grep gpio
./dev/gpio0
./sys/bus/platform/devices/gpio-keys-polled
./sys/bus/platform/devices/1e000600.gpio
./sys/bus/platform/drivers/mt7621_gpio
./sys/bus/platform/drivers/mt7621_gpio/bind
./sys/bus/platform/drivers/mt7621_gpio/uevent
./sys/bus/platform/drivers/mt7621_gpio/unbind
./sys/bus/platform/drivers/mt7621_gpio/1e000600.gpio
./sys/bus/platform/drivers/poweroff-gpio
./sys/bus/platform/drivers/poweroff-gpio/bind
./sys/bus/platform/drivers/poweroff-gpio/uevent
./sys/bus/platform/drivers/poweroff-gpio/unbind
./sys/devices/platform/1e000000.palmbus/1e000600.gpio
./sys/devices/platform/1e000000.palmbus/1e000600.gpio/gpio
./sys/devices/platform/1e000000.palmbus/1e000600.gpio/gpio/gpiochip32
./sys/devices/platform/1e000000.palmbus/1e000600.gpio/gpio/gpiochip32/base
./sys/devices/platform/1e000000.palmbus/1e000600.gpio/gpio/gpiochip32/label
./sys/devices/platform/1e000000.palmbus/1e000600.gpio/gpio/gpiochip32/ngpio
./sys/devices/platform/1e000000.palmbus/1e000600.gpio/gpio/gpiochip32/device
./sys/devices/platform/1e000000.palmbus/1e000600.gpio/gpio/gpiochip32/subsystem
./sys/devices/platform/1e000000.palmbus/1e000600.gpio/gpio/gpiochip32/uevent
./sys/devices/platform/1e000000.palmbus/1e000600.gpio/gpio/gpiochip64
./sys/devices/platform/1e000000.palmbus/1e000600.gpio/gpio/gpiochip64/base
./sys/devices/platform/1e000000.palmbus/1e000600.gpio/gpio/gpiochip64/label
./sys/devices/platform/1e000000.palmbus/1e000600.gpio/gpio/gpiochip64/ngpio
./sys/devices/platform/1e000000.palmbus/1e000600.gpio/gpio/gpiochip64/device
./sys/devices/platform/1e000000.palmbus/1e000600.gpio/gpio/gpiochip64/subsystem
./sys/devices/platform/1e000000.palmbus/1e000600.gpio/gpio/gpiochip64/uevent
./sys/devices/platform/1e000000.palmbus/1e000600.gpio/gpio/gpio13
./sys/devices/platform/1e000000.palmbus/1e000600.gpio/gpio/gpio13/edge
./sys/devices/platform/1e000000.palmbus/1e000600.gpio/gpio/gpio13/value
./sys/devices/platform/1e000000.palmbus/1e000600.gpio/gpio/gpio13/active_low
./sys/devices/platform/1e000000.palmbus/1e000600.gpio/gpio/gpio13/device
./sys/devices/platform/1e000000.palmbus/1e000600.gpio/gpio/gpio13/subsystem
./sys/devices/platform/1e000000.palmbus/1e000600.gpio/gpio/gpio13/uevent
./sys/devices/platform/1e000000.palmbus/1e000600.gpio/gpio/gpio13/direction
./sys/devices/platform/1e000000.palmbus/1e000600.gpio/gpio/gpio14
./sys/devices/platform/1e000000.palmbus/1e000600.gpio/gpio/gpio14/edge
./sys/devices/platform/1e000000.palmbus/1e000600.gpio/gpio/gpio14/value
./sys/devices/platform/1e000000.palmbus/1e000600.gpio/gpio/gpio14/active_low
./sys/devices/platform/1e000000.palmbus/1e000600.gpio/gpio/gpio14/device
./sys/devices/platform/1e000000.palmbus/1e000600.gpio/gpio/gpio14/subsystem
./sys/devices/platform/1e000000.palmbus/1e000600.gpio/gpio/gpio14/uevent
./sys/devices/platform/1e000000.palmbus/1e000600.gpio/gpio/gpio14/direction
./sys/devices/platform/1e000000.palmbus/1e000600.gpio/gpio/gpio15
./sys/devices/platform/1e000000.palmbus/1e000600.gpio/gpio/gpio15/edge
./sys/devices/platform/1e000000.palmbus/1e000600.gpio/gpio/gpio15/value
./sys/devices/platform/1e000000.palmbus/1e000600.gpio/gpio/gpio15/active_low
./sys/devices/platform/1e000000.palmbus/1e000600.gpio/gpio/gpio15/device
./sys/devices/platform/1e000000.palmbus/1e000600.gpio/gpio/gpio15/subsystem
./sys/devices/platform/1e000000.palmbus/1e000600.gpio/gpio/gpio15/uevent
./sys/devices/platform/1e000000.palmbus/1e000600.gpio/gpio/gpio15/direction
./sys/devices/platform/1e000000.palmbus/1e000600.gpio/gpio/gpio16
./sys/devices/platform/1e000000.palmbus/1e000600.gpio/gpio/gpio16/edge
./sys/devices/platform/1e000000.palmbus/1e000600.gpio/gpio/gpio16/value
./sys/devices/platform/1e000000.palmbus/1e000600.gpio/gpio/gpio16/active_low
[... 57 lines cut ...]
./sys/firmware/devicetree/base/palmbus@1e000000/gpio@600/bank@0/compatible
./sys/firmware/devicetree/base/palmbus@1e000000/gpio@600/bank@0/phandle
./sys/firmware/devicetree/base/palmbus@1e000000/gpio@600/bank@0/gpio-controller
./sys/firmware/devicetree/base/palmbus@1e000000/gpio@600/bank@0/linux,phandle
./sys/firmware/devicetree/base/palmbus@1e000000/gpio@600/bank@1
./sys/firmware/devicetree/base/palmbus@1e000000/gpio@600/bank@1/reg
./sys/firmware/devicetree/base/palmbus@1e000000/gpio@600/bank@1/name
./sys/firmware/devicetree/base/palmbus@1e000000/gpio@600/bank@1/#gpio-cells
./sys/firmware/devicetree/base/palmbus@1e000000/gpio@600/bank@1/compatible
./sys/firmware/devicetree/base/palmbus@1e000000/gpio@600/bank@1/gpio-controller
./sys/firmware/devicetree/base/palmbus@1e000000/gpio@600/bank@2
./sys/firmware/devicetree/base/palmbus@1e000000/gpio@600/bank@2/reg
./sys/firmware/devicetree/base/palmbus@1e000000/gpio@600/bank@2/name
./sys/firmware/devicetree/base/palmbus@1e000000/gpio@600/bank@2/#gpio-cells
./sys/firmware/devicetree/base/palmbus@1e000000/gpio@600/bank@2/compatible
./sys/firmware/devicetree/base/palmbus@1e000000/gpio@600/bank@2/gpio-controller
./sys/firmware/devicetree/base/palmbus@1e000000/gpio@600/#address-cells
./sys/firmware/devicetree/base/palmbus@1e000000/gpio@600/#size-cells
./sys/firmware/devicetree/base/palmbus@1e000000/gpio@600/interrupt-parent
./sys/firmware/devicetree/base/pinctrl/pinctrl0/gpio
./sys/firmware/devicetree/base/pinctrl/pinctrl0/gpio/name
./sys/firmware/devicetree/base/pinctrl/pinctrl0/gpio/mtk,function
./sys/firmware/devicetree/base/pinctrl/pinctrl0/gpio/mtk,group
./sys/kernel/debug/gpio
./sys/kernel/debug/pinctrl/pinctrl/gpio-ranges`

brada4, 2023-12-03:
It is just same signals handled :-(

lukasz1992, 2023-12-04:
> Yes, it work in OEM software, you can seee here: (WPS button behavior = Turn LED off/on) [...]

Please login to router on OEM software and run the command:
`nvram show | grep ^led | sort`

maxozerov, 2023-12-26:
> Please login to router on OEM software and run the command:
> nvram show | grep ^led | sort

It may be useful here, just in case I will give the 'output of this command':
- OEM software (With LED Off) (WPS button behavior = Turn LED off/on)
```bash
size: 38184 bytes (23256 left)
led_2g_gpio=255
led_5g_gpio=255
led_all_gpio=255
led_ctrl_cap=4
led_lan1_gpio=255
led_lan2_gpio=255
led_lan3_gpio=255
led_lan4_gpio=255
led_lan_gpio=255
led_logo_gpio=255
led_pwr_gpio=4109
led_usb3_gpio=255
led_usb_gpio=4110
led_val=0
led_wan_gpio=255
led_wps_gpio=4109
```
- OEM software (With LED On) (WPS button behavior = Turn LED off/on)
```bash
size: 38186 bytes (23254 left)
led_2g_gpio=255
led_5g_gpio=255
led_all_gpio=255
led_ctrl_cap=4
led_lan1_gpio=255
led_lan2_gpio=255
led_lan3_gpio=255
led_lan4_gpio=255
led_lan_gpio=255
led_logo_gpio=255
led_pwr_gpio=4109
led_usb3_gpio=255
led_usb_gpio=4110
led_val=255
led_wan_gpio=255
led_wps_gpio=4109
```
- OEM software LED On (switching from Off to On) (logs via dmesg)
```
[263161.995527]  Set_Led_Proc ==> arg = 00-00-00-00-02-00-00-00
[263162.003941] 
[263162.003941] Set_Led_Proc
[263162.009519] 00
[263162.012654] 00
[263162.015462] 00
[263162.017648] 00
[263162.019476] 02
[263162.021251] 00
[263162.023102] 00
[263162.024926] 00
[263162.042330] 
[263162.042330]  Set_Led_Proc ==> arg = 01-00-00-00-02-00-00-00
[263162.049610] 
[263162.049610] Set_Led_Proc
[263162.053782] 01
[263162.055520] 00
[263162.057330] 00
[263162.059108] 00
[263162.060866] 02
[263162.062615] 00
[263162.064362] 00
[263162.066322] 00
```

fertil20, 2024-01-11:
Bump
I got the same thing, can't switch LED's to turn off

brada4, 2024-01-11:
led_val=255 -> 0
Set_Led_Proc ==> arg = 01-00-00-00-02-00-00-00
Set_Led_Proc ==> arg = 00-00-00-00-02-00-00-00

leds 13-16 seem mapped to 493-496 and 19 is a button
either there is some gpio at least 8 steps off the existing gpio control  that turns off whole led block or that is some driver parameter that masks all LEDs with its override value. There is wildly similar AX54 (no U) that has 5 ports, maybe there are 4-5 continuous gpios for individual leds. Who wants to risk zeroing them, wrong one will power off USB or wifi or lock up CPU etc.

lukasz1992, 2024-01-11:
Set_Led_Proc is only from the wifi driver, something else must handle it.

fertil20, 2024-03-08:
So, what's the issue status? I still not able to turn off WAN and LAN leds, so I have very annoying blinks on 4 leds

rozhuk-im, 2024-03-28:
https://github.com/rozhuk-im/openwrt/commit/de960ce92d042b1624f597d8de5b408e3025ced9

I done it few years ago, but does not do PR.

brada4, 2024-03-28:
Seems never made into openwrt

rozhuk-im, 2024-03-28:
http://www.netlab.linkpc.net/download/software/openwrt/ASUS%20RT-AX53U/20240328/

hasselh, 2024-05-02:
Same issue here. Blue WAN + LAN leds are blinking annoying (especially during night time). Looks like @rozhuk-im found a solution. 👍 What is required to get it into main branch ? 🙏

jsapede, 2024-05-24:
thought i was alone with this problem !
is there a way to include this in openwrt release ?

Azuriye, 2024-07-06:
Any idea how to get rozhuk-im's changes into main branch? really useful feature.

jsapede, 2024-07-07:
Yes wold je awesome

martinbarez, 2024-07-28:
[This comment gives a good explanation about why those changes were not merged.](https://github.com/freifunk-gluon/gluon/issues/2756#issuecomment-1416950822) TLDR: if you register them as configurable, either they don't light up in some cases or performance drops.  
I've tried the hacky way they mention through mdio and stumbled into [this](https://forum.openwrt.org/t/getting-back-full-led-control-on-e7350/173189/13). TLDR: you can access the LED registers on similar switches, but not on the one used by RT-AX53U.

So it's either your own kernel patch or nothing.

brada4, 2024-07-28:
Link up down  indication should and/or wor just fine.

hasselh, 2024-07-28:
Link (up/down) indicator is how rt-ax53u OpenWRT image is currently set up. But no possibility to switch these bright LEDs off - like in factory image.. Neither via /sys/class/gpio nor via mdio tool (I spent a few hrs but did not succeed) - so I am still using ducktape instead.. :((

hasselh, 2024-08-27:
> [rozhuk-im@32a123c](https://github.com/rozhuk-im/openwrt/commit/32a123cabdeb7b9fdaba6f577374d4e9deaf0b23)
> I done it few years ago.
> 
> PR was: #10400

@rozhuk-im , can you please help us, what is required to get respective GPIO to the main branch of Asus rt-ax53u, allowing us to turn off these LEDs e.g. via /sys/class/gpio or mdio tool ? 🙏

hasselh, 2026-01-09:
OK I found a fix for Netgear WAX206 LEDs, that is also working for Asus RT-AX53U. And yes, I was brave (desperate ;o) enough to test it... and I can confirm that it is working on my Asus RT-AX53U running OpenWrt 24.10.5, r29087-d9c5716d1d

> opkg update; opkg install mdio-tools
> mdio mt7530-0 0:0x1f raw 0x21 0x0008

Source: #https://github.com/openwrt/openwrt/pull/11363#issuecomment-1379660930
As there is now a proven way to fix it.. Is there any way to get this fix into the main branch ?