openwrt/openwrt #13805: ip route get fails with ipv6

axet, 2023-10-27:
### Describe the bug

```
# ip route get 2001:470:1:c84::20
ip: RTNETLINK answers: Network unreachable

# ping 2001:470:1:c84::20
PING 2001:470:1:c84::20 (2001:470:1:c84::20): 56 data bytes
64 bytes from 2001:470:1:c84::20: seq=0 ttl=48 time=186.757 ms

# ip route get ::1
local ::1 from :: dev lo table local  src ::1  metric 0
```

### OpenWrt version

23.05.0

### OpenWrt target/subtarget

ath79/generic

### Device

TP-Link TL-WR842N/ND v2

### Image kind

Self-built image

### Expected behaviour

get proper route info

### Diffconfig

```text
CONFIG_TARGET_ath79=y
CONFIG_TARGET_ath79_generic=y
CONFIG_TARGET_ath79_generic_DEVICE_tplink_tl-wr842n-v2=y
CONFIG_DEVEL=y
# CONFIG_KERNEL_DEBUG_INFO is not set
# CONFIG_KERNEL_ELF_CORE is not set
# CONFIG_KERNEL_KALLSYMS is not set
# CONFIG_KERNEL_MAGIC_SYSRQ is not set
# CONFIG_KERNEL_PRINTK is not set
# CONFIG_KERNEL_PRINTK_TIME is not set
CONFIG_PACKAGE_6in4=y
# CONFIG_PACKAGE_MAC80211_DEBUGFS is not set
# CONFIG_PACKAGE_MAC80211_MESH is not set
CONFIG_PACKAGE_avahi-nodbus-daemon=y
CONFIG_PACKAGE_coreutils=y
CONFIG_PACKAGE_kmod-crypto-kpp=y
CONFIG_PACKAGE_kmod-crypto-lib-chacha20=y
CONFIG_PACKAGE_kmod-crypto-lib-chacha20poly1305=y
CONFIG_PACKAGE_kmod-crypto-lib-curve25519=y
CONFIG_PACKAGE_kmod-crypto-lib-poly1305=y
CONFIG_PACKAGE_kmod-ip6tables=y
CONFIG_PACKAGE_kmod-ipt-conntrack=y
CONFIG_PACKAGE_kmod-ipt-core=y
CONFIG_PACKAGE_kmod-ipt-nat=y
CONFIG_PACKAGE_kmod-ipt-nat6=y
CONFIG_PACKAGE_kmod-iptunnel=y
CONFIG_PACKAGE_kmod-iptunnel4=y
# CONFIG_PACKAGE_kmod-lib-crc-ccitt is not set
CONFIG_PACKAGE_kmod-libphy=y
CONFIG_PACKAGE_kmod-nf-ipt=y
CONFIG_PACKAGE_kmod-nf-ipt6=y
CONFIG_PACKAGE_kmod-nf-nat6=y
# CONFIG_PACKAGE_kmod-ppp is not set
CONFIG_PACKAGE_kmod-sit=y
CONFIG_PACKAGE_kmod-swconfig=y
CONFIG_PACKAGE_kmod-switch-ip17xx=y
CONFIG_PACKAGE_kmod-switch-rtl8366-smi=y
CONFIG_PACKAGE_kmod-switch-rtl8366s=y
CONFIG_PACKAGE_kmod-udptunnel4=y
CONFIG_PACKAGE_kmod-udptunnel6=y
CONFIG_PACKAGE_kmod-usb-printer=y
CONFIG_PACKAGE_kmod-wireguard=y
CONFIG_PACKAGE_kmod-zram=y
CONFIG_PACKAGE_libavahi-nodbus-support=y
CONFIG_PACKAGE_libcap-ng=y
CONFIG_PACKAGE_libdaemon=y
CONFIG_PACKAGE_libexpat=y
CONFIG_PACKAGE_libncurses=y
CONFIG_PACKAGE_libpcap=y
CONFIG_PACKAGE_libpopt=y
CONFIG_PACKAGE_librt=y
CONFIG_PACKAGE_libuuid=y
CONFIG_PACKAGE_miniupnpd-nftables=y
CONFIG_PACKAGE_muninlite=y
# CONFIG_PACKAGE_opkg is not set
CONFIG_PACKAGE_p910nd=y
# CONFIG_PACKAGE_ppp is not set
CONFIG_PACKAGE_rsync=y
CONFIG_PACKAGE_sexpect=y
CONFIG_PACKAGE_socat=y
CONFIG_PACKAGE_tcpdump=y
CONFIG_PACKAGE_telnet-bsd=y
CONFIG_PACKAGE_terminfo=y
# CONFIG_PACKAGE_ucode-mod-nl80211 is not set
# CONFIG_PACKAGE_ucode-mod-rtnl is not set
# CONFIG_PACKAGE_ucode-mod-uloop is not set
CONFIG_PACKAGE_wireguard-tools=y
# CONFIG_PACKAGE_wpad-basic-mbedtls is not set
CONFIG_PACKAGE_xinetd=y
CONFIG_PACKAGE_zlib=y
CONFIG_PACKAGE_zram-swap=y
CONFIG_STRIP_KERNEL_EXPORTS=y
CONFIG_USE_MKLIBS=y
```
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

danpawlik, 2023-10-27:
Is it not similar to: https://github.com/openwrt/openwrt/issues/13785 ?

axet, 2023-10-27:
It is closed. I left comment at issues you mentioned. 'ip route' suppose to tell which dev / ip will be used by kernel to route to specified address. And this functionality is broken for openwrt. Keyboard 'from' it is a different 'route' functionally.

jow-, 2023-11-01:
IPv6 is multi homed by nature, there can be multiple downstreams and multiple upstreams. In order to ensure that the correct upstream is used for the correct downstream network, default routes are scoped by subnets (`from xxx`). In order to query the default route for IPv6 on OpenWrt you need to specify a `from` selector to give the kernel a hint for which source address you want to query the route.

This is the normal expected behavior and not a bug. Duplicate of #13785

closed 2023-11-01 by jow-