openwrt/openwrt #13815: netifd: WiFi interface initialization race condition on boot

Vavooon, 2023-10-28:
### Describe the bug

Recently I've faced an issue with built-in WiFi module initialization on Raspberry Pi on boot resulting in no WiFi connectivity:

You can see from the log that netifd WiFi initialization happens before BCM4345 firmware is fully loaded and ready.
```
Thu Oct 26 19:28:13 2023 kern.info kernel: [    5.813626] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
Thu Oct 26 19:28:13 2023 kern.info kernel: [    5.813876] usbcore: registered new interface driver brcmfmac
Thu Oct 26 19:28:13 2023 kern.warn kernel: [    5.818033] brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43455-sdio.raspberrypi,4-model-b.bin failed with error -2
Thu Oct 26 19:28:13 2023 kern.warn kernel: [    5.818051] brcmfmac mmc1:0001:1: Falling back to sysfs fallback for: brcm/brcmfmac43455-sdio.raspberrypi,4-model-b.bin
Thu Oct 26 19:28:14 2023 daemon.notice netifd: radio0 (2192): WARNING: Variable 'data' does not exist or is not an array/object
Thu Oct 26 19:28:14 2023 daemon.notice netifd: radio0 (2192): Phy not found
Thu Oct 26 19:28:14 2023 daemon.notice netifd: radio0 (2192): Could not find PHY for device 'radio0'
Thu Oct 26 19:28:14 2023 daemon.notice netifd: Wireless device 'radio0' set retry=0
Thu Oct 26 19:28:14 2023 daemon.crit netifd: Wireless device 'radio0' setup failed, retry=0
Thu Oct 26 19:28:14 2023 daemon.notice netifd: radio0 (2345): WARNING: Variable 'data' does not exist or is not an array/object
Thu Oct 26 19:28:14 2023 daemon.notice netifd: radio0 (2345): Bug: PHY is undefined for device 'radio0'
Thu Oct 26 19:28:14 2023 daemon.notice netifd: Wireless device 'radio0' is now down
Thu Oct 26 19:28:15 2023 daemon.err procd: Could not find firmware /lib/firmware/brcm/brcmfmac43455-sdio.raspberrypi,4-model-b.bin: No such file or directory
Thu Oct 26 19:28:15 2023 daemon.err procd: Could not find firmware /lib/firmware/brcm/brcmfmac43455-sdio.raspberrypi,4-model-b.clm_blob: No such file or directory
Thu Oct 26 19:28:15 2023 kern.info kernel: [    8.406330] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Jul 29 2022 02:15:20 version 7.45.250 (70e9766 CY) FWID 01-e53e306b
Thu Oct 26 19:28:15 2023 daemon.info ModemManager[3204]: hotplug: add network interface wlan0: event processed
```

I investigated the issue and have found that it happens to me only when I have a LTE modem plugged in on boot. For example, here are logs with no modem and with a modem plugged in respectively:

```[    0.090337] usbcore: registered new interface driver usbfs
[    0.090386] usbcore: registered new interface driver hub
[    0.090437] usbcore: registered new device driver usb
[    0.090577] usb_phy_generic phy: supply vcc not found, using dummy regulator
[    0.090706] usb_phy_generic phy: dummy supplies not allowed for exclusive requests
[    0.472733] usbcore: registered new interface driver lan78xx
[    0.472802] usbcore: registered new interface driver smsc95xx
[    0.524068] usb usb1: New USB device found, idVendor=1d6b, idProduct=0002, bcdDevice= 5.15
[    0.524099] usb usb1: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    0.524123] usb usb1: Product: xHCI Host Controller
[    0.524141] usb usb1: Manufacturer: Linux 5.15.134 xhci-hcd
[    0.524159] usb usb1: SerialNumber: 0000:01:00.0
[    0.525095] usb usb2: New USB device found, idVendor=1d6b, idProduct=0003, bcdDevice= 5.15
[    0.525127] usb usb2: New USB device strings: Mfr=3, Product=2, SerialNumber=1
[    0.525152] usb usb2: Product: xHCI Host Controller
[    0.525170] usb usb2: Manufacturer: Linux 5.15.134 xhci-hcd
[    0.525187] usb usb2: SerialNumber: 0000:01:00.0
[    0.526949] usbcore: registered new interface driver uas
[    0.527028] usbcore: registered new interface driver usb-storage
[    0.821857] usb 1-1: new high-speed USB device number 2 using xhci_hcd
[    1.004433] usb 1-1: New USB device found, idVendor=2109, idProduct=3431, bcdDevice= 4.21
[    1.004458] usb 1-1: New USB device strings: Mfr=0, Product=1, SerialNumber=0
[    1.004467] usb 1-1: Product: USB2.0 Hub
[    5.804651] usbcore: registered new interface driver cdc_acm
[    5.805069] usbcore: registered new interface driver cdc_ether
[    5.805729] usbcore: registered new interface driver cdc_ncm
[    5.806318] usbcore: registered new interface driver cdc_wdm
[    5.807690] usbcore: registered new interface driver qmi_wwan
[    5.808925] usbcore: registered new interface driver usbhid
[    5.808943] usbhid: USB HID core driver
[    5.809803] usbcore: registered new interface driver usbserial_generic
[    5.809834] usbserial: USB Serial support registered for generic
[    5.811738] usbcore: registered new interface driver uvcvideo
[    5.812239] usbcore: registered new interface driver cdc_mbim
[    5.819420] usbcore: registered new interface driver cp210x
[    5.819453] usbserial: USB Serial support registered for cp210x
[    5.843699] brcmfmac: brcmf_fw_alloc_request: using brcm/brcmfmac43455-sdio for chip BCM4345/6
[    5.843874] brcmfmac mmc1:0001:1: Direct firmware load for brcm/brcmfmac43455-sdio.raspberrypi,4-model-b.bin failed with error -2
[    5.843892] brcmfmac mmc1:0001:1: Falling back to sysfs fallback for: brcm/brcmfmac43455-sdio.raspberrypi,4-model-b.bin
[    5.844039] usbcore: registered new interface driver brcmfmac
[    5.844500] usbcore: registered new interface driver option
[    5.844528] usbserial: USB Serial support registered for GSM modem (1-port)
[    5.846682] usbcore: registered new interface driver rtl8192cu
[    6.945386] brcmfmac: brcmf_c_preinit_dcmds: Firmware: BCM4345/6 wl0: Jul 29 2022 02:15:20 version 7.45.250 (70e9766 CY) FWID 01-e53e306b
[... 88 lines cut ...]
kmod-mii
kmod-usb-net
kmod-usb-wdm
kmod-usb-net-qmi-wwan
kmod-usb-net-cdc-mbim
kmod-usb-serial-option
kmod-usb-serial
kmod-usb-serial-wwan
luci-proto-modemmanager
```

2. Attach SIMCOM SIM7600G-H modem to USB.
3. Boot the system.

### Actual behaviour

Built-in WiFi interface fails to initialize, resulting in no connectivity until the interface is manually restarted from CLI/Web interface.

### Expected behaviour

WiFi interface should be available after boot.

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

dr-ni, 2023-10-29:
see also https://github.com/openwrt/openwrt/pull/13781

ShadowFlare, 2024-01-30:
I've seen the same issue on boot with the same set of netifd log messages about not being able to find the PHY for radioX. In my case my version is currently r24895-3cf1fe5508 (built myself from source) but I also saw the issue on 23.05.2 with an official image, on x86_64 with 4 cards with the QCN9074 chip supported by ath11k. The ones that are successfully initialized on boot are consistently in the same order (by PCI ID or something like that), though the number of successes varies between 2 and 3. Like the original poster, it works after the interface is manually restarted after boot.

thiakil, 2024-07-08:
Similar here with Qemu VM + pcie passthrough - 2 wifi cards (one ath9k, one brcmfmac), the one that takes longer to init always fails to start on boot (or trigger events that start it). Gets marked as `retry=0`

https://gist.github.com/thiakil/78dd51a29aa976a781dcc29764c36978