openwrt/openwrt #13822: [ipq40xx] FB 7520/7530 bricks on v22.03

maurerle, 2023-10-29:
### Describe the bug

Our Freifunk Community had the issue, that all 7520/7530 devices were bricked when updating to gluon v2023.1.1
See here: https://github.com/freifunk-gluon/gluon/issues/3023

I then tracked down the issue in gluon to these commits:
https://github.com/openwrt/openwrt/compare/1ec274a204a65c7f8543cf750a0b027a685818cb...419218af13b4c036b3f27e450855727005b8d141

I then did build openwrt from:
https://github.com/openwrt/openwrt/commit/1ec274a204a65c7f8543cf750a0b027a685818cb (works)
and
https://github.com/openwrt/openwrt/commit/419218af13b4c036b3f27e450855727005b8d141 (failes)

I could also reproduce this behavior on latest https://github.com/openwrt/openwrt/tree/openwrt-22.03
c2921044e75fb05d666d76850cb922f8f67c1784

This does not occur on v23.05

Unfortunately, I did not test the latest official builds.

### OpenWrt version

419218af13

### OpenWrt target/subtarget

ipq40xx/generic

### Device

AVM FRITZ!Box 7530

### Image kind

Self-built image

### Steps to reproduce

Flash image using `sysupgrade`. Device does not boot properly.

### Actual behaviour

@Djfe did check the Serial output, which is basically stuck on:
```
Read 1024 bytes from 02bd800
maca pointer invalid: a5472102
Using machid 0x8010001 from environment
 
Starting kernel ...
```
https://pastebin.com/0NiNpjqr

I did not check this every time I bricked and unbricked my device - but the behavior is the same

### Expected behaviour

Device should boot properly

### Additional info

I did not find anything on the forum which I found very weird - as it would state that no other member is using this device to test the latest v22.03 image..?

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

robimarko, 2023-10-29:
Its hard to guess without `earlyprintk` being enabled so we can see if kernel is even starting.

Djfe, 2023-10-29:
We are going to test that tomorrow evening.
A guess on IRC so far is, that maybe the DT is at a different address now. so it fails to load it.

robimarko, 2023-10-29:
That shouldn't matter as U-Boot is responsible to put the DTB at 64bit aligned memory address.
As far as I can see address of DTB is 64 bit aligned, that was an issue on some other boards.

maurerle, 2023-12-18:
This issue does not appear on the latest 22.03.6:
https://firmware-selector.openwrt.org/#?version=22.03.6&target=ipq40xx%2Fgeneric&id=avm_fritzbox-7530

So whatever this was, it can probably be closed.

closed 2023-12-18 by maurerle