openwrt/openwrt #13829: Asus RT-N13U ver.B1 23.05 request

alicaccs, 2023-10-30:
### Describe the bug

RT-N13U had 2 versions of hardware: the original with 4MB ROM and 32MB RAM, and later B1 revision with 8MB ROM and 64MB RAM. https://openwrt.org/toh/asus/rt-n13u
The B1 should be able to run 23.05 like other devices of similar configuration (eg. Fonera 2.0N), but its image compilation was disabled along with the original version of RT-N13U. Please allow RT-N13U ver.B1 to upgrade to 23.05.

### OpenWrt version

23.05

### OpenWrt target/subtarget

ramips/rt305x

### Device

RT-N13U

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

BartOrd, 2024-06-09:
use this image to flash by ASUS Firmware Restoration(worked for me)

[https://archive.openwrt.org/releases/23.05.0/targets/ramips/rt305x/openwrt-23.05.0-ramips-rt305x-poray_ip2202-squashfs-sysupgrade.bin](https://archive.openwrt.org/releases/23.05.0/targets/ramips/rt305x/openwrt-23.05.0-ramips-rt305x-poray_ip2202-squashfs-sysupgrade.bin)

BartOrd, 2024-06-09:
ps. if it's not work create backup for version from forum and restore to new version

RomanV0703, 2024-10-18:
Worked for me, thanks for the tips!
More details in case somebody else is interested:
1. Put the router into "Rescue Mode":
https://www.asus.com/support/faq/1000814/#a1
1.1. Follow the instructions (make sure to change the IP to 192.168.1.10)
3. Download and use "ASUS Firmware Restoration" tool:
https://dlcdnets.asus.com/pub/ASUS/wireless/4G-AC53U/Rescue_2102.zip?model=RT-N13U%20(VER.B1)
5. Flash Poray firmware:
https://archive.openwrt.org/releases/23.05.5/targets/ramips/rt305x/openwrt-23.05.5-ramips-rt305x-poray_ip2202-squashfs-sysupgrade.bin
(I used version 23.05.5, latest at the time).