openwrt/openwrt #13830: pcie_mhi not work. Module has pre-installed into router before building firmware.

ChaingTsung, 2023-10-30:
### Describe the bug

I used the https://github.com/ChaingTsung/Quectel_QConnector and https://github.com/ChaingTsung/quectel_mhi_rm50x to build driver for Quectel RM50X 5G module. After the firmware built successfully, it did not  work! But the driver source is offered by the Quectel official FAE and all of them has passwd on Ubuntu20.04 x64 with kernel 5.15, Ubuntu22.04, Ubuntu 18.04, ArchLinux with kernel 6.4.x.(Just clone these code and use the makefile in dir `src` to nake.)

Of course, these packages also passed on openwrt with branch `lede`, it worked fine on LEDE.(https://github.com/coolsnowwolf/lede)

I wonder why it can not work on the official openwrt code.

### OpenWrt version

`main` branch

### OpenWrt target/subtarget

lastest

### Device

banana pi r3

### Image kind

Self-built image

### Steps to reproduce

1.clone code from `main` branch
2.cd `openwrt /package` dir and clone some custom packages (https://github.com/kenzok8/openwrt-packages and https://github.com/kenzok8/small)
3.clone quectel rm50x driver and dail tools code.(https://github.com/ChaingTsung/quectel_mhi_rm50x and https://github.com/ChaingTsung/Quectel_QConnector)
4. run `./scripts/feeds update -a && ./scripts/feeds install` in openwrt root dir
5. run ` make menuconfig` in order to choose packages I need
6. run `make download -j8` and `make -j4 || make -j1 V=s`
7. flash the built firmware into sdcard and install sdcard into router, and then boot router
 

### Actual behaviour

After Router boot, some info are in here.
> boot info
```log
...
[   16.000926] kmodloader: loading kernel modules from /tmp/overlay/upper/etc/modules-boot.d/*
[   16.010262] kmodloader: done loading kernel modules from /tmp/overlay/upper/etc/modules-boot.d/*
[   16.652163] UBIFS (ubi0:2): Mounting in unauthenticated mode
[   16.657924] UBIFS (ubi0:2): background thread "ubifs_bgt0_2" started, PID 784
[   17.128576] UBIFS (ubi0:2): UBIFS: mounted UBI device 0, volume 2, name "rootfs_data"
[   17.136402] UBIFS (ubi0:2): LEB size: 126976 bytes (124 KiB), min./max. I/O unit sizes: 2048 bytes/2048 bytes
[   17.146297] UBIFS (ubi0:2): FS size: 27680768 bytes (26 MiB, 218 LEBs), max 228 LEBs, journal size 1396736 bytes (1 MiB, 11 LEBs)
[   17.157926] UBIFS (ubi0:2): reserved for root: 1307430 bytes (1276 KiB)
[   17.164524] UBIFS (ubi0:2): media format: w5/r0 (latest is w5/r0), UUID 52931B17-F339-4DF1-B562-E22CBC6CB8AC, small LPT model
[   17.187120] block: attempting to load /tmp/ubifs_cfg/upper/etc/config/fstab
[   17.229573] block: extroot: not configured
[   17.244860] UBIFS (ubi0:2): un-mount UBI device 0
[   17.249564] UBIFS (ubi0:2): background thread "ubifs_bgt0_2" stops
[   17.261960] mount_root: switching to f2fs overlay
[   17.271913] overlayfs: null uuid detected in lower fs '/', falling back to xino=off,index=off,nfs_export=off.
[   17.283941] urandom-seed: Seeding with /etc/urandom.seed
[   17.880289] mtdblock: MTD device 'reserved' is NAND, please consider using UBI block devices instead.
[   18.320907] mtdblock: MTD device 'ubi' is NAND, please consider using UBI block devices instead.
[   19.044089] mtdblock: MTD device 'bl2' is NAND, please consider using UBI block devices instead.
[   19.217329] mtdblock: MTD device 'fip' is NAND, please consider using UBI block devices instead.
[   19.656724] procd: - early -
[   19.659677] procd: - watchdog -
[   20.216184] procd: - watchdog -
[   20.220479] procd: - ubus -
[   20.282239] procd: - init -
[   21.006523] urngd: v1.0.2 started.
[   21.042092] kmodloader: loading kernel modules from /etc/modules.d/*
[   21.062910] trusted_key: encrypted_key: failed to alloc_cipher (-2)
[   21.199581] NET: Registered PF_ALG protocol family
[... 399 lines cut ...]
                L1SubCtl1: PCI-PM_L1.2- PCI-PM_L1.1- ASPM_L1.2- ASPM_L1.1-
                           T_CommonMode=0us LTR1.2_Threshold=98304ns
                L1SubCtl2: T_PwrOn=52us
        Capabilities: [240 v1] Data Link Feature <?>
        Kernel driver in use: pcie_mhi
```

### Additional info

# MHI virtual channel function description

| Number |  mhi descriptor | function |
|:-----: | :-----:|:-----: |
| 1 | mhi_BHI | Port for downloading firehose file |
| 2 | mhi_DUN | Port for AT commands |
| 3 | mhi_LOOPBACK | Port for testing loopback |
| 4 | mhi_DIAG | Port for catching up QXDM log |
| 5 | mhi_QMI0 | Port for transporting QMI message |
| 6 |  mhi_QMI1 | Reserved |
| 7 |  mhi_GNSS | Port for getting & uploading GPS NEMA data |
| 8 |  mhi_EDL | Port for firmware  upgrade ( starting to enum when the module is in edl mode) |

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

ChaingTsung, 2023-10-30:
[Here is diffconfig](https://github.com/openwrt/openwrt/files/13205573/seeds.txt)
And the custom packages are not conflict with the driver & dial tools and do not Influence the driver & dial tools.

robimarko, 2023-10-30:
OpenWrt has nothing to do with those Quectel drivers, there are mainline versions to select in OpenWrt.

ChaingTsung, 2023-10-31:
But it is weird. The driver does not work in fact.

ChaingTsung, 2023-10-31:
None of the drivers offered by OpenWRT or Quectel FAE works in the mainline branch! Although driver module named pcie_mhi.ko has been inserted into kernel and loaded, it doesnot work.There is no device description character generated in directory '/dev'.
Actually, as you can see, this means the module does not be identified successfully.

robimarko, 2023-10-31:
Again, OpenWrt does not ship custom vendor drivers nor we care about it.
Mainline MHI drivers work for sure, but forget about the QCA downstream MHI devices as upstream driver will expose the WWAN device for ModemManager to use directly

closed 2023-10-31 by ChaingTsung

suyuan168, 2024-03-15:
As far as I know, no matter how you select all mhi drivers in make menuconfig. Use mhi directly under openwrt in the 6.1 kernel. My modules are RM520NGL and FM150-AE.
lspci is able to see the device regarding Qualcomm X55.
But cannot use Modemmanager correctly. It is also impossible to use qmi to discover the device.
ls /dev can see about
wwan0mbime/
wwan0qcdm0/
wwan0qmi0
Please tell me how to correctly use the 5G module of the pcie interface under openwrt.

suyuan168, 2024-03-15:
https://forum.openwrt.org/t/uses-pcie-to-run-rm520n-gl-5g-module-and-uses-mhi-pcie-how-to-use-modemmanager/185755/5
https://forum.openwrt.org/t/i-have-a-pcie-quectel-rm520n-gl-and-fibocom-fm150-ae-and-fm160-ae/191435
I've already posted it, and it's been a long time. I'm really sorry to bother you all. This problem has troubled me for a long time.

suyuan168, 2024-03-15:
> 没有必要跳到别人的话题上。这是关于第 3 方 kmod 的，几乎肯定根本无法工作。

Sorry, I really couldn't find openwrt's own mhi on how the 5G pcie module works.
[ 1.539913] mhi-pci-generic 0000:01:00.0: no of_node; not parsing pinctrl DT
[ 1.539972] mhi-pci-generic 0000:01:00.0: MHI PCI device found: qcom-sdx55m
[1.539991] mhi-pci-generic 0000:01:00.0: BAR 0: assigned [mem 0x20300000-0x20300fff 64bit]
[ 1.540061] mhi-pci-generic 0000:01:00.0: enabling device (0000 -> 0002)
[ 1.541553] mhi mhi0: Requested to power ON
[3.750723] mhi mhi0: Power on setup success
[ 3.750968] mhi mhi0: Wait for device to enter SBL or Mission mode

ChaingTsung, 2025-01-04:
Public MHI kmod is not suitable. As many APIs(kernel) has changed,  3rd party kmods of 5G modules needs repaired or updated  source by their manufacturers.

robimarko, 2025-01-04:
That is just not the case, kernel does not handle the configuration nor connection itself nor it cares if your modem is LTE or 5G, that is what ModemManager or whater else youa actually use for connection management.

So mainline drivers work just fine with 5G modems

ynezz, 2025-01-04:
> Public MHI kmod is not suitable.

It works just fine.

> My modules are RM520NGL and FM150-AE.

These are my bit older notes from the 5G modem testing.

#### Banana PI-R4, RM520N-GL, OpenWrt 06b37a5856a, 6.6.40 kernel, upstream MHI driver

<details>
<summary>dmesg</summary>

```
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x411fd090]
[    0.000000] Linux version 6.6.40 (ynezz@ubuntu.legion) (aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 13.3.0 r26608-b2a84c86e3a2) 13.3.0, GNU ld (GNU Binutils) 2.42) #0 SMP Thu Jul 18 12:20:58 2024
[    0.000000] Machine model: Bananapi BPI-R4
[    0.000000] OF: reserved mem: 0x0000000042ff0000..0x0000000042ffffff (64 KiB) map non-reusable ramoops@42ff0000
[    0.000000] OF: reserved mem: 0x0000000043000000..0x000000004304ffff (320 KiB) nomap non-reusable secmon@43000000
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000040000000-0x00000000ffffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   [mem 0x0000000100000000-0x000000013fffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000040000000-0x0000000042ffffff]
[    0.000000]   node   0: [mem 0x0000000043000000-0x000000004304ffff]
[    0.000000]   node   0: [mem 0x0000000043050000-0x000000013fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000040000000-0x000000013fffffff]
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.1 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.4
[    0.000000] percpu: Embedded 18 pages/cpu s34536 r8192 d31000 u73728
[    0.000000] pcpu-alloc: s34536 r8192 d31000 u73728 alloc=18*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] CPU features: detected: GIC system register CPU interface
[    0.000000] CPU features: detected: Spectre-BHB
[    0.000000] CPU features: kernel page table isolation disabled by kernel configuration
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Kernel command line: console=ttyS0,115200n1 pci=pcie_bus_perf root=/dev/fit0 rootwait
[    0.000000] Dentry cache hash table entries: 524288 (order: 10, 4194304 bytes, linear)
[    0.000000] Inode-cache hash table entries: 262144 (order: 9, 2097152 bytes, linear)
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 1032192
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] software IO TLB: area num 4.
[    0.000000] software IO TLB: mapped [mem 0x00000000fa7de000-0x00000000fe7de000] (64MB)
[    0.000000] Memory: 4034432K/4194304K available (8512K kernel code, 894K rwdata, 2508K rodata, 384K init, 287K bss, 159872K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000]  Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] GICv3: GIC: Using split EOI/Deactivate mode
[    0.000000] GICv3: 416 SPIs implemented
[    0.000000] GICv3: 0 Extended SPIs implemented
[    0.000000] Root IRQ handler: gic_handle_irq
[    0.000000] GICv3: GICv3 features: 16 PPIs
[    0.000000] GICv3: CPU0: found redistributor 0 region 0:0x000000000c080000
[    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
[    0.000000] arch_timer: cp15 timer(s) running at 13.00MHz (phys).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x2ff89eacb, max_idle_ns: 440795202429 ns
[    0.000000] sched_clock: 56 bits at 13MHz, resolution 76ns, wraps every 4398046511101ns
[    0.000058] Calibrating delay loop (skipped), value calculated using timer frequency.. 26.00 BogoMIPS (lpj=130000)
[    0.000065] pid_max: default: 32768 minimum: 301
[    0.002155] Mount-cache hash table entries: 8192 (order: 4, 65536 bytes, linear)
[    0.002168] Mountpoint-cache hash table entries: 8192 (order: 4, 65536 bytes, linear)
[    0.003653] cacheinfo: Unable to detect cache hierarchy for CPU 0
[... 713 lines cut ...]
           |                          utran-19, ngran-1, ngran-2, ngran-3, ngran-5, ngran-7, ngran-8,
           |                          ngran-12, ngran-13, ngran-14, ngran-18, ngran-20, ngran-25, ngran-26,
           |                          ngran-28, ngran-29, ngran-30, ngran-38, ngran-40, ngran-41, ngran-48,
           |                          ngran-66, ngran-70, ngran-71, ngran-75, ngran-76, ngran-77, ngran-78,
           |                          ngran-79
           |                 current: utran-1, utran-4, utran-5, utran-8, utran-2, eutran-1,
           |                          eutran-2, eutran-3, eutran-4, eutran-5, eutran-7, eutran-8,
           |                          eutran-12, eutran-13, eutran-14, eutran-17, eutran-18, eutran-19,
           |                          eutran-20, eutran-25, eutran-26, eutran-28, eutran-29, eutran-30,
           |                          eutran-32, eutran-34, eutran-38, eutran-39, eutran-40, eutran-41,
           |                          eutran-42, eutran-43, eutran-46, eutran-48, eutran-66, eutran-71,
[... cut ...]
           |                          ngran-66, ngran-70, ngran-71, ngran-75, ngran-76, ngran-77, ngran-78,
           |                          ngran-79
  -----------------------------------
  IP       |               supported: ipv4, ipv6, ipv4v6
  -----------------------------------
  3GPP     |                    imei: <redacted>
  -----------------------------------
  SIM      |          sim slot paths: slot 1: none (active)
           |                          slot 2: none
```
</details>