openwrt/openwrt #13831: IPQ8074: Zyxel NBG7815: "kernel: backport nvmem v6.6 fixes and v6.7 changes" patch causing random MAC on each boot

pwned-pixel, 2023-10-30:
### Describe the bug

After applying this commit: 

https://github.com/openwrt/openwrt/commit/b5956700702b5649ec70bfa0d115af5b439ce6d1
kernel: backport nvmem v6.6 fixes and v6.7 changes

the MAC addresses are changing on every reboot on this device.

`[    4.171248] GMAC2(ffffff8003ffxxxx) Invalid MAC@ - using 82:3f:e0:52:de:8b
[    4.174842] GMAC3(ffffff8003c8xxxx) Invalid MAC@ - using 16:86:7f:39:0c:1d
[    4.181661] GMAC4(ffffff8003ffxxxx) Invalid MAC@ - using 1e:b9:7f:62:e1:3d
[    4.188461] GMAC5(ffffff8003d7xxxx) Invalid MAC@ - using f2:f6:f7:59:51:28
[    4.395754] GMAC6(ffffff8004d9xxxx) Invalid MAC@ - using 86:a2:36:58:1e:65`

This kernel message is not present on earlier versions. I've tracked it down to this commit. If you need the whole dmesg just let me know.

### OpenWrt version

r24256-1d6f9bdfc0

### OpenWrt target/subtarget

qualcommax/ipq807x

### Device

Zyxel NBG7815

### Image kind

Self-built image

### Expected behaviour

MAC's should be stable on each boot.

### Diffconfig

```text
CONFIG_TARGET_qualcommax=y
CONFIG_TARGET_qualcommax_ipq807x=y
CONFIG_TARGET_MULTI_PROFILE=y
CONFIG_TARGET_DEVICE_qualcommax_ipq807x_DEVICE_xiaomi_ax3600=y
CONFIG_TARGET_DEVICE_PACKAGES_qualcommax_ipq807x_DEVICE_xiaomi_ax3600=""
CONFIG_TARGET_DEVICE_qualcommax_ipq807x_DEVICE_zyxel_nbg7815=y
CONFIG_TARGET_DEVICE_PACKAGES_qualcommax_ipq807x_DEVICE_zyxel_nbg7815=""
CONFIG_TARGET_PER_DEVICE_ROOTFS=y
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

pwned-pixel, 2023-10-30:
I've tried to start narrowing things down and I've just deleted 816-v6.7-0004-Revert-nvmem-add-new-config-option.patch. It is fixing the issue. But I don't know if and what sideeffects this could introduce.

robimarko, 2023-10-30:
@rmilecki

john-tho, 2023-10-30:
Nice work narrowing this down. Rafal will fix, but if you wanted to keep digging in the meantime, could you check if there are any message from the `u-boot,env` driver in dmesg (like: `Failed to get`)?
[This line in `u_boot_env_probe`](https://elixir.bootlin.com/linux/v6.6/source/drivers/nvmem/u-boot-env.c#L221) looks suspicious
```C
	priv->mtd = of_get_mtd_device_by_node(np);
	if (IS_ERR(priv->mtd)) {
		dev_err_probe(dev, PTR_ERR(priv->mtd), "Failed to get %pOF MTD\n", np);
		return PTR_ERR(priv->mtd);
	}
```
given we are modifying the `of_node` https://github.com/openwrt/openwrt/blob/b5956700702b5649ec70bfa0d115af5b439ce6d1/target/linux/generic/backport-6.1/816-v6.7-0004-Revert-nvmem-add-new-config-option.patch#L50-L58

If you cannot find anything in dmesg, would you be able to check the DT node `u-boot,env` is using in `u_boot_env_probe` with and without the breaking change? Could edit this file inplace under `openwrt/build_dir/target-aarch_*/linux-*/linux-6*`, then `make target/linux/install`, but only if easy to recover from a bad flash, as easy to make mistakes this way.
```patch
--- drivers/nvmem/u-boot-env.c.orig	2023-10-31 06:44:36.186727737 +1000
+++ drivers/nvmem/u-boot-env.c	2023-10-31 06:47:18.392136608 +1000
@@ -211,6 +211,7 @@
 	struct u_boot_env *priv;
 	int err;
 
+	dev_warn(dev, "DT node: %pOF\n", np);
 	priv = devm_kzalloc(dev, sizeof(*priv), GFP_KERNEL);
 	if (!priv)
 		return -ENOMEM;
```

Cheers

rmilecki, 2023-10-30:
If you can edit kernel sources (like `vim build_dir/target-*/linux-*/linux-*/drivers/nvmem/u-boot-env.c`) please try following patch:
```diff
diff --git a/drivers/nvmem/u-boot-env.c b/drivers/nvmem/u-boot-env.c
index c4ae94af4af7..9ac8cf0b350c 100644
--- a/drivers/nvmem/u-boot-env.c
+++ b/drivers/nvmem/u-boot-env.c
@@ -233,6 +233,7 @@ static int u_boot_env_probe(struct platform_device *pdev)
 	config.ncells = priv->ncells;
 	config.priv = priv;
 	config.size = priv->mtd->size;
+	config.add_legacy_fixed_of_cells = true;
 
 	return PTR_ERR_OR_ZERO(devm_nvmem_register(dev, &config));
 }

```

rmilecki, 2023-10-30:
I'll try to debug that tomorrow. I got some idea (posted comment actually) about `add_legacy_fixed_of_cells` but that was incorrect. I've to spend some extra time on it.

pwned-pixel, 2023-10-31:
> could you check if there are any message from the u-boot,env driver in dmesg (like: Failed to get)?

I've checked. There is no such message. I've attached: dmesg (failed state), z-loader, u-boot and pre-boot log:

[dmesg.txt](https://github.com/openwrt/openwrt/files/13213510/dmesg.txt)
[uboot.txt](https://github.com/openwrt/openwrt/files/13213513/uboot.txt)
[zloader.txt](https://github.com/openwrt/openwrt/files/13213518/zloader.txt)
[pre-boot.txt](https://github.com/openwrt/openwrt/files/13213519/pre-boot.txt)

> If you cannot find anything in dmesg, would you be able to check the DT node u-boot,env is using in u_boot_env_probe with and without the breaking change? Could edit this file inplace under [...]

I'll try it. But probably not until evening.

rmilecki, 2023-10-31:
Can you please compile and run firmware with my debugging patch "kernel: add nvmem u-boot-env debugging patch for 6.1"? I pushed branch with it to the https://git.openwrt.org/?p=openwrt/staging/rmilecki.git;a=shortlog;h=refs/heads/nvmem-u-boot-env-debugging

pwned-pixel, 2023-10-31:
Sure, I'm currently building and I will give feedback as soon as I can. It could be that it will be delayed after lunch.

pwned-pixel, 2023-10-31:
`Starting kernel ...

Jumping to AARCH64 kernel via monitor
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd034]
[    0.000000] Linux version 6.1.60 () (aarch64-openwrt-linux-musl-gcc (OpenWrt GCC 12.3.0 r24256-1d6f9bdfc0) 12.3.0, GNU ld (GNU Binutils) 2.40.0) #0 SMP Mon Oct 30 08:08:39 2023
[    0.000000] Machine model: Zyxel NBG7815
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000040000000-0x000000007fffffff]
[    0.000000]   DMA32    empty
[    0.000000]   Normal   empty
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000040000000-0x0000000040ffffff]
[    0.000000]   node   0: [mem 0x0000000041000000-0x000000004a3fffff]
[    0.000000]   node   0: [mem 0x000000004a400000-0x00000000510fffff]
[    0.000000]   node   0: [mem 0x0000000051100000-0x000000007fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000040000000-0x000000007fffffff]
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: PSCIv1.0 detected in firmware.
[    0.000000] psci: Using standard PSCI v0.2 function IDs
[    0.000000] psci: MIGRATE_INFO_TYPE not supported.
[    0.000000] psci: SMC Calling Convention v1.0
[    0.000000] psci: OSI mode supported.
[    0.000000] psci: failed to set PC mode: -1
[    0.000000] percpu: Embedded 17 pages/cpu s31080 r8192 d30360 u69632
[    0.000000] Detected VIPT I-cache on CPU0
[    0.000000] alternatives: applying boot alternatives
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 258048
[    0.000000] Kernel command line: console=ttyMSM0,115200n8 root=PARTUUID=a0bb3852-36cc-ac69-a1a2-6bd39ff56f0a rootwait
[    0.000000] Dentry cache hash table entries: 131072 (order: 8, 1048576 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 7, 524288 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 887724K/1048576K available (7744K kernel code, 854K rwdata, 2268K rodata, 1280K init, 273K bss, 160852K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 64, nr_irqs: 64, preallocated irqs: 0
[    0.000000] Root IRQ handler: gic_handle_irq
[    0.000000] GICv2m: range[mem 0x0b00a000-0x0b00affc], SPI[448:479]
[    0.000000] rcu: srcu_init: Setting srcu_struct sizes based on contention.
[    0.000000] arch_timer: cp15 and mmio timer(s) running at 19.20MHz (virt/virt).
[    0.000000] clocksource: arch_sys_counter: mask: 0xffffffffffffff max_cycles: 0x46d987e47, max_idle_ns: 440795202767 ns
[    0.000001] sched_clock: 56 bits at 19MHz, resolution 52ns, wraps every 4398046511078ns
[    0.000102] Calibrating delay loop (skipped), value calculated using timer frequency.. 38.40 BogoMIPS (lpj=192000)
[    0.000116] pid_max: default: 32768 minimum: 301
[    0.000592] Mount-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.000606] Mountpoint-cache hash table entries: 2048 (order: 2, 16384 bytes, linear)
[    0.002453] cblist_init_generic: Setting adjustable number of callback queues.
[    0.002462] cblist_init_generic: Setting shift to 2 and lim to 1.
[    0.002657] rcu: Hierarchical SRCU implementation.
[    0.002661] rcu: 	Max phase no-delay instances is 1000.
[    0.003478] smp: Bringing up secondary CPUs ...
[    0.004122] Detected VIPT I-cache on CPU1
[    0.004266] CPU1: Booted secondary processor 0x0000000001 [0x410fd034]
[    0.004940] Detected VIPT I-cache on CPU2
[    0.005043] CPU2: Booted secondary processor 0x0000000002 [0x410fd034]
[    0.005668] Detected VIPT I-cache on CPU3
[    0.005770] CPU3: Booted secondary processor 0x0000000003 [0x410fd034]
[    0.005838] smp: Brought up 1 node, 4 CPUs
[    0.005846] SMP: Total of 4 processors activated.
[    0.005851] CPU features: detected: 32-bit EL0 Support
[    0.005855] CPU features: detected: CRC32 instructions
[    0.005918] CPU features: emulated: Privileged Access Never (PAN) using TTBR0_EL1 switching
[    0.005925] CPU: All CPU(s) started at EL1
[    0.005927] alternatives: applying system-wide alternatives
[    0.016833] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.016861] futex hash table entries: 1024 (order: 4, 65536 bytes, linear)
[    0.017092] pinctrl core: initialized pinctrl subsystem
[    0.018258] NET: Registered PF_NETLINK/PF_ROUTE protocol family
[... 257 lines cut ...]
[   21.595321] I/O error, dev loop0, sector 564 op 0x9:(WRITE_ZEROES) flags 0x800 phys_seg 0 prio class 2
[   22.687460] br-lan: port 1(lan1) entered blocking state
[   22.687509] br-lan: port 1(lan1) entered disabled state
[   22.691731] device lan1 entered promiscuous mode
[   22.704730] br-lan: port 2(lan2) entered blocking state
[   22.704776] br-lan: port 2(lan2) entered disabled state
[   22.709062] device lan2 entered promiscuous mode
[   22.717580] br-lan: port 3(lan3) entered blocking state
[   22.718845] br-lan: port 3(lan3) entered disabled state
[   22.724031] device lan3 entered promiscuous mode
[   22.732207] br-lan: port 4(lan4) entered blocking state
[   22.733860] br-lan: port 4(lan4) entered disabled state
[   22.739094] device lan4 entered promiscuous mode
[... cut ...]
[   25.836123] br-lan: port 4(lan4) entered blocking state
[   25.840781] br-lan: port 4(lan4) entered forwarding state
[   25.846192] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready
[   26.075777] nss-dp 3a001800.dp5 wan: PHY Link up speed: 1000
[   26.075845] IPv6: ADDRCONF(NETDEV_CHANGE): wan: link becomes ready
[   29.995900] nss-dp 3a001000.dp1 lan1: PHY Link up speed: 1000
[   29.995980] br-lan: port 1(lan1) entered blocking state
[   30.000639] br-lan: port 1(lan1) entered forwarding state`

pwned-pixel, 2023-10-31:
I cannot see anything suspicious. :/

closed 2023-10-31 by openwrt-bot

rmilecki, 2023-10-31:
Please give the latest `master` a try.

pwned-pixel, 2023-10-31:
Yep it is working again. Thank you for this fast fix. Kind regards. :)