openwrt/openwrt #13836: compile package  lrzsz failed.

iiio051031448, 2023-10-31:
### Describe the bug

compile package  lrzsz failed with lasted SNAPSHOT.

```
make[3]: Entering directory '/work2/w2/op/bpi-r3/openwrt/build_dir/target-aarch64_cortex-a53_glibc/lrzsz-990823'
cd . && aclocal
aclocal.real: warning: autoconf input should be named 'configure.ac', not 'configure.in'
acinclude.m4:2: warning: underquoted definition of AC_REPLACE_GNU_GETOPT
acinclude.m4:2:   run info Automake 'Extending aclocal'
acinclude.m4:2:   or see https://www.gnu.org/software/automake/manual/automake.html#Extending-aclocal
configure.in:106: error: automatic de-ANSI-fication support has been removed
./lib/autoconf/general.m4:2437: AC_FATAL is expanded from...
/work2/w2/op/bpi-r3/openwrt/staging_dir/host/share/aclocal-1.16/obsolete.m4:26: AM_C_PROTOTYPES is expanded from...
configure.in:106: the top level
autom4te: error: /work2/w2/op/bpi-r3/openwrt/staging_dir/host/bin/m4 failed with exit status: 1
aclocal.real: error: autom4te failed with exit status: 1
make[3]: *** [Makefile:127: aclocal.m4] Error 1
make[3]: Leaving directory '/work2/w2/op/bpi-r3/openwrt/build_dir/target-aarch64_cortex-a53_glibc/lrzsz-990823'
make[2]: *** [Makefile:72: /work2/w2/op/bpi-r3/openwrt/build_dir/target-aarch64_cortex-a53_glibc/lrzsz-990823/.built] Error 2
make[2]: Leaving directory '/work2/w2/op/bpi-r3/openwrt/feeds/packages/utils/lrzsz'
time: package/feeds/packages/lrzsz/compile#1.24#0.05#1.29
    ERROR: package/feeds/packages/lrzsz failed to build.
make[1]: *** [package/Makefile:120: package/feeds/packages/lrzsz/compile] Error 1
make[1]: Leaving directory '/work2/w2/op/bpi-r3/openwrt'
make: *** [/work2/w2/op/bpi-r3/openwrt/include/toplevel.mk:232: package/feeds/packages/lrzsz/compile] Error 2
```

### OpenWrt version

r24257+41-ed5dbbc129

### OpenWrt target/subtarget

mediatek/filogic

### Device

Bananapi BPI-R3

### Image kind

Self-built image

### Steps to reproduce

make ./package/feeds/packages/lrzsz/compile V=s

### Diffconfig

```text
CONFIG_TARGET_mediatek=y
CONFIG_TARGET_mediatek_filogic=y
CONFIG_TARGET_mediatek_filogic_DEVICE_bananapi_bpi-r3=y
CONFIG_DEVEL=y
CONFIG_TOOLCHAINOPTS=y
CONFIG_BUSYBOX_CUSTOM=y
CONFIG_AIRCRACK_NG_OPENSSL=y
CONFIG_ARIA2_ASYNC_DNS=y
CONFIG_ARIA2_BITTORRENT=y
CONFIG_ARIA2_COOKIE=y
CONFIG_ARIA2_LIBXML2=y
CONFIG_ARIA2_METALINK=y
CONFIG_ARIA2_OPENSSL=y
CONFIG_ARIA2_SFTP=y
CONFIG_ARIA2_WEBSOCKET=y
# CONFIG_BUSYBOX_CONFIG_ASH_RANDOM_SUPPORT is not set
CONFIG_BUSYBOX_CONFIG_LSPCI=y
CONFIG_BUSYBOX_CONFIG_LSUSB=y
# CONFIG_BUSYBOX_CONFIG_RESIZE is not set
[... 286 lines cut ...]
CONFIG_SQLITE3_RTREE=y
CONFIG_STRIP_ARGS="--strip-all"
# CONFIG_TARGET_ROOTFS_INITRAMFS is not set
CONFIG_TARGET_ROOTFS_PARTSIZE=7000
CONFIG_TARGET_SUFFIX="gnu"
CONFIG_TESTING_KERNEL=y
CONFIG_USE_GLIBC=y
CONFIG_USE_STRIP=y
CONFIG_WOLFSSL_HAS_NO_HW=y
CONFIG_WPA_ENABLE_WEP=y
CONFIG_WPA_MSG_MIN_PRIORITY=0
CONFIG_boost-compile-visibility-hidden=y
CONFIG_boost-runtime-shared=y
CONFIG_boost-static-and-shared-libs=y
CONFIG_boost-variant-release=y
# CONFIG_LIBCURL_UNIX_SOCKETS is not set
# CONFIG_OPENSSL_WITH_IDEA is not set
# CONFIG_OPENSSL_WITH_MDC2 is not set
# CONFIG_OPENSSL_WITH_SEED is not set
# CONFIG_OPENSSL_WITH_WHIRLPOOL is not set
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

brada4, 2023-10-31:
You  may report packages feed bugs to ../packages/ tracker.
Please try commenting out ansi2knr in configure.in:106. GNU make can handle either headers nowadays.
What is happening - very latest autotools generate ./configure from 15y old config files and encounter now error-deprecated option. It was warning for a while.

krant, 2024-02-20:
@iiio051031448 I've tried your config on the latest master of openwrt and packages - it builds all right

1715173329, 2024-03-06:
this issue seems to be happened "randomly", at least on our buildbot.

dhewg, 2024-03-06:
Builds fine here too, so it's probably the ancient autohell crap that fails randomly?
Dunno if anyone is up to look into that (I'm not for sanity reasons).
I'd say it makes more sense to throw meson at it.

1715173329, 2024-03-06:
> it's probably the ancient autohell crap that fails randomly?

Yes.