openwrt/openwrt #13839: Failed to build libcurl with OpenSSL/without mBedTLS support

MBRjun, 2023-10-31:
### Describe the bug

- ``make -j1``:

  ```
  make[3] -C feeds/packages/net/curl compile
    ERROR: package/feeds/packages/curl failed to build.
  ```
- ``make -j1 V=s``:

  ```
  Package libcurl is missing dependencies for the following libraries:
    libmbedcrypto.so.7
    libmbedtls.so.14
    libmbedx509.so.1
  make[3]: *** [Makefile:197: /home/mbrjun/github/openwrt/bin/targets/x86/64/packages/libcurl4_8.4.0-2_x86_64.ipk] Error 1
  make[3]: Leaving directory '/home/mbrjun/github/openwrt/feeds/packages/net/curl'
  time: package/feeds/packages/curl/compile#0.16#0.02#0.24
    ERROR: package/feeds/packages/curl failed to build.
  make[2]: *** [package/Makefile:120: package/feeds/packages/curl/compile] Error 1
  ```
[image]
[image]

### OpenWrt version

r23497-6637af95aa

### OpenWrt target/subtarget

x86/64

### Device

Hyper-V UEFI Release v4.1 Virtual Machine

### Image kind

Self-built image

### Steps to reproduce

- ``Libraries > libcurl > Selected SSL library``
  Select ``OpenSSL``
- ``Libraries > SSL``
  unselect ``libmbedtls``

### Diffconfig

```text
https://pastebin.com/fSFVDmb1
```

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

xback, 2023-10-31:
I switched from libopenssl to libmbedtls last week with libcurl.
Both builded fine.

Did you do a do a `make` before switching to openssl?
In case `yes`, did you do a `make clean` after switching?

MBRjun, 2023-10-31:
> I switched from libopenssl to libmbedtls last week with libcurl. Both builded fine.
> 
> Did you do a do a `make` before switching to openssl? In case `yes`, did you do a `make clean` after switching?

``make clean`` works. Thanks.

closed 2023-10-31 by MBRjun