openwrt/openwrt #13840: 23.05 wpad-mbedtls does not support radius

farfade, 2023-10-31:
### Describe the bug

Hello,

23.05 with wpad-mbedtls does not support radius. hostapd complains about related unknown parameters.

opkg remove wpad-mbedtls
opkg install wpad-openssl

fixes the problem

I suggest regression testing before such default architecture switches.

### OpenWrt version

23.05

### OpenWrt target/subtarget

ramips/mt7621

### Device

 D-Team Newifi D2

### Image kind

Official downloaded image

### Terms

- [X] I am reporting an issue for OpenWrt, not an unsupported fork.

farfade, 2023-10-31:
Probably linked to https://github.com/openwrt/openwrt/issues/13774

brada4, 2023-10-31:
You need to add some radius  or wpad logs in regard to rejected authorisation....

farfade, 2023-11-01:
I'm sorry @brada4, I've no logs as it is in prod now with openssl.

The messages were like that

```
hostadp : unknown parameter :
auth_server_addr
auth_server_port
 auth_server_shared_secret
Unable to configure EAP authenticator
```

brada4, 2023-11-01:
I think you got confused by packages. Default wpad-basic-mbedtls installed by default does not provide radius support, while wpad-mbedtls is ought to support it. Please check.

farfade, 2023-11-02:
Sounds fair, but I can't test easily anymore... and I'm happy with the openssl flavour.

I was used with that type of package, it was the same behaviour with wolfssl. 
I am sure I installed, then uninstalled wpad-mbedtls (not basic) before installing wpad-openssl that made it work again. But I also messed with libustream and luci-ssl flavours at the same time...

I close it. Thanks for your support !

closed 2023-11-02 by farfade